ghaerr/elks #551: fdisk -l displays empty output table

georgp24, 2020-04-14:
When you enter `fdisk -l /dev/fd0` or `fdisk -l /dev/bda` it will display a table with CHS information and sector start/size information for four partitions. But all entries are zero, expect for the last in partition 4 which seems to be false. 
I try to determine the size of the disk and there is no `df` command available. Therefore I thought the size in sectors would provide this information.

ghaerr, 2020-04-14:
Remember, `fdisk` can only be used on disks with MBR partition tables. Thus, using it with `/dev/fd0` is user error.

There is no standard method for determining whether there is an MBR on the hard drive `/dev/bda`. Thus, `fdisk` can't be made to issue an error when used on a 'flat' hard drive (no MBR).

I'm pretty certain that your use is on a 'flat' partition. It's likely `fdisk` was inspecting the non-MBR boot sector for a Minix boot image, which has zeros in most of what would be the MBR area, and keeps some data towards the end.

georgp24, 2020-04-14:
Yes, the disks had no MBR. Thank you for the information

closed 2020-04-14 by georgp24