ghaerr/elks #552: How to slow the mouse movement?

georgp24, 2020-04-14:
I configured the serial port in qemu.sh to support a serial mouse to run the nx** demos. However, the mouse moves so fast you cannot select anything. If you make a very tiny movement it jumps across the entire screen. How do you set the mouse to move slowly?

When I run the nxworld demo I see a world map. What can you do with that, what actions can the user execute then?

ghaerr, 2020-04-14:
@georgp24, I'll tell you if you promise to document it :) Just kidding.

> How do you set the mouse to move slowly?

The scaling factor and acceleration are specified in nanox/drivers/ser_mouse.c using the `SCALE` and `THRESH` defines.

> I run the nxworld demo I see a world map. What can you do with that

The map is rescaled larger or smaller by dragging the mouse with a finger down. More options are specified in nanox/demos/world.c function 'dokeydown'. These include the keystrokes 's', 'm', ESC, 'q', and 'r' (space).

georgp24, 2020-04-18:
I tested it with the latest version now and it works much better. However:
- nxterm always shows `<-[6n` before the `#` prompt
- it is difficult to quit the nxlandmine and nxdemo programs with the mouse. You have to click several times the left and the right mouse button to terminate the program. The click to draw a line in the nxdemo program works fine.

ghaerr, 2020-04-18:
Hi @georgp24, thanks for the testing. Did you end up changing the defaults in mou_ser.c to get your mouse to work better?

> nxterm always shows `<-[6n` before the `#` prompt

That's a bug due to the way that 'linenoise' queries a remote shell for the window size. I'll fix it.

> it is difficult to quit the nxlandmine and nxdemo programs with the mouse. 

All demos can also be exited with ESC. Did you try that? It isn't shown/said anywhere that ESC quits nano-X.

> You have to click several times the left and the right mouse button to terminate the program. 

Try holding the left button down, then while holding it down, click the right button.

georgp24, 2020-04-18:
Yes, the demos exit with ESC or cursor down - did not try more keys yet. ELKS uses the ESC and displays [B on the screen - now I understand that.

> Try holding the left button down, then while holding it down, click the right button.

This is not intuitive and not required on Linux.

ghaerr, 2020-04-18:
> This is not intuitive and not required on Linux.

Never mind, I thought you were trying to use that sequence to quit nano-X. It does nothing on ELKS.

> it is difficult to quit the nxlandmine and nxdemo programs with the mouse.

I have found the reason for this: a recent change starts `/bin/getty` on /dev/ttyS0 at boot (run `ps` to see), the same serial port your mouse is connected to. So getty and nano-X are both trying to read/write the serial port. This is a problem.

Not sure if it's better to have the standard ELKS image use '/etc/inittab' with getty running on ttyS0, or leave the port open for possible mouse usage. In any case, there needs to be a way to switch between the use cases. We could use `init` to switch run levels to turn off getty before running a nano-X demo, but run levels don't currently work.

georgp24, 2020-04-21:
I tried the latest version now. When I enable the following line in qemu.sh:
`SERIAL="-chardev msmouse,id=chardev1 -device isa-serial,chardev=chardev1,id=serial1"`
I get this message when I try to run nxdemo:
`Error 16 opening serial mouse type ms on port /dev/ttyS0`
`Cannot initialise mouse: Device or resource busy`
`cannot open graphics`
Qemu.sh now has both serial lines commented out for default. If I leave it like that I get `Error 19` with the same message as above.

ghaerr, 2020-04-21:
@georgp24, thanks for your testing.

Yes, access to the serial ports was corrected in PR #569, and qemu.sh changed, so that users wouldn't mistakenly try running a mouse program when the system was also running a getty/login on the serial port at the same time.

This was all explained din PR #596, please take a look at it. Since ELKS now includes multiuser support for multiple consoles (tty1, tty2, tty3), and serial ports, `init` was changed to allow multiple run levels. 

To fix your mouse problem, you should re-enable the mouse serial driver in qemu.sh, and then switch to single-user console mode in elks, by running `init 1` at the prompt.

georgp24, 2020-04-21:
After `init 1` the nx programs work and the mouse works ok now. However, if I click on the large brown box or the grey box in `nxdemo,` ELKS hangs.

ghaerr, 2020-04-21:
>  if I click on the large brown box or the grey box in nxdemo, ELKS hangs.

Are you sure ELKS is hung? Mine changes the cursor, when clicking in the large brown box. But pressing the ESC key exits the program normally.

georgp24, 2020-04-22:
The cursor changes but when I click on the brown box the box gets to the foreground and the cursor does not move any more. Also ESC and further keys do not work. I have to enter CTRL-ALT-G to close the Qemu window.

ghaerr, 2020-07-18:
Hello @georgp24,

The issues you identified and found in this thread are fixed in PR #675.  After you have tested it, please close this issue.

Thanks!

closed 2020-08-11 by ghaerr