ghaerr/elks #555: Mtools obsolete?

georgp24, 2020-04-16:
ELKS supports Minix and FAT12 floppy disks now. Therefore I think the Mtools commands should no longer be added to the generated disk images. They were intended to support to read and write data to MS-DOS formatted floppy disks using ELKS.

ghaerr, 2020-04-16:
The Mtools commands can be excluded from the built disk image in `menuconfig` by answering N to 'mtools' under 'Applications'. It is currently defaulted to Y.

The commands could possibly be useful for accessing a FAT disk when the kernel is compiled without FAT filesystem support. It seems that the above config option should probably be changed to default N though.

Mellvik, 2020-04-16:
I second this.
Mtools are useful when there is no space in the kernel for FAT. Default OFF is good.

—M

georgp24, 2020-04-16:
Greg, can you please set it to "N" then with your next commit?

ghaerr, 2020-04-16:
Default to 'N' added in #557.

closed 2020-04-16 by ghaerr