ghaerr/elks #562: tty & task struct

Mellvik, 2020-04-17:
A ELKS gets more complete and more stable, new needs arise, such as the tty(1) command recently ported.
While testing system access via multiple serial lines, gettys, init loops & more over the past week, I've frequently missed the ability to see (from a ps(1) listing), which process belongs to which tty. I've spent some time digging around inside the system, and come to the (possibly wrong) conclusion that it is next to impossible to find the name of the controlling tty from the task structure.
Thus the question - would it be an idea to add an entry similar to 
`dev_t u_ttyd ` (from the BSD u-structure)
to the ELKS task_structure?

--Mellvik

ghaerr, 2020-04-17:
> I've frequently missed the ability to see (from a ps(1) listing), which process belongs to which tty.

Agreed.

> it is next to impossible to find the name of the controlling tty from the task structure.

The `task->tty` pointer, if non-NULL, points to the controlling TTY in the structure `struct tty ttys[]`. The 'minor' member of `struct tty` can be used to display the TTY for ps. The `ttys` structure can be accessed in the same segment as the `task` struct, that is, the kernel DS retrieved using MEM_GETDS.

I'll add this to my list, unless you want to do it.

Mellvik, 2020-04-17:
Thanks,
and no thanks. I tried to get my head around this today and hit the wall when trying to get from minor to name … (and compiler related issues like including kernel-oriented header files (ntty.h) effectively in a user application). 

—M

closed 2020-04-20 by Mellvik