ghaerr/elks #566: PCem emulator testing

ghaerr, 2020-04-18:
So I just gave a try to fd360-minix.bin in PCem with my 1512 profile and it works great. The only thing you need to do to use these images with PCem is change the file extension to 'img' to load them in. Everything in PCem can be set up from the GUI like VirtualBox so there's not really a big need for a config file. But there is one for each virtual PC config one might brew anyway.

[image]

[image]

[image]

I went and ls'ed and cat'ed around the file system and everything works great so far. 'reboot' works too. The system seems to run as fast as I remember MSDOS to run on a real 1512. I could be wrong but the 1512 doesn't have NEC V30 CPU, but an AMD 8086. Maybe PCem got it wrong. The serial ports are indeed two 16550A on the 1512.

Building binutils/gcc with build.sh works great as well. I was able to build a custom 32mb hd image of ELKS after fiddling a bit in menuconfig. However the geometry automatically detected by PCem for the ELKS hd image isn't supported by the MFM controller emulated by PCem that I chose. Going to try out other controllers. This is what I get so far with the hd image in PCem:

[image]

_Originally posted by @cjsthompson in https://github.com/jbruchon/elks/issues/372#issuecomment-615870788_

ghaerr, 2020-04-18:
Hi @cjsthompson, I moved your issue over here to keep it separate from the IRQ discussions.
Thanks for your testing!!

It's interesting looking at a different emulator boot ELKS. I notice that it's found two LPT ports at the same locations as the serial ports. Also, the system seems to think there's one hard drive and 0 floppy drives when booting from floppy, but works anyways.

The reported CHS of 305,8,17 can't always be trusted. some emulator BIOS's calculate that from the real CHS. What settings did you use in `menuconfig`?

Have you tried running "cd images; make images" to make the extra images and booting the standard HD image 'hd32-minix.bin'?

cjsthompson, 2020-04-18:
I left everything to defaults as far as the HD image settings when I built it. It seems 32mb isn't a standard size for MFM drives. It was 10,20,30,40 mbs. My 1512 has a 40mb MFM Seagate in it.

I am as we speak running menuconfig again to alter the geometry with one I know works with the Xebec MFM controller emulated by PCem. I just need to figure out how to calculate the disk size in 1k blocks for the geometry I put in (17 sectors, 8 heads, 306 cylinders).

ghaerr, 2020-04-18:
> It seems 32mb isn't a standard size for MFM drives. 

Definitely not, the 32MB was built to test a large (for ELKS anyways) hard drive and force FAT32 for testing.

> I just need to figure out how to calculate the disk size in 1k blocks for the geometry I put in

Size in 1k blocks = (C x H x S) / 1024, rounded up.

cjsthompson, 2020-04-18:
With this setup, now I can boot the ELKS hd image in PCem with no problem. 

[image]

There are interrupt problems however. But I get to the login prompt, can log in, and use commands.

[image]

ghaerr, 2020-04-18:
There was a recent change to enable COM3 and its associated IRQ 5 by default. This IRQ is used by other devices, so it looks like this should be turned off by default.

Modify 'elks/include/arch/ports.h', and comment out the \#define CONFIG_NEED_IRQ5 on line 44, that will fix it.

cjsthompson, 2020-04-18:
Now everything is working as it should.

https://www.youtube.com/watch?v=MlcYiYBTEHY

closed 2020-04-19 by ghaerr