ghaerr/elks #573: Minor curses/elvis problem

Mellvik, 2020-04-20:
If the last visible line on the screen (say line 25 on my 80x25 window) is long (longer than the screen), scroll down doesn't work. Only the last line gets replaced while scrolling down, until a short line appears. From then on scrolling is OK (but the screen content wrong). 

Scrolling up works fine.

Attached file demonstrates the effect.

--Mellvik
[xx.gz](https://github.com/jbruchon/elks/files/4503499/xx.gz)

ghaerr, 2020-04-21:
@Mellvik: I'm pretty sure this is as a result of the ANSI console enhancements I made in #554, where the console now does not auto-scroll in the last column, until another character is received.

Thus, the :am: (auto-margins) entry should be removed from /etc/termcap for the ansi= entry (middle one). Can you test remove that and see whether that fixes the problem? I tested and it seems to fix the problem but would like more testing. (This can be removed within ELKS using `vi /etc/termcap` if you don't want to rebuild).

If that fixes the problem, I'll patch it with my next PR. I'm knee-deep in a bad `malloc` libc bug that I've almost got cornered...

Mellvik, 2020-04-21:
Tested and verified, @ghaerr. Thanks.

Not that I know elvis is actually using termcap (and not tinycap), I can install something more efficient (backscrolling) to work with macOS terminal.

Good luck with malloc. I’m having a ball with serial.c - not really a consolation, is it?

-M

closed 2020-04-21 by Mellvik

Mellvik, 2020-04-21:
> Now that I know elvis is actually using termcap (and not tinycap), I can install something more efficient (backscrolling) to work with macOS terminal.
> That should be working now, are you talking about a new termcap entry that uses scroll regions?
> My previous PR added VT100 DSR (device status report) to vi, but it only works half the time on QEMU because QEMU loses serial characters. I haven't been able to test on real serial lines, and there seems to be at least one major bug left in serial (the mouse crash on @pawosm's hardware).
> Is that for the LNEXT implementation? I'm also looking for the FIFO code detection fix.
> We're getting to the point where adding features to the kernel needs to be carefully thought through, as we're running out of code space, and there is no easy way around that, which could cause new kernel features to be hard to implement without removing something else. FYI, I'm actually starting to look for unused kernel code to remove at this point (code not controlled via config #defines), just in case.

@ghaerr,
I'll check out your vt100 entry. It has been ages since I played with termcap. I think it's a good thing I don't use QEMU all that much - widens the 'exposure' for elks.

No, I'm not on ntty yet, I believe it is important to get to the end of the line with serial, with soft & hard flow control. More on that in its own venue.
SPACE: Yes, I'm all in, I have some serial related ideas in that dept, although config related.

M

Mellvik, 2020-04-22:
@ghaerr, final comment on this issue: Turns out that both the PC console and xterm windows support AL, dl and DL termcap entries - all 3 very helpful at slow (serial) line speeds (less repaint). Is it an idea to just put them into the ansi entry in ELKS?

M

ghaerr, 2020-04-22:
> Turns out that both the PC console and xterm windows support AL, dl and DL termcap entries - all 3 very helpful at slow (serial) line speeds (less repaint). Is it an idea to just put them into the ansi entry in ELKS?

Yes, that would be great. Please test the operation of vi on both console and serial to make sure they work without introducing other issues. `kilo` won't be affected, as it uses another (slow) mechanism.

I haven't checked whether the elks ANSI Console supports the sequences you're talking about - if not, then this won't work, and we'd probably want to have another entry used instead, which would be a duplicate of the ansi entry, called 'xterm' for use on serial or external emulators.

I'll wait for a PR from you on this, and go ahead and include the deleted :am: entry as well.