ghaerr/elks #583: mv() incomplete

Mellvik, 2020-04-25:
This may be one someone's list already, it certainly is on mine. But for the record:

mv(1) is incomplete in that it (possibly among other things) cannot move directories. Probably an issue in rename() or link(), and hopefully an easy fix.

ghaerr, 2020-04-26:
I looked at the Minix filesystem source. In elks/fs/namei.c::sys_rename(), `rename` calls `link` then `unlink`, which works for files. The `link` system call does not allow a directory to be hard-linked, and that's why this isn't working (permitted).

Unfortunately, ELKS doesn't have any filesystem entry points for `rename`, which means that all filesystems' file_operations function arrays would have to be modified, and then a function could be written, separately for each filesystem. That'll take a fair amount of work, and all the filesystems would have to be tested, as the file_operations structs are all over the place, but I'll look into it. This probably won't happen quickly.

toncho11, 2020-11-03:
Is it possible to just do it another way? Create a folder, copy the files, delete the old one.

Mellvik, 2020-11-03:
@toncho11 - 
that is indeed possible, but the 'raison d'être'  for `mv` is to change names and avoid copying - which will happen only when moving files across file systems. `cp` should do that what you suggest, but `cp` (like `mv` and many other utilities) is still functionally incomplete in ELKS, so there is work to be done for both.

--Mellvik

ghaerr, 2020-11-03:
Hello @toncho11,

> Is it possible to just do it another way? Create a folder, copy the files, delete the old one.

Nice idea, thanks. The files could be linked, rather than copied, which would make this operation fairly fast. It is a bit of a kluge to implement this in the kernel though. The "proper" way is to implement a new "rename" filesystem entry point, but that's complicated in its own right. I'll take a look at it, now that we have more code space available in the kernel.

Hello @Mellvik,

> cp should do that what you suggest, but cp (like mv and many other utilities) is still functionally incomplete in ELKS, 

What are the unimplemented features in cp and mv? We would need a recursive version of cp that uses links instead of copying, to do what you suggest. That would better be handled by ln, I suppose. However, handling this in userspace rather than the kernel is probably a good idea, until a proper fix to the kernel can be made.

Thank you!

Mellvik, 2020-11-03:
> cp should do that what you suggest, but cp (like mv and many other utilities) is still functionally incomplete in ELKS,
>
> What are the unimplemented features in cp and mv? We would need a recursive [...]
@ghaerr,

your interpretation of my suggestion was a lot more literal that my intention. My point was that mv is for move/rename, cp is for copying.

As to cp in ELKS, it takes no options at all, how much is missing depends on preferences. My personal gripe with the current cp is -rp, which I use a lot. Then -f and -n — oposites in 'force' and 'no-clobber', plus -i as in interactive.

As to mv, #583 makes the primary point as @thoncho11 points out. Then its inability to overwrite the target, which bites me all the time.

And I agree with you, handling this in user space at leas for now, is the way to go.

—Mellvik

ghaerr, 2020-11-03:
> Then its inability to overwrite the target, which bites me all the time.

What do you mean, I don't understand?

Mellvik, 2020-11-03:
> 
> Then its inability to overwrite the target, which bites me all the time.
> 
> What do you mean, I don't understand?
> 
> 
Like this… — mv -f being the default behaviour

# touch xx
# touch yy
# mv yy xx
xx: File exists 
# 

I'm guessing many of these commands come from busybox and not from Minix (or Linux), which in a sense is good since we don't inherit a lot of possibly marginally needed behaviour. On the other hand it means we have to implement whatever gets priority.

—Mellvik

ghaerr, 2020-11-03:
> \# mv yy xx
xx: File exists

I'll fix that. After thinking about this discussion, I'll add @toncho11's suggested fix to mv as well. PR coming shortly.

> I'm guessing many of these commands come from busybox and not from Minix (or Linux)

No, almost none of the base set of commands are from busybox. The only commands from Minix are in the minix\* directories. All the base commands come from David Bell who wrote sash, and most of the commands were separated out from sash.

ghaerr, 2020-11-04:
@Mellvik, if the latest `mv` enhancements solve this issue for you, lets close this. I have the kernel rename issue in my long list of other TODO items; that won't be implemented soon. Thanks!

ghaerr, 2020-11-08:
@Mellvik, did my recent fixes to mv solve this issue?

Mellvik, 2020-11-08:
Thank you, @ghaerr -
I agree with you, this is as far as mv can be reasonably pushed until more underlying issues can be fixed.

Great work! 
-M

closed 2020-11-08 by Mellvik