ghaerr/elks #584: German keyboard support no longer available

georgp24, 2020-04-25:
I use ELKS set to the German keyboard driver with `make menuconfig`. Now when I set the latest version to the German keyboard driver, ELKS still works with the keyboard set to US.
Maybe this is a regression problem with PR #571.

ghaerr, 2020-04-25:
Thanks @georgp24, I'll take a look at it. If not #571, one of my other PR's is likely responsible.

Can this be repeated with QEMU? I'm not running on real hardware.

georgp24, 2020-04-25:
Thank you Greg. I am using mostly Qemu and I observed it this way.

ghaerr, 2020-04-25:
@georgp24, I can't get this to repeat.

The problem is likely not in #571, just looked at it. Can you make sure to do a 'rm .config; make defconfig; make menuconfig' and select German keyboard? Then check .config and make sure it is set (CONFIG_KEYMAP_DE).

When I set the German keyboard, different characters are output on the console, for instance with the ';' key. (However, they are showing up as white blocks).

georgp24, 2020-04-26:
The problem is the message after `make menuconfig`. It says:
```
Saving your kernel configuration...
 * End of ELKS kernel configuration.
 *
 * You may now run `make` to build
```
After a `make clean` and a `make` it works now. Thank you!

closed 2020-04-26 by georgp24