ghaerr/elks #586: Regression? Boot stopped after "VFS: Mounted root (minix filesystem)"

pawosm-arm, 2020-04-25:
I wasn't building 'master' for last few days, so it must have been introduced by one of very recent changes.

pawosm-arm, 2020-04-25:
Right, master updated again today. Let me check if it's any better.

pawosm-arm, 2020-04-25:
Unfortunately, it's still stuck right after VFS.

ghaerr, 2020-04-25:
Can you give more details, perhaps a photo/screenshot? Is this on real hardware?

There have been changes to the serial port code, and the /etc/rc.d/rc.sys script is run immediately after VFS mount. It could be dying running that script, which starts "Running /etc/rc.d/rc.sys".

Are the serial ports identified on the boot screen?

`init` is running at that point, may need to turn on init debugging.

ghaerr, 2020-04-25:
More messages about exactly what is happening can be output by editing 'elkscmd/sys_utils/init.c' and setting DEBUG 1 on line 40, then remaking and rebooting.

pawosm-arm, 2020-04-25:
I don't get it. I changed the console to 'serial' to get some logs for you, and it booted!

There's one thing I need to check though. In order to connect null-modem cable, I had to disconnect serial mouse. Maybe it should not be connected at boot time?!

ghaerr, 2020-04-25:
> I had to disconnect serial mouse.

The default /etc/inittab now runs getty on /dev/ttyS0. That means that getty will interact with a mouse if there's one on it. Its possible something strange is happening there. Turn on init DEBUG as described above and we can see exactly what's happening.

It is possible that the mouse interaction is an issue, but init shouldn't be running getty until AFTER the rc.sys script. So we need to see the debug output from init.

You can also set the default init level to 1 in elkscmd/rootfs_template/etc/inittab by changing the first line to 'id:1:initdefault', which will boot into level 1 rather than 3. This will keep the serial ports out of the picture.

pawosm-arm, 2020-04-25:
Nah, my suspicion was wrong. With nothing connected to serial port it didn't boot either. Also booting with null modem cable connected and minicom started on the other side did not help. Booting stuck after VFS and nothing appeared in minicom (set to 9600bps).
I'll try to build with DEBUG.

pawosm-arm, 2020-04-25:
This change did not reveal anything useful:
```
diff --git a/elkscmd/rootfs_template/etc/inittab b/elkscmd/rootfs_template/etc/inittab
index a4193446..18476c44 100644
--- a/elkscmd/rootfs_template/etc/inittab
+++ b/elkscmd/rootfs_template/etc/inittab
@@ -20,5 +20,5 @@ si::sysinit:/etc/rc.d/rc.sys
 t1:1356:respawn:/bin/getty /dev/tty1
 t2:56:respawn:/bin/getty /dev/tty2
 t3:56:respawn:/bin/getty /dev/tty3
-s0:2346:respawn:/bin/getty /dev/ttyS0 9600
-s1:46:respawn:/bin/getty /dev/ttyS1 9600
+# s0:2346:respawn:/bin/getty /dev/ttyS0 9600
+# s1:46:respawn:/bin/getty /dev/ttyS1 9600
diff --git a/elkscmd/sys_utils/init.c b/elkscmd/sys_utils/init.c
index 493dae45..fe6d8ea4 100644
--- a/elkscmd/sys_utils/init.c
+++ b/elkscmd/sys_utils/init.c
@@ -37,7 +37,7 @@
 #include <autoconf.h>          /* get CONFIG_CONSOLE_SERIAL*/
 
 #define USE_UTMP       0       /* =1 to use /var/run/utmp*/
-#define DEBUG          0       /* =1 for debug messages*/
+#define DEBUG          1       /* =1 for debug messages*/
 
 /* debug and sysinit/respawn sh -e output goes here*/
 #ifdef CONFIG_CONSOLE_SERIAL
```
No new debug messages, and it still hangs... The only way to boot ELKS now is to select serial console in the ELKS menuconfig.

ghaerr, 2020-04-25:
No change in debug messages sounds like init may not be able to open /dev/tty1.

Immediately after mounting VFS, /bin/init is run.

The first thing it does is open /dev/tty1 (or /dev/ttyS0 in the serial console case) and says "starting".

I suspect there may be an issue with the direct console code. Are you running a "make defconfig" .config file? Make sure you've removed .config and started from scratch.

Another option is to remove /bin/init from the image and boot. The system will then try to open /dev/tty1 and run /bin/sh. If that doesn't run, which I suspect it won't, then the problem is in the TTY open or console code. I'll take a look.

pawosm-arm, 2020-04-25:
I'm not using direct console, I'm using BIOS console (AFAIR, direct console caused its own problems last time I checked). And I'm using menuconfig (started by build.sh) to configure ELKS.
I also changed default runlevel as such:
```
diff --git a/elkscmd/rootfs_template/etc/inittab b/elkscmd/rootfs_template/etc/inittab
index a4193446..c6166857 100644
--- a/elkscmd/rootfs_template/etc/inittab
+++ b/elkscmd/rootfs_template/etc/inittab
@@ -1,4 +1,4 @@
-id:3:initdefault:
+id:1:initdefault:
 
 si::sysinit:/etc/rc.d/rc.sys
 
@@ -20,5 +20,5 @@ si::sysinit:/etc/rc.d/rc.sys
 t1:1356:respawn:/bin/getty /dev/tty1
 t2:56:respawn:/bin/getty /dev/tty2
 t3:56:respawn:/bin/getty /dev/tty3
-s0:2346:respawn:/bin/getty /dev/ttyS0 9600
-s1:46:respawn:/bin/getty /dev/ttyS1 9600
+# s0:2346:respawn:/bin/getty /dev/ttyS0 9600
+# s1:46:respawn:/bin/getty /dev/ttyS1 9600
```
Still no luck. More tests tomorrow.

ghaerr, 2020-04-25:
> I'm not using direct console, I'm using BIOS console

Ok, then this is definitely the problem. The enhancement to dynamically allocate the TTY queues did not update the BIOS console, as I thought it was broken. I'll submit a PR.

>  (AFAIR, direct console caused its own problems last time I checked). 

No, the direct console works great, AFAIK.

closed 2020-04-25 by ghaerr

pawosm-arm, 2020-04-26:
Some observations from today:
- BIOS console driver works ok now, booting finally completes.
- direct console works ok on Amstrad PC 2086; yet it's not the machine on which I've experienced glitches with dirtect console, it was experienced last year on the Turbo XT (the one with rare ne2k RJ45 ISA8 card), to which I don't have access nowadays due to the lockdown.
- the last two lines in /etc/inittab (serial console settings) are generally bad idea: it makes serial programs either hang (like ktcp) or complaining about device being in use (miniterm). Also when I left this machine turned on for somewhat more than a 0.5h, I found it dead when I returned, which didn't happen before this change was introduced
- similarly, I left 'mouse' test for 0.5h (not moving or clicking the mouse at that time) and when I returned, it was dead; yet I must admit, 'mouse' does not die so quickly as it used to
- yet still, nxlandmine dies after a few mouse motions
- I was able to use ktcp for a while at speed 300bps, I could telnet my Linux box on which my custom made telnet server was started.
- I was trying to use telnetd on ELKS, no luck despite the partch below:
```
diff --git a/elkscmd/inet/telnetd/Makefile b/elkscmd/inet/telnetd/Makefile
index 5de66cf4..203ea084 100644
--- a/elkscmd/inet/telnetd/Makefile
+++ b/elkscmd/inet/telnetd/Makefile
@@ -14,6 +14,7 @@ include $(BASEDIR)/Make.rules
 
 SRCS = telnetd.c
 OBJS = $(SRCS:.c=.o) 
+LDFLAGS += -maout-chmem=0xc000
 
 all:   telnetd
 
```
As soon as client gets connected either `ktcp` or `telnetd` (it's not clear which, but it's rather `telnetd`) complains:
```
sys_brk failed: brk 6ad8 > eneseg 6910
sys_brk failed: brk 6ade > endseg 6910
sys_brk failed: brk 695e > endseg 6910
```
so I didn't even get to the point where I could test connection stability.

ghaerr, 2020-04-26:
Hello @pawosm-arm, thanks for your continued comments and testing!

> * direct console works ok on Amstrad PC 2086; yet it's not the machine on which I've experienced glitches with dirtect console, it was experienced last year on the Turbo XT (the one with rare ne2k RJ45 ISA8 card), to which I don't have access nowadays due to the lockdown.

Keep me posted, I've been using and enhancing direct console with no issues.

On another note, I've noticed that BIOS and direct console should/could only differ in the way they output to the screen (direct vs int 10), instead of all of the ANSI sequence processing, which should be shared. Currently the BIOS driver has fallen behind in the duplicated code required for the ANSI processing. I was thinking of combining the two drivers into one, with just the low-level output method being config-selected.

> * the last two lines in /etc/inittab (serial console settings) are generally bad idea: it makes serial programs either hang (like ktcp) or complaining about device being in use (miniterm).

Good point. It was left on so that serial ports would get tested, and comments are coming in. 
After we've finished up with the serial testing, moving the init default to level 1, which would turn off serial getty's would solve this. After we get through the next round of serial testing without dropping data, serial testing the network utilities can begin in earnest.

> Also when I left this machine turned on for somewhat more than a 0.5h, I found it dead when I returned, which didn't happen before this change was introduced

There's been no fundamental changes to hardware or IRQ processing, except FIFO management. 

Perhaps try running `init 1` to switch to single-user non-serial mode, then let the system sit. 

> * similarly, I left 'mouse' test for 0.5h (not moving or clicking the mouse at that time) and when I returned, it was dead; yet I must admit, 'mouse' does not die so quickly as it used to

Strange. But then the system appears to die when not running mouse also.

> * yet still, nxlandmine dies after a few mouse motions

That sounds like the original problem... not fixed.

> * I was able to use ktcp for a while at speed 300bps, I could telnet my Linux box on which my custom made telnet server was started.

300 baud? Is that all? Geez. Was this not possible before?

I have made extensive changes in VMIN/VTIME that should definitely help SLIP processing, but these changes have not been introduced to `ktcp` yet. Should I add them for your testing? I also plan on adding the dash options you requested so that `stty` doesn't have to be run separately.

A current problem is that I haven't yet setup to run QEMU with SLIP to a network stack, so I can't test any networking yet.

> * I was trying to use telnetd on ELKS, no luck despite the partch below:

Patch looks correct.
> As soon as client gets connected either `ktcp` or `telnetd` (it's not clear which, but it's rather `telnetd`) complains:

I'm wondering if that sys_brk message is coming from /bin/sh that telnetd execs...
Adding a process number to the kernel message, and then checking with `ps` could verify that.

Are you saying that the message could possibly be coming from `ktcp`? That seems more likely. Perhaps add the chmem link option to ktcp also.

pawosm-arm, 2020-04-26:
> Currently the BIOS driver has fallen behind in the duplicated code required for the ANSI processing. I was thinking of combining the two drivers into one, with just the low-level output method being config-selected.

Every duplication is a source of evil...

> 300 baud? Is that all? Geez. Was this not possible before?

Not for that many lines of text sent back and forth... Yes it is slow, but at least it's more or less stable.

> I have made extensive changes in VMIN/VTIME that should definitely help SLIP processing, but these changes have not been introduced to ktcp yet. Should I add them for your testing? I also plan on adding the dash options you requested so that stty doesn't have to be run separately.

Whenever you're ready. Both changes are welcomed.

> Are you saying that the message could possibly be coming from ktcp? That seems more likely. Perhaps add the chmem link option to ktcp also.

After you added `sh` to the picture, I'm not that sure about that. `sys_brk` should definitely print affected pids.

Mellvik, 2020-04-26:
@ghaerr, @pawosm-arm,
I'd like to throw a couple of comments into this interesting and useful exchange
the last two lines in /etc/inittab (serial console settings) are generally bad idea: it makes serial programs either hang (like ktcp) or complaining about device being in use (miniterm).

> Good point. It was left on so that serial ports would get tested, and comments are coming in.

From my perspective, this is a bug. Getty being open on a serial device should not block others processes from taking over, more or less suspending getty. After spawning login, the open should be exclusive. @ghaerr, I guess this is somewhat related to the minor device # issue we're discussing in a different thread.

As to uptime, FWIW - I've had a physical system (with active login on a serial line) running for 4 days plus. Not much load, but sporadic use. Actually, I'm missing the uptime command right now.

-M

ghaerr, 2020-04-26:
> From my perspective, this is a bug. Getty being open on a serial device should not block others processes from taking over

Hold on a minute here! A few weeks ago it was mentioned that multiple processes should be able to open a serial port. So the system was changed from exclusive-only access to allowing any process to open the serial port, even when another process has it already open, like `getty`.

Then, O_EXCL was added as a result of a user being confused by running a mouse-based program, which was granted access and didn't work since getty was also reading serial data, due to the new policy. Communication programs like `miniterm`, nano-X mouse-based programs, and `mouse` now use the O_EXCL flag on `open` so they know they are the exclusive user, or are given an appropriate error message (Device busy).

Are we saying that is a bug? Which brings us back full circle. 

> should not block others processes from taking over, more or less suspending getty. 

Getty is already suspended waiting for a `read` on the port, it can't be 'suspended' while another process also reads on the same port. The kernel wakes up all processes waiting for a read, and we can't change that mechanism.

> After spawning login, the open should be exclusive.

Neither getty nor login open the port, init does. And the mouse problem remains with getty running with non-exclusive use.

If getty uses O_EXCL, then we're back full circle to only allowing a single process to ever open a port.

I still think the best answer is to have the system work the way it works now, but have the default boot run level be '1', which is solves the problem brought up by @pawosm-arm. Then, the sysadmin can run mouse programs without special knowledge, or, turn on a multi-user mode by issuing an 'init [3456]'. I've only left the init live to spawn serial ports to ease with testing, especially since most users don't follow what we're doing that closely.

> this is somewhat related to the minor device # issue we're discussing in a different thread.

Yes, but more device names for the same port will make things even more complicated.

ghaerr, 2020-04-26:
> it makes serial programs either hang (like ktcp) or complaining about device being in use (miniterm).

I am about to make @pawosm-arm's requested changes to `ktcp`, which will include setting the baud rate on the command line, and also adding the VMIN/VTIME changes that should allow it to run at much higher baud rates with the new serial receiver, FIFO, and TTY driver throughput changes. The device open will also have O_EXCL added, so that `ktcp` is guaranteed an exclusive serial port. 

Since ktcp is run at sys.rc time, another possibility is to go full circle and have getty also request O_EXCL. This would cause those getty's to open fail when ktcp is running, and init would then shut them down after a few tries. However, with O_EXCL set, a redirected `stty < /dev/ttyS0` may not work, that would need to be tested.

Mellvik, 2020-04-27:
@ghaerr,
In all honesty I don't understand your response. As far as I can tell, we're saying exactly the same thing.
> ﻿
> From my perspective, this is a bug. Getty being open on a serial device should not block others processes from taking over
>
> Hold on a minute here! A few weeks ago it was mentioned that multiple [...]
No, this is the way it's supposed to work, but doesn't according to @pawosm-arm. IMHO the mentioned lines in inittab should not create problems for anyone if things work right.

> should not block others processes from taking over, more or less suspending getty.
>
> Getty is already suspended waiting for a read on the port, it can't be 'suspended' while another process also [...]
I must be missing the point here. I was pointing out that the way the system works is indeed the way it should work, and that the observed problem must be considered a bug.
> Then, the sysadmin can run mouse programs without special knowledge, or, turn on a multi-user mode by issuing an 'init [3456]'. I've only left the init live to spawn serial ports to ease with [...]
IMO - This is not really about more device names but a clean handling of different modes. And I don't find any of this really complicated. That said, we seem to be using different tribe languages at times, which is quite normal - if confusing. We can handle that.
> 

My point when flagging this as possible bug instead of a problem, was that a live getty on a serial line (non exclusive) should be blocked by any other (root) process asking for exclusive access. And then ensuring that `stty < line ` still works. The way I'm reading you, you have the same opinion. 

--Mellvik

ghaerr, 2020-04-27:
Hi Helge, thanks for your response.

> This is not really about more device names but a clean handling of different modes.

Agreed. 

>  And I don't find any of this really complicated.

I think I was writing my response while head deep in other code... making things more complicated then they are. 

>  a live getty on a serial line (non exclusive) should be blocked by any other (root) process asking for exclusive access.

I'm not sure how to to block a process in a read (yet)... but we'll figure something out!

Thanks again for your comments and testing, I feel we're making progress :)

PR #588 (2020-04-25): [kernel] Fix BIOS console driver
Fixes #586.