ghaerr/elks #595: Grep stack overflow

ghaerr, 2020-04-27:
@ghaerr -
I'm still having trouble with grep, although I suspect the problem is somewhere else.
Have a look at the attached screen shot.
After reboot, group works. Then - after `time cat /bin/* > /dev/null`  grep bombs the system, either rebooting or reporting a segmentation fault.

--Mellvik 

[image]

_Originally posted by @Mellvik in https://github.com/jbruchon/elks/pull/582#issuecomment-620014267_

ghaerr, 2020-04-27:
I can't get this to repeat, trying both grep and cat.

In looking at your screenshot, the first grep problem occurred after running time. Can you remember what other commands were executed before that? I see `time` was executed. Nothing after the first stack overflow matters, as the system is corrupted.

Mellvik, 2020-04-28:
@ghaerr, try this one, this is on VB - and entirely complete, the first login prompt is from boot:
[image]

ghaerr, 2020-04-28:
Perfect, its repeatable. I've got it down to running just three commands:

grep...
cat...
grep...

It doesn't repeat on sh, only sash.
I'm guessing, grep, sash wildcard expansion, or sash history.
I tried turning sash history off in /etc/sashrc and it remains on.

Lets continue to investigate.

Mellvik, 2020-04-28:
@ghaerr,
If you remove HISTORY from the rc file, it should be turned off. 
I'll change that so that #0 does the same.

I believe I got the problem with grep using sh too, need to check again.

--M

ghaerr, 2020-04-28:
> If you remove HISTORY from the rc file, it should be turned off.

I commented out both 'setenv' statements in /etc/sashrc, history still on.

> I'll change that so that #0 does the same.

I'm sorry, what is that?

> I believe I got the problem with grep using sh too, need to check again.

Good news, that will get `sash` out of the picture. I looked at grep quickly, no large buffers, could be something else.

closed 2020-04-29 by ghaerr

Mellvik, 2020-04-29:
Great - thanks.

I have a list of fixes for sash for a PR tomorrow. Turns out sash didn't read the rc file until after history init.

-Mellvik

PR #598 (2020-04-29): [cmds] Fix stack overflow in sash grep
Fix stack overflow in internal `sash` grep command (Fixes #595).
Traced down needlessly-large 8K stack allocation in grep, which combined with
a 2.5K heap and overly-large 2K stdio allocation on fopen, caused stack overflow.
ELKS kernel acted normally and caught the stack overflow error, which is
checked during every interrupt, and terminated sash with SIGSEGV.

Various other fixups, found during bug trace:

Match grep 'fgets' buffer size to same as stdio BUFSIZ.
Reduce stdio BUFSIZ from 2048 to 1024.
Add DSEG in `ps` output. (`ps -m` option coming).