ghaerr/elks #597: KDSKBMODE ioctl?

cjsthompson, 2020-04-28:
I'm trying to do both vi style commands for 'more' like @Mellvik would like for 1G but also have Page Down, Home, End work which are translated to escape sequences by the terminal. So far I can have one or the other but not both. After having tried all sorts of schemes involving tcsetattr to get both to work at the same time I'm beginning to think that the only way to get this to work is to have the keyboard in raw mode so Page Down & friends won't be translated to escape sequences anymore and therefore when ESC is pressed I can just go just go straight to vi mode without attempting to detect whether this is a lone ESC for entering vi mode or an escape sequence for some keyboard key. Linux has a KDSKBMODE ioctl to set the keyboard in raw mode, but can't find this in ELKS source. Is there a different way to do it?

ghaerr, 2020-04-28:
> Is there a different way to do it?

Yes. We definitely don't want to write `more` to be dependent on particular keyboard scancodes.

I assume the whole problem is just trying to differentiate between an ESC pressed alone, vs an ESC that is part of an escape sequence?

That problem is solved using recently-implemented termios VMIN an VTIME in ELKS. Here's an informative article: http://unixwiz.net/techtips/termios-vmin-vtime.html

Bottom line is you'll want to set VMIN to 1 and VTIME to 2, and then issue a `read` for at least as long as the longest escape sequence bytes you might process. VMIN 1 says wait indefinitely for 1 character, and AFTER the first character has been received into the TTY queue, wait up to 200 msecs for each subsequent character,  (which will be the ESC sequence remainder, if any). Then, if read returns 1, ESC alone was pressed, otherwise its an escape sequence (you have to check for ESC of course).

Look at the code in kilo.c for handling, it uses the same method. (However, I just noticed that it is setting VMIN to 0, and VTIME to 2, which is improper. That will cause kilo to 'poll' the keyboard at 200 ms intervals, which wastes CPU time. I will be submitting a PR to change VMIN to 1 in kilo.c).

cjsthompson, 2020-04-28:
Thanks for the explanation and the link. I see how I can make that work now. And it even solves another problem where I wasn't sure of the length of the escape sequence and adding more fgetc()'s with VMIN set to 1 so it won't block.

cjsthompson, 2020-05-01:
I've been beating my head trying to get this to work with Linux, but I suspect that Linux serial stuff is either broken or behaves differently for some reason. If you set VMIN to 1 and VTIME to 2 in Linux it will block indefinitely and the VTIME timer will not expire. I think that's why Kilo sets VMIN to 0 and then commented out a statement setting VMIN to 1 because he was trying to get it to work as I have been. I'm going to test in FreeBSD see how it goes there.

ghaerr, 2020-05-01:
> but I suspect that Linux serial stuff is either broken or behaves differently 

I doubt it's broken, but I just implemented VMIN/VTIME over a few commits this past few weeks. So there could be issues.

> If you set VMIN to 1 and VTIME to 2 in Linux it will block indefinitely and the VTIME timer will not expire.

You mean after reading one character, it won't expire after 200 msecs waiting for a second character? Well, that sounds buggy, at least from the UNIX/Linux TTY specifications. ELKS should work correctly, does it? We are using 

> I think that's why Kilo sets VMIN to 0 and then commented out a statement setting VMIN to 1 because he was trying to get it to work as I have been.

Actually I commented out the second version, but that was before I got VMIN/VTIME working (that is, if it is working properly). I would like to move kilo on ELKS to using VMIN = 1, VMAX = 2, but I haven't checked to see that kilo doesn't actually need to poll. I think it just busy-loops in the read statement on line 248.

Here is the original kilo source, if you would like to test it running on Linux. https://github.com/antirez/kilo

cjsthompson, 2020-05-01:
It mostlikely works in ELKS. I was strictly speaking about Linux. Things tend to bitrot in Linux when the devs don't actively use it.

ghaerr, 2020-05-12:
Hi Christoph,

Does this issue need to remain open?

I have checked and VMIN/VTIME definitely work correctly on ELKS now.

With regards to kilo, its code is not as straightforward as I thought to show best use of programmatically separating a single ESC from an arrow key ESC sequence. The linenoise_elks.c file in `ash` does a better job.

cjsthompson, 2020-05-12:
Hi,

No, you can close it. It's a Linux problem anyway.

closed 2020-05-12 by ghaerr