ghaerr/elks #600: Enhancement: add new miniterm option '-l - add LF after every CR received'

pawosm-arm, 2020-04-30:
This option is present in most of serial port terminal programs and I've found it extremely useful (e.g. when sending copy-pasted blocks of text, LF's are never sent to the other side, so the receiving side should add LF after every CR to prevent screen being messy). It would be great to have it in `miniterm` too. The logic is simple. With this option specified: for every 'CR' received, cursor should go to the next line (as if LF was received) before returning to the beginning of the line (which is the result of CR).

ghaerr, 2020-04-30:
Hello @pawosm-arm,

Would you like to take a pass at implementing this, since you'll also be able to test it?

pawosm-arm, 2020-04-30:
Sure, I'll look into it.

pawosm-arm, 2020-04-30:
Putting this aside, I could for a first time observe how miniterm is built:

```
miniterm.c: In function 'main':
miniterm.c:389:27: warning: variable 'enable_dtr' set but not used [-Wunused-but-set-variable]
  bool enable_rts = false, enable_dtr = false;
                           ^~~~~~~~~~
miniterm.c:389:7: warning: variable 'enable_rts' set but not used [-Wunused-but-set-variable]
  bool enable_rts = false, enable_dtr = false;
       ^~~~~~~~~~
```
Looks like miniterm does nothing about RTS and DTR...

ghaerr, 2020-04-30:
> Looks like miniterm does nothing about RTS and DTR...

Correct, those options are controlled by -d and -R. The plan is to use the -r option to turn on hw flow control by unsetting CLOCAL and setting CRTSCTS. The serial driver will then enable DTR and use the RTS line to control incoming data.

None of the options have ever been operational in ELKS, so -r currently does nothing as well.

pawosm-arm, 2020-05-01:
merged.

closed 2020-05-01 by pawosm-arm