ghaerr/elks #608: Misc bugs - ps(), sash(), ash(), tail()

Mellvik, 2020-05-03:
Elvis (and probably there interactive programs)  cannot spawn subshells when stared from sash (reported before, will be fixed). Just as interesting, after attempting such a spawn (e.g. `:!ls`), graces of the attempt lingers in ps. More attempts adds more entries to ps with the same data. Eventually ps() prints lots of garbage - seemingly an array out of bounds type problem. See screenshots. The situation does not affect system stability.

Not related, but ash also fails when programs are forking subshells. While sash complains about '-c not found', which is easy to explain, ash simply reports 'fork failed'.

Finally, tail has a memory (sbrk) problem, try `hd /bin/vi | tail -100`. 
All testing i qemu, build from d5b0d45f.

--Mellvik

[image]

Mellvik, 2020-05-03:
Update:
- when staring with ash as login shell, the behaviour is slightly different:
- forking from elvis via ash works (output somewhat garbled since the console is still in raw mode, it's on my list of elvis problems).
- but after such fork (repeated forks work, even :!ps), and exit from elvis, ps() prints garbage.

ghaerr, 2020-05-03:
hello @Mellvik, there are a number of separate issues in the items you bring up. 

> after attempting such a spawn (e.g. :!ls), graces of the attempt lingers in ps. 

Note the processes are in Zombie state. Their task structure entry will only be cleaned up after a 'wait' command. This usually means after the shell as executed the next command. sh and sash have entirely different methods for waiting, which differs in sash depending on whether it was an internal command, had to run a /bin program, or had to run /bin/sh -c for the command line.
If the parent is dead (subshell, etc), then `init` performs the wait. It's a bit complicated.

More attempts adds more entries to ps with the same data.

These are all Zombie states, waiting to be cleaned up.

> Eventually ps() prints lots of garbage - seemingly an array out of bounds type problem. The situation does not affect system stability.

There are only 15 process slots available, was this what you saw? I don't see the garbage output, except for the SIZE obviously incorrect. This is likely due to the task resources having been released or reallocated. `ps` tries to figure all this out by reading /dev/kmem, so there are limits in what it does.

>  but ash also fails when programs are forking subshells. While sash complains about '-c not found', which is easy to explain, ash simply reports 'fork failed'.

I don't understand the ash issue: is ash failing to fork a subshell? When?

> forking from elvis via ash works (output somewhat garbled since the console is still in raw mode, it's on my list of elvis problems).

Remember that since the linenoise enhancement, ash runs in raw mode. If this happens only when elvis is running on ash, this is not an elvis problem. The linenoise issue complicated. Setting TERM=dumb in ash turns off linenoise.

> but after such fork (repeated forks work, even :!ps), and exit from elvis, ps() prints garbage.

Screenshot please

>  tail has a memory (sbrk) problem, try hd /bin/vi | tail -100.

Try running chmem +XXX /bin/tail to see whether this is a heap allocation issue or something else.

Mellvik, 2020-05-03:
> Eventually ps() prints lots of garbage - seemingly an array out of bounds type problem. The situation does not affect system stability.
>
> There are only 15 process slots available, was this what you [...]
No, what I get is many screenfulls og random data.
> but ash also fails when programs are forking subshells. While sash complains about '-c not found', which is easy to explain, ash simply reports 'fork failed'.
> 
> I don't understand the ash issue: is ash failing to fork a subshell? When?
> 
hd /bin/vi | head -100 > /tmp/xx; vi /tmp/xx
Then :!ls or any other command 
> forking from elvis via ash works (output somewhat garbled since the console is still in raw mode, it's on my list of elvis problems).
> 
Agree, turns out this may vary with the content of elvis, maybe something else. Will test more.
> Remember that since the linenoise enhancement, ash runs in raw mode. If this happens only when elvis is running on ash, this is not an elvis problem. The linenoise issue complicated. Setting TERM=dumb in ash turns off linenoise.
> 
My suspicion was never elvis, but ash...
> but after such fork (repeated forks work, even :!ps), and exit from elvis, ps() prints garbage.
> 
> Screenshot please
> 
Sceenshot is of noe value, just garbage.
> tail has a memory (sbrk) problem, try hd /bin/vi | tail -100.
> 
> Try running chmem +XXX /bin/tail to see whether this is a heap allocation issue or something else.
> 
Tried +2048, no change on this one ... the sample command is repeatable ...

--Mellvik

ghaerr, 2020-05-03:
> Tried +2048, no change on this one ... the sample command is repeatable ...

A quick look at minix3/tail.c shows a call to malloc for 20k bytes. Since the standard heap and stack size combined is 16K default, `tail` will get an `sbrk` error on startup. Congratulations - it seems you're the first person (again, lol) to run that program!

`tail` assumes a 10-line buffer (probably too small), and a 2048 byte line size (ridiculous). These should default 10 say 25 lines and 80 bytes, respectively, IMO. This would allow it to run under many circumstances.

However, any program with the potential to allocate memory should probably be given the chance for a decent heap size. For instance, `sort` is given a 48k heap (and stack). ELKS will allow a heap to come within 4K of the stack, thus, the max heap on default ELKS programs is effectively 12k.

When playing with `chmem`, start high (e.g. +40000) and work downwards. The +2048 adjustment had no effect because of the 20k heap requirement.

Would you like to submit a PR to fix tail?

ghaerr, 2020-05-03:
> No, what I get is many screenfulls og random data.

I think I understand now - `ps` is reporting on a Zombie process, and it's been only luck that the memory segments that WERE allocated have not been reused... and the de-allocated data segment is where `ps` gets the process command line (by looking just below the previous top of stack, where argc/argv/envp are located).

Thus, we have a couple choices - leave as is (since it is convenient in the case of little ELKS activity to see Zombie process command lines), remove the command line for zombie states (definite fix but then we lose good information), or stop the display of command line information after column 80 or so (it will be garbage but contained). 

Which option do you like?

ghaerr, 2020-05-03:
> forking from elvis via ash works (output somewhat garbled since the console is still in raw mode, it's on my list of elvis problems).
Agree, turns out this may vary with the content of elvis, maybe something else. Will test more.

I think this can be fixed in ash by not setting raw mode when processing script files or being invoked during -c operation. I will add this to my fix list.

>  but ash also fails when programs are forking subshells. While sash complains about '-c not found', which is easy to explain, ash simply reports 'fork failed'.
>  I don't understand the ash issue: is ash failing to fork a subshell? When?
hd /bin/vi | head -100 > /tmp/xx; vi /tmp/xx
Then :!ls or any other command

Is this supposed to cause a fork/exec failure? I can't get that to repeat, using that exact sequence.

Mellvik, 2020-05-04:
> but ash also fails when programs are forking subshells. While sash complains about '-c not found', which is easy to explain, ash simply reports 'fork failed'.
> I don't understand the ash issue: is [...]
This turns out to be an out of memory situation, ash simply reports 'Cannot fork'. And this sequence does not repeat the situation unless memory is otherwise spent.

BTW - in order to save memory, wouldn't it make sense to have 'sh -c' exec the subcommand instead of having the shell (uselessly) stick around in memory?

—M

closed 2020-05-04 by Mellvik

ghaerr, 2020-05-04:
> in order to save memory, wouldn't it make sense to have 'sh -c' exec the subcommand instead of having the shell (uselessly) stick around in memory?

`sash` will exec the subcommand, unless it's a builtin. Follow the procedure call chain command() -> trybuiltin() -> runcmd() -> fork() -> exec(). Also, command() has to be called to parse the -c single argument string, as execl and execv require each argument separately.

Mellvik, 2020-05-04:
> ﻿
> in order to save memory, wouldn't it make sense to have 'sh -c' exec the subcommand instead of having the shell (uselessly) stick around in memory?
>
> sash will exec the subcommand, unless it's a [...]
Yes, I was following the path and it didn't match with the ps() I got from within vi:

:!ps
[Press <RETURN> to continue]  PID   GRP  TTY USER STAT CSEG DSEG  HEAP   FREE   SIZE COMMAND
   1     0      root    S 3e46 3ff0  3072  13252  26800 /bin/init 
  16    16      root    S 4dc4 4f61     0  16275  27344 /bin/getty /dev/tty1 
  17    17   S0 root    S 7600 81dd  1167  14940  75648 -/bin/sh 
  30    17   S0 root    S 8878 5471  3072  12958 112096 vi 
  31    17   S0 root    S 7600 97b3  1025  15066  75648 /bin/sh -c ps 
  32    17   S0 root    R 9e4e 6094  1025  14129  26672 ps 

If 31 was a zombie, that would explain it. But it isn't...

-M

ghaerr, 2020-05-04:
Helge - 

Process 30, vi, is requested to run ls. So vi executes system(), which forks, creating process 31, and execs "'/bin/sash' '-c' 'ps'". Sash, process 31, looks at its single argument following -c, deconstructs it into argc/argv, and forks, creating process 32, and then execs /bin/ps. So everything is working perfectly.

Are you suggesting that vi should fork/exec 'ps' directly? It can't do that easily because the user may have type ":!ls -l / | tail -100". That requires a shell to interpret, so the system(3) libc routine always runs /bin/sh -c instead. (actually it is replaced with its own version in elvis, and it also uses the SHELL variable to determine the shell to run).

You will notice that running ":!ls" on sash will operate with one less process, and run ls internally.

Does this explanation help?

Mellvik, 2020-05-05:
Thanks for your through walkthru @ghaerr.

I'm curious, and not really suggesting we do anything differently. Hitting the memory-wall (or rather, the ceiling) all too frequently in ELKS these days, looking for places to save space becomes a (sometimes bad) habit.

What rose my curiosity here, partly because previous knowledge of the issue is buried (or lost) in my own memory, was the apparent uselessness of the shell process lingering after the new process(es) were launched. My real question was why the shell not simply exited after creating the requested process(es).

BTW - my 'clip' from elvis executing ps demonstrated another (minor) problem with elvis: A mixture of printfs and fprintfs some times cause output out of sequence - such as in this case where the 'Press return to continue' comes before the actual output. 

—Mellvik

ghaerr, 2020-05-05:
I think I now understand more of your original thought - why a shell run "-c" doesn't just exec its subcommand (argument). And my explanation above was correct, except for a simple case in the sash shell only. Let me answer your other question first:

> was the apparent uselessness of the shell process lingering after the new process(es) were launched. My real question was why the shell not simply exited after creating the requested process(es).

A shell needs to hang around for a couple reasons: it needs to wait() for the child process to complete, so that other processes don't continue out of order, and also needs to pass the exit status back to the parent process. Since the shell has to fork first, there is no savings in memory and the shell needs to wait for the child exit status.

> A mixture of printfs and fprintfs some times cause output out of sequence - such as in this case where the 'Press return to continue' comes before the actual output.

That out of sequence output was a bug, and should be fixed (is it?). The reason was previously explained that the `wait` elvis executed was buggy and returned before the child process chain completed running.

> In order to save memory, wouldn't it make sense to have 'sh -c' exec the subcommand instead of having the shell (uselessly) stick around in memory?

The usual reason is that the subcommand could be complex: multiple piped commands, async execution, etc that require the shell to wait for completion, OR that `sh` is run instead for seamless integration. However, since `sash` does not allow complex pipes, redirection, etc, I now see where sash doesn't need to fork and can just exec its -c argument. So your initial thought on this is possible. With the "seamless ash integration" (which allows elvis to do ":!ls -l / | tail" for instance, sash would have to do more computation to determine whether or not to fork. I can look more into this if you think its worth it.

Mellvik, 2020-05-05:
Thanks again, @ghaerr - 

> A mixture of printfs and fprintfs some times cause output out of sequence - such as in this case where the 'Press return to continue' comes before the actual output.
>
> That out of sequence output was [...]
This was out of context - but I figured it should be on someone else's bulkiest in addition to min.
> In order to save memory, wouldn't it make sense to have 'sh -c' exec the subcommand instead of having the shell (uselessly) stick around in memory?
>
> The usual reason is that the subcommand could be [...]
I'm unsure about this - it entirely depends on the amount of work required. Right now - given the memory constraints when ktcp is active, any memory improvement is good. Then again, I tend to use ash for that particular reason.

While on sash, I guess it would make sense to have sash check if the first argument (if any, and not -c) is a text file (script) or a binary. Right now it just pretends /bin/ls (for example) is a script and runs it. Interesting results. BSD csh ignores the argument (returns the prompt) if it's not a text file.

-M

ghaerr, 2020-05-05:
> Right now - given the memory constraints when ktcp is active, any memory improvement is good

I'll decrease the default L2 buffer size, that will help greatly.

> While on sash, I guess it would make sense to have sash check if the first argument (if any, and not -c) is a text file (script) or a binary. 

The problem of shell scripting and recognizing shell scripts is far more complicated than that. I have a plan for it but its complicated and needs to start with ash and the kernel, not sash. Sash isn't built for scripting and will exec what its given, except for the ash integration. Its a stand alone shell, useful for very slow or very small systems, and should probably just stay that way.

That said - I'll look into sash not forking in certain cases, for the rare but useful case of needing to run applications with not much memory available.

Mellvik, 2020-05-05:
Thanks - 
IMHO this must be low priority.

-M