ghaerr/elks #612: Regressions fail on serial and kbd drivers

ghaerr, 2020-05-05:
Moved from #610:

- Some time over the last week a change has badly affected the XT keyboard driver. The keyLEDs (capslock, numlock etc. no longer work, keypad send wrong characters).
- In the same timeframe a change in the serial driver (presumably) has made serial access non functional: Output is 'eating' most of the data, less than half of the bytes sent get to the terminal window. Input seems ok but it's hard to tell when output is incomplete,.

More testing on this tomorrow - @ghaerr, if you have any immediate guesses, I'd appreciate it.

ghaerr, 2020-05-06:
> Some time over the last week a change has badly affected the XT keyboard driver. The keyLEDs (capslock, numlock etc. no longer work, keypad send wrong characters).

Are you running direct console or BIOS console? CONFIG_CONSOLE_DIRECT vs _BIOS. I have noticed the BIOS console keyboard driver has problems. Check .config. You might want to rm .config and start over with make defconfig, then make menuconfig.

Did this just start when you updated the source tree, or did it happen in your initial testing of the ethernet, with a changed network card?

Can you check operation without the network card inserted? Just want to make sure there's no hardware issues with it and the keyboard or serial port.

I notice on your second screenshot in network testing you were booting from commit af376b67. Was the serial port and keyboard working on that commit? Any idea what your last working commit was?

On Apr 22, we added the new FIFO detect code, and I also made the change to init the FIFO with UART_FCR_ENABLE_FIFO14 rather than UART_FCR_ENABLE_FIFO8, in rs_open(). You might try changing that back if pulling the network card out doesn't change anything.

The XT keyboard code (only used in Direct Console) was last changed on Apr 19 with the configurable IRQ port.h changes... that was quite a while ago.

ghaerr, 2020-05-06:
Helge, another thought is to recompile the kernel without networking enabled (reset .config to the way it was before you started network testing), just to see whether that has an effect on the serial/kbd problems. There aren't really any big changes to the kbd or serial in the last week, so its possible a new configuration you've selected could also be causing interrupt or other hardware conflicts.

Mellvik, 2020-05-06:
> Some time over the last week a change has badly affected the XT keyboard driver. The keyLEDs (capslock, numlock etc. no longer work, keypad send wrong characters).
>
> Are you running direct console or [...]
I'm using BIOS console.
> Did this just start when you updated the source tree, or did it happen in your initial testing of the ethernet, with a changed network card?
> 
The KBD LED problem has been there for a while - possibly as far back as april 19th, as you suspect. The arrow key problem is more recent. Easy to notice w/ash linenoise.
> Can you check operation without the network card inserted? Just want to make sure there's no hardware issues with it and the keyboard or serial port.
>
> I notice on your second screenshot in network [...]
In this commit serial was fine, KBD was not (LEDs), keypad probably OK.
> On Apr 22, we added the new FIFO detect code, and I also made the change to init the FIFO with UART_FCR_ENABLE_FIFO14 rather than UART_FCR_ENABLE_FIFO8, in rs_open(). You might try changing that back if pulling the network card out doesn't change anything.
> 
I've moved serial connection to the builtin port, it's a 16540, so FIFO is off the table for now. Also, like I poited out before, input seems OK, it is output that fails.
> The XT keyboard code (only used in Direct Console) was last changed on Apr 19 with the configurable IRQ port.h changes... that was quite a while ago.
> 
Still possible - I'll look at that. 
> 
--Mellvik

ghaerr, 2020-05-06:
Try using Direct.console. There are a number of issues with BIOS console, the least of which it does not handle ANSI emulation correctly, and has had none of the updates made for vi to run bug-free.

It sounds like these "new" bugs may have been happening for some time. None of the keyboard arrow key and other enhancements required for serial port use of vi were added to BIOS console, so linenoise won't run properly. Other changes have been made to linenoise that removed hardware-specific key codes, which could be part of the issues here. Applications should not be coded with hard-code scan codes, that was changed a three weeks ago, but only implemented in direct console to emit ANSI kbd sequences.

All of this can be fixed when BIOS and direct console are merged - part of the plan but lower priority, and yet another result of originally duplicated code.

> I've moved serial connection to the builtin port, it's a 16540, so FIFO is off the table for now. Also, like I poited out before, input seems OK, it is output that fails.

I get that, It's not off the table because the new code writes to a FIFO register that may have bad results if the chip detect isn't working properly, so changing the serial port hw is adding yet another variable to debugging this, and could in fact be part of the problem.

I suggest putting the hardware, including serial and network cards, back to the way they were when the system was working, to see whether those are issues or not. Finally, use direct console not BIOS console. Bugs in BIOS console, possibly all of these, can be put in a new issue. I have done almost no testing with BIOS console.

> In this commit serial was fine, KBD was not (LEDs), keypad probably OK.

FYI that commit was the latest commit to ELKS, so sounds like possible serial hardware or other issues caused by changing hardware.

Mellvik, 2020-05-06:
@ghaerr,

the serial problem has been resolved - turned out to be a software update on the 'other side of the line', a raspberrypi, which caused getty to become activated, so it ate most of the output from elks.

The console kbd is more of a problem. Changing to direct console made no difference and I've spent some time with bisect, but cannot get older commits to compile. For various reasons depending on age. 

The last one is from mid april, and stops @:

gcc -I/home/helge/src/elks/include -I/home/helge/src/elks/elks/include -o mkbootloader mkbootloader.c
make[3]: Leaving directory '/home/helge/src/elks/elks/arch/i86/tools'
ia16-elf-as -mtune=i8086  -o boot/bootsect.o boot/bootsect.s
ia16-elf-ld  -s  -M -T /home/helge/src/elks/elks/elks-tiny.ld -o boot/bootsect boot/bootsect.o > boot/bootsect.map
ia16-elf-ld:/home/helge/src/elks/elks/elks-tiny.ld:14: undefined symbol `_start' referenced in expression
Makefile:66: recipe for target 'boot/bootsect' failed
make[2]: *** [boot/bootsect] Error 1
make[2]: Leaving directory '/home/helge/src/elks/elks/arch/i86'
Makefile:73: recipe for target 'Image' failed
make[1]: *** [Image] Error 2
make[1]: Leaving directory '/home/helge/src/elks/elks'
Makefile:11: recipe for target 'all' failed
make: *** [all] Error 2

ahead of this is a rm .config; make clean; make

Any hints?

BTW - I have verified that this is not a hardware issue. The symptoms are the same on two machines, and not present in DOS.

—Mellvik

> 
> In this commit serial was fine, KBD was not (LEDs), keypad probably OK.
> 
> FYI that commit was the latest commit to ELKS, so sounds like possible serial hardware or other issues caused by changing hardware.
> 
> 
>

ghaerr, 2020-05-06:
> the serial problem has been resolved - turned out to be a software update on the 'other side of the line', a raspberrypi, which caused getty to become activated, so it ate most of the output from elks.

That's good to know, you had me worried there for a bit! And sounds like Pi is having the same multiple getty problem! :)

> Changing to direct console made no difference

That I do not understand, because the direct console doesn't attempt to modify the kbd LEDs etc at all. Be certain that you're running Direct vs BIOS, it is displayed as the first kernel message at boot.

Perhaps your git repo is screwed up, I'm not a git bisect expert. Perhaps do a new git clone in another directory and try compiling the latest after setting console direct and verifying it in .config.  The direct console code almost certainly works, although it does replace IRQ1 so make sure your network card is removed for the test, just incase that is an issue.

> ahead of this is a rm .config; make clean; make

Perhaps run make defconfig; make menuconfig ahead of make.

Mellvik, 2020-05-06:
> 
> Changing to direct console made no difference
> 
> That I do not understand, because the direct console doesn't attempt to modify the kbd LEDs etc at all. Be certain that you're running Direct vs BIOS, it is displayed as the first kernel message at boot.
> 
> 
I'm taking a new stab at this with a fresh repo as suggested. However, I get the feeling that the problem is somewhere else, and what we're seeing is an unintended consequence of a change elsewhere. So let me ask you this: As elks boots, it displays various boot messages. Where in this sequence is the console is being initialized? I'm getting response from the keyboard lights until somewhere in the middle of the list of messages. Having slow HW, this may give a clue …

> Perhaps your git repo is screwed up, I'm not a git bisect expert. Perhaps do a new git clone in another directory and try compiling the latest after setting console direct and verifying it in [...]
Yes, I've run tests without the Compaq expansion box, the dual slot ISA box that attaches to the back of the Portable III/386. Which contains the ethernet card and the extra serial card.
> ahead of this is a rm .config; make clean; make
> 
> Perhaps run make defconfig; make menuconfig ahead of make.
> 
> 
I forgot to include that in the list above, actually I've completely forgotten about defconfig so I've been through menuconfig open and save default every time. Thanks for the reminder.

—M

ghaerr, 2020-05-06:
> However, I get the feeling that the problem is somewhere else, and what we're seeing is an unintended consequence of a change elsewhere.

Could certainly be, I know that BIOS console needs work, which is why I suggest using only direct console for the time being. I'll up the priority on the BIOS console duplicate source code mess.

> Where in this sequence is the console is being initialized? 

It's the first message the booted kernel displays, after the boot code. See picture.
[image]

ghaerr, 2020-05-07:
Hello @Mellvik,

After configuring BIOS console driver and using it, I have identified a number of problems, possibly similar to the issues you've found, that make BIOS console driver almost unusable. I don't think these issues are regressions that just happened, but rather a result of lots of improvements made to direct console that have not been updated in BIOS console. 

I think it will be far easier to fix/improve BIOS console by starting from our current source, rather than going back to when it was "working", because of the large number of improvements to `vi`, `kilo`, `sh` with linenoise, etc. Thus, I am going to commit the couple PR's I have been holding awaiting your testing and then submit new PRs to fix the BIOS console, which is obviously not working well.

Let's use this renamed issue for the BIOS console problems, and continue to talk about the serial and network testing in their own threads. 

Thanks!

ghaerr, 2020-05-07:
In the meantime, for testing the serial and networking, I highly suggest only using Direct Console, so as to not introduce more variables and issues into each problem.

Mellvik, 2020-05-07:
> 
> 
> 
> In the meantime, for testing the serial and networking, I highly suggest only using Direct Console, so as to not introduce more variables and issues into each problem.
> 
> 
Will do - I've spent some time today following the hunch that the console kbd problem may originate somewhere else. It's a dead end.
The response to the LED keys disappear just after the 'Console: …' message. 
I'll test the changes in the morning and possibly make another attempt to bisect the problem.

-M

ghaerr, 2020-05-07:
> The response to the LED keys disappear just after the 'Console: …' message.

You mean Console: Direct? Or Console BIOS?

> possibly make another attempt to bisect the problem.

If you do that, you might as well go back a month or longer, just to see how it previously worked. There has been very little, if any, structural changes to the BIOS console driver.

OTOH, I plan on making a number of fixes to BIOS Console shortly, and it can be tested at that time, since it would be nice to get the serial and now network testing to a known state before we go in another direction.

Mellvik, 2020-05-08:
> The response to the LED keys disappear just after the 'Console: …' message.
> 
> You mean Console: Direct? Or Console BIOS?
> 
> 
I was using direct console. See more comments below.
> possibly make another attempt to bisect the problem.
> 
> If you do that, you might as well go back a month or longer, just to see how it previously worked. There has been very little, if any, structural changes to the BIOS console driver.
> 
> 
Yes, I was thinking along the same lines. That's what got me into trouble last time. There have been so many changes of the the past months, not the least in the config section, apparently making it harder to get older commits to compile.
> OTOH, I plan on making a number of fixes to BIOS Console shortly, and it can be tested at that time, since it would be nice to get the serial and now network testing to a known state before we go in another direction.
> 
> 
I've applied those, and the verdict is the BIOS console works, direct does not.
While checking this I discovered a recent error/oversight on my part: Hurrying along, I've been changing the CONFIG_HW_KEYBOARD_BIOS instead of the CONFIG_CONSOLE_BIOS, so I've been using direct console only for several weeks now.
I haven't bisected myself backwards to figure out if the BIOS console was in fact broken, and with all the improvements you have now put in, that may not make sense. The status is, however, that direct is still broken somehow.

This exercise suddenly put CONFIG_HW_KEYBOARD_BIOS on the table. From what I can tell, the only function of this config option is to enable a call to INT16 til set the repeat rate for the system keyboard. Doesn't make much sense, since INT16 is used in several other places uninhibited but this CONFIG option. Maybe a candidate for removal?

@ghaerr, let me know what other tests to do in order to make direct console fully operational. And I apologise for the screwup with the config option. Mea culpa.

Network testing goes in a different thread.

—Mellvik

ghaerr, 2020-05-08:
Ok, so the problem is only with the direct console driver, not BIOS console.

> Some time over the last week a change has badly affected the XT keyboard driver. The keyLEDs (capslock, numlock etc. no longer work, keypad send wrong characters).

Lets start over: exactly what doesn't work? I am running on QEMU and caps lock works for me. There is caps lock handling code in direct console (for key codes, not LEDs), it hasn't been changed. If I can get an exact problem description and replication, perhaps this can be fixed.

The numeric keypad codes intersect with the ins/del/home/end keys and they were changed when linenoise was introduced such that the home and end keys work. Hex key codes are no longer sent to applications for portability. In order to get Home and End working, not all numeric keypad characters will output their respective numbers.

The direct console has the benefit that it allows international keymappings of any scan code to value, which makes some of these issues very hard to test, without testing each of 8 languages.

Finally, how that BIOS console is fully implemented in #615, you might compare numeric keypad with BIOS console if you're looking for it to be operational.

Mellvik, 2020-05-08:
> Ok, so the problem is only with the direct console driver, not BIOS console.
>
> Some time over the last week a change has badly affected the XT keyboard driver. The keyLEDs (capslock, numlock etc. no [...]
@ghaerr, I agree - direct console is the way to go, and compared to a lot of the other challenges that have met the dust recently, this must be quite minor.
The problem - short version: kbd LEDs don't work, and - since I'm primarily interested in the arrows keys, I can confirm that they send something other than expected. Numlock doesn't seem to work, when the shift key is down, the keypad emits numbers.

I can make an exact map of what is being emitted from the keypad - and verify the (non)function of the numlock and shift lock keys..

—M

ghaerr, 2020-05-08:
> The problem - short version: kbd LEDs don't work,

Direct console only hooks INT 1 to service the kbd interrupt directly, and has never attempted to program the LEDs, so this is an enhancement request, not a regression bug.

> and - since I'm primarily interested in the arrows keys, I can confirm that they send something other than expected.

Arrow keys have been heavily tested, they should be sending ESC [ A, ESC [ B, ESC [ C, ESC [ D.
This works on my systems while editing using vi, kilo as well as lineedit in sh.

>  Numlock doesn't seem to work, when the shift key is down, the keypad emits numbers.

That also has never been implemented in direct console, so this is also an enhancement request.

> I can make an exact map of what is being emitted from the keypad - and verify the (non)function of the numlock and shift lock keys..

No need. If you would like this working in direct console, add this to your list and edit xt_key.c to add the enhancements.

Mellvik, 2020-05-08:
OK, I'll take a look. The arrow keys work with BIOS console, the don't work with direct console. The home key seem to work though.

-M

> 8. mai 2020 kl. 18:04 skrev Gregory Haerr <notifications@github.com>:
> 
> 
> The problem - short version: kbd LEDs don't work,
> 
> Direct console only hooks INT 1 to service the kbd interrupt directly, and has never attempted to program the LEDs, so this is an enhancement request, not a regression bug.
> 
> and - since I'm primarily interested in the arrows keys, I can confirm that they send something other than expected.
> 
> Arrow keys have been heavily tested, they should be sending ESC [ A, ESC [ B, ESC [ C, ESC [ D.
> This works on my systems while editing using vi, kilo as well as lineedit in sh.
> 
> Numlock doesn't seem to work, when the shift key is down, the keypad emits numbers.
> 
> That also has never been implemented in direct console, so this is also an enhancement request.
> 
> I can make an exact map of what is being emitted from the keypad - and verify the (non)function of the numlock and shift lock keys..
> 
> No need. If you would like this working in direct console, add this to your list and edit xt_key.c to add the enhancements.
> 
> 
>

ghaerr, 2020-05-08:
> The arrow keys work with BIOS console, the don't work with direct console.

That is very strange... they work fine on QEMU. They have worked for you on VB, right?
So they're not working on real hardware. This could have to do with an invalid keymap.h for your system.

Perhaps you can add a printk in xt_key.c in lines 231 and 275 and I'll help try to determine what the problem is. My guess is that your PC is outputting different scan codes than what is expected in the dircon.c.

ghaerr, 2020-07-17:
PRs #668 and #670 fix all issues brought up here.

closed 2020-07-17 by ghaerr