ghaerr/elks #618: Kilo editor - problem erasing characters

georgp24, 2020-05-09:
You cannot properly remove characters from an input file with kilo.

I observed the following: when I make a new file and enter in there `12345678` and then press Ctrl-s and Ctrl-q, a file with this content is generated. When I load the file again, I press `del` four times so only `5678` remains. After Ctrl-s and Ctrl-q, I load the file again. The file then contains:
```
5678
678
```
However, it should just contain `5678`, the second line is a bug. If you try to erase again further lines will be added when saving.

Mellvik, 2020-05-09:
> You cannot properly remove characters from an input file with kilo.
> 
Interesting coincidence, @georgp24.
I tried kilo (first time ever) yesterday while researching keypad arrow key problems, and found it severely lacking, including some of the errors you report.

Even w/o the editing and cursor positioning problems, kilo is currently useless on anything other than a fast emulator because every cursor movement, every insert, every delete cause repainting the screen - from the cursor position and out. Imagine this on a 9600 or 4800 serial line...

Nothing that cannot be fixed - it's now on my list, and probably on someone else's too.

--Mellvik

> You cannot properly remove characters from an input file with kilo.
> 
>

georgp24, 2020-05-09:
I think for 9600 baud you better use vi as an editor. I think kilo is an interesting alternative for vi to be used at the console.

ghaerr, 2020-05-09:
Hello Georg, thanks for the bug report and actually using ELKS applications :)

Sadly, it seems that backspace to remove characters doesn't work either - take your same file of 12345678, go the end of the line and backspace to remove 2 characters and save. But then again, that's not surprising, since forward-delete uses "move cursor forward by one, then delete character" to work.

The `kilo` editor was created by the same guy that created the interesting `linenoise` project, and I learned a lot from studying his approach to linenoise. I thought kilo was interesting because it is extremely small, knows about C keywords, and displays color. It uses a very interesting approach of completely de-coupling the buffer editing from the screen updates, which use a small set of sequences, unlike vi, which is very large. I like the idea of very small programs doing big things in ELKS. Also, very useful for beginners that don't know about editing, and just want a screen editor to use. I'm well aware of its repainting issues. So I'm with Georg here, which is I thought it would be a useful editor for the console, and it was another example of a small program that required a strict ANSI console in order to work, which resulted in enhanced console code.

However, it's buggy. Anyone have time to take a quick look at it and fix it?

ghaerr, 2020-05-09:
BTW, a quick check of the antirez/kilo source shows that someone has submitted a patch for what could be this problem... https://github.com/antirez/kilo/pull/56/commits/c4c18df1a5d989fcdf44daea4808717ab22543f5

There are also other fixes that could be of interest. https://github.com/antirez/kilo/pull/56/commits/9c03efa274b1a1f1a85d4f7936e14b1476e1ceeb

These should be looked at individually, rather than applying his large patch as a whole.
https://github.com/antirez/kilo/pull/56

georgp24, 2020-05-10:
This PR implements multiline, I have the impression this fixes a problem with DEL and multiline.

Greg, I think you already took a close look at this editor, so it may be easy for you to fix it. Also, it may be a problem on ELKS because I cannot believe that all these developers post PRs at the kilo site and did not observe this very obvious bug.

georgp24, 2020-05-10:
I downloaded kilo and compiled it with gcc on my SUSE Linux. It works fine there and the problem I reported with ELKS is not present.
However, on ELKS you can press the DEL key in the first column and it will erase the character and move the rest of the line to the left. However, on kilo on SUSE nothing happens then. You have to move the cursor to the second character and DEL erases the character to the left of that. So DEL is just translated into a BACKSPACE with kilo on SUSE.

ghaerr, 2020-05-10:
> I downloaded kilo and compiled it with gcc on my SUSE Linux. It works fine there and the problem I reported with ELKS is not present.

Thanks for the testing. There were several changes made to kilo for the port to ELKS. This means one of them created this bug, or that the ELKS ANSI console is still buggy.

I just tested kilo using serial console (which uses the host system's terminal emulator, just like running kilo natively), and found that kilo is buggy using delete character. I'll look in to fixing it.

georgp24, 2020-05-16:
The del key does work now. Thank you!

closed 2020-05-16 by georgp24