ghaerr/elks #622: Problem with wait()

georgp24, 2020-05-10:
I did not develop the `cron` program with the latest ELKS version and now found that it would not work with the latest version. The processes were not terminated and the system run out of memory. 

I found that the problem results from the change made to `wait.c` on the 4th of May. This stopped the `wait()` call in the cron program to work as expected. In the man page for `wait()` it says:
> The call wait(&status) is equivalent to: `waitpid(-1, &status, 0);`

Therefore I recommend to translate `wait()` to `waitpid` in wait.c.

I replaced the `wait()` call with `waitpid(-1, &status, 0)` and the `cron` program works with the latest ELKS version.

ghaerr, 2020-05-10:
Georg,

Thanks for testing. The general `wait` problem is unfortunately more complicated than just translating wait to waitpid. I fixed the libc wait() for another problem where it returned without waiting for a different reason.

It seems I still do not have it right, and will research this further and fix it.

I need to have available the original exact source code statements using `wait` are, so I know when I have it fixed properly. Can you leave the original statements in, commented out? That's the only way I'll be able to determine it working properly.

Thanks!

georgp24, 2020-05-10:
The cron version that does not work is in the PR #620 now. It is the wait in the runjob.c file at line 66. In my local version here I commented this out and made a waitpid() below that instead. I will upload that tomorrow.

georgp24, 2020-05-16:
The `cron` program uses `wait()` or `waitpid()` to reap the child process. When I committed my last version of cron, this worked with `waitpid()` but not with `wait().` With the latest version of ELKS it does not work with `waitpid()` too. So the child processes are no longer removed from the process table and after a while ELKS reports that it runs out of slots. This refers to `wait()` or `waitpid()` in `runjob.c.`

Also zombie processes are not removed by init. I cannot see if these are not properly reparented to `init` or what the reason is. They just stay "forever" in the process table.

ghaerr, 2020-05-16:
Hi @georgp24,

Thank you again for continued testing on this. Geez, it seems I just haven't been able to get `wait` working quite right. I had been seeing zombie processes pile up in `ps`, but tested this and thought `init` was getting them after a while.

> When I committed my last version of cron, this worked with waitpid() but not with wait(). 
> With the latest version of ELKS it does not work with waitpid() too.

I think both `wait` and `waitpid` are working to hold the parent process from running until the child exits in this version, please confirm. The problem is that the child processes are not being reaped, right? That is what I have been occasionally seeing.

I will fix this, thanks.

georgp24, 2020-05-17:
I did some further testing. In `runjob.c` there is this piece of code:
```
    /* wait for all subprocesses to finish */
    //while ( wait(&status) != -1 )
    while ( waitpid(-1, &status, 0) != -1 )
	    ;
    }
    exit(0);
```
The while loop used to terminate and exit(0) was executed. This would result in a `SIGCHLD` signal which triggers this signal handler in `cron.c:`
```
/* SIGCHLD handler -- eat a child status, reset the signal handler, then
 * return.
 */
static void
eat(int sig)
{
    int status;

    waitpid(-1, &status, 0);
    signal(SIGCHLD,eat);
}
```
What I observed now is that the while loop in `runjob.c` is not terminated and thus exit(0) is never executed. This again causes no SIGCHLD signal being generated. So the process remains in the process table. New processes are forked as scheduled and remain in the process table until no more slots are available.

Regarding zombies: if everything works as it had in the past, zombies are removed from the process table but every now and then a zombie is "forgotten" and remains in the process table. If you schedule a new job every minute and let that run for over 30 minutes, the number of zombies gets very high. I will test what problem that causes when wait works again.

I still have to handle in `crontab.c` when no `var/cron` directory is present.

georgp24, 2020-05-17:
Just want to add to the post above: if too many zombies are "forgotten" to be removed, ELKS will report: "Only 0 slots".

ghaerr, 2020-05-17:
Georg - thank you for the testing and clarification on this. I don't quite yet see exactly where the bug is, as either wait above should reap runjob when it terminates, but agree a kernel fix is needed.

The way wait works internally is that orphaned processes are identified and then reparented to init, only during the task list search during the wait call. It is possible that in this scenario, since cron is still running, the zombies are not reparented since cron is their parent. I'm thinking the original wait call returned -1 rather than waiting for the runjob, and that's why the zombies are piling up. When the last task table (there are 15 fixed task/process slots) is used up, the kernel reports "0 slots" and the system is unusable.

ghaerr, 2020-07-18:
Hello @georgp24,

I've fixed the "zombies forever" problem you identified in this thread. PR #673 now properly processes the zombie process, and it won't be shown in `ps` listings.

Both `wait` and `waitpid` operate properly and will remove zombie children, and your original `wait` code can be returned, if desired.

While testing the code with `cron`, I've noticed that the various cron jobs fork multiple times, and it seems that although the zombies are removed, new cron processes start accumulating in the `ps` listing. I would like you to test the wait issue, as well as look into whether your code (see the SIGCHLD handler you discuss above), are operating properly. It seems that the task list will still eventually fill up with cron processes.

I don't understand cron well enough to debug this. For much easier debugging, turn on DEBUG_WAIT in include/linuxmt/debug.h, that will display what each process is doing in regards to waiting.

Thanks!

closed 2020-09-21 by ghaerr