ghaerr/elks #634: Linenoise translates the delete key to backspace

georgp24, 2020-05-16:
As the title says, when you press the delete key, a backspace key is generated. The character at the cursor position should be removed and the rest of the line moved one position to the left.

ghaerr, 2020-05-16:
Hello @georgp24.

> The character at the cursor position should be removed and the rest of the line moved one position to the left.

Ok.

I will raise the priority of my planned fix for this, the full fix is a bit more complicated. 
Looks we need to completely separately handle DEL vs BS. The additional issue is `stty` default settings on serial logins. My macOS laptop doesn't have a separate BS key and they've placed `delete` where Backspace should be, and it sends DEL. For serial logins, Delete doesn't erase characters.

I think the full solution is to separate DEL and BS as suggested, and add a config option (for at least myself, although default OFF) that sends a BS when the Delete key is sent, implemented in the BIOS and Direct console drivers.

georgp24, 2020-05-29:
I use almost exclusively the delete key. Only when erasing mistyped passwords I use backspace. So the conversion of the delete key to a backspace key in linenoise is a nuisance for me.

ghaerr, 2020-05-29:
@georgp24, I committed your PR #636. Please close this issue if that handles this problem, thanks.

georgp24, 2020-05-30:
Great that it works now!

closed 2020-05-30 by georgp24