ghaerr/elks #638: kilo editor cannot load kilo.c

georgp24, 2020-05-17:
Kilo is compiled with `-maout-chmem=0xfff0`. However, when I move the `kilo.c` file to ELKS I cannot load that in kilo, it fails with `sys_brk`. It will only work when I reduce the `kilo.c` file to 149 lines or `5343 bytes`.
I would assume `-maout-chmem=0xfff0` would allow for bigger files than 5k.

ghaerr, 2020-05-17:
Hello @georgp24,

Thank you for actually trying to use ELKS, its nice to try to get it to the point of being actually useful. That said, using ELKS usually means finding more problems, but I think we're making progress, thanks to your and everyone else's help :)

I was aware of the fact that `kilo` can't load large files, but given its data+heap allocation of 65520 bytes and small data/bss usage, it should at least be able to load files larger as large as 32k, and with some changes, ~50k+ or so.

I think this problem has to do with libc `malloc` internally using `sbrk` to allocate memory. The problem that 8086 `sbrk` has is that when passed an allocation request larger than 32767, that value is interpreted as a negative number, and `sbrk` fails, or worse, succeeds and offers "new" memory below the current heap! I will see if I can get malloc to work directly with `brk`, otherwise at least limit malloc allocations to 32k - 1 to prohibit corruption errors.

In debugging `kilo`, I noticed that `ps` shows negative "free" values when editing a blank file, and  a negative heap value when editing a larger file. This is likely due to heap and free values also being treated as signed numbers,  and there could be more to the problem. I'll take a look at that too.

The new "min_stack" coming shortly will also allow for smaller stack requirements, and a larger total heap, for cases like this where a process needs as much heap as it can get. I will consider kilo when doing the min_stack work as well.

Thank you!

georgp24, 2020-05-18:
Maybe Kilo could be compiled with the new medium memory model to have access to more memory.

I looked through the pull requests on the Kilo site, debugged and added a few of these to Kilo. This are delete line, copy line and paste line, plus a separate help page.

ghaerr, 2020-05-19:
Hello @georgp24,

I've looked into the libc memory allocation routines, and they already correctly handle the problem I thought `kilo` might be having. Looking further into kilo, it seems that instead of a single large malloc for the entire file, it malloc's each line separately, along with a number of other allocations. So it seems that kilo uses up tons of memory for small files. I'm not sure what can be done about this.

> Maybe Kilo could be compiled with the new medium memory model to have access to more memory.

The coming medium model support only provides for code segments > 64k, the data segment will still be limited to 64k, so this won't help kilo either.

georgp24, 2020-05-19:
Maybe @tkchia can, while he is at it, also implement the compact model. This has 64k of code and a data segment >64k instead.

georgp24, 2020-05-19:
I think malloc should return a NULL pointer if no memory can be allocated. I tested this and observed that it will return an address and then a sys_brk fail occurs.
This is my test code which lets me crawl the available memory until it fails with sys_brk:
```
void testfunction()
{
    int *r;
    int Kb = 0;
    while ((r=malloc(1<<10))){
        ++Kb;
        printf("Allocated %d Kb total r=%p\r\n", Kb,(void*)r);
        sleep(1);
    }
}
```

ghaerr, 2020-05-19:
> I think malloc should return a NULL pointer if no memory can be allocated. I tested this and observed that it will return an address and then a sys_brk fail occurs.

Good point, and I agree. I just ran your test program and see incorrect behavior.

The sys_brk failure is just a kernel printk, and does not actually reset the break (end of data) when it reports a failure. It seems that the process somehow terminates instead of returning NULL.

I'll have a deeper look at this.

ghaerr, 2020-05-19:
> I just ran your test program and see incorrect behavior.

I'm sorry, I was incorrect.

Your test program runs correctly, and so does malloc. While malloc returns a value (not NULL), printf is called, and then the last time through the loop the kernel sys_brk printk is called (while inside the malloc routine in the sbrk call), malloc returns NULL, and your program exits. It runs exactly as designed.

georgp24, 2020-05-21:
I put the `testfunction()` I posted above into the Kilo program and can call this function with `Ctrl-T` after loading a file. 
ELKS returns `sys_brk(11) fail` even when malloc can still allocate memory. This did irritate me. However, I still do not understand why these sys_brk errors occur, what they mean and what can be done to avoid these. Here is a screenshot:
[image]
In the case of this file malloc returns `000000` as the memory address but not yet the NULL pointer:
[image]

ghaerr, 2020-05-21:
Georg, I'm not sure what you're getting at here: NULL, 0, 0000 and 000000 are all the same thing.

Remember, the sequence is: 1) malloc is called, 2) if there isn't enough memory sbrk is called to get more, 3) if there is no more memory that can be given, the kernel printk is displayed, and 4) malloc returns memory allocated, or else NULL (0, 0000, etc).

In the first picture above, that is exactly what is happening: the printk's occur inside the call to malloc, then it returns NULL.

In the second picture, I've realized the reason that you think ELKS has an error and returning NULL but continuing anyways is actually a BUG in your own program: Using %p for a "pointer" in stdio is for LONG pointers, you should be using %x. Go look at the source for libc/stdio/vfprintf.c and you'll see that using %p is an error on your programming, not in ELKS. In the second picture, malloc is returning a valid 16-bit pointer but your printf is using %p and expecting a long argument. Pointers are only 16-bits in ELKS. Got it?

I can explain more about how process memory works in ELKS, including the data segment, bss, heap (which malloc uses), end of heap, and stack, if you'd like to document it.

ghaerr, 2020-05-21:
Actually looking at this further, I would say that I am incorrect, it is not a bug in your program, %p is supposed to work for variable-sized (either 32-bit or 16-bit) wide pointers. It appears there may be an issue in libc vprintf handling this. So you should try using %x instead. Looking at the top of picture 2, is shows a pointer of value '6', which should never happen - that's six bytes at the start of your program data segment. Thus I think we have a problem with the use of %p instead of %x for pointers.

Another thought: around three months ago, a commit was made that deleted what I thought was a useful file, 'Documentation/text/bin_formats.txt'. (see https://github.com/jbruchon/elks/commit/58867a26d887ea0582a878e02ebd231b8edd4306#diff-c4ea08e4f4bfbdc81d682b32048b6cbd). Take a look at that, it shows some of the memory layout that could be expanded upon for you to understand how the sys_brk works. I propose we undelete that file, and update it so that you and others can understand what is going on.

georgp24, 2020-05-22:
Thank you Greg, I will try using %X instead of %p then.

There is already an "Executable format" document in the Documentation folder which discusses the `aout header.` I will add the additional information in the `bin-formats.txt` document you mentioned to this document. Maybe one could mention the `chmem` program in this document also.

georgp24, 2020-05-22:
> I can explain more about how process memory works in ELKS, including the data segment, bss, heap (which malloc uses), end of heap, and stack, if you'd like to document it.

Yes, I would like to document it. I suggest I extend the `exe_format.html` document with the information I already have and you can then comment on it.

georgp24, 2020-05-22:
I did some further testing. I modified my test function so it is using `%lX` now:
```
int checkAvailableMemory()
{
    long int *r[500];
    int Kb = 0; 
    int i = 0;
    while ((r[i]=(long int*)malloc(1<<10))){
        i++;
        ++Kb;
        printf("Allocated %d Kb total, returned memory address from malloc=%lX\r\n", Kb,*r[i]);
        if (i>499) break;
        sleep(1);
    }
    if (r[i]==NULL) printf("NULL pointer received\r\n");
    printf("Allocated %d Kb total\n",Kb);
    for (int j=0;j<i;j++) free(r[j]);
    sleep(15);
    return Kb;
}
```
So it now uses `%X.` I try to free the allocated memory so I can continue with Kilo and display the free memory in the status line after executing this function. However, the while loop should terminate if malloc returns a `NULL` pointer or `0`. It does not do so even if the printout is `0`. Therefore I suspect the printf is not working well.
This is the result when using the ELKS version before you committed the new chmem version. I did two screenshots to show the entire output:
[image]
[image]
As you can see the printout shows several 0 results before while decides to terminate the loop. Even the second call to malloc returns 0 apparently. However it does terminate the loop while the new version does not.
When using the latest version of ELKS the result is different, but also e.g. the second call to malloc returns zero. There are no longer sys_brk failures but a stack overflow at the end.
[image]
To get some further information I used screen to load kilo in one window and then call meminfo, ps and chmem in another window:
[image]
What I do not understand is why `chmem` reports a heap of 58722 bytes while `ps` reports a heap of zero.

georgp24, 2020-05-22:
Just found the following: after adding a stack of 2048 bytes to kilo using the new chmem, the result is the same again as with the ELKS version before the new chmem was committed.

ghaerr, 2020-05-22:
Georg,

There are too many problems with your test program to enable me to debug the problem you're trying to solve.

First, why are you continuing to use 32-bit pointers, when I just reminded you that ELKS uses 16-bit pointers? In your function, ALL "long int" should be just "int". You should not cast return return from malloc, and you should use %x, not %lx. Further, the printf should use r[i] not *r[i].
Finally, when you declare your "long int \*r[500]" inside the function, it allocates it on the stack. That's 4 \* 500 = 2k on the stack, which is resulting in stack overflow. That declaration should be static or outside the function.

There's no need to make this so complicated with casting and longs - remember everything is 16 bit, period.

ghaerr, 2020-05-22:
> Just found the following: after adding a stack of 2048 bytes to kilo using the new chmem, the result is the same again as with the ELKS version before the new chmem was committed.

Good, you're making progress. Give yourself at least 4k of stack with chmem, remove the longs and other things recommended above, and lets see what that looks like.

ghaerr, 2020-05-22:
> What I do not understand is why chmem reports a heap of 58722 bytes while ps reports a heap of zero.

That is a possible bug in ps display, after you get this other stuff worked out I will look further into it. I am guessing that perhaps this may be a 16-bit overflow condition the ps isn't handling.

ghaerr, 2020-05-22:
The routine in `malloc` that actually gets the memory from ELKS is in a file libc/malloc/__mini_malloc.c, line 31. Put a `printf` for 'size' just above that, and that will allow you to debug the amount of memory that is bring requested fro ELKS. This routine is being called from malloc.c line 281.

While this libc `malloc` is fairly hard to understand, the __mini_malloc is very straightforward, and can be optionally used to allocate memory directly from ELKS. That is, you can call it directly from your test program if you want to skip the other malloc memory management. While not normally recommended, this will allow you to essentially call `sbrk` directly, which ends up calling sys_brk when a program needs more memory.

georgp24, 2020-05-24:
I modified my checkmemory code as you pointed out. Now it looks like that:
```
/* compile with:
 * ia16-elf-gcc checkmem.c -o checkmem -maout-chmem=0xfff0 -melks-libc -mcmodel=small
 * cp ./checkmem elkscmd/rootfs_template/bin
 * make
 * ./qemu.sh
 */
#include <stdlib.h>
#include <stdio.h>

int checkAvailableMemory()
{
    void *r[70];
    int Kb = 0; 
    int i = 0;   
    do {
        r[i]=malloc(1024);
        i++;
        if (i>69) break;
        if (r[i] != NULL) {
            Kb++;
            printf("Allocated %d Kb total, returned memory address from malloc=%x\r\n", Kb,r[i]);
        } else {
            printf("NULL pointer received - terminating\r\n");
        }
    }  while (r[i] != NULL);
    for (int j=0;j<i;j++) free(r[j]);
    return Kb;
}

int main() {
    int Kb = 0; 
    Kb = checkAvailableMemory();
    printf("Allocated %d Kb total\n",Kb);
    return 0;
}
```
However, this way I cannot get any memory, malloc returns zero on the first call.

georgp24, 2020-05-24:
I also tried `sbrk`. Although `malloc` returned zero, `sbrk` returns a number higher by 2048 bytes. From 0d7c to 157c = 800hex. So this got increased although `malloc` returned zero. I used the following code for this:
```
#include <stdlib.h>
#include <stdio.h>
#include <unistd.h>

int checkAvailableMemory()
{
    void *r[70];
    int Kb = 0; 
    int i = 0;
    
    void *p; 
    p = sbrk(0);
    printf("sbrk returned:%p\r\n", p);
    
    do {
        r[i]=malloc(1024);
        i++;
        if (i>69) break;
        if (r[i] != NULL) {
            Kb++;
            printf("Allocated %d Kb total, returned memory address from malloc=%x\r\n", Kb,r[i]);
        } else {
            printf("NULL pointer received - terminating\r\n");
            void *p; 
            p = sbrk(0);
            printf("sbrk returned:%p\r\n", p);
        }
    }  while (r[i] != NULL);
    
    for (int j=0;j<i;j++) free(r[j]);
    return Kb;
}

int main() {
    int Kb = 0; 
    Kb = checkAvailableMemory();
    printf("Allocated %d Kb total\n",Kb);
    return 0;
}
```

ghaerr, 2020-05-25:
Georg,

The reason why both your programs above don't work is that they are buggy: after putting the result of `malloc` into r[i], you increment i, then compare r[i] for NULL, which isn't comparing the result of malloc, but the next array entry, which is uninitialized. For the kinds of things you're trying to do in that program, a for loop is much better than do/while, since the array index can be controlled in a single line (for).

`sbrk(0)` always returns the address just above the top of the current heap, so it will always return a higher address than `malloc`, FYI.

I slightly rewrote your program so that it works correctly. You will find it allocates 59k of memory before running out. You can play around with how much heap memory to assign the program by using `chmem -h <heap> checkmem`. The default heap is 4k, if no -maout-chmem= option is used.
```
/* compile with:
 * ia16-elf-gcc checkmem.c -o checkmem -maout-chmem=0xfff0 -melks-libc -mcmodel=small
 * cp ./checkmem elkscmd/rootfs_template/bin
 * make
 * ./qemu.sh
 */
#include <stdlib.h>
#include <stdio.h>

void *r[64];

int checkAvailableMemory()
{
    int Kb = 0;
    int i = 0;
    for (i=0; i<64; i++) {
        r[i]=malloc(1024);
        if (r[i] != NULL) {
            Kb++;
            printf("Allocated %d Kb total, returned memory address from malloc=%x\r\n", Kb,r[i]);
        } else {
            printf("NULL pointer received - terminating\r\n");
            break;
        }
    }
    for (int j=0;j<i;j++) free(r[j]);
    return Kb;
}

int main() {
    int Kb = 0;
    Kb = checkAvailableMemory();
    printf("Allocated %d Kb total\n",Kb);
    return 0;
}
```

georgp24, 2020-05-27:
Thank you very much Greg, I overlooked this. The reason for my tests is that I am trying to check the memory available while loading files with kilo and stop with that if there is no more memory available because the file is too big.

When I run the version you posted I get the following result:
[image]
So I use chmem to set the stack to 1024 bytes and get the following result:
[image]
Is there a way to avoid these sys_brk failures?
However, without adding a stack with chmem I can get the code to work without problems when adding the variable `char test[64];` like this:
```
#include <stdlib.h>
#include <stdio.h>

void *r[64];
char test[64];

int checkAvailableMemory()
```
[image]
I cannot explain that. If I add a stack of 1024 with chmem then, I still get the sys_brk failures as before.

ghaerr, 2020-05-27:
Hello Georg,

It just so happens that while you are testing this, myself and @tkchia are finishing up the final touches on the enhanced configurable heap/stack feature, and the "maximum heap" function which kilo needs is not quite finished being implemented. I will want you to re-run these tests after myself and @tkchia finish this work in the next few days. I think that is why you are getting the STACK OVERFLOW message. In our final working version, that will not happen.

A little more information on the "sys_brk() fail": that message will always be displayed, possibly a number of times, just prior to malloc failing. This is for two reasons: 1) malloc always calls sbrk which calls sys_brk to get more memory, and when that memory has run out, the kernel currently prints that somewhat informative message. This message has no effect on the malloc return, it is informative only, so that we can see a program running out of memory. The second reason is that the libc malloc tries to be smart (look at the 2nd screenshot above): libc malloc first tries to sbrk a large amount, then 1/2 the large amount, etc, down to 512 bytes when memory is short. When the process is truly out of memory, multiple sys_brk() console messages are displayed, showing the multiple sbrk calls malloc is making trying to get memory. I hope this clarifies how the system operates.

The sys_brk() message could be removed in the future, but I think it probably good to leave in a little while longer; it does not affect when malloc returns NULL.

georgp24, 2020-05-27:
Thank you Greg, I will continue to work on Kilo with small files for now.

ghaerr, 2020-06-01:
@georgp24, kilo is being compiled with max heap requested using -maout-heap=0xffff. I have tested that it is getting maximum memory, and this thread has shown that libc malloc and sbrk appear to function correctly. It seems that kilo just uses up lots of memory, and would require a redesign in order to change much. Thus, this issue should be closed, thanks.

georgp24, 2020-06-07:
I tried to load kilo.c with the kilo version currently in the repo. It will load only 152 lines of the file although there is still 20kb available when the `realloc` call returns NULL. I made a small change to test where the problem is and that is what I get:
[image]

This is the change to the code I made:
```
int checkAvailableMemory()
{
    int i, Kb = 0;
    void *r[64];
    for (i=0; i<64; i++) {
        r[i]=malloc(1024);
        if (r[i] != NULL) {
            Kb++;
        } else {
            break;
        }
    }
    for (int j=0;j<i;j++) free(r[j]);
    return Kb;
}

/* Insert a row at the specified position, shifting the other rows on the bottom
 * if required. */
void editorInsertRow(int at, char *s, size_t len) {
    if (at > E.numrows) return;
    E.row = realloc(E.row,sizeof(erow)*(E.numrows+1));
    if (E.row==NULL) {
      printf("Out of memory at 595 loading line %d requesting %d bytes, Kbyte free:%d\r\n",
        E.numrows,sizeof(erow)*(E.numrows+1),checkAvailableMemory());
      exit(1);
    }
......
```
I really do not know what to do here. Realloc should not return NULL if 20 kb are still available. This was compiled with the latest ELKS version using the `-maout-heap=0xffff` parameter.
The line `E.row = realloc(E.row,sizeof(erow)*(E.numrows+1));` is line number 595 here.

ghaerr, 2020-06-08:
Hello @georgp24,

> I really do not know what to do here. Realloc should not return NULL if 20 kb are still available. 

It seems your code looks correct; there may be a bug in realloc here. However, it is possible that there are 20 "1024 byte" slots available in the heap, while there is not a single "2142 byte" slot available, as realloc needs continuous memory. You might try changing the 1024 in checkAvailableMemory() to 2200, just to check.

I have ported a UNIX V7 malloc to ELKS, its in libc/malloc/v7malloc.c. I did this a while back to try to debug realloc, which I thought was buggy (from "ls") but then decided it was not. I could be wrong. This simpler v7 malloc could be used to print the entire heap of malloc/realloc/free memory slots to completely debug this problem. Since the output list would be long, you would need to run in serial console mode by setting rootfs_template/bootopts to "serial", and change the qemu.sh settings to redirect the serial port to stdio. Have you tried that yet?

georgp24, 2020-06-11:
> However, it is possible that there are 20 "1024 byte" slots available in the heap, while there is not a single "2142 byte" slot available, as realloc needs continuous memory. You might try changing the 1024 in checkAvailableMemory() to 2200, just to check.

I did that. Yes, there are 20 Kb free, but no block of 2200 bytes:
[image]

ghaerr, 2020-06-11:
> I did that. Yes, there are 20 Kb free, but no block of 2200 bytes

I ran my debug version of malloc and saw the same thing. It seems that kilo operates with extreme memory inefficiency, unfortunately. Here's the debug heap after kilo's first malloc fail, when trying to load a 2631 byte file and given a 4k heap:
```
malloc(18) 140 HEAP LIST
1: 07a8 size 0 (skipped)
2: 07aa size 3318 (skipped)
3: 14a2 size 8
4: 14ac size 24
5: 14c6 size 1024
6: 18c8 size 128
7: 194a size 14 (free)
8: 195a size 78
9: 19aa size 78
10: 19fa size 78
11: 1a4a size 28 (free)
12: 1a68 size 74
13: 1ab4 size 74
14: 1b00 size 74
15: 1b4c size 42 (free)
16: 1b78 size 36
17: 1b9e size 36
18: 1bc4 size 36
19: 1bea size 56 (free)
20: 1c24 size 4
21: 1c2a size 4
22: 1c30 size 2
23: 1c34 size 70 (free)
24: 1c7c size 76
25: 1cca size 76
26: 1d18 size 74
27: 1d64 size 84 (free)
28: 1dba size 4
29: 1dc0 size 4
30: 1dc6 size 2
31: 1dca size 98 (free)
32: 1e2e size 70
33: 1e76 size 70
34: 1ebe size 70
35: 1f06 size 112 (free)
36: 1f78 size 4
37: 1f7e size 4
38: 1f84 size 2
39: 1f88 size 126 (free)
40: 2008 size 24
41: 2022 size 24
42: 203c size 24
43: 2056 size 84 (free)
44: 20aa (top)
Total alloc 2348, free 714
```
Notice that the (sorted) heap has many very small allocations, and various free 100+ byte free sections; when kilo needs a larger allocation, it fails because of the pattern of alloc/free and ends up with no larger spaces but still lots of free memory.

I originally ported kilo to ELKS thinking it was a nice case of a useful, small editor that used very few escape sequences, and greatly helped debug the ANSI direct console code. What I didn't realize is that its creation was seemingly to show how small an editor could be written, instead of being written for use on small systems. It is possible this could be fixed with a revised memory allocator that sorted allocations based on size to avoid hashing the heap. For those types that need a non-vi editor, hopefully the existing version will be enough, given its limitation on edit file size.

I'm working on a way to easily use a debug version of malloc so we can better see the internals of ELKS application programs during execution.

georgp24, 2020-06-12:
Kilo makes three malloc calls for each line in the text. One for the characters, one for the screen representation expanding TABs and hiding undisplayable characters and a line to handle syntax highlighting. So I assume the 4,4,2 group is a line consisting of cr/lf. If you load a file with TABs in it, the second memory allocation will be greater than the one before, e.g. 40:[...]24, 41:[...]32, 42:[...]32. I do not know what causes the free memory allocations, maybe getline() allocates some memory internally and frees it again.

If the loaded file uses the entire memory, it will be difficult to edit the file, e.g. add new lines and search in it.

The `include/linuxmt/minix.h` file contains these lines:
```
/* executable with far text, and optionally relocations */
#define SUPL_FARTEXT_HDR_SIZE	sizeof(struct elks_supl_hdr)
#define EXEC_FARTEXT_HDR_SIZE	(EXEC_MINIX_HDR_SIZE + SUPL_FARTEXT_HDR_SIZE)
```
If Kilo could be compiled with this memory model, this would solve the memory problems.

ghaerr, 2020-06-12:
> If Kilo could be compiled with this memory model, this would solve the memory problems.

The FAR text (medium) model only increases code space, not data, so that won't work.

With the addition of PR #660 which checks for kilo memory available, and the debug malloc showing that malloc works but that kilo manages memory poorly, this PR can be closed.

georgp24, 2020-06-24:
Kilo does not load kilo.c yet but I close this.

closed 2020-06-24 by georgp24