ghaerr/elks #641: Memroy management

marcin-laszewski, 2020-05-19:
Can you explain a ELKS memory management?
There are two family: heap_alloc (heap_*) and seg_alloc.
seg_alloc uses heap_alloc.
Both uses "flags" (like Linux).
Both functions offers merge, split blocks so what is the difference?
Is heap_alloc a ELKS version of 'kmalloc' and "seg_alloc"  of "vmalloc" (without MMU?)?

ghaerr, 2020-05-19:
> Both functions offers merge, split blocks so what is the difference?

In order to understand ELKS memory management, one needs to understand the 8086 processor, and segmentation. I suggest reviewing a chapter on how the segment registers are used to address memory, particularly the DS register, as this defines the need for the different memory management routines in ELKS.

https://www.geeksforgeeks.org/memory-segmentation-8086-microprocessor/
https://en.wikipedia.org/wiki/X86_memory_segmentation
https://en.wikipedia.org/wiki/Intel_8086

More particularly, the ELKS kernel runs with a fixed CS (for code) and DS register (for "local" data). The DS register is used by the C compiler to directly access variables and data much like a normal C program. The heap_alloc function allocates memory much like malloc in the kernel DS (data) segment.

Since each segment is limited to 64k, and also to run and load other programs, as well as access more data (buffers), the seg_alloc function is used, which allocates memory like malloc in memory that is not directly addressable by the DS register. That means that memory cannot be directly addressed by usual C variables. All of the "L2" buffers and user programs are managed and allocated using seg_alloc. The implementation of seg_alloc uses linked lists which are allocated by heap_alloc. The implementation of memory allocation routines is another subject, but can be learned about from other books.

You might also want to read up on how a basic memory allocator works.
https://danluu.com/malloc-tutorial/

In general, the flags are used for `meminfo` and not strictly required; they denote the type of use of the memory for informational purposes at this point. Later, those flags may be used to discard or otherwise improve the memory allocation/deallocation process.

ghaerr, 2020-05-31:
@marcin-laszewski, do you need more explanations on this, or can this be closed?

marcin-laszewski, 2020-06-01:
No, your explanation is suffiecient.

closed 2020-06-01 by marcin-laszewski