ghaerr/elks #648: Insert key missing

georgp24, 2020-05-27:
The keyboard driver supports the Home and End plus the  PageUp and PageDown keys which are not on the original XT keyboard. But unfortunately the Insert key is reported as Shift-O.

I am working on the Kilo editor and would like to use the Insert key to toggle insert mode and replace mode when editing a file.

Also I would like to jump to the start of the file with Ctrl-Home and to the end of the file with Ctrl-End. However, Ctrl-Home returns Ctrl-W and Ctrl-End returns Ctrl-Q. The Ctrl-Q combination is already used to termiate Kilo.

ghaerr, 2020-05-27:
Which console are you running, direct console or BIOS console? The keyboard drivers are different, unfortunately.

Since I am not running on actual PC hardware, we will need your testing in order to get this working. I cannot test because my macOS hardware does not work the same way.

If you are running direct console, the scan codes are converted by the various drivers/char/KeyMaps/\*.h files, as well as in some special code in xt_key.c line 261. You will need to understand and decode the xtkb_scan[] array in the KeyMaps/*.h file you are configured for, and then trace which scan code your keyboard is producing when pressing the Insert key, and then pass back an UNUSED value to keyp_E0.

For the ctrl-XXX functions, you will need to do the same thing, but there doesn't appear to be a xtkb_ctrl[] table, which may need creating. You might need to start with ctrl-alt-XXX functionality instead, using the xtkb_scan_ctrl_alt[] table.

Unfortunately this looks like a potentially large project since there are eight different international keyboard variations, although we can probably use the same value for the Insert key in each, once the proper table entry location is found.

Finally, we will need to assign the Insert key (and any other ctrl/Alt keys that you desire, their own ESC [ ... ANSI sequence, and also modify the code in kilo.c to recognize those sequences.

Mellvik, 2020-05-27:
@gregorp24, @ghaerr,
Adding support for the keypad Insert key Should be part of the coming update to the direct keyboard driver, temporarily stashed while taking whatever time available to work the networking issues.

As to the ctrl-variant, I agree with @ghaerr: the driver is in need for an overhaul - possibly borrowing an alternate table structure from minix2.

--Mellvik

georgp24, 2020-05-28:
I looked at the code and I think I will be able to add support for the insert key. However, as far as I understand there is no ANSI sequence for CTRL-Home. So this will not work if you want to keep the Kilo version working on standard Linux.

ghaerr, 2020-05-29:
> I looked at the code and I think I will be able to add support for the insert key.

Ok, good. Please read this XFree86 document on Xterm control sequences: https://www.xfree86.org/current/ctlseqs.html. In the section "PC Style Function Keys", you will see that only Home and End are defined, and ELKS uses those values (ESC [ H and ESC [ F).

Since neither PageUp/Dn or other function keys are defined in that section, I (and kilo) has been using the next section, "VT220 Style Function Keys", which defines values for PageUp, PageDn and Delete (ESC [ 5 ~, ESC [ 6 ~, and ESC [ 3 ~). You should implement Insert as ESC [ 2 ~, as indicated:

[image]

We could implement ctrl-XXX function keys perhaps by getting out of the lower range, and adding perhaps 40 to the sequences, when ctrl is also pressed. If we went that way, then ctrl-Home would send ESC [ 4 1 ~ , ctrl-Insert would send ESC [ 4 2 ~, (and other sequences the same way).

> this will not work if you want to keep the Kilo version working on standard Linux.

I'm not real worried about that, the values would be hard-coded in kilo.c function editorReadKey(), just like they are now.

The proper way to keep everything completely portable is to assign keyboard key names in termcap/terminfo, and then read their values in rather than hardcoding them. See below for an example of the names used in the above document. I don't think ELKS needs to go this far for kilo.c, although note that our current `vi` does go to the trouble.

[image]

closed 2020-06-06 by georgp24