ghaerr/elks #666: ELKS Wiki

ghaerr, 2020-07-12:
Hello Jody (@jbruchon),

I have been thinking that adding a Github Wiki section to this repository would be a good idea, and would like to write some articles on how to use ELKS, as well as educational descriptions of its internals. Hopefully others will catch on and contribute articles and information as well. What do you think of this idea?

If you agree, can you please turn on the Wiki in Settings?

Thank you!

jbruchon, 2020-07-12:
I love the issue number XD done.

ghaerr, 2020-07-12:
Thanks Jody. I smiled myself thinking that the issue number could possibly be related to most programmer's feelings about documentation!

jbruchon, 2020-07-17:
Who knew that the solution to the Antichrist was so absurdly simple? 🤣

ghaerr, 2020-07-22:
The first, very rough draft of the outline for the ELKS Wiki has been posted: https://github.com/jbruchon/elks/wiki, or just click on the "Wiki" tab on the home page.

I thought perhaps to start with "How to use ELKS" so that our current community and others can quickly learn how to get ELKS up and running, and cover the basics of how one can use ELKS to perform a number of essential tasks without recompiling. The idea is to explain how ELKS is configured and adapted from its own runtime, before we get into developer documentation.

I also plan on preparing a section for releases as well as hopefully get binaries to be built automatically and uploaded to GitHub (as described in https://github.com/actions/upload-release-asset so that ELKS is usable without great technical expertise and compilation/setup requirements.

Please comment on this thread regarding the kinds of things you'd like to see in this How-To section of the Wiki, or other future sections, thanks!

Mellvik, 2020-07-22:
Great (and important) contribution, @ghaerr.

Suggestions:
1) A developer section with technical stuff, recommendations, references, experiences, guidelines, what ELKS is not and why (!), hardsware supported, etc.
2) A section with man-pages. Since ELKS is (still) floppy only (from a distro point of view), with limited space for docs, a place to get easy access to man pages would be great. For creating them and using them.

--Mellvik

ghaerr, 2020-07-25:
The Wiki section on How to Use ELKS is now complete, with the exception of some links. It is meant to be a quick-start guide for ELKS users, take a look and let me know what else should be covered.

I plan to add a developer section in the future. I would like to get a download section, preferably from the GitHub auto-builds, so that users don't have to be developers in order to try ELKS.

closed 2020-08-11 by ghaerr

Mellvik, 2020-08-12:
@ghaerr,
I realize I should have commented on this before, my apologies.

Since I just recently started using qemu networking and serial, I’ve fresh experience: While the qemu.sh script is self-explanatory, it took me a while to figure out how to connect to the virtual serial port qemu creates. A line or two in the wiki about that would be useful.

Also, for qemu networking, and this may be because I’m using qemu on a mac, not linux: I can connect til ELKS via ‘localhost', but the mac has no idea about the 10.0 network. Is there a trick to this?

—Mellvik

ghaerr, 2020-08-12:
> I can connect til ELKS via ‘localhost', but the mac has no idea about the 10.0 network. Is there a trick to this?

I have found the following to work. Please test and I'll document both in the wiki:

QEMU needs host forwarding for incoming packets, but will forward outgoing packets by default.
To connect to ELKS on QEMU from macOS, using the host forwarding line "hostfwd=tcp:127.0.0.1:2323-10.0.2.15:23" already set in qemu.sh:

    telnet localhost 2323            (telnet from macOS to ELKS)

To telnet out from ELKS on QEMU to macOS, no host forwarding is needed as QEMU will auto-forward all outgoing packets, so use the IP address of your macOS system. 

Ensure that a telnetd server is running on macOS. I use port 2424 so as to not possibly get confused with other things:

    /usr/local/opt/telnetd/sbin/telnetd -debug 2424 &     (start homebrew telnetd on port 2424 on macOS)

Then use the macOS IP address from within ELKS:

    telnet 192.168.179.176 2424             (connect to macOS IP using port 2424)

To telnet to ELKS telnetd from within ELKS, which now works great and supports multiple connections, connect to the ELKS IP directly, since gethostbyname localhost is not yet implemented:

    telnet 10.0.2.15              (connect to ELKS from ELKS)

Mellvik, 2020-08-12:
> ﻿
> To connect to ELKS on QEMU from macOS, using the host forwarding line "hostfwd=tcp:127.0.0.1:2323-10.0.2.15:23" already set in qemu.sh:
> 
> telnet localhost 2323            (telnet from macOS to ELKS)
Yes, this is my point. telnet localhost etc. works fine, but that's not what the wiki says :-)
> To telnet out from ELKS on QEMU to macOS, no host forwarding is needed as QEMU will auto-forward all outgoing packets, so use the IP address of your macOS system.
> 
Actually, no need to do that. Thanks to the qemu cmd line, you can do telnet 10.0.2.2 from elks and you're fine (plus port number, but using the default 23 should be fine).
> Ensure that a telnetd server is running on macOS. I use port 2424 so as to not possibly get confused with other things:
>
> /usr/local/opt/telnetd/sbin/telnetd -debug 2424 &     (start homebrew telnetd [...]
Great, I haven't tested this one yet .
> 
--Mellvik

ghaerr, 2020-08-12:
>  it took me a while to figure out how to connect to the virtual serial port qemu creates. A line or two in the wiki about that would be useful.

Actually, I've haven't figured out how to connect to a QEMU virtual serial port from macOS. Can you give an example so I can put it in the Wiki? Thanks!

Mellvik, 2020-08-15:
OK, like most things it's simple when it has been figured out. Logical - that's an open question.

ANyway - ELKS /dev/ttyS0 is the window from which you start qemu. 

———————————
➜  elks git:(ether) ✗ pwd
/Users/helge/src/mellvik/elks
➜  elks git:(ether) ✗ ./qemu.sh 
Using QEMU: qemu-system-x86_64
Using disk image: -fda image/fd1440.bin
char device redirected to /dev/ttys006 (label chardev1)
WARNING: Image format was not specified for 'image/fd1440.bin' and probing guessed raw.
         Automatically detecting the format is dangerous for raw images, write operations on block 0 will be restricted.
         Specify the 'raw' format explicitly to remove the restrictions.
ttyS0 at 0x3f8, irq 4 is a 16550A
ttyS1 at 0x2f8, irq 3 is a 16550A
lp: no ports found
eth: NE2K at 0x300, irq 12, MAC 52:54:00:12:34:56
bioshd: 1 floppy drive and 0 hard drives
device_setup: BIOS drive 0x0, root device 0x380
PC/AT class machine, Unknown x86 CPU, 639K base RAM.
ELKS kernel 0.3.0 (57584 text + 7472 data + 52848 bss + 5216 heap)
Kernel text at c0:0000, data at ecf:0000, 515K for user processes.
fd: found valid ELKS disk parameters on /dev/fd0 boot sector
fd: /dev/fd0 probably has 18 sectors, 2 heads, and 80 cylinders
MINIX-fs: mounting unchecked file system, running fsck is recommended.
VFS: Mounted root (minix filesystem).
Running /etc/rc.d/rc.sys script
Starting networking on eth
ktcp -b 10.0.2.15 10.0.2.2 255.255.255.0
ktcp: ip 10.0.2.15, gateway 10.0.2.2, netmask 255.255.255.0
ktcp: ethernet 52.54.00.12.34.56 mtu 1024
Starting daemons telnetd httpd 
ELKS built from commit 061d4bed
Sat Aug 15 11:17:55 2020

ELKS 0.3.0

login: 

————————————————————————————————

Notice the message from qemu at the start, where it say char device redirected to etc.
This is the local (mac) device for elks /dev/ttyS1:

screen /dev/ttys006

and you're connected….

—Mellvik

> 12. aug. 2020 kl. 22:15 skrev Gregory Haerr <notifications@github.com>:
> 
> 
> it took me a while to figure out how to connect to the virtual serial port qemu creates. A line or two in the wiki about that would be useful.
> 
> Actually, I've haven't figured out how to connect to a QEMU virtual serial port from macOS. Can you give an example so I can put it in the Wiki? Thanks!
> 
> 
>

ghaerr, 2020-08-15:
Thanks @Mellvik.

I'm not sure what exactly we're talking about here - is it redirecting the emulated ELKS /dev/ttyS0 (I know how to do that, by using the CONSOLE= line below), or are you talking about redirecting the emulated ELKS /dev/ttyS1?

I don't understand how to redirect a secondary (not /dev/ttyS0) emulated ELKS serial port to a specified host /dev/ serial device.

Where in the qemu.sh script does it specify to redirect the emulated /dev/ttyS1 serial output to /dev/ttys006? Following is the (standard) line in qemu.sh:
```
#SERIAL="-chardev pty,id=chardev1 -device isa-serial,chardev=chardev1,id=serial1"
#SERIAL="-chardev msmouse,id=chardev1 -device isa-serial,chardev=chardev1,id=serial1"
```
Both of these lines are commented out. I understand the second line, and usually use the following to redirect ELKS /dev/ttyS0 to the terminal:
```
CONSOLE="-serial stdio"
```

My QEMU installation never states "char device redirected to ..."

What I was initially asking about would be the ability to redirect the QEMU stdout to, say, a SLIP reader so that one could somehow test ELKS SLIP configured to /dev/ttyS0 and running something like "qemu.sh | slip-reader". I don't have a slip-reader too and I think using the CONSOLE= line would work...

Mellvik, 2020-08-15:
> 
> Where in the qemu.sh script does it specify to redirect the emulated /dev/ttyS1 serial output to /dev/ttys006?
> 
It doesn't. AFAIK there is no way to do that, possibly a feature courtesy of macos. Qemu tells you which device was allocated from the host OS (my take).
> Following is the (standard) line in qemu.sh:
> 
> #SERIAL="-chardev pty,id=chardev1 -device isa-serial,chardev=chardev1,id=serial1"
> #SERIAL="-chardev msmouse,id=chardev1 -device isa-serial,chardev=chardev1,id=serial1"
In my setting I uncomment the first of these.
> Both of these lines are commented out. I understand the second line, and usually use the following to redirect ELKS /dev/ttyS0 to the terminal:
>
> CONSOLE="-serial stdio"
> My QEMU installation never [...]
I think the above would work for slip if you use the tty device qemu lists at 'boot' time ...

--Mellvik