ghaerr/elks #697: [INFORMATION] How to transfer files to/from elks

Mellvik, 2020-08-12:
##Intro
This is an informational post, not a bug report. 

ELKS is rapidly getting closer to being self-sufficient. being able to transfer files into and out of the systems in an important milestone towards that goal. Recent improvements in networking support and serial port reliability & speed have made that possible.

Transferring limited size ascii files can be done with reasonable reliability via serial lines and terminal emulation. Mechanisms to make such transfers binary and reliable are forthcoming. In the meanwhile, the `urlget` command is available to import any type file of any size if the network is up and running - ethernet or (presumably, untested) slip.

Similarly, files can be transferred from ELKS using the http server which is started by default on boot. [untested]

Using `urlget` has the distinct disadvantage of requiring a http-server at the other end. No big deal, but an extra step and a complication for 'regular users'. When the ELKS `httpd` has been enhanced to support HTTP PUT, which I believe should have high priority, this situation changes, and is simpler (both technically and usage) than the TFTP solution we've discussed before.
`urlget`does not support filename globbing, use `tar` to transfer multiple files and directory trees. Stay away from `tar` compression, ELKS doesn't support that yet.

### HOWTO
**QEMU:**
- No changes to the qemu.sh script
- Starting a python3 http-server is simplest:
  - On a mac, run `python3 -m http.server 8000 --bind 127.0.0.1`, this will make the current directory available for file transfers. On other platforms, check which version of python is installed (`python -V`). Using `python`instead of `python3` may be the only change you need.
- Put the file(s) you need to transfer into that directory (or a subdirectory).
- Run `urlget http://10.0.2.2:8000/filename > local_filename` or `urlget http://10.0.2.2:8000/dir1/dir2/filename > local_filename`

**Physical hardware:**
The same python based http-server can be used, or install a real http-server like `lighttpd`(which is what I did) - on linux (raspbian). The default  home directory for the server is `/var/www/html`, for convenience add a subdirectory (e.g. `elks`) to hold files to be transferred. The transfer cmd will be `urlget http://10.0.2.2/elks/filename > local_file`(assuming `lighttpd` is using port 80 (the default).

ghaerr, 2020-08-14:
> On a mac, run python3 -m http.server 8000 --bind 127.0.0.1

I tried this on my system, running macOS 10.14.6, and it only has python 2.7.16 installed. I'm afraid to upgrade python since I have other projects dependent on that version of python.

It would be most convenient to transfer files from the current directory, rather than having to copy them to a web server directory, before pulling them from a running ELKS system.

For initiating file transfers from a host to ELKS, I have looked into implementing POST on ELKS httpd. Unfortunately, it seems even the current distribution BusyBox for Linux doesn't implement POST, but uses a CGI-based shell script to make that work. I'm thinking that perhaps HTTP isn't the method that should be used for file transfers, but something more like rcp, which would easily support globbing on the host side for pushing files, with a minimal rlogind on ELKS.

You mentioned that "hd -r" doesn't always work, can you explain in more detail where the bug is?

Thank you!

Mellvik, 2020-08-14:
> On a mac, run python3 -m http.server 8000 --bind 127.0.0.1
> 
> I tried this on my system, running macOS 10.14.6, and it only has python 2.7.16 installed. I'm afraid to upgrade python since I have other projects dependent on that version of python.
> 
I have the same. That's why I'm using the python3 command. I forget whether it's included ot brew installed...
> It would be most convenient to transfer files from the current directory, rather than having to copy them to a web server directory, before pulling them from a running ELKS system.
> 
Yeah, that's what the python version is doing. That said, this is not about convenience, but about opportunity - right now. This is possible,certainly not convenient.
> For initiating file transfers from a host to ELKS, I have looked into implementing POST on ELKS httpd. Unfortunately, it seems even the current distribution BusyBox for Linux doesn't implement POST, [...]
Right now we just need something that works. Urlget works inbound, curl/wget outbound seems fine (limited testing). This is not for the faint of heart. Rcp may be a good idea, regular ftp may be so too... The pre with ftp is that the other end is readily available, r-commands sort of a forgotten species for security reasons.
> You mentioned that "hd -r" doesn't always work, can you explain in more detail where the bug is?
> 
Need more testing on this one. Try it locally on elks both ways, had trouble with that yesterday, then pursued other problems instead.

--M

Mellvik, 2020-08-25:
I've updated the file transfer doc - it's more complete and cover transfer both ways. At some point it may be good to move it to the wiki.

--mellvik

closed 2020-09-15 by Mellvik