ghaerr/elks #699: Technical Blog - First Article

ghaerr, 2020-08-13:
I've written the first of several planned technical articles describing in detail how ELKS works internally.

It's up on the [Wiki Blog](https://github.com/jbruchon/elks/wiki/Technical-Blog) under [System Calls and Interrupts](https://github.com/jbruchon/elks/wiki/System-calls-and-interrupts).

The [Wiki](https://github.com/jbruchon/elks/wiki) has also been updated on the topics of Boot Options using /bootopts, Networking with QEMU, and Device Names.

Let me know if this sort of thing is useful, and what technical level to strive for.

It would be nice to get this information out on the general [ELKS Website](http://elks.sourceforge.net) also, which remains highly outdated.

Mellvik, 2020-08-14:
Very inspiring, @ghaerr. An important contribution - and I urge everyone on the list to read and comment.
Documentation like this is a game changer AFAIK. On one hand it lowers the threshold for serious contributions. On the other it enables to use of ELKS as a teaching tool.
My intent is to 'read to understand', which will inevitably lead to a bunchload of questions and comments. We may all be developers at some level, but our frames of reference will be different - so what's obvious to one may be unintelligible to the next.
How do you want to play this @ghaerr - general questions or specific comments (down to the sentence if you'd like ...)??

//Mellvik

> ﻿
> I've written the first of several planned technical articles describing in detail how ELKS works internally.
>
> It's up on the Wiki Blog under System Calls and Interrupts.
>
> The Wiki has also been [...]

ghaerr, 2020-08-14:
With any operating system, there are a number of concepts whose operation must be understood in order to see how the whole system works. For the first few articles, rather than discussing the system as a whole at a high level, I thought it might be interesting to just dive directly into an important low-level routine, to show how the kernel effectively comes to life for its primary purpose - providing services to multiple applications. After reading it, there may be many questions that spring to mind.

I welcome questions or comments directly on the article itself, how well (or not) it explains the actual code being analyzed, and separately suggestions of other topics to be covered, which could be written by commenting on more source code. Ultimately, understanding the kernel is best done by reading kernel source code. 

Thank you!

closed 2020-10-15 by ghaerr