ghaerr/elks #704: telnetd: stack overflow

pawosm-arm, 2020-08-15:
It's been a while since I last tested telnet over ktcp. What I noticed now is that although one can login, exit and login again (which wasn't possible last time), none of the commands (except `exit`) completes. E.g. soon after `meminfo` stuck at the end, I could see following on the ELKS side:
```
SRC NULL
(17)STACK OVER MINSTACK by 33958 BYTES
(17)STACK OVERFLOW BY 15673 BYTES
```

ghaerr, 2020-08-15:
Hi @pawosm-arm,

Very interesting. Are you running the latest commit(s) from yesterday?

Can you tell me exactly what commands you are running since boot, and configuration? It seems your telnet'ing in from Linux to ELKS, logging in, exit, login again, then running meminfo and then you get the above message?

I can't get this to repeat - I have been running with CONFIG_BOOTOPTS=y, and with or without the /bootopts file. I have been using /bootopts with the second line uncommented, which sets serial console and starts the network at boot, and set init level 3. Even without all that, I can't duplicate it: boot, login to console, run "net start", telnet in from remote twice, using ^D to exit, then third time login as root and run meminfo. At the moment, this is all on QEMU. Are you running on real hardware over your 4800 baud line, using "net start slip"?

BTW, the "SRC NULL" is a printf compiled in to ktcp, which displays when memcpy is called with either SRC or DEST a NULL pointer. So your testing has (finally) identified one of the last remaining memory corruptions, but it isn't repeating over here (yet). The ktcp code allows the memcpy to complete, even with a NULL pointer, so anything afterwards is subject to corruption, including the telnetd stack overflow.

pawosm-arm, 2020-08-15:
Hi @ghaerr, it's today's master SHA fd1778268f216bd75111e2db59734ab191a8a5c2

I've tried two commands (when telneting Linux to ELKS): 'ps' and 'meminfo', both several times, and they never completed... It does not matter if I log-in, exit and log-in again (it was just another exercise I was testing).

I'm running it on my real hardware as usual (Amstrad PC 2086), I was doing 9600bps as last time it worked just fine.

I've noticed this in .config:
```
CONFIG_EXEC_MMODEL=y
```
Was it added recently?

ghaerr, 2020-08-15:
Question: does ps and meminfo operate normally, when logging into ELKS from the console? Have you seen anything funny at all when NOT operating from a network session? I want to be sure some other modifications are not involved.

If neither ps nor meminfo run, even on the first remote login, but do work fine from the console, then this issue should probably be renamed "Can't execute commands on inbound telnet" or something like that.

I assume the "SRC NULL" and telnetd STACK OVERFLOW only happened once or a few times, right? Otherwise no error messages and ps or meminfo just hang?

I will test the current system on serial on real hardware as well, thank you very much for your testing.

Note: the default serial driver has been changed from CONFIG_FAST_IRQ4 back to CONFIG_NEED_IRQ4. It is possible, although unlikely, that there are issues here.

It is also remotely possible that a change I made to decrease time with interrupts disabled could affect this. In the file elks/arch/i86/irqtab.S, the following lines:
```
//
// This path will return directly to user space
//
        sti                     // Enable interrupts to help fast devices
        call    schedule        // Task switch
        call    do_signal       // Check signals
        cli
```
The 'sti' should be commented out, to see whether that makes any difference.

CONFIG_EXEC_MMODEL was added at least a few months ago, its something that @tkchia and I are working on to support medium model for applications and the kernel. It would have no effect on this issue.

ghaerr, 2020-08-16:
Hi @pawosm-arm,

Sorry, never mind with the above comments. I have gotten the problem to repeat on real hardware (over slip) and it's not either of what I suggested above. I'm working on finding what the problem is.

ghaerr, 2020-08-16:
@pawosm-arm,

Thanks to you finding and reporting this and the nice selection of network debugging statements we've got in the kernel and ktcp, I was able to track this problem down in about half an hour.

A stack buffer for slip packet assembly was declared as 128 bytes and getting 140, thus corrupting the stack in ktcp. Its amazing this didn't produce problems earlier, but it seems that it only pops up during long listings (such as meminfo, ps and ls -lR), for which the system produces 140 byte IP packets. I have recently rewritten the kernel locking with regards to multiple processes, so perhaps the read/writes are longer now. I'm not yet sure why we're only generating 140 byte IP packets, but I'll look into that later.

It's all fixed with PR #706.  I tested telnetting both in and out simultaneous, and disconnecting and reconnecting three times each, on real hardware and networking is starting to look pretty good!

(Note: with full debugging on, I did find a strange repeatable issue when reconnecting when telnetting in, where the login prompt was showed, but it went directly to a \# root prompt after typing just one character. Strange, but not repeating with debug statements off. I'll be looking in to this, it sounds like something to do with raw mode not being reset for some reason).

pawosm-arm, 2020-08-16:
Hi @ghaerr, thanks for the fix, it works now! I could maintain couple of telnet sessions in parallel and all of them worked.
Some observations:
- for the first time I turned my Linux box into a gateway, so ELKS could 'go outside' for the very first time. I could telnet external web servers (to `GET` some plain-text HTML content), yet I had to use bare IPv4 addresses. I've checked (on some other machine in my local network) that nameservers listed in `/etc/resolv.conf` are usable, yet still telnet could not cope with turning name address into IPv4 address...
- CSLIP is still causing bad TCP checksums and retransmissions (though connections don't stuck anymore in such situation). Compression seems to be important issue with serial connection which tends to slow down dramatically whenever more than one thing tries to make use of the established SLIP connection.

pawosm-arm, 2020-08-16:
BTW, what is the origin of the `/etc/resolv.conf` content? Shouldn't it have an option in `.config`?

pawosm-arm, 2020-08-16:
Closing it as the original issue is fixed now.

closed 2020-08-16 by pawosm-arm

ghaerr, 2020-08-16:
> I could maintain couple of telnet sessions in parallel and all of them worked.

> for the first time I turned my Linux box into a gateway, so ELKS could 'go outside' for the very first time.

That's great news! It's been a long haul getting networking working and stabilized, I'm glad to hear ELKS networking is actually becoming useful!

> yet I had to use bare IPv4 addresses. I've checked (on some other machine in my local network) that nameservers listed in /etc/resolv.conf are usable, yet still telnet could not cope with turning name address into IPv4 address...

Being able to convert hostnames to IP addresses is on my list. I've got some ideas, but haven't decided whether to implement `gethostbyname` or another method. ELKS networking doesn't use standard networking libc routines, and I am evaluating the best way to modify the many (non-standard) network commands ELKS already has.

I'm thinking of converting "localhost" without having to open a file, for speed, but using /etc/hosts (not /etc/resolv.conf) for the name-to-IP-address resolution. /etc/resolv.conv is supposed to be used for name servers only.

There is no DNS or dynamic name server resolution code in ELKS, it could be contributed by someone after the basic gethostbyname method is running, which I'll up the priority on.

> what is the origin of the /etc/resolv.conf content? Shouldn't it have an option in .config?

The only code that uses /etc/resolv.conf is the little-used `nslookup` utility. The file /etc/resolv.conf is copied to the distribution image only when networking is selected in .config, which is done either by CONFIG_APP_KTCP, or by selecting a disk image size that includes ":net" in the select-application-by-image-size configuration file elkscmd/Applications.

> CSLIP is still causing bad TCP checksums and retransmissions (though connections don't stuck anymore in such situation).

Yes, I'm aware of this. The CSLIP protocol doesn't actually transfer checksums over the link so there's something fishy going on somewhere. From a few of the bugs I fixed in the last CSLIP commit, which unfortunately didn't fix the overall problem, I can't see how CSLIP ever worked on ELKS. I am guessing it never did, but with no error messages, and the fact that it does somehow recover eventually, this is how CSLIP has always operated on ELKS.

I'm looking for volunteers to help on debugging CSLIP; I haven't had the time to chase down exactly what both sides of the connection think is happening, which will point to where the error is.