ghaerr/elks #705: Discussion on far procedures in the kernel

ghaerr, 2020-08-15:
Hello @tkchia,

I opened up a new issue to keep from straying too far from the point of PR #703.  

I have analyzed the code produced by gcc using the `far_section` attribute applied to C procedures. Unfortunately, for far procedures calling near procedures, as you mentioned, the inefficiency is such it cannot likely be used in the kernel. I hadn't thought of the problem of having to rewrite the stack parameters due to their offsets being incorrect.

It seems it might be far faster, both in execution and programmer time, to just compile and port the asm files to medium model, than trying to use a mixed model approach. However, before making that decision, I thought to ask a few more questions after having looked at the code produced.

Here's the far call thunk source (for easy reference):
```
        /* These routines must go into the near text segment, if at all.  */
        .text

        .global __ia16_far_call_thunk.bx.si
        .type   __ia16_far_call_thunk.bx.si, @function
        .global __ia16_far_call_thunk.bx.di
        .type   __ia16_far_call_thunk.bx.di, @function
__ia16_far_call_thunk.bx.si:
        pushw   %bp                     /* set up stack frame */
        movw    %sp,    %bp
        pushw   %bx                     /* preserve %bx and %cx */
        pushw   %cx
        movw    %si,    %cx             /* move stack args. size to %cx */
        jmp     0f
__ia16_far_call_thunk.bx.di:
        pushw   %bp
        movw    %sp,    %bp
        pushw   %bx
        pushw   %cx
        movw    %di,    %cx
0:                                      /* re-push stack arguments */
        jcxz    2f
        leaw    4(%bp), %bx
        addw    %cx,    %bx
        shrw    %cx
1:
        pushw   %ss:(%bx)
        decw    %bx
        decw    %bx
        loop    1b
2:
        movw    -4(%bp), %cx            /* reload %cx for callee */
        callw   *-2(%bp)                /* call function */
        movw    %bp,    %sp             /* tear down stack frame and return */
        popw    %bp
        lretw
```
And the kernel C test routines:
```
void nearproc(void)
{           
    boot_console = 0;           
}  
            
void __far __attribute__ ((far_section, noinline)) farproc(void)
{  
    nearproc();
}
```
And the generated code:
```
00000000 <farproc>:
   0:   57                      push   %di
   1:   bb 00 00                mov    $0x0,%bx
            2: R_386_16 nearproc
   4:   31 ff                   xor    %di,%di
   6:   9a 00 00 00 00          lcall  $0x0,$0x0
            7: R_386_16 __ia16_far_call_thunk.bx.di
            9: R_386_OZSEG16    __ia16_far_call_thunk.bx.di
   b:   5f                      pop    %di
   c:   cb                      lret
```
From the look of it, it seems that the compiler doesn't handle the special case of no parameters, and instead always calls the `__ia16_far_call_thunk.bx.di` handler when calling a near procedure.

Does the compiler know that it is calling a near function (with no parameters)? Or is the decision of which segment a near procedure is in looked at only at link time?

It would seem that if `farproc`, were calling `nearproc`, and it were in the same code segment, a single `call nearproc` would work. If it's not in the same segment (or unknown, since the compilation of the entire file above was not specified to be in a single segment), then an lcallw nearproc would work, and skip saving di and calling the thunk entirely.

Since it appears an entire file cannot be labeled such that all its procedures (near and far) are placed into a single text segment, then each and every (kernel) procedure would have to be individually labeled. And then, without call analysis, only static procedures could be easily analyzed. For kernel near procedures with parameters called by far procs, it's way faster (at runtime) to declare them far, as, even with a zero-parameter optimization suggested above, their parameters would have to be re-pushed.

Conclusion: It is faster and smaller to run the entire kernel medium model, or at least declare all the routines ever called to or from a "linked-in module" as far (which would be many, given the f_ops proc arrays). The discussed possible improvement for near procs with no parameters won't help the situation, as most kernel routines have parameters.

tkchia, 2020-08-16:
Hello @ghaerr,

> It seems it might be far faster, both in execution and programmer time, to just compile and port the asm files to medium model, than trying to use a mixed model approach. 

Well, OK.

> I have analyzed the code produced by gcc using the `far_section` attribute applied to C procedures. Unfortunately, for far procedures calling near procedures, as you mentioned, the inefficiency is such it cannot likely be used in the kernel.

By way of explanation:  I added support for far → near procedure calls early last year, when there were no known use cases yet for this, so I was more focused then on getting something simple and correct to work.

I guess perhaps this is a good time to think about making it fast.  So, I have just made some changes to `gcc-ia16` so that far → near calls with a small fixed number of parameters now go through a routine along the lines of
```
	pushw	%bp
	movw	%sp,	%bp
	...
	pushw	8(%bp)
	pushw	6(%bp)
	callw	*%bx
	movw	%bp,	%sp
	popw	%bp
	lretw
```
which should be (somewhat) more efficient than always going through `__ia16_far_call_thunk.bx.si` or `__ia16_far_call_thunk.bx.di`.

In the future I hope to look into further optimizations.  All these compiler changes, of course, will need to be tested.

> The discussed possible improvement for near procs with no parameters won't help the situation, as most kernel routines have parameters.

An additional possibility --- if we are going to port the assembly routines anyway --- is to switch to the `regparmcall` calling convention, so that parameters can be passed in registers, and fewer stack parameters need to be copied around.  This should be useful whether the kernel uses the small model or the medium model.

(I tried to make this change some time ago, but I was not sure if this would break any userland stuff that depended on kernel internals, e.g. `ps` and `ktcp`.)

Thank you!

ghaerr, 2020-08-16:
Hello @tkchia,

> So, I have just made some changes to gcc-ia16 so that far → near calls with a small fixed number of parameters now go through a routine

That is a huge speed improvement, thank you!

>> It seems it might be far faster, both in execution and programmer time, to just compile and port the asm files to medium model, than trying to use a mixed model approach.

> Well, OK.

There may be another way - as proposed in https://github.com/tkchia/gcc-ia16/issues/63#issuecomment-674546850: A method where effectively only the "inbound" interface to a module is far-called, while all known near routines within the module are compiled into the same text segment as the interface, so that no thunking is needed within a module, except for callbacks. I'm not sure whether the compiler supports module-specified segments.

> An additional possibility --- if we are going to port the assembly routines anyway --- is to switch to the regparmcall calling convention, so that parameters can be passed in registers, and fewer [...]

I had thought of this also, but then declined to mention it, wondering whether the lack of extra registers in the 8086 allows for efficient compilation of multiple arguments. I will have to see what kind of code is generated to best judge whether this would work well in a -Os compilation environment.

> (I tried to make this change some time ago, but I was not sure if this would break any userland stuff that depended on kernel internals, e.g. ps and ktcp.)

User applications do not depend on any internal kernel procedure calling convention, although some are tied to kernel include files. `ps` uses /dev/kmem to read the task structure, so changes in `struct task` require recompilation. `ktcp` uses the /dev/tcpdev character device file with complicated semaphore management and the kernel read and write routines to pass information from the kernel to ktcp and back. I don't think there's much to worry about as far as breaking applications if one wanted to consider changing the kernel internal calling convention. 

However - there is a big requirement that the system call and hardware interrupt entry routine (_irqit) save the registers in a special ordering which is not easily changed, the order of which is later then used as arguments to every sys_xxx system call entry point, with the ordering matching the libc syscall stubs that move the passed parameters to registers before the INT 80h system call. (See my recent article https://github.com/jbruchon/elks/wiki/System-calls-and-interrupts on how _irqit works, with the parameter stack correctly configured at the time of the `call syscall` which table-jumps to each sys_xxx entry point).

Thank you!

ghaerr, 2020-09-05:
Hello @tkchia,

I have been making progress testing the medium model support now in ELKS. I have been experimenting with a small C program, deciding to get the compilation specifics worked out in user mode before implementing the same idea as kernel "linked-in modules", in an effort to define a single separate far text section to open up more space in the kernel (near) .text code segment.

I have a couple questions about the linker script(s) and compiler model switches used to compile this basic program, which attempts to mirror what a linked-in module might look like. Because of the required thunks otherwise generated, I'm coming to the conclusion that it would likely be better to compile each "linked-in module" completely in medium model, and link them to the rest of the small model kernel. However, when I try this with the application, the program crashes.

Here is the Makefile. If `mmprocs.o` is compiled -mcmodel=medium, the program crashes:
```
include $(TOPDIR)/Make.defs
  
################################################################################
INCLUDES=-I$(TOPDIR)/include -I$(TOPDIR)/libc/include -I$(TOPDIR)/elks/include
CC=ia16-elf-gcc
CFLBASE=-fno-inline -melks-libc -mtune=i8086 -Wall -Os
MODEL=-msegment-relocation-stuff
WARNINGS=
CFLAGS=$(CFLBASE) $(MODEL) $(WARNINGS) $(LOCALFLAGS) $(INCLUDES)
LD=ia16-elf-gcc
LDFLAGS=$(CFLBASE) $(MODEL)
AS=ia16-elf-as
ASFLAGS=-mtune=i8086
################################################################################

PRGS=mmtest

all: $(PRGS)

clean:
    rm -f *.o $(PRGS)

mmtest: clean
    $(CC) $(CFLAGS) -c -mcmodel=small -o mmtest.o mmtest.c
    $(CC) $(CFLAGS) -c -mcmodel=small -o mmprocs.o mmprocs.c. <-- problem in this line with =medium model
    $(LD) $(LDFLAGS) -mcmodel=small -o mmtest mmtest.o mmprocs.o $(LDLIBS)
```
Where does gcc get the linker script by default, it doesn't appear to be using the INCLUDE= directories, as they don't specify elks/elks, which is where the .ld scripts are. Is that hard-coded?

Being unable to compile a group of source files in medium model means that each procedure would have to be separately specified as FARTEXT, as they otherwise would get placed in the near .text segment (including static procedures, I presume). I am hoping this can be resolved through another means.

Can this problem be resolved with an enhanced linker script? I don't yet understand the reason for the application crash, even though the link seems to complete without error.

For reference, here's the rest of the source code:
```
/* mmtest.h*/
#define FARTEXT __far __attribute__ ((far_section, noinline))
#define NEARTEXT

void NEARTEXT near_noparms(void);
void NEARTEXT near_parm1(int i);
void FARTEXT far_noparms(void); 
void FARTEXT far_parm1(int i);
```
```
/* mmprocs.c*/
#include <stdio.h>
#include "mmtest.h"

void NEARTEXT near_noparms(void)
{
    printf("Hello!\n");
}

void NEARTEXT near_parm1(int i)
{
    printf("Hello %d\n", i);
}

void FARTEXT far_noparms(void)
{
    printf("Hello!\n");
}

void FARTEXT far_parm1(int i)
{
    printf("Hello %d\n", i);
    far_noparms();
    near_noparms();
    near_parm1(2);
}
```
```
/* mmtest.c*/
#include <stdio.h>
#include "mmtest.h"

int main(int ac, char **av)
{
    far_noparms();
    near_noparms();
    far_parm1(1);
    near_parm1(1);
    return 0;
}   
```
Everything compiles and runs fine as-is, which is encouraging. However, lots of thunk code is generated to by mmproc.c routines to both near procedures without parameters (which could be optimized out as a possible enhancement) and with parameters (needed for small model, but not generated if compiled medium model). This is why I would like to compile the entire mmproc.c module in medium model.

After getting this worked out, I was planning on writing a complementary relocation routine which, rather than reading in each relocation entry separately, uses a far pointer to all relocation entries in memory, and another far pointer to the text segment, and does all the relocation in one go. This could then be used in both the existing sys_exec() routine (at the cost of more initial memory) as well as called from setup.S to relocate the .fartext kernel items.

Thank you!

tkchia, 2020-09-05:
Hello @ghaerr,

> Where does gcc get the linker script by default, it doesn't appear to be using the INCLUDE= directories, as they don't specify elks/elks, which is where the .ld scripts are. Is that hard-coded?

The linker script is taken from the installation directory, which will be somewhere inside `cross/` in an ELKS build --- e.g. `cross/ia16-elf/lib/elkslibc/medium/elks-medium.ld`.  You can choose a different linker script by specifying a `-T` _script_ option to `ia16-elf-gcc`.

Anyway, my hunch is that
  * the medium-model side was trying to call `printf` as a far function --- with a 32-bit return address --- since the medium model assumes that _all_ functions are far;
  * but `printf` was (of course) near, since you were linking the program as small-model with a small-model library, so it thought the return address was 16-bit;
  * and so badness ensued.

If so, then one possible way to work around it might be to write a sort of far wrappers for near functions on the small-model side --- say, things like this (!):
```
int __far far_printf(const char *fmt, ...)
{
  va_list ap;
  int r;
  va_start(ap, fmt);
  r = vprintf(fmt, ap);
  va_end(ap);
  return r;
}
```
(This is, of course, not particularly neat, but there we go. 😑)

Thank you!

ghaerr, 2020-09-05:
> The medium-model side was trying to call printf as a far function --- with a 32-bit return address --- since the medium model assumes that all functions are far;

Thank you for your response, @tkchia.

I'll assume your hunch is correct - as, in any case, this means that compiling a whole "linked-in module" as medium model is potentially too dangerous as it assumes everything, including gcc library routines, will be medium model. I hadn't thought of that.

So perhaps a different direction might work better, now that I see that we can link various __far procedures into small model without much difficulty. That means, one can, with as much precision as desired, effectively move individual C functions out of the near .text segment, thus reducing its 64k code space as required/desired.

My first thoughts on all of this would be to move **all** kernel initialization code to far procedures. This simultaneously frees up space in .text while also combining init routines all into a single segment that could be discarded after use. Many benefits could be achieved relatively easily. The additional thunk code required by any calls by init routines to near procedures would be part of the discardable .fartext initialization section, but in general most all called init functions would be far, and then not required.

This could all be achieved fairly easily with just the following:
- Loading /linux, including the extended header and relocation section(s) by the bootstrap loader and/or setup.S. (isn't this already be loaded by the 2nd-stage boot /linux loader, just not moved by setup.S?)
- Writing a C relocation function `relocate_far_text(__far *textseg, __far *relocation_table)` used by both setup.S to relocate the kernel as well as sys_exec().
- Marking initialization routines with __far __attribute((far_section, noinline))
- Later, freeing the entire .fartext segment in memory after startup.
- (Later): setup.S's code segment could also be .fartext so as to be combined with the other init code and be discarded?

What do you think of this approach to lowering kernel .text size, and can you think of anything else that would need doing?

Thank you!

ghaerr, 2020-09-06:
Hello (again) @tkchia,

I got a kernel with a few far text routines compiled, linked, and booting! So far its just reading the extended supplementary header, but things look promising. I haven't written the relocation routine yet (in setup), but it's coming. I'm following the code in sys_exec to understand the executable file structure. It appears that although the .text, .fartext, and ,data sections are packed immediately together in the executable, each segment needs to be aligned on a paragraph boundary when copying from SYSSEG (the executable inside /linux) to REL_SYSSEG (the final location of the kernel code/data). After the final copy, the relocations will be done.

I am displaying the relocation data from sys_exec loading the mmtest program, and see S_TEXT and S_FTEXT relocations, but have yet to see any S_DATA relocations, despite putting a far procedure into a (data segment) function pointer. When are .data segment relocations emitted by the linker ?

The other problem I've run into is that the thunk routines aren't yet in the kernel arch/i86/lib library yet. I'm wondering what the best way to do this might be, should they be copied out of the cross/build/gcc-src/libgcc/config/ia16 directory and ported? There are a number of defines like '__IA16_CMODEL_IS_FAR_TEXT', in ia16-preamble.h, and I'm a little worried about they can be ported directly or need rewriting. Will these always be near-called by the compiler?

So, I am wondering whether the approach I'm taking, that of modifying setup.S to read supplementary or normal headers, read in text, far text and relocate everything, will work once many kernel far functions are compiled. Currently, a far proc calling any near function ends up with a kernel link error missing the thunk routine(s). Are there other routines, and/or do we need to worry about the other kernel lib (udivmod3, etc) routines being called far, in the case they're needed by a function declared far?

Just trying to get my hands around the problem, now that a kernel is booting :)

Thanks for your help on this! I can upload a patch if you want to see more of what I'm doing. I ultimately plan on submitting a PR for your review once I get things further along.

ghaerr, 2020-09-06:
I've moved forward with the approach discussed above with PR #743, meant for review.

closed 2020-09-07 by ghaerr