ghaerr/elks #718: ELKS for testing

toncho11, 2020-08-24:
Please provide a 360kb image of ELKS for testing.

I will use a 360kb floppy on my IBM XT 8088 (IBM PC) or my IBM compatible Olivetti 8086 computer.

I will use under DOS the following method to write the raw 360kb image:
dskimage.com elks.img 0:40:2:9

Mellvik, 2020-08-24:
thoncho11,

Thank you for your interest.

I will make available an image for download for you tomorrow (my time, CET-1).

--Mellvik

ghaerr, 2020-08-24:
Hi @Mellvik,

I just tried building the 360k Minix image using "cd image; make images" and it failed (again) running out of space on the floppy. Everything has been slowly getting bigger. I'm going to produce a PR to fix this and will upload a 360k binary for @toncho11.

I would like to add the "make images" to the nightly build and somehow allow all images produced to be downloadable from Github, but I haven't figured out how to do that yet.

And @toncho11, welcome back!

ghaerr, 2020-08-24:
> I just tried building the 360k Minix image using "cd image; make images" and it failed (again) running out of space on the floppy. Everything has been slowly getting bigger.

Sorry - my rootfs_template/root directory was filled with extra test files, so "make images" works after all.

@toncho11, here's the disk image for a 360k floppy, zipped. Let us know if you have any problems getting this booted. 

[fd360-minix.bin.zip](https://github.com/jbruchon/elks/files/5119732/fd360-minix.bin.zip)

toncho11, 2020-08-24:
Hi, thanks!!!

It works. I tried on Amstrad 1640. It is indeed a very good 8086 machine. Here are my impressions and questions:

1) There was graphical demo program, but I could find it on this image. 
2) I tried a SMC ISA network card, but it was not recognized. I guess it works only for one two specific network cards?
3) Can I list attached hard drives? Which command?
4) Can I mount fat16 hard disk?
5) Can I mount fat32 hard disk?
6) Can I switch to another 360kb floppy disk? If I write some programs (such as the demo program) on a second floppy disk then would it be possible to ... unmount the current one ... mount the second one ... wait but the mount command will be still on the first one ... Programs use something like LIBC or are self-contained? Would the demo program work from a second floppy disk if I provide these next to the executable?

ghaerr, 2020-08-24:
> It works. Good job! I tried on Amstrad 1640. 

Glad to hear it's working. Thanks for the testing.

> 1. There was graphical demo program, but I could find it on this image.

The Nano-X demos all start with nx, such as nxclock, nxlandmine, etc. Unfortunately, the 360k disk image its out of space to add much anything else. You'll have to use a drive B or perhaps a USB drive to get more of the applications.

We also now produce MSDOS (FAT-16) images which contain almost all the programs. This could be copied to an MSDOS hard drive and then mounted.

> 2. I tried a SMC ISA network card, but it was not recognized. I guess it works only for one two specific network cards?

Yes, currently only the NE2000 card is supported. The kernel you have has networking built in, but you also need the "ktcp" suite of network programs in order to run "net start" which starts networking.

> 3. Can I list attached hard drives? Which command?

Not really. But you can "ls /dev" which will show you the devices that you can mount filesystems on to.

Please read the Wiki for information on how to use ELKS filesystems. That is at https://github.com/jbruchon/elks/wiki#Filesystems.

> 4. Can I mount fat16 hard disk?
> 5. Can I mount fat32 hard disk?

Yes  for both. You'll use "mount -t msdos /dev/bda1a" or something like that. See the wiki above for details.

> 6. Can I switch to another 360kb floppy disk? If I write some programs (such as the demo program) on a second floppy disk then would it be possible to ... unmount the current one ... mount the second one ... wait but the mount command will be still on the first one ...

We don't (yet) support switching floppies on the same drive.

> Programs use something like LIBC or are self-contained? 

All programs are self-contained.

> Would the demo program work from a second floppy disk if I provide these next to the executable?

Yes. Here's a 1440k boot floppy in FAT-16 format which has all the commands on it. You can try using MSDOS to copy this to your hard drive, then boot from the originally floppy and mount the hard drive to run everything.

[fd1440-fat.bin.zip](https://github.com/jbruchon/elks/files/5120346/fd1440-fat.bin.zip)

toncho11, 2020-08-25:
I have used dd to put this last image on my hard drive. But it is not bootable.
Also I can not moun it with:
mount -t msdos /dev/hdb /mnt

It detects something because it prints some numbers, but then it says 
**FAT: unsupported format**

Where is the problem?

ghaerr, 2020-08-25:
> I have used dd to put this last image on my hard drive. But it is not bootable.

The last image is a 1.4MM floppy image, it will not work to boot from HD, as it lacks a master boot record (MBR).

I thought you had the ability to boot from MSDOS HD, in which case you could copy the files from a floppy with that image, then mount the HD FAT image. It sounds like you only have a 360k floppy and an unused HD, which now has an incorrect filesystem written on it.

> Also I can not moun it with:
> mount -t msdos /dev/hdb /mnt

If the HD is C: (first hard drive), then "mount -t msdos /dev/bda /mnt" (you might try /dev/bda1 also).
If the HD is D: (second hard drive), then "mount -t msdos /dev/bdb /mnt"
Unfortunately, ELKS is quite confusing in using /dev/bdXX rather than /dev/hdXX, I think this should be changed.
See the Wiki page I mentioned earlier for details. 

> It detects something because it prints some numbers, but then it says
> **FAT: unsupported format**
> 
> Where is the problem?

I would have to see the numbers. I suspect its because you're trying to mount a floppy image as a hard drive image. Things get a little complicated without knowing whether there's an MBR on your HD.

If you want to just boot from HD, you need to use the hd32mbr-minix.bin image. I have attached it. This must be copied onto the drive, starting at sector 1, cylinder 0, overwriting the MBR master boot record. Be careful.

[hd32mbr-minix.bin.zip](https://github.com/jbruchon/elks/files/5125413/hd32mbr-minix.bin.zip)

toncho11, 2020-08-25:
Thanks for all the info. It makes sense. Yes, I decided to go with an empty drive in order not to damage my main drive.

Now I have created a FAT partition of 50 MBs in Windows. It should be fat16. Then I extracted the elks programs from fd1440-fat.bin.zip and copied them on this new partition. Next I tried to mount it with -t msdos from within ELKS, but it did not work. Same error.
It says: 
FAT: me=44.csz=0,#f=0,floc=900,fsz=16690,rloc=900,#d=192,dloc=912,#s=16640,ts=538976288
FAT: unsupported format

ghaerr, 2020-08-25:
The reason for the error is that the mount code is reading the wrong portion of a disk, which is garbage instead of a FAT header.

What device are you mounting? (exact mount command)

And what partition number is it?

It must be an older-style MBR partition.

You can run "fdisk /dev/bda1" (or whatever) to check the partition correctness.

I can't debug unless I know the details, as there's lots of potential issues with HD MBRs.

Here are the mount instructions and device names from the Wiki:
#### Mounting file systems

To mount a filesystem after booting ELKS, use the `mount` command and specify the filesystem type, e.g.:

```
# mount /dev/fd1 /mnt          [mount MINIX floppy B on /mnt]
# mount /dev/bda1 /mnt         [mount MINIX partition 1 on /mnt]
# mount -t msdos /dev/fd1 /mnt [mount FAT floppy B on /mnt]
```
Be sure to `sync` the data on drives before rebooting, and use `umount /dev/fd1` to unmount (eject) the floppy after use.

#### Device Names
Here's a useful list of block device names, useful for mounting, or using the root= option in /bootopts:
```
/dev/bda       first hard drive, no partitions
/dev/bda1      first hard drive, partition 1
/dev/bda2      first hard drive, partition 2
/dev/bda3      first hard drive, partition 3
/dev/bda4      first hard drive, partition 4
/dev/bdb       second hard drive, no partitions
/deb/bdb1      second hard drive, partition 1
/dev/bdc       usually USB drive
/dev/fd0       floppy A
/dev/fd1       floppy B
```

toncho11, 2020-08-25:
1) I am using: mount -t msdos /dev/bda /mnt
bda1 does not work at all.

2) I also tried hd32mbr-minix.bin.zip and it did start to load but it halted with:
..................ELKS found......................4!
Press key

and then it restarts. I used win32diskimager windows program to write it directly to the hdd.

ghaerr, 2020-08-25:
> I am using: mount -t msdos /dev/bda /mnt
bda1 does not work at all.

You didn't answer my question about which partition # you created. You/I need to know the partition number you created.

/dev/bda is for "flat" (non-partitioned devices), so that won't work.

/dev/bda1 is for partition 1. It seems you don't have an MBR partition 1.

The hd32mbr-minix.bin image I sent you is a MINIX partition, not FAT. So you don't use the -t msdos option.

I have attached a "flat" MINIX partition, you can try that, but its not bootable. You would use "mount /dev/bda /mnt" for this after booting from ELKS floppy.

[hd32-minix.bin.zip](https://github.com/jbruchon/elks/files/5125853/hd32-minix.bin.zip)

toncho11, 2020-08-25:
I got it working. I used MiniTool Partition Wizard to create the 50 MB fat16 parition and got bda1 in ELKS. Then I was able to mount it successfully with all the programs.

ghaerr, 2020-08-25:
Good news @toncho11!

It sounded a little painful, but I'm glad you got it all figured out.

Please play around with the system, including the graphical programs, and let us know whether you find problems, or portions of the system you like.

The basic operation of serial ports, mice, etc is documented in the Wiki on Github (https://github.com/jbruchon/elks/wiki).

I hope to get binaries uploaded automatically on the nightly build so that it is easier for you to get binaries, I haven't figured out how to do that yet.

toncho11, 2020-08-25:
Thanks. 

The graphical demos do no start. I just see a blinking cursor top left. I can still kill them with ctrl+C. One of them printed some memory messages after a while. Do I need a VGA card for them? I am using the Amstrad built-in video card currently.

I also tried booting ELKS on my IBM XT 5160. It does not load at all. The XT defaults to Basic. The same machine boots Svarog86 (FreeDOS) and I have used the same floppy disk that I am using for my Amstrad and it does work on the Amstrad.

ghaerr, 2020-08-25:
The nano-X demos require VGA. There is support for mono hercules, but that requires recompiling.

Which graphical program prints memory messages? That may be yet another bug! Are they sys_brk messages?

> I also tried booting ELKS on my IBM XT 5160. It does not load at all. The XT defaults to Basic. 

Hmmm... does it display anything at all? Perhaps it needs some BIOS configuration to load from floppy rather than run Basic?

toncho11, 2020-08-25:
For the IBM it is probably the way it checks for a bootable floppy. I suppose it reads the first sectors to identify a bootable disk, but it still checks differently than the Amstrad. How many ways there to indicate that a floppy disk is bootable?

ghaerr, 2020-08-25:
Generally, just one - 0xAA55 in the last two bytes of the first (512-byte) sector, which is read and copied to RAM at boot. It has to read it into memory, usually to location 0x7c00 before checking, then jumps to location 0x7c00 if the magic signature is there.

You might listen closely to see if the drive spins up and a read takes place or not. If you can find a boot floppy that does work, perhaps upload the first sector.

toncho11, 2020-08-25:
The nxclock and nxdemo are OK.
The nxworld starts and then gives 4 messages:
(13)STACK OVER MINISTACK BY 2214 BYTES

toncho11, 2020-08-26:
I can not amount a fat32 parition, only fat 16. I think it could be a bug.

ghaerr, 2020-08-26:
> I can not amount a fat32 parition, only fat 16. I think it could be a bug.

FAT32 is definitely supported, but that's not saying there aren't any further problems. Can you take a picture or screenshot of the kernel message(s) displayed just after the mount fail?

What OS created the image?

closed 2020-09-21 by ghaerr