ghaerr/elks #729: Kernel stated as 'non re-entrant" ?

mfld-fr, 2020-08-27:
I am a bit surprised to read in PR #695 that the ELKS kernel is stated as 'not re-entrant'. As far as I can see in the interrupt handling code, an asynchronous call can occur at any moment upon an HW interrupt, and a task switch can then occur when the interrupt level goes back to 0 at the end of the interrupt handler. It means that the kernel code executed to serve a system call, except between CLI and STI, can be interrupted at any moment. So why this statement ?

ghaerr, 2020-08-27:
Hello @mfld-fr,

It appears from after the first submission of this issue, through the subsequent edits, you have figured out that the ELKS kernel is indeed non-reentrant, in that hardware interrupts during kernel code execution will never result in a task switch, by design. This means that a kernel system call will never be interrupted, which has big benefits, explained below.

> As far as I can see in the interrupt handling code, an asynchronous call can occur at any moment upon an HW interrupt, but a task switch can only occur when the global interrupt level goes back to 0 at the end of the interrupt handler.

That is correct. In addition, any interrupt occurring during kernel execution, or during another interrupt (same thing), executes on a special "interrupt stack" instead of the kernel stack in the task structure.

The exact details of how this works are explained in two articles I wrote on the wiki: [System Calls & Interrupts Part I,](https://github.com/jbruchon/elks/wiki/System-calls-and-interrupts) [Part 2](https://github.com/jbruchon/elks/wiki/Syscalls-&-interrupts,-Part-II).

> It means that the kernel code is actually re-entrant for HW interrupt handling, but not for system calls.

Only kernel routines registered through `request_irq` need worry about reentrancy, along with routines called from them, which are quite limited. The vast majority of the kernel is not reentrant, nor needs to be, the major benefit being not having to be protected from reentrancy. Kernel interrupt routines can never call the sleep routines, for instance.

> This looks like to be an unnecessary & strong limitation, 

It is indeed a strong limitation. For single-CPU, non-realtime systems, there aren't any good reasons for the kernel to be reentrant. Should the clock tick during a system call, the process won't get rescheduled, but allowed to run until the next clock tick occurring in "user mode". 

Some of the many benefits of non-reentrancy include:
- Simplicity. Very little of the kernel needs protection using semaphores for kernel structures. In a fully-reentrant system, all kernel structures would need protection. Kernel programmers need not worry about the complex issues required with multi-processor systems.
- Speed. Kernel semaphores are slow, particularly the wakeup portion. Atomic instructions, lock prefixes, complex sleep/wait handling, all unneeded.

The long history of the non-reentrant early kernels for Linux is described in "O'Reilly - Understanding the Linux Kernel, First Version" (available on the web). The Linux kernel v2.0, while having just added support for SMP, still used a global lock protecting major portions of the kernel, while years were taken reducing critical section sizes, replaced with spin locks etc, for fast multiple-processor benefits.

> because I see no reason in the interrupt handler to forbid a task switch when returning from any HW interrupt to any system call ?

Allowing a task switch after any HW interrupt has a number of problems - the first of which is that the system isn't running on the kernel stack, and the saved register stack wouldn't be in the expected location for many kernel routines, along with the problem of unlimited kernel stack usage, etc. The other problems revolve around the need to protect every kernel resource with more code, explained above and in the O'Reilly book.

It took me a while to fully understand how the ELKS kernel works in this fashion. The precise control realized by `_irqit` allows for a much simpler kernel programming paradigm, suitable for small systems like ELKS, and especially useful for the sleep/wait subsystem and device drivers.

Thank you!

mfld-fr, 2020-08-27:
I have several concerns with the current interrupt handler implementation. The proposed code below does not constain the scheduling policy, that is encapsulated into the C interrupt procedure (up to that procedure to decide if the kernel code should be re-entrant or not), reduces the size of the interrupt handler / uninterruptible sections, reduces the size of the kernel stacks, and last but not least is more simple so readable, at the cost of accessing the caller registers with a far pointer (but not a big deal with the @tkchia version of GCC-IA16).
```
/*

// Offsets in stack frame

#define stack_di     2*0x0
#define stack_si     2*0x1
#define stack_bp     2*0x2
#define stack_sp     2*0x3  // dummy
#define stack_bx     2*0x4
#define stack_dx     2*0x5
#define stack_cx     2*0x6
#define stack_ax     2*0x7
#define stack_ds     2*0x8
#define stack_es     2*0x9
#define stack_num    2*0xA
#define stack_ip     2*0xB
#define stack_cs     2*0xC
#define stack_fl     2*0xD

// Task structure

#define KERN_STACK_SIZE 256

struct task {
	reg_t user_sp;
	seg_t user_ss;

	reg_t kern_sp;

	word_t kern_stack [KERN_STACK_SIZE];
	};

// System call vector with kernel CS

#define vect_system 0x80

// Global vector with kernel DS

#define vect_global 0xFF
#define kern_ds (4*vect_global+2)

*/

	.code16

	.text

//------------------------------------------------------------------------------

	.global vect_init

vect_init:

	push    %es
	xor     %ax,%ax
	mov     %ax,%es
	mov     %ds,%es:kern_ds
	pop     %es
	ret

//------------------------------------------------------------------------------

#define INT_SLOT(n) \
int_##n: \
	call    int_hand \
	.word   n

INT_SLOT(0x08)  // timer
INT_SLOT(0x80)  // system call

//------------------------------------------------------------------------------

	.extern int_proc

int_hand:

	sti

	// Save registers to stack

	push    %es
	push    %ds

#ifdef CPU_186
	pusha
#else
	push    %ax
	push    %cx
	push    %dx
	push    %bx
	sub     $2,%sp  // skip SP
	push    %bp
	push    %si
	push    %di
#endif

	// Switch to kernel data segment

	xor     %dx,%dx
	mov     %dx,%ds
	mov     kern_ds,%dx
	mov     %dx,%ds

	// Save far pointer to registers

	mov     %sp,%bp
	mov     %ss,%ax
	mov     %ax,%es

	// Check & increment system level
	// =0: user mode interrupted
	// >0: kernel mode interrupted

	cli
	mov     sys_level,%ax
	inc     %ax
	mov     %ax,sys_level
	cmp     $1,%ax
	jne     1f

	// User mode interrupted
	// Switch to kernel stack

	mov     task_cur,%bx
	mov     %sp,user_sp(%bx)
	mov    	%ss,user_ss(%bx)
	lea     kern_stack+KERN_STACK_SIZE(%bx),%sp
	mov     %dx,%ss

	// Get interrupt number

1:	sti
	mov     %es:stack_num(%bp),%si
	mov     %cs:(%si),%ax

	// Call interrupt procedure

	push    %es
	mov     %dx,%es
	push    %bp
	push    %ax
	call    int_proc
	add     $2*3,%sp

	// Decrement & check system level

	cli
	mov     sys_level,%ax
	dec     %ax
	mov     %ax,sys_level
	jnz     1f

	// Back to user mode
	// Switch to user stack
	// Automatic CLI

int_exit:

	mov     task_cur,%bx
	mov     %user_ss(%bx),%ss
	mov     %user_sp(%bx),%sp

	// Restore registers from stack

1:	sti

#ifdef CPU_186
	popa
#else
	pop     %di
	pop     %si
	pop     %bp
	add     $2,%sp  // skip SP
	pop     %bx
	pop     %dx
	pop     %cx
	pop     %ax
#endif

	pop    %ds
	pop    %es

	add    $2,%sp  // skip int_num *

	iret

//------------------------------------------------------------------------------

	.global	task_switch

// Task switch in kernel mode only
// within the same CS DS SS

// Synchronous call so AX BX CX DX not saved
// as scratch registers for GCC-IA16

// arg1: task prev *
// arg2: task next *

stack_switch:

	push	%bp
	push	%es
	push	%si
	push	%di

	mov     4(%bp),%bx  // arg1
	mov     %sp,kern_sp(%bx)

	mov     6(%bp),%bx  // arg2
	mov     kern_sp(%bx),%sp

	pop	    %di
	pop	    %si
	pop	    %es
	pop	    %bp
	ret

//------------------------------------------------------------------------------

	.global task_init_kern

// arg1: task *
// arg2: kernel IP

task_init_kern:

	push    %bp
	mov     %sp,%bp
	push    %es
	push    %di

	mov     4(%bp),%bx  // arg1

	// Setup kernel stack
	// to exit from task_switch()
	// and branch to provided IP

	mov     %ds,%ax
	mov     %ax,%es
	lea     kern_stack+KERN_STACK_SIZE(%bx),%di
	sub     $2,%di
	std

	mov     6(%bx),%ax  // arg2
	stosw
	xor     %ax,%ax
	stosw   // BP
	mov     %ds,%ax
	stosw   // ES
	xor     %ax,%ax
	stosw   // SI
	stosw   // DI

	cld
	add     $2,%di
	mov     %di,kern_sp(%bx)

	pop     %di
	pop     %es
	pop     %bp
	ret

//------------------------------------------------------------------------------

	.global task_init_user

// arg1: task *
// arg2: user IP
// arg3: user CS
// arg4: user DS ES
// arg5: user SP
// arg6: user SS

task_init_user:

	push    %bp
	mov     %sp,%bp
	push    %es
	push    %di

	mov     4(%bp),%bx  // arg1

	// Setup kernel stack
	// to exit from task_switch()
	// and branch to int_exit

	push    6(%bp)      // arg2
	push    %bx
	call    task_init_kern
	add     $2*2,%sp

	// Setup user stack
	// to exit from int_hand

	mov     12(%bp),%di  // arg5
	mov     14(%bp),%es  // arg6
	sub     $2,%di
	std

	pushf
	pop     %ax
	stosw   // FL
	mov     8(%bp),%ax  // arg3
	stosw   // CS
	mov     6(%bp),%ax  // arg2
	stosw   // IP
	xor     %ax,%ax
	stosw   // int num *
	mov     10(%bp),%ax  // arg4
	stosw   // ES
	stosw   // DS
	xor     %ax,%ax
	mov     $8,%cx  // 8 general registers
	rep
	stosw   // AX CX DX BX dummy-SP BP SI DI

	cld
	add     $2,%di
	mov     %di,user_sp(%bx)
	mov     %es,user_ss(%bx)

	pop     %di
	pop     %es
	pop     %bp
	ret

//------------------------------------------------------------------------------

	.data

	.extern sys_level

//------------------------------------------------------------------------------

	.bss

	.extern task_cur

//------------------------------------------------------------------------------
```

ghaerr, 2020-08-27:
Well, that's a lot of code to understand quickly. From my quick take of it, this isn't tested and working, but rather just thoughts about how to rewrite the lowest levels of a kernel. There are many more issues, currently working in ELKS, (strace, restartable system calls, signal execution, rescheduling during interrupt checks, proper SS==DS checks, next stable version, etc) that would have to be implemented and tested before such a thing could be seriously considered.

I don't yet see any benefits - would this be for allowing the kernel to be reentrant? Why? What benefit is there on a single-processor 8086, given the effort of the then-required locking, semaphores, wakeup, etc, that aren't needed now? Not to mention the added complexity of writing device drivers and having to worry about reentrancy issues. There have been several PRs introduced, by me, cleaning up the kernel of not required semaphores/locks, etc., due to the kernel interrupt process not being fully understood, which have both simplified the code and sped up execution. If the reason is for more complex scheduling capabilities, I think, with only 16 process slots and a basic round-robin scheduler with no priority handling, that is both unneeded and the process not even started.

> The proposed code below does not constain the scheduling policy, that is encapsulated into the C interrupt procedure (up to that procedure to decide if the kernel code should be re-entrant or not), [...]

I don't see where the proposed code gives these advantages, can you point them out more precisely? I do see the reduction in kernel stack requirement, at the expense of using the user stack instead, but can't recommend that design.

A problem with the proposed design is that it uses the user stack to save all the registers - a bad idea, since stack space isn't guaranteed, the kernel could crash if interrupted during a time where stack space was not available (and outside its control). Going along that line of thought, that of using user space for stacks - why have a kernel stack at all? Everything could be saved on the user stack, which is presumed to have space. The conclusion to this would be ensuring user applications have sufficient stack space for a tinier kernel and no kernel stacks - simpler, but not reliable. The tradeoff here is a possible simpler system, and no separate stacks since there is no user and kernel mode on the 8086 anyways, at the cost of requiring user applications to allocate stack space for interrupts (including system calls).

ACTUALLY - as I write this, I realize the whole idea above, but also including the proposed code, won't work: should a running application be interrupted via a HW interrupt that ends up causing a task switch that ends up causing a process to exit, the system will be in trouble - since an interrupted process's registers will be saved on a user stack that exits and memory deallocated. This is why the proposed code that saves the registers onto the user stack, and just passes a pointer to them, won't work in that case. The registers must be saved onto either the current kernel stack or another interrupt stack, outside of any process's address space.

Storing kern_ds at some offset of segment 0 doesn't make much sense - since CS is known at all interrupt times, why not just use a CS segment overwrite and store it in the kernel code segment, which is faster anyways.

Thank you!

mfld-fr, 2020-08-28:
1. The kernel stack is mandatory merely because the kernel C code works in the small memory model, where DS=SS. So for any C data pointer to be valid, the kernel has to switch to its own stack inside its data segment before executing that C code.

2. Even if the C code works in the large memory model, where DS != SS, the kernel stack is also mandatory because, as you pointed out, only the user context sould be pushed to the user stack, as that user stack could be discarded on process deletion.

3. Except in the tiny memory model, CS != DS, so kernel CS and DS must be stored in two different locations. The kernel CS is already stored in the 'system call' vector (80h). The idea is to store the kernel DS in another vector for consistency. Kernel DS cannot be stored in kernel code segment in case that segment is read only (see the ROM kernel code option in the configuration menu).

4. Even saving some registers in the kernel stack won't save in case the user stack is full, because a part of the user context is always pushed as IP CS FL on the user stack. The idea here is to push all the user context on the user stack, to simplify and speed up the interrupt handling code, at the cost of reserving 20 more bytes on the user stack, balanced with the benefit to save 16*20 = 320 bytes in the kernel data segment.

5. This redesign does not impact the current scheduling and locking policies as the interrupt procedure can still switch tasks only when system level = 1 before returning. But it allows any contributor to change that policies in the C code, without touching the interrupt handling in assembly code.

6. The proposed code could be improved by storing the system level in the stack structure, to allow task switching at any level. This could allow to use the 'break point' then the 'step by step' interrupts to switch to a debug task and monitor the execution of all the kernel code (except the timer interrupt). This technique is demonstrated in my MON86 project.

7. It could be also improved by reintroducing the interrupt stack for level > 2, to save more space in the kernel stacks, with only a few more assembly lines.

8. Discussing the ELKS mainstream priorities is not in the scope of my question. This issue is intended to discuss the pros & cons of a design change proposal from a pure technical standpoint, without considering it fits or not in the 'community driven' schedule. If it does not fit, no problem, I will implement and evaluate it in my fork after being improved through that discussion, and I could push it to mainstream later.

tkchia, 2020-08-28:
Hello @mfld-fr,

Well, I am not that well-versed (yet) in the kernel internals, but I think there is just no easy way to make a non-reentrant kernel reentrant, and **still maintain correctness**.  Such a change will have to cut across the entire kernel code base.  Of course, you are free to try.

Having said that...

  * I think there might be value in allowing a _limited, well-understood_ set of kernel functions to run on a userland stack.  For example, rather than having the `getpid` and `gettimeofday` syscalls switch to another stack, perhaps they can just retrieve the needed information and quickly get out of the way.

    (MS-DOS --- which is largely non-reentrant --- kind of works this way.  There are a few DOS functions which can be called even when another DOS function is running.)

  * Recently I had added to `gcc-ia16` a new `no_assume_ss_data` attribute for functions, which allows one to write C functions that run on a stack that is outside the data segment (`%ss` != `.data`).  It is usable, but still quite experimental.

Thank you!

mfld-fr, 2020-08-28:
@tkchia : That looks like an interesting idea...

So having two different kernel entry points, one fully synchronous (INT 81h) that only switches to the kernel data segment to access internal structures, and does not allow any task switching, for the best performance, and the other (INT 80h) that allows task switching at the cost of more code (saving the whole user context and switching stacks).

First one would be for the non-blocking operations (get / set), and the second one for the blocking operations (read / write).

mfld-fr, 2020-08-29:
A progressive way to make some parts of the kernel code re-entrant but not impacting the other parts is to use a global flag that disables the task switching at any level.

Optimizing the kernel stacks with a common interrupt stack is only possible when no task switching is allowed during an asynchronous nested interrupt. Not compatible with the monitor task as described above for debugging.

mfld-fr, 2020-08-30:
Will evaluate the proposed design in another sub-project. Closing this for now.

closed 2020-08-30 by mfld-fr

mfld-fr, 2020-09-02:
For anyone interested in, I tuned and tested the proposed interrupt handler and related stuff here: https://github.com/mfld-fr/sys86/blob/master/int-hand.S

mfld-fr, 2020-09-20:
For anyone interested, the conclusion of a benchmark between the two design options here: https://github.com/mfld-fr/sys86/issues/2#issuecomment-695773009