ghaerr/elks #730: Partition tables [again]

Mellvik, 2020-08-28:
This issue has been discussed before, and recently it became my (!) problem:

The BIOS HD driver gets physical CHS values from the system NVRAM/CMOS. Then - when reading the partition table from the MBR, it decides to skip the partition if the difference is significant. Such as
```Partitions: bda:(0,614880)  bda1:(63,997857) skipped```

While this may seem like an error in the MBR, it's not. Some old clunkers have a fixed table of drive types to choose from, which may not match what we put in them today. In my case (compaq is probably the worst offender in this), I have a 512MB CF card with CHS parameters that match nothing in the Compaq list. 

My first idea was that the CF card, being what it is, must be reprogrammable - so I set out to change the CHS values. No cigar, it turns out to not be possible, at least for the Transcend cards I have.

Fortunately, the BIOS uses the CMOS CHS values only for booting, so if the heads and sectors per track values matches, the C doesn't matter.  So the CMOS tells the system I have a 300MB drive, which DOS happily boots and disk tells me the real story, 512MB. As does disk on ELKS. But mount on ELKS fail because we've decided that the CMOS is more reliable than the MBR.
```# fdisk /dev/bda
Geometry: 610 cylinders, 16 heads, 63 sectors.
Command (? for help): p

                           START              END          SECTOR
Device           #:ID   Cyl Head Sect    Cyl Head Sect  Start   Size

/dev/bda1       *1:06     0    1    1     989   15   63     63 997857
/dev/bda2        2:00     0    0    0      0    0    0      0      0
/dev/bda3        3:00     0    0    0      0    0    0      0      0
/dev/bda4        4:00     0    0    0      0    0    0      0      0
Command (? for help): 
```

There may be more to this, but my suggestion is to either choose MBR over CMOS or put in an additional sanity check that it is only the cylinder value that differs.

Opinions solicited.

--Mellvik

ghaerr, 2020-08-28:
Hi @Mellvik,

Interesting.

I'm not sure if this is the answer you're looking for, but if the basic problem (explained well above) is that ELKS should not second-guess a partition size and incorrectly skip it, this is easily fixed:

Put an '\#if 0' around the checking code at lines 74-80 in drivers/block/genhd.c. I added that checking code back when reimplementing partitions to try to force users to only mount "valid" partitions by disabling the /dev/bdaX partition. It won't hurt to remove the check, except that it seems there are quite a few invalid partitions out there! Should a user try to mount an invalid partition, the MINIX driver will almost always refuse to do so and return a mount error.

Mellvik, 2020-08-28:
> Hi @Mellvik <https://github.com/Mellvik>,
>
> Interesting.
>
> I'm not sure if this is the answer you're looking for, but if the basic problem (explained well above) is that ELKS should not second-guess a [...]
I like that answer. It's exactly what I did :-) works like a charm.

Are you saying it's OK to make this change permanently (like a PR)? FWIW - that would make my life easier.

Talking about making life easier - and totally unrelated: I'd like to introduce a new user in the default /etc/passwd:

'tty' which executes /bin/tty as the login shell, then exits. The fast way to figure out which tty line is this without going through a full login. Courtesy BSD 2.X - maybe even Unix V7… 

And yes, you have to have 3 or more serial lines in order to need it. I do. 

—Mellvik

ghaerr, 2020-08-28:
> Are you saying it's OK to make this change permanently (like a PR)? FWIW - that would make my life easier.

Yes.

Add the comment "/* partition skipping disabled as virtual cylinder values sometimes needed for CF cards*/"
on the \#if 0 line.

> I'd like to introduce a new user in the default /etc/passwd:
'tty' which executes /bin/tty as the login shell, then exits. The fast way to figure out which tty line is this without going through a full login. Courtesy BSD 2.X - maybe even Unix V7…

Put that in the PR too.

PR #732 (2020-08-30): [kernel] Allow MBR CHS data to deviate from CMOS data (#730)
- Remove 'sanity check' on partition tables to allow real disk capacity to deviate from BIOS CHS data. Allows CF disks to be fully utilized on old systems with a fixed set of BIOS-supported disk drives. This closes #730.

- Also add user 'tty' to /etc/password - which executes /bin/tty and exits.

This fix for #730 brings up a followup issue: Now /dev/bda and (say) /dev/bda1 mismatch, and /dev/bda is no longer a useable access point for the entire disk drive. Would it not make sense to update the date for the 'master' device when det [... cut]
closed 2020-08-30 by ghaerr