ghaerr/elks #731: [minix fs] mkfs question

Mellvik, 2020-08-28:
Minix fs `mkfs` reports MAXSIZE, which is calculated like this:
```
        MAXSIZE = (7+512+512L*512L)*1024L;
```
Other than being printed, the value is not used anywhere. Does anyone have a clue as to what this is?
```
# mkfs /dev/fd1
Usage: mkfs /dev/name blocks (Max blocks=65535)
# mkfs /dev/fd1 2400
800 inodes
2400 blocks
Firstdatazone=29 (29)
Zonesize=1024
Maxsize=268966912
```

--Mellvik

ghaerr, 2020-08-28:
> MAXSIZE = (7+512+512L*512L)*1024L;

Yes, this is the maximum size of a MINIX filesystem, in Kbytes. The above math is showing the inode calculations file storage in 1k blocks for the 7 directly accessible blocks stored in the inode itself, and the 512 indirect blocks stored in the single indirect inode block (2 bytes/block number). Double indirect blocks are not implemented in ELKS.

> Max blocks=65535

That's 65535k, or 65.535MB, the maximum size of a hard disk partition supported on ELKS Minix filesystems.

Mellvik, 2020-08-28:
> MAXSIZE = (7+512+512L*512L)*1024L;
>
> Yes, this is the maximum size of a MINIX filesystem, in Kbytes. The above math is showing the inode calculations file storage in 1k blocks for the 7 directly [...]
I get it. Now, does it make sense to print this from mkfs? I mean, it confused the hell out of me …

—Mellvik

ghaerr, 2020-08-28:
> Usage: mkfs /dev/name blocks (Max blocks=65535)

It's a helpful message, once one realizes it's trying to say that 65535 (1k blocks) are the most that can be passed as the blocks parameter.

Mellvik, 2020-08-28:
The problem is, that's not what it says:

# mkfs /dev/fd1 2400
800 inodes
2400 blocks
Firstdatazone=29 (29)
Zonesize=1024
Maxsize=268966912

And that's why it seems nonsensical to me, probably more so to the average user ...

-Mellvik

ghaerr, 2020-08-29:
> Maxsize=268966912

The above number is equal to 7 + 512 + 512 * 512 * 1024...

>  MAXSIZE = (7+512+512L*512L)*1024L;

... which is the maximum size of a MINIX filesystem in bytes. Converting to blocks by dividing by 1024 gives

262,663 which is the max number of blocks in a MINIX filesystem.

ELKS using 16-bit block numbers, so max blocks is 65535, which multiplied by 1024 gives the 65.535Mb drive limitation above. However, it seems that Minix might be using 32-bit block numbers, which would allow 262,663 blocks, or just over 256Mb drives to be used.

> And that's why it seems nonsensical to me, probably more so to the average user ...

Yep!

Mellvik, 2020-08-30:
So my suggestion is to remove the Maxsize item from the printout when creating a minix filesystem, limiting the information to what is actually useful...

--Mellvik

ghaerr, 2020-08-30:
> remove the Maxsize item from the printout when creating a minix filesystem, limiting the information to what is actually useful...

Ok, agreed. Or at least show the maximum size of the just-created volume (in bytes).

I was going to make a quick fix and modify this, but looking further, it seems that this value MAXSIZE (incorrectly calculated by mkfs, mfs and also reported incorrectly by fsck and mfsck) is stuffed into the MINIX super block s_max_size field. This field, always currently set to just over 265Mb, is incorrect for any MINIX filesystems usable by ELKS.

The s_max_size value is supposed to be the maximum file size for the device, which isn't quite the same as the block count set by mkfs. In addition, neither ELKS nor any of these utilities actually use the field.

I'll submit a PR for this shortly. Thank you!

Mellvik, 2020-08-30:
Great, @ghaerr.

And - now that we're on file system issues, what does it take to get fsck (really) operational?

Currently fsck prints lots of messages but doesn't seem to do much. At least it does not remove the waring from mount that the fs needs file system check. Admittedly I haven't looked at the sources yet, which may be an educational experience.

—Mellvik

ghaerr, 2020-08-30:
I’m aware that fsck is broken. It looks like its from the same vintage as the working mfsck
that I ported from an older Linux source tree though, so getting it working shouldn’t be
too bad. 

My guess as to the biggest issue is that Linux is 32 bit, while ELKS compiles fsck as 16 bit.
So that should be looked at.

The next big item is to determine whether to use the later “mfsck.c” sources out of the elks/tools
directory, rather than the original in elkscmd/disk_utils/. I diff’d them a few months ago and came
to the conclusion they are close. If that’s really the case, I’d start with mfsck.c from the tools
directory, since we KNOW that version works, and port it to 16 bit.

Mellvik, 2020-08-31:
> I’m aware that fsck is broken. It looks like its from the same vintage as the working mfsck
> that I ported from an older Linux source tree though, so getting it working shouldn’t be
> too bad.
>
> My [...]
Yes! That sounds like a plan. Let me know if I can contribute.

-M

ghaerr, 2020-09-01:
@Mellvik, this issue is fixed in #733, try out mkfs and fsck and if satisfied can close this issue.

Mellvik, 2020-09-02:
Thanks, @ghaerr -

this is looking good. The fixed fsck did not repair a floppy filesystem created with the previous version of mkfs, but I guess that was to be expected. This is a major improvement!

The testing reminded me of the issue of the disk probe messages.
While df() is now ok, fsck, mkfs and mount still repeats the initial probe message on every open. Thinking loud, would it be an idea to make and store a hash or a checksum of the first sector of the device, and use it to check for media change?

Also, not entirely letting go of the maxsize message from mkfs, how about adding a metric to it [blocks], to make it clearer what it is?
And, on fsck, maybe move the ID (version) message to the top, even in verbose mode.

Thanks again!

—Mellvik

# umount /dev/fd1
# mkfs /dev/fd1 2400
fd: probing disc in /dev/fd1
                            fd: /dev/fd1 probably has 15 sectors, 2 heads, and 8
0 cylinders
           800 inodes
2400 blocks
Firstdatazone=29 (29)
Zonesize=1024
Maxsize=2457600
  
# fsck /dev/fd1
fd: probing disc in /dev/fd1
                            fd: /dev/fd1 probably has 15 sectors, 2 heads, and 80 cylinders
           fsck, 1.0.1 - 09/01/20
# mount /dev/fd1 /mnt/fd

…

# fsck -fsv /dev/fd1
fd: probing disc in /dev/fd1
                            fd: /dev/fd1 probably has 15 sectors, 2 heads, and 80 cylinders
           800 inodes
2400 blocks
Pre-zone blocks 2+1+1+25 29
Firstdatazone=29 (29)
Zonesize=1024
Maxsize=2457600
Filesystem state=1
namelen=14
  
fsck, 1.0.1 - 09/01/20
Forcing filesystem check on /dev/fd1.
Root inode is a directory.
/ size 112 040755 0/0 links 2 Z29 0 0 0 0 0 0 0 0
/linux size 68672 100755 0/0 links 1 Z30 31 32 33 34 35 36 37 0
/linux1 size 68576 100755 0/0 links 1 Z99 100 101 102 103 104 105 106 0
/nlinux size 68560 100755 0/0 links 1 Z167 168 169 170 171 172 173 174 0
/xlinux size 68688 100755 0/0 links 1 Z235 236 237 238 239 240 241 242 0
/ylinux size 68608 100755 0/0 links 1 Z304 305 306 307 308 309 310 311 0
  
     6 inodes used (0%)
   372 zones used (15%)
  
     5 regular files
     1 directories
     0 character device files
     0 block device files
     0 links
     0 symbolic links
------
     6 files
#

ghaerr, 2020-09-02:
> The fixed fsck did not repair a floppy filesystem created with the previous version of mkfs, but I guess that was to be expected.

What exactly is not working? I haven't tested the repair capabilities, thought I'd leave that to you, but mkfs wasn't modified in any significant way, only the max_size super block field is written differently.

>  Thinking loud, would it be an idea to make and store a hash or a checksum of the first sector of the device, and use it to check for media change?

If you're going to read the disk for 1-2k to calculate a checksum, you might as well just do a disk probe, since changing media while mounted is not supported in the kernel. Perhaps just removing the kernel probe message would be better. The message now serves as a good indicator for when probing is occurring, which, after the last fix is essentially correct, and could be removed before the next version release after testing.

> Also, not entirely letting go of the maxsize message from mkfs, how about adding a metric to it [blocks], to make it clearer what it is?

It's in bytes, and matches the image size. I'll add "bytes".

> And, on fsck, maybe move the ID (version) message to the top, even in verbose mode.

Ok.

Mellvik, 2020-09-02:
> The fixed fsck did not repair a floppy filesystem created with the previous version of mkfs, but I guess that was to be expected.
>
> What exactly is not working? I haven't tested the repair [...]
It may be my interpretation of things - it goes like this:

# fsck /dev/fd1
fd: probing disc in /dev/fd1
                            fd: /dev/fd1 probably has 15 sectors, 2 heads, and 8
0 cylinders
           fsck, 1.0.1 - 09/01/20
Root inode is a directory.
/ size 112 040755 0/0 links 2 Z29 0 0 0 0 0 0 0 0
/linux size 68672 100666 0/0 links 1 Z30 31 32 33 34 35 36 37 0
/nlinux size 68560 100666 0/0 links 1 Z99 100 101 102 103 104 105 106 0
/linux1 size 68576 100755 0/0 links 1 Z167 168 169 170 171 172 173 174 0
/xlinux size 68688 100666 0/0 links 1 Z235 236 237 238 239 240 241 242 0
/ylinux size 68608 100666 0/0 links 1 Z355 356 357 358 359 360 361 362 0
Inode 6 not used, marked used in the bitmap.
Zone 304: marked in use, no file uses it.
Zone 305: marked in use, no file uses it.
Zone 306: marked in use, no file uses it.
Zone 307: marked in use, no file uses it.
Zone 308: marked in use, no file uses it.
Zone 309: marked in use, no file uses it.
Zone 310: marked in use, no file uses it.
Zone 311: marked in use, no file uses it.
Zone 312: marked in use, no file uses it.
Zone 313: marked in use, no file uses it.
Zone 314: marked in use, no file uses it.
Zone 315: marked in use, no file uses it.
Zone 316: marked in use, no file uses it.
Zone 317: marked in use, no file uses it.
Zone 318: marked in use, no file uses it.
Zone 319: marked in use, no file uses it.
Zone 320: marked in use, no file uses it.
Zone 321: marked in use, no file uses it.
Zone 322: marked in use, no file uses it.
Zone 323: marked in use, no file uses it.
Zone 324: marked in use, no file uses it.
Zone 325: marked in use, no file uses it.
Zone 326: marked in use, no file uses it.
Zone 327: marked in use, no file uses it.
Zone 328: marked in use, no file uses it.
Zone 329: marked in use, no file uses it.
Zone 330: marked in use, no file uses it.
Zone 331: marked in use, no file uses it.
Zone 332: marked in use, no file uses it.
Zone 333: marked in use, no file uses it.
Zone 334: marked in use, no file uses it.
Zone 335: marked in use, no file uses it.
Zone 336: marked in use, no file uses it.
Zone 337: marked in use, no file uses it.
Zone 338: marked in use, no file uses it.
Zone 339: marked in use, no file uses it.
Zone 340: marked in use, no file uses it.
Zone 341: marked in use, no file uses it.
Zone 342: marked in use, no file uses it.
Zone 343: marked in use, no file uses it.
Zone 344: marked in use, no file uses it.
Zone 345: marked in use, no file uses it.
Zone 346: marked in use, no file uses it.
Zone 347: marked in use, no file uses it.
Zone 348: marked in use, no file uses it.
Zone 349: marked in use, no file uses it.
Zone 350: marked in use, no file uses it.
Zone 351: marked in use, no file uses it.
Zone 352: marked in use, no file uses it.
Zone 353: marked in use, no file uses it.
Zone 354: marked in use, no file uses it.
#
Repeated fsck’s gave the sam result and mounting the fs elicited a warning about the need to run fsck.

I overwrote the FS using the fixed mkfs and everything was fine and mount did not complain. This is a significant improvement. Previously, even newly created filesystems on floppy would cause fsck to barf.
> Thinking loud, would it be an idea to make and store a hash or a checksum of the first sector of the device, and use it to check for media change?
> 
> If you're going to read the disk for 1-2k to calculate a checksum,
> 
I was thinking the first sector, which would be very much faster than the seeks correctly done. The speed is as important as the message actually.
> you might as well just do a disk probe, since changing media while mounted is not supported in the kernel. Perhaps just removing the kernel probe message would be better. The message now serves as a [...]
I agree as to the value of the probe message - so I’m resting my case :-), it’s easy enough to take it out when the time comes to be really concerned about ‘regular users’.
> Also, not entirely letting go of the maxsize message from mkfs, how about adding a metric to it [blocks], to make it clearer what it is?
> 
> It's in bytes, and matches the image size. I'll add "bytes".
> 
> And, on fsck, maybe move the ID (version) message to the top, even in verbose mode.
> 
> Ok.
> 
> 
> 
Thanks. 
And - BTW - just slightly related (partitions and probes this time): I’m testing out the IDE probe code from directed.c in bioshd.c in order to get (yet) another source of info for the real CHS values. I hit the wall with /dev/bda not being the entire disk because I’m transferring an entire disk image to and from elks via Ethernet. So instead of brute forcing it, I figured let’s try 'IDE identify’ and see what we get. Should work for all systems AT and newer.

—Mellvik

ghaerr, 2020-09-02:
> Inode 6 not used, marked used in the bitmap.
Zone 304: marked in use, no file uses it.

I saw that sequence of errors, I think they were the result of the fsck bug. There weren't any changes made to mkfs that could affect that, but everything seems to be working now.

> I’m testing out the IDE probe code from directed.c in bioshd.c in order to get (yet) another source of info for the real CHS values. I hit the wall with /dev/bda not being the entire disk because [...]

I was going to ask where to find the IDE probe code. You'll want to test this on a number of drives as well as CF cards, as well as enable it only for PC/AT, which is done by testing "arch_cpu > 5". In addition, if this gets added permanently, we need to think about whether a config option is necessary around it all, as the BIOS driver is not supposed to query hardware directly, for those systems that might not actually be PC compatible.

That said, if an IDE query can solve the problem of setting the proper size of /dev/bda, that would be a good solution rather than having to use MBR partitions to size the raw device, given the large number of strange MBR partition tables I have already run into testing partitions.

Mellvik, 2020-09-03:
> I’m testing out the IDE probe code from directed.c in bioshd.c in order to get (yet) another source of info for the real CHS values. I hit the wall with /dev/bda not being the entire disk because [...]
There are a number of sanity checks that can ensure the validity of the IDE data, I'm coming back to that. The code is working, and I need some advice:
The get-identity IDE command needs a 512 byte temporary buffer. Directhd allocates that on the stack, which doesn't work for us. Since this is @ early boot time, there must be a statically alllocated buffer we can use for the purpose. Any suggestions?

--Mellvik

ghaerr, 2020-09-03:
> Since this is @ early boot time, there must be a statically alllocated buffer we can use for the purpose.

I assume you're doing this in `init_bioshd()`. Finding a static buffer (although there are many) to properly use for this is proving harder than I thought. Perhaps for now you can use the kernel local heap allocator:
```
#include <linuxmt/heap.h>
...
char *buffer = heap_alloc(512, 0);
...
heap_free(buffer);
```
This will allocate a kernel-addressable buffer off the local heap. The last parameter to heap_alloc is a usage tag, since this is very temporary, 0 will do. When we find a usable static buffer, it will be simpler to use that and remove this.

Mellvik, 2020-09-03:
Thanks @ghaerr,

exactly what was needed!

-M

> 3. sep. 2020 kl. 16:30 skrev Gregory Haerr <notifications@github.com>:
> 
> #include <linuxmt/heap.h>
> ...
> char *buffer = heap_alloc(512, 0);
> ...
> heap_free(buffer);

ghaerr, 2020-09-03:
Great. Lets close this issue and address further fsck issues on #464.

closed 2020-09-04 by ghaerr