ghaerr/elks #740: Enhancement: support more of 8390-based network adapters in ELKS

pawosm-arm, 2020-09-05:
Currently, only the ne2k cards are supported. As discussed in #716, ELKS ne2k driver is all about talking to 8390 chip. As more network cards are based on this chip, this driver can be extended to support more adapters (namely, the one I have, which is ISA 8-bit card SMC8003WC a.k.a WD8003). Alternatively, we could take Linux approach which exposed 8390-specific code common to all of the drivers (for 8390-based adapters) in the `lib8390.c` library.

ghaerr, 2020-09-05:
Sounds good! Are you going to take the first shot at this, it would be great to support more adapters!

I don't think it necessary to worry first about using a library, just use @Mellvik's driver as a start, and seeing what modifications are needed to start talking to the card, which might be a bit different. The core 8390 code, if needed, can be moved to a library once the card is working.

pawosm-arm, 2020-09-12:
Back to Manchester. I've installed this SMC8003WC card into my Amstrad PC2086 and although it behaves the same way under DOS (e.g. I can start and access `ftpsrv` of FreeDOS), there's a huge (surprisingly positive) difference in how it behaves under ELKS comparing to my other XT machine (the good old Turbo-XT). Back on Turbo-XT when I started `ktcp`, it always stuck on following loop in `ne2k_reset()` function in `ne2k-asm.S`:

```
nr_loop:
        // wait for reset
        // without too much CPU

        hlt

        in      %dx,%al
        test    $0x80,%al       // Wait for RST bit to set
        jz      nr_loop
```

Note that it wasn't the `hlt` instruction. When I removed `jz nr_loop` line, `ktcp` continued until it stuck again somewhere in `ktcp_run()`.
It's much better on Amstrad though: `ktcp` never blocks! ...but it doesn't work properly either. I have made following changes to find out what it does (and where it blocks when it blocks):

```
diff --git a/elkscmd/ktcp/config.h b/elkscmd/ktcp/config.h
index 8fd04236..aa6c856f 100644
--- a/elkscmd/ktcp/config.h
+++ b/elkscmd/ktcp/config.h
@@ -5,11 +5,11 @@
 #define CSLIP          1       /* compile in CSLIP support*/
 
 /* turn these on for ELKS debugging*/
-#define DEBUG_TCP      0
-#define DEBUG_IP       0
-#define DEBUG_ARP      0
-#define DEBUG_ETH      0
-#define DEBUG_CSLIP    0
+#define DEBUG_TCP      1
+#define DEBUG_IP       1
+#define DEBUG_ARP      1
+#define DEBUG_ETH      1
+#define DEBUG_CSLIP    1
 
 /* leave these off for now - too much info*/
 #define DEBUG_MEM      0               /* debug memory allocations*/
diff --git a/elkscmd/ktcp/ktcp.c b/elkscmd/ktcp/ktcp.c
index ea31cb2d..4957a180 100644
--- a/elkscmd/ktcp/ktcp.c
+++ b/elkscmd/ktcp/ktcp.c
@@ -215,13 +215,13 @@ printp();
        setsid();
     }
 
-    arp_init ();
-    ip_init();
-    icmp_init();
-    tcp_init();
-    netconf_init();
+    printf("arp_init()\n"); arp_init ();
+    printf("ip_init()\n"); ip_init();
+    printf("icmp_init()\n"); icmp_init();
+    printf("tcp_init()\n"); tcp_init();
+    printf("netconf_init()\n"); netconf_init();
 
-    ktcp_run();
+    printf("ktcp_run()\n"); ktcp_run();
 
     exit(0);
 }
```

What I'm observing is, ELKS isn't pingable from the local network, and any attempt to telnet remote host in my local network results in `arp` finding MAC of any remote IP being `ff:ff:ff:ff:ff:ff`. So it seems it doesn't talk to the card properly.

Also note that I had to change reading routine of local MAC address as it is completely different for ne2k than for smc/wd cards, e.g. for my card the MAC can be read like this (assuming `0x240` I/O port base):

```
for (i = 0; i < 6; i++)
                printk(" %2.2X", inb(0x240 + 8U + i));
```

I'll try to investigate it later, yet any suggestions from the current driver's author would be helpful.

Mellvik, 2020-09-12:
HI @pawosm-arm,

yes, this is indeed encouraging.

I assume your testing under DOS verifies both the operational status of the card and that the I/O address and the IRQ are what you expected them to be.

 
The I/O address you're reading the MAC address from indicates a different register structure than the ne2k, which is somewhat suspicious - if we're assuming that the controller chip is indeed the same. That said, it is encouraging that the reading of the command register seems to work.

Could you take a picture of the card close enough for the text on the chips to be readable? I'd like to see how much buffer memory is on the card and the actual make/part# of the controller chip, that would help a lot - in particular since we don't have any docs available yet.

Also be aware that the current driver assumes the card to be a 16bit card, it will not read/write data correctly to/from and 8 bit card. That's a very small change to make though, when the basics have been figured out. What I'm saying is that even if you get the address stuff right, there are other changes necessary in order to get ping to work. Actually, When ping works, you're in really good shape ...

—Mellvik

PS: I'd like to see the mac address you're getting from the card too…

pawosm-arm, 2020-09-12:
> I assume your testing under DOS verifies both the operational status of the card and that the I/O address and the IRQ are what you expected them to be.

Indeed. I'm also using DOS utilities to configure the card (I've selected SOFT config jumpers as all of the standard jumper-selectet options are conflicting with other devices on those XT machines).

What DOS driver (`PKT8000`) prints at start is:

```
Packet driver for SMC 8xxx adapters, version 11.3.1
Portions Copyright 1994, Standard Microsystems Corporation
Packet driver skeleton copyright 1988-93, Crynwr Software.
Licensed by SMC for distribution without complete source.

System: 8088/8086 processor, ISA bus
Packet driver software interrupt is 0x60 (96)
Frame type: Ethernet II (class 1), IEEE 802.3 (class 11)
Adapter: 8003WC
I/O port 0x240 (576)
Interrupt number 0x2 (2)
Memory address 0xCCE00 (839168)
Ethernet adapter, BNC media.
My Ethernet address is 00:00:C0:16:8F:43
```

Note that XT has only one IRQ controller (IRQs 0 to 7, usually all of them in use), hence IRQ2 is a normal interrupt line, not a cascade as on AT and later machines.

One thing this driver has wrong is that it's not `BNC` media, it's `UTP` media with `RJ-45` connector.

> The I/O address you're reading the MAC address from indicates a different register structure than the ne2k, which is somewhat suspicious - if we're assuming that the controller chip is indeed the same. That said, it is encouraging that the reading of the command register seems to work.

I've taken this method from Linux driver. The Linux driver is split into two parts: `lib8390` common for all of the 8390-based network cards (ne1k, ne2k, 3com 3c503, smc/wd 8003 and various clones). The MAC reading part differs across those cards hence it's not a part of `lib8390`, I've taken MAC reading code that works with my card from the `probe1()` function of `wd.c`, while ELKS ne2k's driver takes similar approach as in the `probe1()` function of `ne.c`.

> Also be aware that the current driver assumes the card to be a 16bit card, it will not read/write data correctly to/from and 8 bit card.

Ah, that's one of the thing I've forgotten to verify. Note that Linux driver is full of `if (ei_status.word16)` and alike.

> PS: I'd like to see the mac address you're getting from the card too…

With the method borrowed from `wd.c` it's the same as DOS finds, which is `00:00:C0:16:8F:43`. If I don't change the method and rely on what `ne2k-asm.S` originally has, it's `01:01:01:01:01:01`.

The picture of the card is attached to this message.

[image]

pawosm-arm, 2020-09-12:
Some more details.
To reflect hardware settings, I've modified ELKS as such:

```
diff --git a/elks/include/arch/ports.h b/elks/include/arch/ports.h
index a1666221..243643d8 100644
--- a/elks/include/arch/ports.h
+++ b/elks/include/arch/ports.h
@@ -53,8 +53,9 @@
 //#define CONFIG_NEED_IRQ8
 
 #ifdef CONFIG_ETH_NE2K
+#define CONFIG_NEED_IRQ2
 //#define CONFIG_NEED_IRQ9
-#define CONFIG_NEED_IRQ12
+//#define CONFIG_NEED_IRQ12
 #endif
 
 /* unused*/
@@ -99,8 +100,8 @@
 #define COM4_IRQ       2               /* unregistered unless COM4_PORT found*/
 
 /* ne2k, ne2k.c */
-#define NE2K_IRQ       12
-#define NE2K_PORT      0x300
+#define NE2K_IRQ       2
+#define NE2K_PORT      0x240
 
 /* obsolete - experimental IDE hard drive, directhd.c (broken)*/
 #define HD1_PORT       0x1f0
```

I've also tried to avoid 16-bit I/O accesses, yet I've only found the obvious places and modified them as such

```
diff --git a/elks/arch/i86/drivers/net/ne2k-asm.S b/elks/arch/i86/drivers/net/ne2k-asm.S
index 74c194f2..78053479 100644
--- a/elks/arch/i86/drivers/net/ne2k-asm.S
+++ b/elks/arch/i86/drivers/net/ne2k-asm.S
@@ -168,7 +168,7 @@ dma_write:
        inc     %cx     // make byte count even
        and     $0xfffe,%cx
        call    dma_init
-       shr     %cx     // half -> word size transf
+//     shr     %cx     // half -> word size transf
 
 
        // start DMA write
@@ -184,8 +184,8 @@ dma_write:
 
 emw_loop:
 
-       lodsw
-       out     %ax,%dx
+       lodsb
+       out     %al,%dx
        loop    emw_loop
 
        // wait for DMA completed
@@ -276,7 +276,7 @@ dma_read:
        inc     %cx     // make byte count even
        and     $0xfffe,%cx
        call    dma_init
-       shr     %cx     // half -> word size transf
+       // shr     %cx     // half -> word size transf
 
        // start DMA read
 
@@ -293,8 +293,8 @@ dma_read:
        cld
 
 emr_loop:
-       in      %dx,%ax
-       stosw
+       in      %dx,%al
+       stosb
        loop    emr_loop
 
        // wait for DMA to complete
```

Unfortunately, it didn't help much. Using `init 3` in ELKS (similar feature as `ctty COM1:` on DOS), I managed to produce log of what `ktcp` is doing as I tried to telnet the IP address 192.168.1.3:

```
ELKS 0.3.0

login: root
# ktcp -m 1500 -p eth -l /dev/eth 192.168.1.44 192.168.1.254 255.255.255.0
ktcp: ip 192.168.1.44, gateway 192.168.1.254, netmask 255.255.255.0
ktcp: ethernet 00.00.c0.16.8f.43 mtu 1500
arp_init()
ip_init()
icmp_init()
tcp_init()
netconf_init()
ktcp_run()
ktcp: update 0,0
ktcp: update 0,0
ktcp: update 0,0
IP: send 192.168.1.44 -> 192.168.1.3 v4 hl:5 tos:0 len:44 id:1 fo:0 chk:0 ttl:64 prot:6
TCP: send src:1024 dst:22 flags:2 seq:2f34d526 ack:0 win:255 urg:0 chk:0 len:24
arp: no cached entry for 192.168.1.3
arp: adding cache entry for 192.168.1.3, valid=0
eth: queueing packet len 44
arp: SEND request
ARP: op:1 ll:00.00.c0.16.8f.43->ff.ff.ff.ff.ff.ff 192.168.1.44->192.168.1.3 eth:00.00.c0.16.8f.43->00.00.00.00.00.00
eth: write 42 bytes: ff ff ff ff ff ff 00 00 c0 16 8f 43 08 06 00 01
08 00 06 04 00 01 00 00 c0 16 8f 43 c0 a8 01 2c
00 00 00 00 00 00 c0 a8 01 03
rtt 0 RTT 16 RTO 32
```

Mellvik, 2020-09-12:
Thanks @pawosm-arm,

this helps. And yes, the mechanism for reading MAC address will doffer from on card type to the next so the method (very complicated btw) of the ne2k will be useless in your case.

For now, why don't you just comment out the ne2k hw address routine and use your own.

Also, the 8bit change is very simple less than 10 lines to change in ne2k-asm.S
635 - change 49 to 48
297 - change to stosb
187 - change to lodsb
comment out lines 168, 169, 171, 276, 277, 279

That's it.
Then tx_first, rx_first, rx_last will certainly have to be changed, and many of the addresses above that.

The picture confirms the controller chip, which is another reason for optimism regarding reuse of most of the current driver. It has 8k bytes of buffer memory, which is half of what most n2k cards has, but no big deal.

I'll take another look at the linux driver to see if there are other obvious things to check out and change. 

—Mellvik

Mellvik, 2020-09-12:
NO this will not help, but you're moving in the right direction - see my other mail.

You need to figure out the differences in the register structure (most are the same, but such as the reset register (not a register, just a flip-flop on the card)) and the addresses of the shared (rx_start etc.)

I'll see what I can find out.

—M

Mellvik, 2020-09-12:
> 12. sep. 2020 kl. 18:53 skrev Helge Skrivervik <helge@mymayday.com>:

@pawosm-asm,
I've looked at the wd.c code, this is going to take some work.
It turns out the i/o mechanism is entirely different from the ne2k, using memory mapping instead of i/o port read/writes.
Not complicated, just different. I'll see if I can come up with some advice if you need it.

-m

pawosm-arm, 2020-09-12:
For what it's worth, I've found that I'm having the sources for the DOS packet driver. I've managed to google them as well: https://www.icc-al.org/iccenglish/support/hd%20web/crynwr/smcpkt5.zip - I'm using `PKT8000.COM` driver from this set which is apparently compiled from `UMAC.ASM` sources with `DR_TYPE` set to 106.

Also I was wondering how this card could behave differently on two different XT machines (hanging vs. not hanging on reset loop) running the same code, and this reminded me some article from long time ago about subtle differences between 8088 and 8086. Turbo-XT has original 8088 (which is externally an 8-bit CPU) while Amstrad PC2086 has 8086 clone (which is entirely 16-bit).

Mellvik, 2020-09-13:
Thanks @pawosm-arm,
This not only helps, but contributes to something else I've been contemplating - writing an elks driver for the xircom parallell adapter, which needs reverse engineering a packet driver. Seeing the source for one such beast is useful. The challenge has an interesting twist: Most parallell ports are one-way, write only, there is no provision for sending data. The trickery to get around this is rather interesting, and - as it turns out, reasonably efficient. 

@ghaerr, instead of searching for it, I'm taking the shortcut and asking you: How do we efficiently - in kernel space - transfer data to/from upper memory (like the shared memory on this card @ CCE00)?

--Mellvik

pawosm-arm, 2020-09-13:
OT:

> The challenge has an interesting twist: Most parallell ports are one-way, write only, there is no provision for sending data. The trickery to get around this is rather interesting, and - as it turns out, reasonably efficient.

Even the simplest uni-directional parallel port in PC has five input lines (ack, busy, paper end, select, error). Long time ago I was using specially crafted parallel port cable as a faster and more reliable replacement to null-modem serial communication. AFAIR, the full-duplex protocol was splitting bytes in halfs to send them over four of the lines, the fifth line was used as strobe signal, the acknowledgement received from the other side required two lines (ack and data ready signals, the level of data ready signal could change the state of the other side from sending to receiving) to achieve two-way communication.

pawosm-arm, 2020-09-13:
> (like the shared memory on this card @ CCE00)

I'm not sure where this 0xCCE00 comes from, `ezsetup` that I'm using to configure this card has it at 0xCE000 (the only free piece of the upper address space I've got, after lo-tech's XT-IDE CF adapter at C8, yet before 64k UMB at D0, 64k EMS window at E0 and 64k of ROM at F0).

```
c:\smc8003>ezsetup
SMC LAN Adapter Setup Program -- Version 1.22 (960731)

Board Type:   8003WC
Node Address: 0000C0168F43

                     Current Setup

I/O Base Address     240
IRQ                  2
RAM Size             8  K
WIN Size             8 K
RAM Base Address     0CE000
Add Wait States      Yes
Network Connection   BNC or 10BaseT

ROM Size             Disabled
ROM Base Address     Disabled

Do you want to change the setup ? (y) -> n

c:\smc8003>
```

Mellvik, 2020-09-13:
OK, sounds like you’ll have to experiment with this in the code, which is a lot faster than trying to debug a probe routine (a task for later).

I’ll look more at the sample wd.c code tomorrow.

—Mellvik

ghaerr, 2020-09-13:
> How do we efficiently - in kernel space - transfer data to/from upper memory (like the shared memory on this card @ CCE00)?

`fmemcpyw` or `fmemcpyb` can be used for that, as well as possibly other functions. For these two, the segment and offset of the source and destinations are passed separately.

Mellvik, 2020-09-14:
Thanks @ghaerr.
@pawosm-arm - I suggest the following progression towards a working elks wd driver:
-- NIC initialization: when interrupts start ticking, initialization is OK
-- read ICMP packets
-- write ICMP packets -> ping works
-- At this stage it is likely that TCP works too, but it has never (?) been tested on such a 'small' machine before.

I will try to assist you the best I can along the way. 

--Mellvik

pawosm-arm, 2020-09-14:
@Mellvik is there a reason why substantial part of current ne2k driver in ELKS is written in assembly? IMHO it would be easier to rewrite it in C...

Mellvik, 2020-09-14:
@pawosm-arm,
there is a substantial benefit in terms of performance and space, both of which are particularly important for ELKS, given the limited resources of the hardware it is supposed to support.

—Mellvik

pawosm-arm, 2020-09-14:
To make it easier for me, I'll create a new driver next to the existing one and write it in C and see how far I can go with this:

```
diff --git a/elks/arch/i86/drivers/net/Makefile b/elks/arch/i86/drivers/net/Makefile
index 8755af9d..dffb45f7 100644
--- a/elks/arch/i86/drivers/net/Makefile
+++ b/elks/arch/i86/drivers/net/Makefile
@@ -31,6 +31,9 @@ OBJS = net_drv.o
 ifeq ($(CONFIG_ETH_NE2K), y)
 OBJS += ne2k-asm.o ne2k.o
 endif
+ifeq ($(CONFIG_ETH_WD), y)
+OBJS += wd.o
+endif
 
 all: net_drv.a
 
diff --git a/elks/arch/i86/drivers/net/config.in b/elks/arch/i86/drivers/net/config.in
index ffee503c..5ca46112 100644
--- a/elks/arch/i86/drivers/net/config.in
+++ b/elks/arch/i86/drivers/net/config.in
@@ -7,7 +7,9 @@ mainmenu_option next_comment
        if [ "$CONFIG_ARCH_SIBO" != "y" ]; then
                bool 'Ethernet device support' CONFIG_ETH n
                if [ "$CONFIG_ETH" = "y" ]; then
-                       bool 'NE2K device driver' CONFIG_ETH_NE2K n
+                       choice 'Select Ethernet device' \
+                               "ne2k   CONFIG_ETH_NE2K \
+                                WD/SMC CONFIG_ETH_WD" ne2k
                fi
        fi
 endmenu
diff --git a/elks/arch/i86/drivers/net/net_drv.c b/elks/arch/i86/drivers/net/net_drv.c
index c9c7f7a8..c4233cee 100644
--- a/elks/arch/i86/drivers/net/net_drv.c
+++ b/elks/arch/i86/drivers/net/net_drv.c
@@ -8,4 +8,7 @@ void eth_init(void)
 #ifdef CONFIG_ETH_NE2K
        eth_drv_init();
 #endif
+#ifdef CONFIG_ETH_WD
+       eth_drv_init();
+#endif
 }
diff --git a/elks/arch/i86/drivers/net/wd.h b/elks/arch/i86/drivers/net/wd.h
new file mode 100644
index 00000000..dd32717f
--- /dev/null
+++ b/elks/arch/i86/drivers/net/wd.h
@@ -0,0 +1,6 @@
+#ifndef WD_H
+#define WD_H
+
+#include <arch/ports.h>
+
+#endif /* WD_H */
diff --git a/elks/include/arch/ports.h b/elks/include/arch/ports.h
index a1666221..04065b0a 100644
--- a/elks/include/arch/ports.h
+++ b/elks/include/arch/ports.h
@@ -6,7 +6,7 @@
  *  IRQ        ELKS Device         Config Option           Status
  *  0   Timer                                       Required
  *  1   Keyboard            CONFIG_CONSOLE_DIRECT   Optional
- *  2*  Cascade -> 9 AT, Unused on XT               Optional on XT, not available on AT
+ *  2*  Cascade -> 9 on AT  CONFIG_ETH_WD           Optional on XT, not available on AT
  *  3   Com2 (/dev/ttyS1)   CONFIG_CHAR_DEV_RS      Optional
  *  4   Com1 (/dev/ttyS0)   CONFIG_CHAR_DEV_RS      Optional
  *  5*  Unused
@@ -57,6 +57,10 @@
 #define CONFIG_NEED_IRQ12
 #endif
 
+#ifdef CONFIG_ETH_WD
+#define CONFIG_NEED_IRQ2
+#endif
+
 /* unused*/
 //#define CONFIG_NEED_IRQ10
 //#define CONFIG_NEED_IRQ11
@@ -102,6 +106,10 @@
 #define NE2K_IRQ       12
 #define NE2K_PORT      0x300
 
+/* wd, wd.c*/
+#define WD_IRQ         2
+#define WD_PORT                0x240
+
 /* obsolete - experimental IDE hard drive, directhd.c (broken)*/
 #define HD1_PORT       0x1f0
 #define HD2_PORT       0x170
```

Mellvik, 2020-09-14:
@pawosm-arm,
That is your choice to make. I do not think it's a good idea, but I'll try to help you when I can.

- Mellvik

ghaerr, 2020-09-15:
Hello @pawosm-arm,

The above structure is a nice starting framework, regardless of whether you write the driver in C or ASM. I'm not sure you'll need a separate wd.h file, if your driver is contained in a single wd.c file.

I don't know whether writing the driver in C is a good or bad idea, as I haven't really studied how much bit-twiddling the driver does. It certainly would be more understandable and portable in C though. You can always check the ASM output from your C driver by using `ia16-elf-objdump -D -r -Mi8086` on wd.o, which I have setup as an alias to make it easy to use.

pawosm-arm, 2020-09-15:
> The above structure is a nice starting framework, regardless of whether you write the driver in C or ASM.

At least (the C part missing above) it responds to `ktcp`'s `ioctl`s and doesn't make it hang.

> I'm not sure you'll need a separate wd.h file, if your driver is contained in a single wd.c file.

I've noticed that and removed `wd.h` already.

> I don't know whether writing the driver in C is a good or bad idea, as I haven't really studied how much bit-twiddling the driver does.

I can always put inline asm for things C can't do. Also ELKS offers macros with inline assembly that I may find useful.

pawosm-arm, 2020-09-15:
The IRQ handler in `lib8390` (Linux), which is `ei_interrupt()` function is a loop like this:

```
    while ((interrupts = inb_p(e8390_base + EN0_ISR)) != 0
                   && ++nr_serviced < MAX_SERVICE) {
      ....
    }
```

...where `MAX_SERVICE` is 12.

There's no such loop in ELKS ne2k driver, so it behaves as if MAX_SERVICE was set to 1. The interrupts on ISA bus are edge triggered. Would it lead to a case of missing interrupts?

pawosm-arm, 2020-09-15:
...and ne2k IRQ handler in ELKS does not handle 8390 error statuses!

ghaerr, 2020-09-15:
> The IRQ handler in lib8390 (Linux), which is ei_interrupt() function is a loop like this

Linux supports shared interrupts between devices, I think that is the reason for such code. ELKS is much simpler, and no loop will be necessary. All interrupts will go to the routine specified in `request_irq`. As was heavily discussed in @Mellvik's development of the NE2K driver, be careful: the interrupt routine may be executed while the kernel is in the middle of your other driver code, so that code must be either well thought out, or protected with cli/sti pairs. (See arch/irq.h for C versions).

pawosm-arm, 2020-09-16:
I'm stuck. Following code (as seen in `wd.c`, part of `open` operation in `file_operations`) makes entire system hang right after starting `ktcp`:

```
#define WD_PORT 0x240
#define WD_IO_EXTENT    32U
#define WD_8390_OFFSET  16U
#define WD_8390_PORT    (WD_PORT + WD_8390_OFFSET)

        outb_p(E8390_NODMA | E8390_PAGE0 | E8390_STOP,
                WD_8390_PORT + E8390_CMD);
```

The only thing that happens before that is RESET:
```
#define WD_RESET        0x80

        outb(WD_RESET, WD_PORT);
```

pawosm-arm, 2020-09-16:
I also tried to replace `outb_p()` with `outb()`, but it didn't change my bad luck... still something causes the system to hang.

ghaerr, 2020-09-16:
I can't see the value of E8390_CMD, is it set properly? I am unaware of the 8390's register stack and memory map.

Use `ia16-elf-objdump -D -r -Mi8086 wd.o` to check it is exactly the code you are expecting.

Perhaps use literal values rather than symbol + symbol to see if that changes the compiled code output.

pawosm-arm, 2020-09-16:
> I can't see the value of E8390_CMD, is it set properly? I am unaware of the 8390's register stack and memory map.

Ah, I've forgotten to paste that:

```
#define WD_SHMEM        0xce000U

#define WD_STAT_RX      0x0001U /* packet received */
#define WD_STAT_TX      0x0002U /* packet sent */
#define WD_STAT_OF      0x0010U /* RX ring overflow */

#define WD_START_PG     0x00U   /* First page of TX buffer */
#define WD_STOP_PG      0x20U   /* Last page + 1 of RX ring */

#define TX_2X_PAGES     12U
#define TX_1X_PAGES     6U
#define TX_PAGES        TX_2X_PAGES

#define WD_RESET        0x80U   /* Board reset */
#define WD_MEMENB       0x40U   /* Enable the shared memory */
#define WD_IO_EXTENT    32U
#define WD_8390_OFFSET  16U
#define WD_8390_PORT    (WD_PORT + WD_8390_OFFSET)

#define E8390_RXCONFIG  0x04U   /* EN0_RXCR: broadcasts, no multicast,errors */
#define E8390_RXOFF     0x20U   /* EN0_RXCR: Accept no packets */
#define E8390_TXCONFIG  0x00U   /* EN0_TXCR: Normal transmit mode */
#define E8390_TXOFF     0x02U   /* EN0_TXCR: Transmitter off */
#define E8390_STOP      0x01U   /* Stop and reset the chip */
#define E8390_START     0x02U   /* Start the chip, clear reset */
#define E8390_TRANS     0x04U   /* Transmit a frame */
#define E8390_RREAD     0x08U   /* Remote read */
#define E8390_RWRITE    0x10U   /* Remote write  */
#define E8390_NODMA     0x20U   /* Remote DMA */
#define E8390_PAGE0     0x00U   /* Select page chip registers */
#define E8390_PAGE1     0x40U   /* using the two high-order bits */
#define E8390_PAGE2     0x80U   /* Page 3 is invalid */
#define E8390_CMD       0x00U   /* The command register (for all pages) */

/* For register EN0_ISR. */
#define E8390_TX_IRQ_MASK       0x0aU
#define E8390_RX_IRQ_MASK       0x05U

/* Page 0 register offsets. */
#define EN0_CLDALO      0x01U   /* Low byte of current local dma addr  RD */
#define EN0_STARTPG     0x01U   /* Starting page of ring bfr WR */
#define EN0_CLDAHI      0x02U   /* High byte of current local dma addr  RD */
#define EN0_STOPPG      0x02U   /* Ending page +1 of ring bfr WR */
#define EN0_BOUNDARY    0x03U   /* Boundary page of ring bfr RD WR */
#define EN0_TSR         0x04U   /* Transmit status reg RD */
#define EN0_TPSR        0x04U   /* Transmit starting page WR */
#define EN0_NCR         0x05U   /* Number of collision reg RD */
#define EN0_TCNTLO      0x05U   /* Low  byte of tx byte count WR */
#define EN0_FIFO        0x06U   /* FIFO RD */
#define EN0_TCNTHI      0x06U   /* High byte of tx byte count WR */
#define EN0_ISR         0x07U   /* Interrupt status reg RD WR */
#define EN0_CRDALO      0x08U   /* low byte of current remote dma address RD */
#define EN0_RSARLO      0x08U   /* Remote start address reg 0 */
#define EN0_CRDAHI      0x09U   /* high byte, current remote dma address RD */
#define EN0_RSARHI      0x09U   /* Remote start address reg 1 */
#define EN0_RCNTLO      0x0aU   /* Remote byte count reg WR */
#define EN0_RCNTHI      0x0bU   /* Remote byte count reg WR */
#define EN0_RSR         0x0cU   /* rx status reg RD */
#define EN0_RXCR        0x0cU   /* RX configuration reg WR */
#define EN0_TXCR        0x0dU   /* TX configuration reg WR */
#define EN0_COUNTER0    0x0dU   /* Rcv alignment error counter RD */
#define EN0_DCFG        0x0eU   /* Data configuration reg WR */
#define EN0_COUNTER1    0x0eU   /* Rcv CRC error counter RD */
#define EN0_IMR         0x0fU   /* Interrupt mask reg WR */
#define EN0_COUNTER2    0x0fU   /* Rcv missed frame error counter RD */

/* Page 1 register offsets. */
#define EN1_PHYS        0x01U   /* This board's physical enet addr RD WR */
#define EN1_CURPAG      0x07U   /* Current memory page RD WR */
#define EN1_MULT        0x08U   /* Multicast filter mask array (8 bytes) RDWR */

/* Bits in EN0_ISR - Interrupt status register. */
#define ENISR_RX        0x01U   /* Receiver, no error */
#define ENISR_TX        0x02U   /* Transmitter, no error */
#define ENISR_RX_ERR    0x04U   /* Receiver, with error */
#define ENISR_TX_ERR    0x08U   /* Transmitter, with error */
#define ENISR_OVER      0x10U   /* Receiver overwrote the ring */
#define ENISR_COUNTERS  0x20U   /* Counters need emptying */
#define ENISR_RDC       0x40U   /* remote dma complete */
#define ENISR_RESET     0x80U   /* Reset completed */
#define ENISR_ALL       0x3fU   /* Interrupts we will enable */
```

Mellvik, 2020-09-16:
@pawosm-arm,

you need to check the WD_RESET address,
it's most definitely wrong. If I remember correctly, IO port 0x80  is used in all kinds of contexts as a dummy port. The reset port should be a port in the IO address space of the card.

—Mellvik

pawosm-arm, 2020-09-16:
> If I remember correctly, IO port 0x80  is used in all kinds of contexts as a dummy port.

`0x80` is a value here, not the port address:

```
#define outb(value,port) \
__asm__ ("outb %%al,%%dx"::"Ral" ((unsigned char)(value)),"d" (port))
```

ghaerr, 2020-09-16:
> you need to check the WD_RESET address,
> it's most definitely wrong. If I remember correctly, IO port 0x80  is used in all kinds of contexts as a dummy port. The reset port should be a port in the IO address space of the card.

A couple different things going on here: in @pawosm-arm's code, he's sending 0x80 to WD_PORT, which is 0x250 (0x240 + 16).
Secondly, the outb_p() macro, when used, outputs AL to port 0x80 (not used here since outb() being used).

I would suggest removing both outb() and making sure no crash. Then adding the first (reset), then the second call.
Once confirmed the second call is failing, use ia16-elf-objdump to ensure proper code being generated.

pawosm-arm, 2020-09-16:
Oh my... something else was missing!

```
        outb(((WD_SHMEM >> 13U) & 0x3fU) | WD_MEMENB, WD_PORT);
```
...should be right after `WD_RESET`.

pawosm-arm, 2020-09-16:
Ok then, I'm attaching something that doesn't explode and prints `/01` when pinged from remote host (it's alive!). It fits to the 'framework' posted above.
[wd.c.gz](https://github.com/jbruchon/elks/files/5233642/wd.c.gz)

Now I need to design how to fit the rest of Linux's `wd.c` into that. Any comments and suggestions welcomed.

Mellvik, 2020-09-16:
Ahhh, I missed that.
Then, here's another suggestion:
The normal (my experience) is that the reset port is a HW flopflop toggling a physical line on the card to reset everything. It doesn't matter what you write to it, it's the write operation that matters. The WD_RESET value is most likely the value you need to wait for in the command reg after toggling the reset port.

This procedure is the same for the ne2k card and the 3c509 I'm working on.

--Mellvik

pawosm-arm, 2020-09-16:
> The WD_RESET value is most likely the value you need to wait for in the command reg after toggling the reset port.

Waiting in a loop (as in `ne2k.c`) made `ktcp` hang while I was testing it on my other PC (Turbo-XT), and `wd.c` on Linux doesn't do that...

Mellvik, 2020-09-16:
Correction:
The reset complete (0x80) should be read from the interrupt status reg, not the command reg.

From looking at the defines, when you have figured out the shared mem read write, the rest seems equal to the ne2k, as it should be since it's the same chip..

-M

pawosm-arm, 2020-09-22:
Got stuck again... Following code:

```
typedef struct {
        unsigned char status;   /* status */
        unsigned char next;     /* pointer to next packet */
        unsigned short count;   /* header + packet length in bytes */
} __attribute__((packed)) e8390_pkt_hdr;

static e8390_pkt_hdr rx_hdr;
static byte_t recv_buf[MAX_PACKET_ETH + 4U];
static byte_t send_buf[MAX_PACKET_ETH];

...

                /* Get the rx page (incoming packet pointer). */
                outb_p(E8390_NODMA | E8390_PAGE1, WD_8390_PORT + E8390_CMD);
                rxing_page = inb_p(WD_8390_PORT + EN1_CURPAG);
                outb_p(E8390_NODMA | E8390_PAGE0, WD_8390_PORT + E8390_CMD);
                /* Remove one frame from the ring. */
                /* Boundary is always a page behind. */
                this_frame = inb_p(WD_8390_PORT + EN0_BOUNDARY) + 1U;
                if (this_frame >= WD_STOP_PG)
                        this_frame = WD_FIRST_RX_PG;
                if (this_frame != current_rx_page)
                        debug_eth("Mismatched read page pointers %2x vs %2x.\n",
                                this_frame, current_rx_page);
                if (this_frame == rxing_page) /* Read all the frames? */
                        break; /* Done for now */
                hdr_start = WD_SHMEM + ((this_frame - WD_START_PG) << 8U);
                fmemcpyb((byte_t *)(&rx_hdr), kernel_ds,
                        (byte_t *)(_FP_OFF(hdr_start)), _FP_SEG(hdr_start),
                        sizeof(e8390_pkt_hdr));
                next_frame = this_frame + 1U + (rx_hdr.count >> 8U);
                /* Check for bogosity warned by 3c503 book. */
                if (rx_hdr.next != next_frame
                        && rx_hdr.next != (next_frame + 1U)
                        && rx_hdr.next != (next_frame - RX_PAGES)
                        && rx_hdr.next != (next_frame + 1U - RX_PAGES)) {
                        current_rx_page = rxing_page;
                        outb(current_rx_page - 1U, WD_8390_PORT + EN0_BOUNDARY);
                        continue;
                }
                if ((rx_hdr.count < 64U) ||
                    (rx_hdr.count > (MAX_PACKET_ETH + sizeof(e8390_pkt_hdr)))) {
                        debug_eth("Eth: bogus packet size: %d, "
                                "status = %#2x nxpg = %#2x.\n",
                                rx_hdr.count, rx_hdr.status, rx_hdr.next);
                        break;
                } else if ((rx_hdr.status & 0x0fU) == ENRSR_RXOK) {
                        fmemcpyb(pack, kernel_ds,
                                (byte_t *)(_FP_OFF(hdr_start)),
                                _FP_SEG(hdr_start),
                                rx_hdr.count);
                } else {
                        debug_eth("Eth: bogus packet: "
                                "status = %#2x nxpg = %#2x size = %d\n",
                                rx_hdr.status, rx_hdr.next, rx_hdr.count);
                }
```

...produces following error:

```
Eth: bogus packet size: -15229, status = 9b nxpg = b6.
```

Which suggests, `fmemcpyb` does not do what I expected it to do...

As a side note, I've tried two `WD_SHMEM` values (`0xce000`, `0xcce00`), it resulted in different values printed with `bogus packet size` message, yet they were still terribly incorrect.

ghaerr, 2020-09-22:
Hello @pawosm-arm,

> Which suggests, fmemcpyb does not do what I expected it to do...

Actually, your code looks pretty good, and mostly correct. fmemcpyb is working, but I'm guessing you're probably not sending it the correct segment and offset values.

> I've tried two WD_SHMEM values (0xce000, 0xcce00)

It seems you're using "linear" (32-bit) rather than segment values for WD_SHMEM. For ELKS, it is better to keep the segment and offset values separate, e.g.:
```
#define WD_SHMEMSEG    0xce00      // segment (not linear) value for WD shared memory (NOT 0xce000)
word_t hdr_start_off = (this_frame - WD_START_PG) << 8;
...
  fmemcpyb((byte_t *)(&rx_hdr), kernel_ds,
                        (byte_t *)hdr_start_off, WD_SHMEMSEG,
                        sizeof(e8390_pkt_hdr));
```
You'll need to convert any other Linux 32-bit (pointer) values in the same way to ELKS segment:offset. Remember that a segment value is the upper 16 bits of a 32-bit word, shifted right 4.

You might also be interested in using the `hd` program to directly inspect PC memory, it can now be passed a segment:offset directly, which really helps seeing what's going on inside ELKS:
```
hd ce00:0#512    <- will dump 512 bytes of hex at ce00:0000 to console
```

I hope this helps.

Thank you!

Mellvik, 2020-09-22:
@pawosm-arm,
Looks like you're making progress. I was going to suggest doing some tests to verify the correct memory address, and @ghaerr not only beat me to it, but taught me a much more elegant way. Thanks @ghaerr.

To me the code seems overly complicated. I suggest you kick out the 3c503 'suggestions'. Your next_frame should always be set to rx_hdr.next, and NEVER change the front ('current') pointer!. This is what the NIC chip is good at, if these don't work right, the card cannot possibly work.
Also, if you initialize the NIC to drop error frames and frames with odd sizes you can skip those tests too.

--M

pawosm-arm, 2020-09-22:
Thanks @ghaerr, now ktcp receives arps and MAC addresses of remote hosts pinging ELKS are transferred correctly. I also followed @Mellvik's suggestion and kicked out 3c503 checkups (next frame is always `rx_hdr.next` now) - no observed changes though. Now the problem is on the other side: send does not seem to do anything:

```
static int wd_pack_put(byte_t * pack, word_t len)
{
        fmemcpyb((byte_t *)((WD_FIRST_TX_PG - WD_START_PG) << 8U), WD_SHMEMSEG,
                pack, kernel_ds, len);
        outb_p(E8390_NODMA | E8390_PAGE0, WD_8390_PORT + E8390_CMD);
        if (inb_p(WD_8390_PORT + E8390_CMD) & E8390_TRANS) {
                printk("Eth: attempted send with the transmitter busy.\n");
                return -EIO;
        }
        outb_p(len & 0xffU, WD_8390_PORT + EN0_TCNTLO);
        outb_p(len >> 8U, WD_8390_PORT + EN0_TCNTHI);
        outb_p(WD_FIRST_TX_PG, WD_8390_PORT + EN0_TPSR);
        outb_p(E8390_NODMA | E8390_TRANS | E8390_START,
                WD_8390_PORT + E8390_CMD);
        return 0;
}

static word_t wd_tx_stat(void)
{
        return (inb_p(WD_8390_PORT + E8390_CMD) & E8390_TRANS) ? WD_STAT_TX :
                0U;
}

static size_t wd_write(struct inode * inode, struct file * file,
        char * data, size_t len)
{
        size_t res = 0U;

        do {
                prepare_to_wait_interruptible(&txwait);
                if (wd_tx_stat() != WD_STAT_TX) {
                        if (file->f_flags & O_NONBLOCK) {
                                res = -EAGAIN;
                                break;
                        }
                        do_wait();
                        if(current->signal) {
                                res = -EINTR;
                                break;
                        }
                }
                if (len > MAX_PACKET_ETH)
                        len = MAX_PACKET_ETH;
                memcpy_fromfs(send_buf, data, len);
                res = len;
                if (len < 64U) len = 64U;  /* issue #133 */
                if (wd_pack_put(send_buf, len)) {
                        res = -EIO;
                        break;
                }
        } while (0);
        finish_wait(&txwait);
        return res;
}
```

pawosm-arm, 2020-09-22:
`wd_tx_stat()` seems to have inverted logic... I'll try to reverse it.

pawosm-arm, 2020-09-22:
IT WOOOORKS!!!!

pawosm-arm, 2020-09-22:
Attached is the code that works for me.
[wd.c.gz](https://github.com/jbruchon/elks/files/5263783/wd.c.gz)

```
$ telnet 192.168.1.44
Trying 192.168.1.44...
Connected to 192.168.1.44.
Escape character is '^]'.
login: root
# ps
  PID   GRP  TTY USER STAT CSEG DSEG  HEAP   FREE   SIZE COMMAND
    1     0      root    S 3ee1 408c  3072   2014  15648 /bin/init 
   12    12    1 root    S 4b19 5654  1166   6746  64544 -/bin/sh 
   19    17      root    R 473a 48ba     0   1976  12336 telnetd 
   15    15      root    S 5adb 79e3  9376  26402  75376 ktcp -b -d -m 1500 -p eth -l /dev/eth 192.168.1.44 192.168.1.254 255.255.255.0 
   17    17      root    S 473a 42b3     0   2006  12336 telnetd 
   20    20   p0 root    S 4b19 8f42  1166   6818  64544 -/bin/sh 
   21    20   p0 root    R 4436 8c36  1024   6428  18592 ps 
# meminfo 
  HEAP   TYPE  SIZE    SEG   TYPE    SIZE  CNT
  d9de   SEG     16   1ee1   BUF    65536    1
  d9fa   SEG     16   2ee1   BUF    65536    1
  da16   SEG     16   3ee1   CSEG    6832    1
  da32   SEG     16   408c   DSEG    8816    1
  da4e   SEG     16   42b3   DSEG    6192    1
  da6a   TTY    128
  daf6   TTY     64
  db42   SEG     16   5f76   free  108240    0
  db5e   SEG     16   79e3   DSEG   56512    1
  db7a   SEG     16   4b19   CSEG   46000    2
  db96   SEG     16   5654   DSEG   18544    1
  dbb2   SEG     16   87af   free   18544    0
  dbce   SEG     16   5adb   CSEG   18864    1
  dbea   SEG     16   473a   CSEG    6144    2
  dc06   SEG     16   48ba   DSEG    6192    1
  dc22   SEG     16   4a3d   free    3520    0
  dc3e   SEG     16   4436   CSEG    4544    1
  dc5a   SEG     16   93c9   free   50032    0
  dc76   SEG     16   8f42   DSEG   18544    1
  dc92   TTY     64
  dcde   TTY    256
  ddea   SEG     16   8c36   DSEG   12112    1
  de06   SEG     16   4552   free    7808    0
  de22   SEG     16   8f2b   free     368    0
  de3e   free  8642
  Total heap   9774     Total mem  528880
  Memory usage  516KB total,  332KB used,  184KB free
#
```

I guess it's a good candidate for development of a wider 8390-based driver.

ghaerr, 2020-09-22:
Wow @pawosm-arm!!!!! TOTALLY COOL!!!!!!

ghaerr, 2020-09-22:
Hello @pawosm-arm,

I have reviewed your wd.c, and I have to say, I am quite impressed. A new ethernet driver in a single .c file. Very nice.

Being able to get this done so quickly is a testament to the portability of well-written C code, your programming prowess, @Mellvik's work on getting the first driver done and a lot of TCP/IP testing and bug fixes, and to ELKS in general for actually starting to work well.

I think I had as much fun reading "IT WOOOORKS" as you did :) I'm really pleased to see highly-complex code coming in from Linux, and usable within our ELKS framework, thanks to you. I can't wait to have this committed!

Thank you!!

pawosm-arm, 2020-09-22:
@ghaerr it's #760

ghaerr, 2021-10-19:
Hello @pawosm-arm,

Are you still lurking out there? Given the support for `netport=` recently in #974, I was wondering if the same enhancement would be useful for the WD series NICs, or if they're almost always at 0x240. It looks pretty straightforward to code it up, which I can do, but I don't have any cards with which to test. What do you think?

Thank you!

pawosm-arm, 2021-10-19:
Hi @ghaerr, there is a DOS utility I have mentioned earlier for this card that can change any hardware setting of it, including I/O port base. Thing is, after inserting all of those cards into an XT machine, there's not much ports left, so I'm always using 0x240.
Other than that, not much time for playing with old computers recently :(

ghaerr, 2021-10-19:
> so I'm always using 0x240.

Ok, we'll leave the variable netport= for WD8003's in my long TODO list for now then.

> Other than that, not much time for playing with old computers recently :(

Me too, but I did happen to win a "brand new" Compaq LTE/20 (386) from 1990 on eBay the other day for $100, which seems to want to boot. I've ordered a floppy emulator for it, and plan on it being the first ELKS laptop. (Pictures coming when complete):
[image]

closed 2022-06-16 by ghaerr