ghaerr/elks #755: TCP retransmit issue

Mellvik, 2020-09-21:
While debugging some odd network send issues that may be ne2k driver related, I came across this one. And as much as I'd like to just fix it and go on, I figure it's useful with some advice.

Here's the scenario:
The outside server is pulling data (lots) from the httpd on ELKS. This was reasonably stable a while back, now I'm seeing regular problems, and this is one of them:
 - transfer is running smoothly, one packet in, one out.
 - then ELKS gets an ARP, and replies in stride
 - then two packets later trouble starts and the sequence goes like this (numbers just to illustrate the sequence):
```
...
ELKS                        server
data 101 ------------->
        <----------------   ACK 101
        <----------------   ARP req
data 102 ------------->
        <----------------    ACK 102
ARP reply ------------>
data 103 ------------->
       <-----------------   ACK 103
data 104 -------------> 
       <-----------------   ACK 104
data 103 ------------->                           retransmit
       <-----------------   ACK 104
data 105 ------------->
       <-----------------   ACK 105
data 104 ------------->                          retransmit
      <-----------------    ACK 105
data 106 ------------->
     <-----------------     ACK 106
data 105 ------------->                         retransmit
     <-----------------     ACK 106
...
```
... and so on a little longer until `ktcp: panic in write` terminates transmission.
For every retransmit there is a `retrans retry #1 rto 4 mem 360` message with increasing numbers, starting @ 369, ending @ 1988.

Any hints very welcome.

--M

ghaerr, 2020-09-21:
Hi @Mellvik,

> This was reasonably stable a while back, now I'm seeing regular problems, and this is one of them

I'm pretty sure this is the same nagging problem ktcp has had all along - for some reason, whenever ktcp gets tripped up and has to start doing retransmits, everything goes to heck. I have mentioned earlier that, for instance, when ktcp is forced to drop a packet (when either its ring buffer is full or an application hasn't run long enough to empty that buffer), ktcp drops the packet and the same thing happens.

> For every retransmit there is a retrans retry #1 rto 4 mem 360 message with increasing numbers

I hadn't noticed that the retransmits end up allocating yet more memory - they should just retransmit from the stored TCP packet, that's it. Although what could be happening is ALL the subsequent packets end up going through a alloc/retrans process.

The retransmit stuff is all in tcp_output.c. The function tcp_retrans() is called by the ktcp main loop every half second or so to check for retransmit needed, and then calls tcp_reoutput if needed. Taking a quick look here shows that it is NOT reallocating the packet, but just calling ip_sendpacket on it (and the cycle repeats, it seems).

In order to fix this longstanding problem, I'm thinking we need to determine exactly why it starts. I think what is happening is that the ACK/expected ACK don't match, although not sure why. The exact place where this is checked is in tcp_output.c, in the tcp_retrans() function:
```
    n = retrans_list;
    while (n != NULL) {
        datalen = n->len - TCP_DATAOFF(&n->tcphdr[0]);

        //debug_tcp("retrans: %lx %lx", ntohl(n->tcphdr[0].seqnum) + datalen,n->cb->send_una);
        if (SEQ_LEQ(ntohl(n->tcphdr[0].seqnum) + datalen, n->cb->send_una)) { <---- HERE
            if (n->retrans_num == 0) {
                rtt = Now - n->first_trans;
                if (rtt > 0)
                    n->cb->rtt = (TCP_RTT_ALPHA * n->cb->rtt + (100 - TCP_RTT_ALPHA) * rtt) / 100;
                debug_tcp("rtt %d RTT %ld RTO %ld\n", rtt, n->cb->rtt, n->rto);
            }
            n = rmv_from_retrans(n);
            continue;
        }

//debug_tcp("retrans %d mem %d\n", tcp_timeruse, tcp_retrans_memory);
        if (TIME_GEQ(Now, n->next_retrans)) {  <--- this times out for a bit then retransmits the non-ACK'd packet
            tcp_reoutput(n);
            return;
        }
        n = n->next;
    }
```
There are a series of SEQ_LEQ etc macros that are used to compare the 32-bit sequence numbers, and retransmit accordingly.

The received ACK'd sequence numbers are updated in the tcp.c code in the tcp_established() state handler:
```
    if (h->flags & TF_ACK) {    /*update unacked*/
        acknum = ntohl(h->acknum);
        if (SEQ_LT(cb->send_una, acknum))
            cb->send_una = acknum;   <- this updates the ACKed data on ktcp's side
    }
...
    cb->rcv_nxt += datasize;
    cb->flags = TF_ACK;
    cb->datalen = 0;
    tcp_output(cb);   <- this sends the next ACK to the other side
```

I am guessing but it could be possible that there is an off-by-one error when a packet is dropped, not sure.

And, it seems that you're saying that for some reason an ARP disrupts things, perhaps even loses a packet.

We might try putting some printf's in these places to show more accurately exactly what is happening with both sent and received ACK sequence numbers. That should allow you to trace down to the very first error above.

Thanks!

Mellvik, 2020-09-21:
Thanks, @ghaerr.
Well, yes, this may be a great opportunity to find out where it starts. It makes no sense that the ARP exchange - being isolated in itself - should trigger this. Then again that's exactly what it does, however indirectly.

> I'm pretty sure this is the same nagging problem ktcp has had all along - for some reason, whenever ktcp gets tripped up and has to start doing retransmits, everything goes to heck. I have mentioned [...]
OK, some challenge - let' give it a shot... 
> For every retransmit there is a retrans retry #1 rto 4 mem 360 message with increasing numbers
>
> I hadn't noticed that the retransmits end up allocating yet more memory - they should just retransmit [...]
Good point, I'll keep an eye on that as testing continues. Again memory corruption is a possiblility I guess.
> The retransmit stuff is all in tcp_output.c. The function tcp_retrans() is called by the ktcp main loop every half second or so to check for retransmit needed, and then calls tcp_reoutput if needed. [...]
Good point!
> And, it seems that you're saying that for some reason an ARP disrupts things, perhaps even loses a packet.
> 
ARP does indeed disrupt somehow, but no lost packets in the packettraces so far ...
> We might try putting some printf's in these places to show more accurately exactly what is happening with both sent and received ACK sequence numbers. That should allow you to trace down to the very first error above.
> 
Will do, thanks.  More on this issue tomorrow.

--Mellvik

ghaerr, 2020-09-22:
@Mellvik,

> ARP does indeed disrupt somehow, but no lost packets in the packettraces so far ...

Very strange. Lets use your packet tracer to match up the ACK's on both ends, and add printfs to ktcp to display them on ELKS as well. Since no packets are seemingly being lost, this may have to do with timing or data corruption. We should be able to match the ELKS ACK sequence numbers up to the packet tracer. 

I think the best way to debug this will be to ONLY trace until the first retransmit or otherwise ACK match error. It seems things just get worse after that. It is entirely possible that a received ARP overwrites a ktcp buffer - it's not like we haven't seen that problem at least a couple of times so far!!!

Thank you!

ghaerr, 2020-09-22:
@Mellvik - You may not want to pull down the latest source, but just work from what you have. Pulling down the latest will end up requiring you to perform a full gcc tools recompile. I also just found an additional compiler problem, which will mean another round of tools recompiling is likely.

ghaerr, 2020-09-26:
Hello @Mellvik,

Is there a way to duplicate this ktcp issue using QEMU?

Mellvik, 2020-09-26:
> Hello @Mellvik,
> 
> Is there a way to duplicate this ktcp issue using QEMU?

I've tried real hard, but no cigar. 
If there is a way to slow down QEMU to the speed of a 20MHz machine it may be possible... We need to somehow trigger retransmits to get it started.

There seem to be more than one problem at work  here, the primary being that a retransmitted packet that has been ack'ed does not always get removed from the retransmit queue. It does not get retransmitted again, but it does take up space. And not always. As I add printf's, the timing changes and the behaviour change with it. 

There was some real work (!) that had to take priority :-) , so I haven't touched this since wednesday, getting back on it this weekend.

--Mellvik

ghaerr, 2020-09-29:
Hello @Mellvik,

> There seem to be more than one problem at work  here, the primary being that a retransmitted packet that has been ack'ed does not always get removed from the retransmit queue. It does not get retransmitted again, but it does take up space. And not always. 

I'm trying to hone into the problem here, but can't get anything to fail on QEMU, even when I add special code to drop a received or sent packet on occasion - the system recovers as it should by a retransmit. I don't have real hardware running a NIC yet.

I think your observation above is interesting, but I can't see anything immediately wrong with the tcp_retrans() function that would cause that behavior. There are two commented out debug_tcp statements in that routine you might add back in.

Perhaps we should concentrate on your originally-described problem: everything works great until an ARP packet is received and replied, then things go bad. Is this the only way to cause the problem, or have you seen it start in other ways? Perhaps the ARP processing is in fact corrupting memory, and that is the actual bug. I would like to know if the problem always starts with a single ARP packet received, or whether it only sometimes does.

Thanks!

ghaerr, 2020-09-29:
There is this code that is ktcp executes on an ARP reply received:
```
        case ARP_REPLY:
                debug_arp("arp: incoming REPLY\n");
                /* update cache, check for queued packet*/
                entry = arp_cache_update(arp->ip_src, arp->eth_src);
                if (entry) {
                        if (entry->qpacket) {
                                debug_arp("arp: sending queued packet\n");
                                eth_sendpacket(entry->qpacket, entry->len, entry->eth_addr);
                                free(entry->qpacket - sizeof(struct ip_ll));
                                entry->qpacket = NULL;
                        }
                }
                netstats.arprcvreplycnt++;
```
I added this code to fix the ARP kluge problem a while back. Basically, on an ARP reply, after updating the cache, if there is a queued packet (that is, a packet that was waiting to be sent to an address for the first time before an ARP ever occurred), then it resends that packet after having received the remote address. 

I suggest we turn on DEBUG_ARP and look for "arp: sending queued packet" when you see the failure, to see if this is the problem. It could be possible that an old queued packet is being sent again, which is screwing things up.

ghaerr, 2020-09-29:
> that there may be a problem with send & receive priority: When ICMP or ARP or something else disturbs the regular send/get_ack pairs, the ack gets delayed, and more and more so as there is more 'extraneous' traffic.

There is no send or receive priority in ELKS or ktcp: everything is processed in order of the ethernet packet received.

I would like to concentrate on your original description to this issue. I can't work on it until you get it repeatable. We can then capture packet sequence number information.

> Perhaps we should concentrate on your originally-described problem: everything works great until an ARP packet is received and replied, then things go bad. Is this the only way to cause the problem, or have you seen it start in other ways? 

> There are two commented out debug_tcp statements in that routine you might add back in.

>  I would like to know if the problem always starts with a single ARP packet received, or whether it only sometimes does.

> I suggest we turn on DEBUG_ARP and look for "arp: sending queued packet" when you see the failure, to see if this is the problem. It could be possible that an old queued packet is being sent again, which is screwing things up.

All four of the above will help me debug this.

Mellvik, 2020-09-30:
@ghaerr,

First,  ARP is off the hook - for now. The pattern we're seeing is usually triggered by ARP, but it turns out that any disturbance (ICMP, ARP, other traffic …) will trigger the delicate balance between send and receive.

The scenarios is (again) send large amounts of data from elks (via httpd). What happens is that ktcp seems to prioritize writes over reads (or rather, not prioritize reads when they're piling up), and never catches up with reads/outstanding packets. So incoming pings or arps disturb the flow of send/ack/send/ack pairs, outgoing packets get ahead of the ACKs which fall more and more behind, until the retransmit buffer overflows. Depending on the amount of disturbance, the scenario may or may not trigger retransmits. The scenario described below has some, I can make a similar example without retransmits at all, the result is the same - ACKs falling behind and the retransmit buffer eventually overflowing. 

The packet trace starts just before the initial ICMP packet which starts the 'disturbance'. This is completely repeatable. The retransmitted packets have 'highlighted' seq# (**).

It seems to me that the underlying problem may be overload: There is more traffic than ELKS can handle - most of it generated by ELKS itself (send), and the inability to prioritize reads seemingly aggravates the situation. That said, the retransmission mechanism in most cases seems to behave well - except for the one thing that has been mentioned a few times recently: It seems ACKed packets don't always get purged as they should. If there is a timed cleanup, maybe that gets disturbed by the heavy load too? Just speculating. Also - if this is really overload, maybe a means for ktcp to slow down sends would be helpful, having the same effect as (almost) closing the window when receiving too fast?
```
16:52:47.073740 IP (tos 0x0, ttl 64, id 41010, offset 0, flags [DF], proto TCP (6), length 40)
    10.0.2.2.43932 > 10.0.2.15.http: Flags [.], cksum 0x182b (incorrect -> 0x0944), ack 31848, win 29200, length 0
16:52:47.149318 IP (tos 0x0, ttl 64, id 1759, offset 0, flags [none], proto TCP (6), length 140)
    10.0.2.15.http > 10.0.2.2.43932: Flags [P.], cksum 0x79e9 (correct), seq 31848:31948, ack 76, win 255, length 100: HTTP
16:52:47.149642 IP (tos 0x0, ttl 64, id 41011, offset 0, flags [DF], proto TCP (6), length 40)
    10.0.2.2.43932 > 10.0.2.15.http: Flags [.], cksum 0x182b (incorrect -> 0x08e0), ack 31948, win 29200, length 0
16:52:47.149905 IP (tos 0x0, ttl 64, id 44636, offset 0, flags [DF], proto ICMP (1), length 828)
    10.0.2.2 > 10.0.2.15: ICMP echo request, id 8509, seq 1, length 808
16:52:47.224763 IP (tos 0x0, ttl 64, id 1760, offset 0, flags [none], proto TCP (6), length 140)
    10.0.2.15.http > 10.0.2.2.43932: Flags [P.], cksum 0x7985 (correct), seq 31948:32048, ack 76, win 255, length 100: HTTP
16:52:47.225046 IP (tos 0x0, ttl 64, id 41012, offset 0, flags [DF], proto TCP (6), length 40)
    10.0.2.2.43932 > 10.0.2.15.http: Flags [.], cksum 0x182b (incorrect -> 0x087c), ack 32048, win 29200, length 0
16:52:47.304285 IP (tos 0x0, ttl 64, id 1761, offset 0, flags [none], proto ICMP (1), length 828)
    10.0.2.15 > 10.0.2.2: ICMP echo reply, id 8509, seq 1, length 808
16:52:47.350165 IP (tos 0x0, ttl 64, id 44655, offset 0, flags [DF], proto ICMP (1), length 828)
    10.0.2.2 > 10.0.2.15: ICMP echo request, id 8509, seq 2, length 808
16:52:47.376996 IP (tos 0x0, ttl 64, id 1762, offset 0, flags [none], proto TCP (6), length 140)
    10.0.2.15.http > 10.0.2.2.43932: Flags [P.], cksum 0x7921 (correct), seq 32048:32148, ack 76, win 255, length 100: HTTP
16:52:47.377235 IP (tos 0x0, ttl 64, id 41013, offset 0, flags [DF], proto TCP (6), length 40)
    10.0.2.2.43932 > 10.0.2.15.http: Flags [.], cksum 0x182b (incorrect -> 0x0818), ack 32148, win 29200, length 0
16:52:47.452602 IP (tos 0x0, ttl 64, id 1763, offset 0, flags [none], proto TCP (6), length 140)
    10.0.2.15.http > 10.0.2.2.43932: Flags [P.], cksum 0x78bd (correct), seq 32148:32248, ack 76, win 255, length 100: HTTP
16:52:47.452813 IP (tos 0x0, ttl 64, id 41014, offset 0, flags [DF], proto TCP (6), length 40)
    10.0.2.2.43932 > 10.0.2.15.http: Flags [.], cksum 0x182b (incorrect -> 0x07b4), ack 32248, win 29200, length 0
16:52:47.554246 IP (tos 0x0, ttl 64, id 44665, offset 0, flags [DF], proto ICMP (1), length 828)
    10.0.2.2 > 10.0.2.15: ICMP echo request, id 8509, seq 3, length 808
16:52:47.564251 IP (tos 0x0, ttl 64, id 1764, offset 0, flags [none], proto TCP (6), length 140)
    10.0.2.15.http > 10.0.2.2.43932: Flags [P.], cksum 0x7921 (correct), seq **32048:32148**, ack 76, win 255, length 100: HTTP
16:52:47.564335 IP (tos 0x0, ttl 64, id 41015, offset 0, flags [DF], proto TCP (6), length 40)
    10.0.2.2.43932 > 10.0.2.15.http: Flags [.], cksum 0x182b (incorrect -> 0x07b4), ack 32248, win 29200, length 0
16:52:47.643048 IP (tos 0x0, ttl 64, id 1765, offset 0, flags [none], proto ICMP (1), length 828)
    10.0.2.15 > 10.0.2.2: ICMP echo reply, id 8509, seq 2, length 808
16:52:47.710106 IP (tos 0x0, ttl 64, id 1766, offset 0, flags [none], proto TCP (6), length 140)
    10.0.2.15.http > 10.0.2.2.43932: Flags [P.], cksum 0x7859 (correct), seq 32248:32348, ack 76, win 255, length 100: HTTP
16:52:47.710212 IP (tos 0x0, ttl 64, id 41016, offset 0, flags [DF], proto TCP (6), length 40)
    10.0.2.2.43932 > 10.0.2.15.http: Flags [.], cksum 0x182b (incorrect -> 0x0750), ack 32348, win 29200, length 0
[... cut ...]
 - ktcp is persistent if anything, and sends more data, again as if everything is just fine. No flags except ACK.
    - This is where the trace above stops
 - Then immediately repeats the last send, and the client tries again to have it shut up. (RST)
 - This continues for 4 rounds, then ktcp starts the final chapter: 
    - Another data packet which is sent once, receiving another RST
    - repeat 10 times
    - the goes into CLOSE_WAIT and does the normal cleanup.

Hopefully, the observations may be useful - they are to me as I proceed from symptoms to (possible) cure ...

—Mellvik

ghaerr, 2020-09-30:
Hello @Mellvik,

Thanks for your fantastic debugging information. I'm going to hold on the latter information about the termination issues with RST and ACK for just a bit and concentrate on the first problem for now...

> What happens is that ktcp seems to prioritize writes over reads (or rather, not prioritize reads when they're piling up), and never catches up with reads/outstanding packets. So incoming pings or [...]

Ok - I finally understand what you're talking about, and think I know what the problem is!

Now that I realize this problem is (so far) only occurring on HTTP GETs *from* ELKS, a quick look at httpd and the kernel code for handling socket writes shows the problem:

- httpd uses a 2048 byte buffer to read the file requested, and calls write with the same 2048 buffer repeatedly in `process_request()`.
- the kernel `inet_write() routine gets that 2048 byte buffer, and then divvies up the write count to the TCPDEV driver's max write length allowed to pass to ktcp. This is TDB_WRITE_MAX, which is 100 bytes (!!) (I've mentioned this small number before, but didn't realize the ramifications).
- tcp's kernel process *stays* in inet_write sending 100 byte increments of the 2048 buffer to ktcp. This means that httpd's calls to write doesn't return until after more than 20 ktcp <-> tcpdev transfers.

I'm not sure exactly why the backup occurs yet, but I am guessing that the httpd process that is in a tight loop for tcpdev transfers to ktcp is getting scheduled by the kernel more often, and tcpdev is a bottleneck, which sets up the conditions you are seeing. Thus, the large httpd write and subsequent kernel write loop forces lots of data through tcpdev at once, which only works when ktcp or another process doesn't need tcpdev for another reason, since its backed up constantly sending data to ktcp.

A quick temporary fix would be to change the BUF_SIZ define in httpd.c from 2048 to 100 (no larger!). Then recompile httpd and see whether this problem persists. I'm pretty sure that will solve the problem. I have some thoughts on longer term fixes if that helps.

Mellvik, 2020-10-01:
Thanks, @ghaerr,
I'm following your logic. It did  the trick - to some degree: The transfer is now more robust, it takes more than an ARP request or a minimal ICMP to trigger the problem. A larger og more frequent ping and we're back - which I guess is what you expected.
SO I'm eagerly awaiting the next sted … :-) 
BTW is there a simple way to increase the 100 bytes to - say - 400 now that we've more available space in the kernel (snd_buf and rcp_buf bot gone …)?

The httpd file size problem got its fix too (file size was an int, not an off_t). BTW - I stated incorrectly that curl defaults to ignore file size on incoming files on Linux. That is not the case, I just hadn't noticed the problem before because I've been using symbolic links and devices…

—Mellvik

ghaerr, 2020-10-01:
The next step will be determining exactly why the ACKs don't get processed correctly. Ktcp source shows that received packets should be processed prior to reading tcpdev in the main loop, so something more involved is happening. 

We need to trace the received ACK packet handling now in more detail to find an ACK being received, then dropped or not processed. It is possible a received packet is being overwritten, not sure.

> It seems ACKed packets don't always get purged as they should. 

Is there some kind of trace that could show a received ACK being displayed, but then not removed in tcp_retrans()? There are two commented-out debug_tcp statements in tcp_retrans, lets put them back in. We want to focus on the SEQ_LEQ compare statement, which removes a retransmit, and the later TIME_GEQ, possibly adding a sequence number to display, which starts a retransmit.

> If there is a timed cleanup, maybe that gets disturbed by the heavy load too? 

Yes. The above tcp_retrans handles all retransmit starts and removals. It is called in the main loop in ktcp.c line 122 with
"if (tcp_timeruse > 0)". Perhaps comment that if statement out (tcp_timeruse should definitely be > 0 already). The tcp_retrans routine traverses through the entire retrans list and either 1) uses SEQ_LEQ to remove the retrans item if the sequence number is less than or equal to the acknowledged sequence number (n->cb->seq_una), or 2) sends the next retrans if the timeout wait for that has expired using TIME_GEQ and n->next_retrans.

It is also possible this entire problem is related to the timeout of TIME_GEQ above, where the retrans starts, only because too much time has passed with the retrans packet waiting in the retrans queue. The following code in `add_for_retrans` calculates the first retransmission:
```
    n->rto = cb->rtt << 1;
    if (linkprotocol != LINK_ETHER && n->rto < TIMEOUT_MIN_SLIP)
        n->rto = TIMEOUT_MIN_SLIP;              /* 1/2 sec min retrans timeout for slip/cslip*/
    n->next_retrans = Now + n->rto;
```
That code doubles the RTT time (which could be small) and uses that for a start retransmit time. It seems that I have previously fixed a similar problem for SLIP timeouts, and enforce a minimum SLIP timeout (of 8 = 1/2 sec). Perhaps change the 2nd line to:
```
    if (n->rto < TIMEOUT_MIN_SLIP)
```
The more I look at this, the more I think this may fix our issue entirely.

> is there a simple way to increase the 100 bytes to - say - 400 

Yes, we will worry about getting that back to a reasonable size after determining the real problem. The large transmit queue isn't necessarily the problem, it is only triggering the real problem - ACK retransmits getting out of sync.

I will produce a PR for httpd with file size and buffer size changed after debugging.

ghaerr, 2020-10-01:
I'm pretty sure we've just found the real problem: the retransmits start happening merely because the system is moving data so quickly, that the RTO/RTT calculation is quite low, which then sets the first retrans to start too quickly. Once started, this continues retransmitting, rather than removing retrans packets until the system is stopped by running out of retrans memory.

tcp.h shows that I have reduced the retrans memory to 2048 bytes, (so this problem wouldn't cycle all the way up to the original 16k limit). However, the 2048 byte value is the same as the write size in httpd. So, even though the RTO fix above will likely fix the problem, we should experiment with a slightly higher (say 4096) byte retrans memory limit TCP_RETRANS_MAXMEM to see whether that allows the original "outstanding data" time to recover or not.

Does this all make sense?

Mellvik, 2020-10-01:
@ghaerr,
I noticed the enxt message, but I'd like to start with this one anyway.

Thanks for 'thinking loudly' like this, it's very effective (aka sharing thoughs, I like thinking loudly better).
Let's continue the brainstorming:
> 
> The next step will be determining exactly why the ACKs don't get processed correctly. Ktcp source shows that received packets should be processed prior to reading tcpdev in the main loop, so something more involved is happening.
> 
It's important to keep in mind that the sequence of packets on the wire (and their timing) is different from in ELKS. The incoming packets end up in the ring buffer, then get read, then processed etc. So what I'm seeing is that the acks may have arrived but elks is too busy sending to read them (in time). And when it does, the retransmit is already a fact.
> We need to trace the received ACK packet handling now in more detail to find an ACK being received, then dropped or not processed. It is possible a received packet is being overwritten, not sure.
> 
Timing is a big part of the problem. It's hard to get dependable traces because every trace added changes the behaviour. I've been pulling my hair out in order to get really valuable data on this. We're getting there (btw - I've been bald for aeons) slowly.
> It seems ACKed packets don't always get purged as they should.
> 
> Is there some kind of trace that could show a received ACK being displayed, but then not removed in tcp_retrans()?
> 
Yes, I do have traces showing this. Getting back on that.
> There are two commented-out debug_tcp statements in tcp_retrans, lets put them back in. We want to focus on the SEQ_LEQ compare statement, which removes a retransmit, and the later TIME_GEQ, possibly adding a sequence number to display, which starts a retransmit.
> 
Will look closely at that, thanks.
> If there is a timed cleanup, maybe that gets disturbed by the heavy load too?
>
> Yes. The above tcp_retrans handles all retransmit starts and removals. It is called in the main loop in ktcp.c line 122 [...]
I'm enthusiastic about this. Will test at the first possible opportunity. Re timing - think about this: (from before changing the httpd read size):
Httpd sends 20 packets of 100 bytes each, then one @ 48 bytes, then waits 1.8-3.1 seconds (seriously!) for the floppy to get 4 more sectors ... during which network interrupts arrive but no reads or writes.
It's really hard to set timeouts that work in such scenarios. I think we have to accept that some times the timeouts will be far off, and the mechanism will still have to recover ...
> 
--Mellvik

ghaerr, 2020-10-02:
@Mellvik,

I've figured out what the problem is, thanks to your excellent description from a few days ago. Lowering httpd's buffer size to 100 (for now) helps a little, and we definitely need to change the if statement in add_for_retrans to "if (n->rto  < TIMEOUT_MIN_SLIP)" which will set a half second ACK delay before any retransmit, but neither are the full solution to the problem.

Here's the problem: there is no send window management in ktcp at all. That is, ktcp doesn't EVER stop sending outbound packets at full speed (if an application writes to a socket, like httpd GET does with a large file), regardless of whether any packets are ACKed, lost, or whatever. The massive sending works for a while, but various situations can cause the "send window" to get so big, that a timeout finally occurs on the first packet in the retransmit queue, and then all packets in the retrans queue will eventually get retransmitted, as the system is in fact falling farther and farther behind.

It's not quite that ktcp hasn't gotten around to reading the ACK packets - it reads and processes any received packet during every main loop cycle. It is that there is absolutely no throttling of the packet sending, and every sent packet adds another retransmit (for that packet) to the outstanding queue. Should the receiver get behind just a little, ktcp doesn't care, it keeps sending packets non-stop. Finally the timeout which causes the retransmit to start, as explained above, and the both sides get further behind, while ktcp still sends outgoing packets non-stop. Increasing that timeout WILL help. TIMEOUT_MIN_SLIP is 8, which means 8/16 = 1/2 second timeout before retransmitting a non-ACKed packet. You may want to play with increasing that, possibly, after changing the if statement to apply to ethernet link level as well (I coded it for SLIP only, now I remember fixing this same kind of issue with SLIP).

Here's what's actually buggy, and it has to do with the fact that ktcp has to get its incoming socket data (to send) from tcpdev: ktcp cannot say "NO" to an application that wants to send data to a socket. That is, after the httpd GET request, regardless of the httpd `write` size, httpd sits in a tight loop and writes data to the outbound socket. The kernel gets those writes, repackages them into smaller writes (100 bytes at this time, but it doesn't matter how big or small), and then writes tcpdev to communicate the fact that there is socket data needing to be written. Every main loop cycle, ktcp hangs in select until either an ethernet packet has arrived, OR tcpdev needs attention. Tcpdev handles both reading and writing, so it CANNOT be ignored or stopped. In our case, tcpdev ALWAYS has data ready to be read - so what does ktcp do? It doesn't buffer write data (only read data), so it takes the data from tcpdev, forms a TCP packet, and sends it. It is exactly like being forced to drink water from a fire hose.

Thus, every main loop cycle, if httpd has data, of ANY length (which it will for a large file), ktcp takes that data, mallocs space for a retransmit packet, then send a packet of that data. It then checks for received packets. If there isn't a received packet, then ktcp has just fallen behind a bit. This process continues until the retransmit timer for the first packet in the retransmit queue times out.

The "bug" is a design problem - ktcp cannot ignore the input data sent to it, as that would be tantamount to actually dropping file data. The reason there is no send window throttling is such a thing would be impossible to implement in this design, unless ktcp malloc'd all the data sent to it when over its send window size, and its data limit is a mere 64k, so that would never work for large files.

The "solution": test and implement the idea presented above, to show that it improves things a bit. Then, we are going to have to change the tcpdev "write" interface to the same way I changed the "read" interface: instead of just writing, the kernel is going to have to ask ktcp how much data it can take (for a given socket). If ktcp says 0, then the kernel will suspend the calling process (httpd in this case), until ktcp says it is ready for data. ktcp will have to be modified to look at its outstanding send window by looking at the current sequence number versus the first seqno in the retransmit queue, and limit write data to a certain size. That's all going to take a bit to think it through, and also we need to come up with an algorithm for calculating a workable send window size.

In the meantime: although is seems counter-intuitive, the solution is to *decrease*, rather than increase, ktcp's "bite size". This effectively means slowing things down a touch and letting all processes involved spin like clock gears a little bit more. This will work, because when ktcp gets smaller packets, the send window is decreased, and the amount of memory ktcp has to allocate for retransmit packets is less, etc etc. If we just increased the retransmit memory, it would just prolong the ultimate desync for large files when this issue arises. It is also for this reason that ktcp currently publishes a receive window of 255 bytes, FYI. Advertising a larger value would just require more ring buffer space in ktcp, and take longer time for applications to process received data in larger buffer sizes etc.

[... cut ...]
Yep. I think the answer may be to do what I've done with all the other ELKS utilities: set the buffer size equal to the disk sector size, which is 1024. At this point, it might be a good thing for ELKS to have to wait for a disk read to slow the sending, until the larger problem is fixed. I suppose, after you've got the other minor helper fixes in, it could be worth playing around with the buffer size a bit to see how much throughput you gain, while still not exceeding the point where the real bug comes into play.

Let me know your testing, and I'll start working on a fix for all this.

ghaerr, 2020-10-02:
Also, uncomment the following line in tcpdev.c, is will show the send window size which can be looked at when the system breaks down:
```
//printf("ktcpdev WRITE sock %x window %ld timer %d\n", sock, cb->send_nxt - cb->send_una, tcp_timeruse);
```

Mellvik, 2020-10-02:
Thanks @ghaerr, I think you're spot on.

I'll be back with more on sunday or monday.

--M

ghaerr, 2020-10-06:
@Mellvik,

The retransmit and httpd issues should be fixed from #772 and #773, please test.

The RST issues have not been looked at yet.

Thank you!

Mellvik, 2020-10-10:
Quick test today delivers impressive results. Curl test (outgoing) is now up to 75kBps, which means we're 3 times higher than 3 weeks ago. I'd say that's a cork-popper if anything. 
RST (curl ^C) is fine, the symbolic link issue us resolved.

While there may still be nuances that need attention in TCP retransmit, I've not been able to push it so that it breaks after @ghaerr's numerous fixes and enhancements. Time to close this issue.

THANKS @ghaerr

closed 2020-10-10 by Mellvik