ghaerr/elks #756: Likely compiler bug

ghaerr, 2020-09-22:
Hello @tkchia,

I'm sorry to bombard you with so many issues... and excuse me for the very long post.

In addition to the `elf2elks: error: R_386_SEGRELATIVE relocation has bad in-place addend 0` error discussed in https://github.com/jbruchon/elks/pull/743#issuecomment-696282140, it seems we have an even bigger problem with the compiler:

Short version:
It seems ia16-elf-gcc is emitting an "add $0x0a,%sp" out of nowhere. In a kernel built with far text, this later ends up trashing the segment return value on the stack, and the system crashes. It turns out that the same add instruction is also emitted without any far text whatsoever (i.e. in normal compiles), but doesn't crash because of the nuances of exactly what the stack looks like.

Long version:
Rather than waiting for a fix to the above relocation addend error, I decided to go ahead with a small change to the kernel, that of rearranging some start_kernel code and putting all the kernel init calls in a single far text kernel_init routine. It didn't work. After lots of debugging, I find that the system crashes on LRET from the far text procedure. As it turns out, the reason for the crash is that the stack is trashed by a seemingly errant "add $0x0a,%sp" instruction.

The kernel_init function is far, but calls lots of near routines, so there are lots of thunk calls. I've matched them all up and determined that my relocating loader is patching them correctly. I'm going to list everything out visually, but have attached a patch file so you can easily see the compiler error for yourself.

Here's the entire patch:
```
diff --git a/elks/arch/i86/Makefile b/elks/arch/i86/Makefile
index a3997c80..89b6133f 100644
--- a/elks/arch/i86/Makefile
+++ b/elks/arch/i86/Makefile
@@ -113,6 +113,7 @@ boot/system:        $(XINCLUDE) $(AARCHIVES) $(ADRIVERS) boot/crt0.o
        (cd $(BASEDIR) ; $(LD) $(LDFLAGS) -M $(ARCH_LD) -T $(TOPDIR)/elks/elks-small.ld \
                $(ARCH_DIR)/boot/crt0.o \
                init/main.o '-(' $(ARCHIVES) $(DRIVERS) '-)' \
+               "`$(CC) $(CFLAGS) -print-libgcc-file-name`" \
                -o $(ARCH_DIR)/boot/system > $(ARCH_DIR)/boot/system.map ; \
                $(POSTLINK) $(ARCH_DIR)/boot/system)
 #              sort -k3,4 $(ARCH_DIR)/boot/system.tmp > $(ARCH_DIR)/boot/system.map ; \
diff --git a/elks/init/main.c b/elks/init/main.c
index a9ca3646..34d0ebe4 100644
--- a/elks/init/main.c
+++ b/elks/init/main.c
@@ -45,6 +45,8 @@ static char *envp_init[MAX_INIT_ENVS+1];
 static unsigned char options[256];
 
 extern int boot_rootdev;
+//#define INITPROC     /* uncomment for all near code*/
+INITPROC void kernel_init(void);
 static int parse_options(void);
 static void finalize_options(void);
 static char *option(char *s);
@@ -54,7 +56,29 @@ static void init_task(void);
 extern int run_init_process(char *cmd);
 extern int run_init_process_sptr(char *cmd, char *sptr, int slen);
 
+
 void start_kernel(void)
+{
+early_printk("BEFORE\n");
+       kernel_init();
+printk("AFTER\n");
+
+    kfork_proc(init_task);
+    wake_up_process(&task[1]);
+
+    /*
+     * We are now the idle task. We won't run unless no other process can run.
+     */
+    while (1) {
+        schedule();
+
+#ifdef CONFIG_IDLE_HALT
+        idle_halt ();
+#endif
+    }
+}
+
+INITPROC void kernel_init(void)
 {
     seg_t base, end;
[... 202 lines cut ...]
            a7: R_386_16    __ia16_far_call_thunk.bx.4
            a7: R_386_SUB16 __ia16_far_call_thunk.bx.4!
            a9: R_386_SEG16 __ia16_far_call_thunk.bx.4!
  ab:   b8 c4 00                mov    $0xc4,%ax
            ac: R_386_16    .rodata.str1.1
            ac: R_386_SUB16 .rodata.str1.1!
  ae:   50                      push   %ax
  af:   bb 00 00                mov    $0x0,%bx
            b0: R_386_16    printk
            b0: R_386_SUB16 printk!
  b2:   9a 00 00 00 00          lcall  $0x0,$0x0
            b3: R_386_16    __ia16_far_call_thunk.bx.2
            b3: R_386_SUB16 __ia16_far_call_thunk.bx.2!
            b5: R_386_SEG16 __ia16_far_call_thunk.bx.2!
  b7:   89 ec                   mov    %bp,%sp      <--- BP being unprotected from SP now trashed by above calls
  b9:   5d                      pop    %bp
  ba:   cb                      lret       <--- CRASH!
```

It turns out that this bug isn't related to .fartext - uncommenting the "\#define INITPROC" in the patch will produce all-near code with the same ADD instruction.

Please excuse the long post.

Thank you!
[gccerror.patch.zip](https://github.com/jbruchon/elks/files/5258748/gccerror.patch.zip)

tkchia, 2020-09-22:
Hello @ghaerr,

The `add $0xa, %sp` is correct.  Since the kernel still uses the `cdecl` function calling convention, it means that `kernel_init()` needs to clean up any stack parameters it passes to the functions it calls.  In this case it pushes 5 shortwords of arguments in total --- to `setup_arch`, `mm_init`, and `set_console`.  Instead of cleaning the stack after each call, GCC chooses to pop everything at one go, as an optimization.

tkchia, 2020-09-22:
Anyway, using `qemu-system-i386 -d in_asm,cpu -D` _log-file_ (which, incidentally, produces _gobs_ of log output), I found that the stack pointer was restored correctly, but `kernel_init()` was returning to a bogus address.

Instead of returning to, say, `0xc0:0x434`, which was in `.text`, it would return to `0xecf:0x434`, which was in the data segment (!).  By adding `early_printk` calls I found that the corruption already happened right at the first function call, i.e. `sched_init`.  Then I found that `kfork_proc(NULL)` was trying to set up an initial stack on `task[0]` --- but the kernel startup code was already running on this very stack.

Thank you!

ghaerr, 2020-09-22:
Hello @tkchia,

Very interesting. So it turns out that task 0 doesn't ever actually use the `ret_from_syscall` mechanism faked up by `arch_build_stack` at all, which now makes sense since task 0 never runs as a user mode process. 

This also means that ES is never initialized for task 0. (Does ia16-elf-gcc ever assume ES==DS, or instead just push ES before use as an extra register?)

I'll fixup the now known-incorrect comment about this above ret_from_syscall in irqtab.S and initialize ES from DS in crt0.S, for  correctness, now that we understand what is really happening.

Your finding that the kernel stack was being overwritten as early as the call to sched_init solves a lot of problems I have been having getting the far segment kernel operational with my much larger patch introducing the .fartext.init section. I'm sure this is the reason for it not working since inception.

Thank you very much for your brilliant help. Sometimes I think working with ELKS is like playing pac-man - there can be gremlins around every corner, and if you don't watch very carefully, they will eat you. I learned a lot watching a master pull out his sword and decisively slay a gremlin that started eating me. We are definitely reducing the gremlin count in ELKS.

Thank you!!

ghaerr, 2020-09-22:
> This also means that ES is never initialized for task 0. 

I stand corrected, ES is initialized (along with DS) in setup.S

closed 2020-09-22 by ghaerr