ghaerr/elks #763: GCC 10 fixes in upstream binutils-ia16

xenoxaos, 2020-09-26:
There's issues with binutils-ia16. I've already created a PR with the fixes in it:
https://github.com/tkchia/binutils-ia16/pull/8

If I remember, I'll create a PR with an updated tools/Makefile to point to the commit, but I might forget once it's merged. So, creating an issue for one of us to check it out

tkchia, 2020-09-26:
Hello @ghaerr,

In sum, it seems there were some errors when building `binutils-ia16` with GCC 10.  (I did not encounter them as I was building with GCC 9.)  Do you think you would like to bump up the `binutils-ia16` version to use, to address this?  Thanks!

ghaerr, 2020-09-26:
Hello @tkchia,

Considering that Mike required it for his build, you might as well bump the version number as well in your fix for it in #764, along with a message that a rebuild is not required unless one is using an upstream compiler version. For our users that have not yet rebuilt their toolchain for the new ld.gold linker support, they'll get the all the fixes.

Thank you!

tkchia, 2020-09-26:
Hello @ghaerr: OK, I have updated the PR.  Thanks!

ghaerr, 2020-11-19:
Hello @tkchia,

Is this issue finalized?

I'd like to close it unless there's a reason to keep it open across v0.4.0.

Thank you!

tkchia, 2020-11-19:
Hello @ghaerr,

> Is this issue finalized?

Not quite --- if I remember correctly, the `binutils-ia16` version used in ELKS has not yet been bumped up to a newer version.  For now though, this only affects people trying to build ELKS and the toolchain under newer versions of host GCC.

Thank you!

tkchia, 2020-11-30:
With the PR https://github.com/jbruchon/elks/pull/884 , this problem should be fixed now.  Thanks!

closed 2020-11-30 by tkchia