ghaerr/elks #774: Tools not compiling

Mellvik, 2020-10-06:
`tools/build.sh`  no longer builds on ELKS.

The final output is
```
checking how to run the C preprocessor... gcc -E
checking build system compiler gcc... rm: conftest.dSYM: is a directory
rm: conftest.dSYM: is a directory
no
configure: error: Specified CC_FOR_BUILD doesn't seem to work
make[2]: *** [configure-gmp] Error 1
make[1]: *** [all] Error 2
make: *** [/Users/helge/src/mellvik/elks/cross/build/.gcc.build] Error 2
```
... but there are a number of similar situations before this `rm <some directory>.dSYM: is a directory`

Clues anyone? This is on a mac, updated Xcode tools.

--Mellvik

ghaerr, 2020-10-06:
Hello @Mellvik,

I'm on macOS Mohave 10.14.6 and the tools build works for me.

I'm running gcc (Apple LLVM) 9.0.0 (clang-900.0.38).
The following versions are in my tools/Makefile:
BINUTILS_VER=f19a905d1434da03940951913514a9ef12f0d7dd
GCC_VER=45605f97ac4d313f8783e1ba9924d0aec0b1d6e8

We'll see what @tkchia has to say.

ghaerr, 2020-10-06:
Looking at elks/cross/build/gcc-src/gmp/configure, I can see several places where "rm -f conftest\*" or similar are used, which could be an issue for you if your system failed some of the configure tests, and the rm was executed. You may want to try editing that file, search for "conftest" and replace the number of "rm -f ..." with "rm -rf ..." for the conftest\* and conftest.\* used since .dSYMs are directories on macOS.

Then restart the build. I'm not quite sure how to restart the build using make instead of tools/build.sh, but I think something like "make -C tools all" will do it.

tkchia, 2020-10-06:
Hello @Mellvik, hello @ghaerr,

Very interesting --- and weird --- error.

It seems that this happens not when building GCC proper, but when building the GNU MP library (which GCC depends on).

I am not sure the `rm` was causing the build to fail, since you mentioned that there were several similar such `rm` messages just before the failure.

If the build still does not work, consider checking `cross/build/gcc-build/gmp/config.log`, which should give detailed logs from the `configure` run.  This should show more clearly what was leading to the "`Specified CC_FOR_BUILD doesn't seem to work`" message.

Thank you!

Mellvik, 2020-10-06:
Thanks,
I’m looking at the configure script right now, Funny thing is that in every failure (there are just a few), the file that cannot be removed, which is actually a directory, has the extension dSYM.

Changing all rm’s (wow, there are many, including the $rm macro) changes things but does not fix them.

The log file tells me little, but may still be helpful

Test compile: __builtin_alloca availability
configure:4831: gcc -g -O2 -DNO_ASM conftest.c >&5
conftest.c:1:43: warning: non-void function does not return a value [-Wreturn-type]
int k; int foo () { __builtin_alloca (k); }
                                          ^
1 warning generated.
Undefined symbols for architecture x86_64:
  "_main", referenced from:
     implicit entry/start for main executable
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
configure:4834: $? = 1
failed program was:
int k; int foo () { __builtin_alloca (k); }

there may be other errors, I’ll have to look at that in the morning.

—mellvik

ghaerr, 2020-10-06:
> Changing all rm’s (wow, there are many, including the $rm macro) changes things but does not fix them.

Look at the failing line in your compile - is it an "rm -rf conftest\* ..."? If the -r option isn't there, then you missed one.

I'm thinking the problem IS in the removal of the .dSYMS file on macOS, because I've had this very problem on other projects on macOS.

> Test compile: __builtin_alloca availability

The test(s) you mention will have failures - you're just observing the method configure uses to determine whether a system has thus-and-such a capability: it compiles up code and checks the linker return value. So these are not actually errors that stop the configure script.

If still in doubt, send over the patched configure file.

ghaerr, 2020-10-06:
Try this configure script in the gmp directory. It replaces all "rm -f conftest\*" with "rm -rf conftest\*".

[configure.zip](https://github.com/jbruchon/elks/files/5336887/configure.zip)

tkchia, 2020-10-07:
Hello @Mellvik,

> The log file tells me little, but may still be helpful
> ```
> Test compile: __builtin_alloca availability
> configure:4831: gcc -g -O2 -DNO_ASM conftest.c >&5
> ...
> ```

Try looking further down the log.  There should be log entries for the particular check that caused the failure, i.e. "`checking build system compiler gcc`".  Thank you!

Mellvik, 2020-10-07:
Thanks @ghaerr,
I tried that one too - and got almost the same results.
The difference being that my edited version took everything ‘rm’, regardless of the file/dir-name. Slightly different error messages, same result. 

It turns out there are many different file names, the common denominator not being ‘conftest’ but the .dSYM extension.

Now, even in the case where every single rm (and rm macro) in configure has been changed to rm -rf, there is still a rm-problem, supposedly indicating that there is more to it than the configure file.

Will keep you posted.

-M

tkchia, 2020-10-07:
Hello @Mellvik,

What does `cross/build/gcc-build/gmp/config.log` say about the "`checking build system compiler gcc`" test specifically?  Thanks!

Mellvik, 2020-10-07:
Hi @tkchia,
thanks for following up.

you may  be on to something here, excerpt below. Seems non-existent include file ac_nonexistent.h is creating some havoc in the configuration process.

—M

...
configure:8322: checking how to run the C preprocessor
configure:8362: gcc -E  conftest.c
configure:8368: $? = 0
configure:8399: gcc -E  conftest.c
conftest.c:12:10: fatal error: 'ac_nonexistent.h' file not found
#include <ac_nonexistent.h>
         ^~~~~~~~~~~~~~~~~~
1 error generated.
configure:8405: $? = 1
configure: failed program was:
| /* confdefs.h.  */
| #define PACKAGE_NAME "GNU MP"
| #define PACKAGE_TARNAME "gmp"
| #define PACKAGE_VERSION "4.3.2"
| #define PACKAGE_STRING "GNU MP 4.3.2"
| #define PACKAGE_BUGREPORT "gmp-bugs@gmplib.org"
| #define PACKAGE "gmp"
| #define VERSION "4.3.2"
| #define WANT_FFT 1
| #define HAVE_HOST_CPU_none 1
| /* end confdefs.h.  */
| #include <ac_nonexistent.h>
configure:8438: result: gcc -E
configure:8467: gcc -E  conftest.c
configure:8473: $? = 0
configure:8504: gcc -E  conftest.c
conftest.c:12:10: fatal error: 'ac_nonexistent.h' file not found
#include <ac_nonexistent.h>
         ^~~~~~~~~~~~~~~~~~
1 error generated.
configure:8510: $? = 1
configure: failed program was:
| /* confdefs.h.  */
| #define PACKAGE_NAME "GNU MP"
| #define PACKAGE_TARNAME "gmp"
| #define PACKAGE_VERSION "4.3.2"
| #define PACKAGE_STRING "GNU MP 4.3.2"
| #define PACKAGE_BUGREPORT "gmp-bugs@gmplib.org"
| #define PACKAGE "gmp"
| #define VERSION "4.3.2"
| #define WANT_FFT 1
| #define HAVE_HOST_CPU_none 1
| /* end confdefs.h.  */
| #include <ac_nonexistent.h>
configure:8584: gcc -c -g -O2  conftest.c >&5
configure:8590: $? = 0
configure:8613: checking build system compiler gcc
configure:8626: gcc conftest.c
conftest.c:4:3: error: implicitly declaring library function 'exit' with type 'void (int) __attribute__((noreturn))' [-Werror,-Wimplicit-function-declaration]
  exit(0);
  ^
conftest.c:4:3: note: include the header <stdlib.h> or explicitly provide a declaration for 'exit'
1 error generated.
configure:8629: $? = 1
configure:8636: result: no
configure:8641: error: Specified CC_FOR_BUILD doesn't seem to work

## ---------------- ##
## Cache variables. ##

tkchia, 2020-10-07:
Hello @Mellvik,

Thanks!  That is very useful information.

I just checked out the `configure` script at around line 8,613.  The log is a bit confusing --- the `<ac_nonexistent.h>` thing applies to the _preceding_test ("`checking how to run the C preprocessor`").

Apparently `configure` then later tries to compile a _different_ program, which is not shown in the log (argh!).  The program is simply this:
```
int
main ()
{
  exit(0);
}
```

The C compiler does not like this because
```
conftest.c:4:3: error: implicitly declaring library function 'exit' 
with type 'void (int) __attribute__((noreturn))' [-Werror,-Wimplicit-function-declaration]
  exit(0);
  ^
conftest.c:4:3: note: include the header <stdlib.h> or explicitly provide a declaration for 'exit'
```

I.e. the test program called `exit(0);` without declaring a prototype for `exit(.)`.  For some reason `-Werror` was turned on in your system, so that things that GCC would normally only warn about are now treated as fatal errors.

(I get a similar message on my system, but it was only a warning.)

I guess this can be considered a bit of a bug in the GNU MP 4.3.2...?  Maybe I will check if newer versions of GNU MP have addressed this.

Thank you!

Mellvik, 2020-10-07:
Thank you @tkchia,
again much appreciated.

Here’s some potentially useful info:

I moved the tool chain building to my laptop, where it compiles fine. There are plenty complaints about not being able to remove the dSYM directories, so that does not seem to be the problem.

What’s different between the two is the gcc version. The laptop has an older version than the iMAC:

HS-MB-Pro:elks helge$ gcc -v
Configured with: --prefix=/Library/Developer/CommandLineTools/usr --with-gxx-include-dir=/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/c++/4.2.1
Apple clang version 11.0.3 (clang-1103.0.32.62)
Target: x86_64-apple-darwin19.6.0
Thread model: posix
InstalledDir: /Library/Developer/CommandLineTools/usr/bin

➜  elks git:(ethtest) ✗ gcc -v
Configured with: --prefix=/Applications/Xcode.app/Contents/Developer/usr --with-gxx-include-dir=/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/c++/4.2.1
Apple clang version 12.0.0 (clang-1200.0.32.2)
Target: x86_64-apple-darwin19.6.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin

Does this provide any clues?

—Mellvik

tkchia, 2020-10-07:
Hello @Mellvik,

> Does this provide any clues?

Thanks.  It seems that the newer LLVM-GCC compiler is (for some reason) becoming more strict when deciding which C programs it would accept, while the very old version of GNU MP happens plays a bit fast and loose with the C language, as I have explained.

> > I guess this can be considered a bit of a bug in the GNU MP 4.3.2 configuration script...? Maybe I will check if newer versions of GNU MP have addressed this.

I see that GNU MP 6.1.2 --- which I have been using for `gcc-ia16` builds outside of ELKS --- does indeed address this problem.  It says
```
int
main ()
{
  return 0;
}
```
(which avoids the missing prototype problem), instead of
```
int
main ()
{
  exit(0);
}
```
I can think of a few ways to solve the build failure, each with its trade-offs:

  * Somehow pass the flag `-Wno-error` via the `CC_FOR_BUILD` environment variable to the GMP configuration script.
    * My main worry is that this may mess up `gcc-ia16`'s (and GMP's) compiler auto-detection.
    * Also, I am not completely sure how `gcc-ia16`'s top-level configuration script passes environment variables to GMP's.
  * Or, I can modify the `contrib/download_prerequisites` script inside the `gcc-ia16` source tree so that it downloads a newer GMP.
    * ELKS's `tools/Makefile` currently uses this script to download `gcc-ia16`'s dependencies.
    * But this means we need to bump up the `gcc-ia16` version used by ELKS, and you will need to download the source tree again.
  * Alternatively we can modify ELKS's `tools/Makefile` to not use `download_prerequisites`, but instead run its own commands to do the downloading.
    * This might be a more flexible arrangement.  It should allow us to bump up the versions of the GMP, MPFR, and MPC libraries as and when needed.
    * It does mean though `tools/Makefile` needs to do a bit more work.

Thank you!

tkchia, 2020-10-07:
Hello @Mellvik,

In the meantime, try modifying GMP's `configure` script to replace all
```
  exit(0);
```
with
```
  return 0;
```
This should hopefully improve things on your desktop iMac.

Thank you!

Mellvik, 2020-10-07:
Thanks @tkchia -

I’m probably messing up, I couldn’t get that to work. If there is anything I can submit or test to further help the diagnosing, let me know.

I transferred the entire cross directory from the MB Pro to the iMac, and I have something that works for now.

—Mellvik

tkchia, 2020-10-17:
Hello @Mellvik, hello @ghaerr,

I have sent in a pull request (https://github.com/jbruchon/elks/pull/799) to try to resolve this problem.  Thanks!

tkchia, 2020-10-18:
Hello @Mellvik,

May I know if the updated ELKS tree fixes the build problem you had on your iMac?  Thank you!

Mellvik, 2020-10-18:
Thank you @tkchia,

if tools/build.sh is what it takes to pull and test, the test was successful! [it did pull gmp, mpfr and mpc plus some emu86 stuff.]

If a full rebuild of cross is required, let me know.

—mellvik

tkchia, 2020-10-18:
Hello @Mellvik,

> if tools/build.sh is what it takes to pull and test, the test was successful! [it did pull gmp, mpfr and mpc plus some emu86 stuff.]

Thanks.  Besides pulling GMP, MPFR, and MPC, did `tools/build.sh` also try to rebuild them (as well as GCC)?

If it did try to rebuild those, then yes, it means the new setup is indeed working.

Thank you!

Mellvik, 2020-10-18:
@tkchia,
looking closer, only the emu86 changes were actually compiled.

So I'm removing the cross dir and doing a new build.

—Mellvik

ghaerr, 2020-10-18:
Hello @tkchia,

> Besides pulling GMP, MPFR, and MPC, did tools/build.sh also try to rebuild them (as well as GCC)?

I can second @Mellvik's statement that your most recent update to tools/build.sh does update the GMP .zip file but does NOT rebuilt GMP nor GCC. This was tested when I verified the EMU86 change I made yesterday.

Thank you!

tkchia, 2020-10-18:
Hello @Mellvik,

I have put out another pull request (https://github.com/jbruchon/elks/pull/804) to try to fix this.  Sorry and thanks!

Mellvik, 2020-10-18:
OK,
FWIW - the cross dir rebuilds fine from scratch…

—M

tkchia, 2020-10-19:
Hello @Mellvik,

Thanks!  Good to know that the build is working now.  In that case, will it be OK to close this issue?

Mellvik, 2020-10-19:
Confirming that the tools now rebuild automatically. Thanks.

ghaerr, 2021-05-17:
Thanks for re-opening this @Mellvik. Since you're saying the current errors you are receiving is exactly the same, it sounds like perhaps the GNU MP library dependency needs to be updated again. I would jump in but can't at the moment because I need my tools working given the current activity on ELKS. Hang in there and I'll try to find a spare system!

Mellvik, 2021-05-18:
@ghaerr, thanks for the followup - and my apologies for reopening without submitting the description. I was interrupted - and have not had the chance to dive back in yet.

The same 'exit(0)' issue in gmp configure is part of the problem.
More later.

@mellvik

Mellvik, 2021-05-18:
@ghaerr, @tkchia - 
Seems we didn't entirely succeed in killing this beast the last time. Whether the reason is a new version of MacOS or something else, I don't know - the symptoms are exactly like before:
```
checking for gcc option to accept ISO C99... none needed
checking for gcc option to accept ISO Standard C... (cached) none needed
checking how to run the C preprocessor... gcc -E
checking build system compiler gcc... rm: conftest.dSYM: is a directory
rm: conftest.dSYM: is a directory
no
configure: error: Specified CC_FOR_BUILD doesn't seem to work
make[2]: *** [configure-gmp] Error 1
make[1]: *** [all] Error 2
make: *** [/Users/helge/src/mellvik/elks/cross/build/.gcc.build] Error 2
```
Following the recipe from last time, the remaining `exit(0)` occurrences in the gmp configure script were replaced by `return 0`, which helped. We're getting a lot further in the process. Gmp is now fine, moving to gcc - where suddenly `g++` is forced - probably erroneously - to compile C code `clang: warning: treating 'c' input as 'c++' when in C++ mode, this behavior is deprecated [-Wdeprecated]`:
```
g++ -c   -g -O2 -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE  -fno-strict-aliasing -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -DGENERATOR_FILE -fno-PIE -I. -Ibuild -I../../gcc-src/gcc -I../../gcc-src/gcc/build -I../../gcc-src/gcc/../include  -I../../gcc-src/gcc/../libcpp/include  \
		-o build/genmddeps.o ../../gcc-src/gcc/genmddeps.c
clang: warning: treating 'c' input as 'c++' when in C++ mode, this behavior is deprecated [-Wdeprecated]
In file included from ../../gcc-src/gcc/genmddeps.c:19:
In file included from ../../gcc-src/gcc/system.h:236:
In file included from /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/c++/v1/new:90:
In file included from /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/c++/v1/exception:81:
In file included from /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/c++/v1/cstdlib:85:
In file included from /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/c++/v1/stdlib.h:93:
In file included from /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/stdlib.h:66:
In file included from /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/sys/wait.h:110:
/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/sys/resource.h:89:25: error: 'long type-name' is invalid
typedef __uint64_t      rlim_t;
                        ^
./auto-host.h:2321:16: note: expanded from macro 'rlim_t'
#define rlim_t long
               ^
In file included from ../../gcc-src/gcc/genmddeps.c:19:
../../gcc-src/gcc/system.h:540:20: error: functions that differ only in their return type cannot be overloaded
extern const char *strsignal (int);
             ~~~~~~^
/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/string.h:134:7: note: previous declaration is here
char    *strsignal(int __sig);
~~~~~~~~~^
In file included from ../../gcc-src/gcc/genmddeps.c:20:
../../gcc-src/gcc/coretypes.h:62:1: warning: class 'rtx_def' was previously declared as a struct; this is valid, but may result in linker errors under the Microsoft C++ ABI [-Wmismatched-tags]
class rtx_def;
^
../../gcc-src/gcc/coretypes.h:55:8: note: previous use is here
struct rtx_def;
       ^
1 warning and 2 errors generated.
make[3]: *** [build/genmddeps.o] Error 1
make[2]: *** [all-gcc] Error 2
make[1]: *** [all] Error 2
make: *** [/Users/helge/src/mellvik/elks/cross/build/.gcc.build] Error 2
```
I've been digging though the config.log files without finding anything of interest. Hints welcome.

FWIW - `cross` builds fine as is on MacOS Mojave.

--Mellvik

tkchia, 2021-05-18:
Hello @Mellvik ,

Strange.  It looks like there was some sort of weird mismatch between your C(++) compiler environment when running `configure`, and your compiler environment when running `make` (i.e. when actually building `gcc-ia16`).  As for why this is happening, I am still not sure.

Since the errors now occur when building `gcc-ia16` proper, this time round you probably want to look at `cross/build/gcc-build/gcc/config.log`, for detailed logs of the `configure` run for the `gcc/` subtree.  Try looking especially at the parts where `configure` tries to look for `sys/resource.h` and `strsignal`.

> ```
> /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/sys/resource.h:89:25: error: 'long type-name' is invalid
> typedef __uint64_t      rlim_t;
>                         ^
> ./auto-host.h:2321:16: note: expanded from macro 'rlim_t'
> #define rlim_t long
>                ^
> ```

Here it seems that the auto-generated file `gcc/auto-host.h` supplied its own definition of `rlim_t`, which ended up conflicting with macOS Xcode's definition of `rlim_t`.  But by right, `gcc/auto-host.h` should define `rlim_t` _only_ if the build system (i.e. macOS) does not provide a `<sys/resource.h>` or does not provide its own `rlim_t`.

Anyway, on my Ubuntu build system, the relevant part of `gcc/auto-host.h` comes out as

```
/* Define to `long' if <sys/resource.h> doesn't define. */
#ifndef USED_FOR_TARGET
/* #undef rlim_t */
#endif
```

The situation for `strsignal` looks similar.  `gcc/system.h` is declare its own prototype of `strsignal`, but by right it should only do that if the build system does not already declare `strsignal`.

> where suddenly `g++` is forced - probably erroneously - to compile C code

This seems to be a deliberate choice on the part of the GCC authors.  The GCC source files have a `.c` extension, but they actually use (a few) C++ language features, so they effectively need to be compiled as C++ code.  The warning should be harmless.

Thank you!

Mellvik, 2021-05-19:
Thank you @thchia,

turns out - as you suspected, that removing resource.h and 'has strsignal' from gcc/configure provides for a successful compile. I haven't figured out how to make that happen automagically yet, will dive back into it later today.

--Mellvik

Mellvik, 2021-05-19:
> Does cross/build/gcc-build/gcc/config.log provide any information on why (or how) the <sys/resource.h> and strsignal tests are failing?
> 
> 
> 
@tkchia,

I have not found any reference to either of them in the log files, will take another stab at that tomorrow.

--M

Mellvik, 2021-05-25:
@tkchia, 
sorry about the delay. Configure does indeed find `sys/resource.h` and the presence of `strsignal`. Preventing it from finding them, would apparently solve the problem.

Since the toolchain compiles and works on my MacMini (running Mojave), I moved my work to that machine for now, until I get the time to dive further into the issue.

Pending that, it makes sense to keep this issue open.

--M

ghaerr, 2021-12-04:
Hello @Mellvik,

Did this issue ever get resolved? And/or is it specific to a single macOS machine/OS version you have?

Thank you!

ghaerr, 2022-02-08:
Hello @Mellvik,

Should this issue remain open? 

Thank you!

Mellvik, 2022-02-08:
Yes, this one's still open.

I'm running ELKS development off of a Mojave based mac,
on newer OSes the toolchain compile fails. A real pain, but not bad enough to dive even deeper yet.

Is your experience differently by any chance?

—M

Mellvik, 2022-06-19:
As of XCODE Version 13.3.1 (13E500a) the ELKS toolchain compiles fine on MacOS Monterey (12.4).

closed 2022-06-19 by Mellvik