ghaerr/elks #776: Kernel Panic when opening symbolic link to FAT file

Mellvik, 2020-10-08:
For all I know this may be a known problem, but here we go:

When https opens a file (for send) and the file is a symbolic link to a file on a mounted FAT filesystem, the kernel crashes like this. The problem is repeatable.
```
login: root
# mount -t fat /dev/bda1 /mnt
FAT: me=f8,csz=16,#f=2,floc=1,fsz=244,rloc=489,#d=512,dloc=521,#s=997857,ts=997857
  FAT: 498M, fat16 format
                         # ls -l /var/www
lrwxrwxrwx  1 root     root           15  Oct 08 15:47 image -> /mnt/elks/image
-rwxr-xr-x  1 root     root          275  Sep 30 12:34 index.html
lrwxrwxrwx  1 root     root            8  Sep 30 18:48 xx -> /dev/bda
# cat /var/www/image > dev/null
#
  panic: Process 26 exceeded kernel stack limit! magic 8

                                                        apparent call stack:
                                                                            Line: Addr    Parameters
                    ~~~~: ~~~~    ~~~~~~~~~~
                                               0: 2A3B => 1400 1C08 1400 1C40 0400 001A 0011
               1: 3694 => 0000 0000 0000 0000 0000 0000 0000
                                                               2: 0000 => 642F 7665 632F 6E6F 6F73 656C 5500
                               3: 0000 => 642F 7665 632F 6E6F 6F73 656C 5500
                                                                               4: 0000 => 642F 7665 632F 6E6F 6F73 656C 5500
                                               5: 0000 => 642F 7665 632F 6E6F 6F73 656C 5500
               6: 0000 => 642F 7665 632F 6E6F 6F73 656C 5500
                                                               7: 0000 => 642F 7665 632F 6E6F 6F73 656C 5500
                               8: 0000 => 642F 7665 632F 6E6F 6F73 656C 5500
                                                                            SYSTEM HALTED - Press CTRL-ALT-DEL to reboot:
```

Mellvik, 2020-10-08:
Update: 
The `cat /var/www/image > /dev/null` works once. The next attempt hangs the system hard, no messages.

ghaerr, 2020-10-08:
Hi @Mellvik,

This crash is the same as the previous kernel stack limit exceeded problem you reported several weeks ago.

The problem is easily fixed by increasing the kernel stack size in sched.h:
```
#define KSTACK_BYTES 600    /* Size of kernel stacks */
```
I've upped the kernel stack size from 512 to 600 so that the system can run with system call traces turned on:

With the cool CONFIG_STRACE debugging option we have now (settable under kernel options), the following trace shows what's going on when running "hd image" in the /var/www directory:
```
[25/4750: 11 execve      ("/bin/hd", 0x2CAE, 94)][25:execve/ret=0,ks=214/288]
[25/2e84: 69 sbrk        (0)][25:sbrk/ret=0,ks=126/288]
[25/2e84: 69 sbrk        (1024)][25:sbrk/ret=0,ks=126/288]
[25/2ea0:  5 open        ("image", 0, 438)][25:open/ret=3,ks=524/524]
[25/2e72: 54 ioctl       (3, 21505, 11906)][25:ioctl/ret=-22,ks=126/524]
[25/2e84: 69 sbrk        (0)][25:sbrk/ret=0,ks=126/524]
[25/2e84: 69 sbrk        (2048)][25:sbrk/ret=0,ks=126/524]
[25/2e34: 54 ioctl       (1, 21505, 11844)][25:ioctl/ret=0,ks=126/524]
```
Just prior to opening "image", the kernel stack is running 126 bytes, with a max of 288. After accessing the FAT filesystem open call through the symlink "/mnt/var/www/image", the kernel stack is 524 bytes, which overflowed the previous max of 512.

So the FAT filesystem is using 398 bytes of extra kernel stack - way too much! I will have to look into this.

Try running at 600 for a while and see whether this solves your problem, it should. It would be nice to see whether your previously reported crash, using various "cp" invocations, is also solved with this stack size.

I previously shrunk the kernel stack size from 988 bytes to 512 in an effort to reduce kernel data space usage since this number is times 16 (processes).

Also, I plan to fix the kernel printk problem on serial lines, (no CR), PR coming.

Mellvik, 2020-10-10:
Thanks for the fix, @ghaerr - confirm that printk problem is also solved. 

Time to close.

closed 2020-10-10 by Mellvik