ghaerr/elks #796: [kernel][fat] Fatal FAT fs bug?

Mellvik, 2020-10-16:
Writing BIG file to FAT fs crashes entire disk:
```
# mount -t fat /dev/bda1 /mnt
# cd /mnt
# dd if=/dev/bda of=10m-file bs=2b count=10000
# echo no problem
# dd if=/dev/bda of=100m-file bs=2b count=100000
# ls
garbage listing
```
The MBR and the partition boot block are both gone. No error messages. The final `ls` seems to mess up the kernel too, a following `ps` does not list all processes.

--M

Mellvik, 2020-10-16:
> This is reading and writing from the same filesystem, so it seems the free space on the FAT filesystem will always be used up.
> 
> Thus, it is hard to tell if this is a write issue with large files, or more likely a problem with running out of FAT filesystem space.
> 
Not really. It's reading from the raw device, not the file system, and it's repeatable on qemu (fs size 512mb). My guess (will check) is that using /dev/zero as the input file will give the same result.
> Further testing could include showing the FAT display after the mount, to know the filesystem specifics, as well as turning on DEBUG_FAT (and ^P before testing to turn on debug messages) to display what is happening in more detail.
> 
Good idea, will include that in the next post.

--M
> —

Mellvik, 2020-10-17:
Probably unrelated, but while working on this (now with a destroyed FAT file system so a good time to do something else with that disk) I repartitioned and created a /dev/bda2 size 65535. Then copied /bin from floppy to the new file system, and df() reports odd numbers:
```
# mkfs /dev/bda2 65535
63424 inodes
65535 blocks
Firstdatazone=2000 (2000)
Zonesize=1024
Maxsize=67107840

# mount /dev/bda2 /mnt
# mkdir /mnt/bin
# cp /bin/* /mnt/bin
# df 
Filesystem    1k-Blocks     free     used    %  FUsed%
/dev/fd0           1440      377     1063  74%    50%
# df /dev/bda2
Filesystem    1k-Blocks     free     used    %  FUsed%
/dev/bda2         65535    38261    27274  42%    40%
# du /mnt
941	/mnt/bin
942	/mnt
# 
```

Mellvik, 2020-10-17:
Update to the previous post - there seem to be something else at work here:
```
# mkfs /dev/bda2 65535
63424 inodes
65535 blocks
Firstdatazone=2000 (2000)
Zonesize=1024
Maxsize=67107840

# mount /dev/bda2 /mnt
# ls /mnt

# mkdir /mnt/bin
# umount /mnt
panic: brelse
apparent call stack:
Line: Addr    Parameters
~~~~: ~~~~    ~~~~~~~~~~
   0: 06C1 => 0302 90A8 8EEE 0000 0000 0F0E 0100
   1: 000A => 0000 0000 0000 0000 0000 0000 0000
   2: 0000 => 736E 6C6F 0065 6E55 6261 656C 7420
   3: 6F63 => 0000 0000 0000 0000 0000 0000 0000
   4: 0000 => 642F 7665 632F 6E6F 6F73 656C 5500
   5: 0000 => 6564 2F76 6F63 736E 6C6F 0065 6E55
   6: 2F00 => 642F 7665 632F 6E6F 6F73 656C 5500
   7: 0000 => 6564 2F76 6F63 736E 6C6F 0065 6E55
   8: 2F00 => 642F 7665 632F 6E6F 6F73 656C 5500
SYSTEM HALTED - Press CTRL-ALT-DEL to reboot:
```
The creation of /mnt/bin turns out to be unimportant: mount and then umount is enough to crash the system. Doing a mount and then umount of a FAT file system works fine.

Maybe there is a clue in the fact that fsck does not like a newly created file system:
```
# mkfs /dev/bdb3 60000
56309 inodes
60000 blocks
Firstdatazone=1777 (1777)
Zonesize=1024
Maxsize=61440000

# fsck -fsv /dev/bdb3
fsck 1.0.1 - 09/01/20
fsck: bad s_imap_blocks field in super-block
#
```

Mellvik, 2020-10-17:
Back to the original issue:

```
ELKS 0.3.0

login: root
# mount -t fat /dev/bda1 /mnt
FAT: me=f8,csz=16,#f=2,floc=1,fsz=244,rloc=489,#d=512,dloc=521,#s=997857,ts=997857
FAT: 498M, fat16 format
# ls /mnt
autoexec.bat  autoexec.n2k  autoexec.old  autoexec.pe3  bin           
command.com   compaq        config.old    config.sys    dos           
drvspace.bin  elks          ibmbio.com    ibmdos.com    io.sys        
kernels       mscdext       msdos.sys     mtcp          nansi.sys     
norton        old_dos.1     pcplus        pkzip         plusdiag      
psfiles       telix         tmp           wina20.386    xx            

# dd if=/dev/zero of=/mnt/elks/100M-file bs=2b count=100000
100000+0 records in
100000+0 records out
102400000 bytes (100000 KiB) copied
# ls /mnt/elks
/mnt/elks: No such file or directory 
# ls /mnt

# umount /mnt
# mount -t fat /dev/bda1 /mnt
FAT: me=0,csz=0,#f=0,floc=0,fsz=0,rloc=0,#d=0,dloc=0,#s=0,ts=0
FAT: Unsupported format
mount failed: Invalid argument 
# 

```

ghaerr, 2020-10-18:
Hello @Mellvik,

```
# mkfs /dev/hda1 65535
63424 inodes
65535 blocks
Firstdatazone=2000 (2000)
Zonesize=1024
Maxsize=67107840

# mount /dev/hda1 /mnt
# mkdir /mnt/bin
# cp /bin/* /mnt/bin
# umount /dev/hda1
```
I tested the above on the latest ELKS and yes, this crashes the system.

I will have to check more deeply, but I think the max MINIX v1 filesystem size for ELKS is 31752 1k-blocks, which equals the disk geometry in image/config.in:

63 sectors \* 16 heads \* 63 cylinders = 63504 disk blocks, which equals 31752 ELKS 1k-blocks.

When I try "mkfs /dev/hda1 31752", I can't get it to fail. Trying 40000, it seems to work. Trying 50000, it fails.

So, I suspect that's what the problem here is.
In order to fully understand more what is happening here, it might be worth testing both FAT and MINIX filesystems that are limited to 31752 1k-blocks (~32Mb) in size first. 

The MINIX v1 free list is a bitmap and is limited in max size. Thus, mkfs should definitely refuse to create disks that are too large to work.

For FAT filesystems, there could very well be a bug in the ELKS filesystem driver not limiting writing beyond either the hardware disk or FAT file chain limit. FAT provides sector clustering which allows larger block sizes using the same FAT chain limitations.

> FAT: me=f8,csz=16,#f=2,floc=1,fsz=244,rloc=489,#d=512,dloc=521,#s=997857,ts=997857
FAT: 498M, fat16 format

Your disk was using cluster size 16 on FAT16 format, which uses 16-bit FAT chains: 65535 \* 16 \* 512 = 512Mb, which is how the 498Mb disk worked on FAT16.

I have been neck-deep in the emu86 emulator and ROM filesystems, but plan to get back to this once I remember the math involved for calculating max ELKS CHS, partition sizes, and MINIX/FAT filesystem sizes.

Thank you!

ghaerr, 2020-10-18:
@Mellvik,

More on MINIX and FAT math:

My statement above that MINIX v1 filesystem is limited to 32MB on ELKS is incorrect. Actually, the MINIX free list bitmap is allowed to be 8 zones of 1024 bytes which = 8 \* 1024 \* 8 = 64k. ELKS uses 16-bit block numbers, so one could have in theory up to 64Mb MINIX partitions.

However, given a block number above 31752, the conversion of block number to CHS will end up being larger than 63/16/63, which older BIOSes may not support. In ELKS, the disk block to CHS conversion happens starting in genhd.c on line 709. That may be not working with QEMU, as the ELKS disk block -> CHS conversion may not match the QEMU CHS -> disk block reconversion.

With regards to FAT, the filesystem allows much larger filesystem sizes through clusters as discussed, but ELKS is still limited to a 16-bit block number, and any block number > 32k will have the same CHS issue above.

It seems the problem may be related to specific BIOS or QEMU handling of CHS conversions of blocks > 32k. I think the DEBUG_BLK option should help debug/show this, let me know if it needs more work.

ghaerr, 2020-10-19:
@Mellvik,

I turned on DEBUG_BLK and also wrote some code to display the CHS values used when "mkfs /dev/hda1 50000" is used to create a filesystem. Although the math above is correct, it seems none of that has to do with the issue here. I've also (re)realized that the cylinders portion of CHS is almost always larger than 63 for most devices... thus my statements about BIOS incompatibility are incorrect.

I haven't yet tracked down the real reason for the kernel panic on MINIX filesystems, and the FAT filesystem issue is completely different. The reason fsck is giving the "bad s_imap_blocks field in super-block" is that the maximum inode blocks on ELKS is 4, and mkfs is creating 5 or more. In addition, fsck is dying half the time just trying to check various larger images. I'm afraid both ELKS fsck and mkfs are buggy (again).

Its been a bit slow going since I can't seem to get QEMU to boot with a floppy and flat (non-partitioned) hard drive, QEMU always tries to boot the hard drive instead of the floppy. We need to continue testing with various volume sizes of 30000, 40000 and 50000 as discussed above, but should use the reliable mfs and fsck tools on the host OS, rather than ELKS versions, to more easily determine the cause(s) of this issue.

ghaerr, 2020-10-19:
@Mellvik,

PR #807 should fix all MINIX bugs relating to this issue. The system should now work with MINIX filesystems up to and including 65535 blocks, created from either mkfs or mfs.

Unfortunately, we will need to get a duplication on the FAT issue in order to debug that. Suggest turning on DEBUG_FAT and possibly DEBUG_BLK (then use ^P to turn on) to get a trace.

Mellvik, 2020-10-19:
Thank you @ghaerr,

Will have a repeatable case for the fat- problem shortly.

-M

Mellvik, 2020-11-03:
More debugging:
There may actually be more than one problem here, so I'll start with the one that hits me first:
When trying to exceed the available space on a FAT file system, the error is reported correctly:
```
# cd elks
# cp gg xx
gg: I/O error 
Failed to copy gg -> xx
#
```
Then removing the partial file not only fails, but it screws up the file system:
```
# ls
bin        gg         hh         image      image.hex  xx         
# rm xx
FAT: delete past EOF# 
```
Then, after a `sync` and `reboot`:
```
ELKS 0.3.0

login: root
# mount -t fat /dev/hda1 /mnt/a
FAT: me=f8,csz=16,#f=2,floc=1,fsz=244,rloc=489,#d=512,dloc=521,#s=997857,ts=997857
FAT: 498M, fat16 format
# mount -t fat /dev/hdb1 /mnt/b
FAT: me=20,csz=54,#f=48,floc=25632,fsz=11822,rloc=3264,#d=12832,dloc=3298,#s=8240,ts=908996142
FAT: 4120k, fat12 format
# 
```
Before the `rm xx` the two file systems were exactly the same - 498MB. (BTW this is on QEMU).

New printouts of the same sequence with DEBUG_BLK and DEBUG_FAT coming shortly.

--Mellvik

Mellvik, 2020-11-03:
More debug output from the above:
There are obviously thousands of lines, here's a hopefully smart selection:

```
# ls -l
BLK 14276 read
dir: '.          ' attr 10
dir: '..         ' attr 10
dir: 'BIN        ' attr 10
dir: 'IMAGE       ' attr 20
dir: 'IMAGE   HEX ' attr 20
dir: 'GG          ' attr 20
dir: 'HH          ' attr 20
dir: '.          ' attr 10
dir: '..         ' attr 10
dir: 'BIN        ' attr 10
BLK 978 read
drwxrwxrwx  3 root     root         8192  Aug 28 13:15 bin
dir: '.          ' attr 10
dir: '..         ' attr 10
dir: 'BIN        ' attr 10
dir: 'IMAGE       ' attr 20
dir: 'IMAGE   HEX ' attr 20
dir: 'GG          ' attr 20
-rwxr-xr-x  1 root     root     73128360  Oct 23 12:24 gg
dir: '.          ' attr 10
dir: '..         ' attr 10
dir: 'BIN        ' attr 10
dir: 'IMAGE       ' attr 20
dir: 'IMAGE   HEX ' attr 20
dir: 'GG          ' attr 20
dir: 'HH          ' attr 20
-rwxr-xr-x  1 root     root     109692541  Oct 23 12:24 hh
dir: '.          ' attr 10
dir: '..         ' attr 10
dir: 'BIN        ' attr 10
dir: 'IMAGE       ' attr 20
-rwxr-xr-x  1 root     root      1451520  Oct 08 15:47 image
dir: '.          ' attr 10
dir: '..         ' attr 10
dir: 'BIN        ' attr 10
dir: 'IMAGE       ' attr 20
dir: 'IMAGE   HEX ' attr 20
-rwxr-xr-x  1 root     root      6094030  Oct 14 09:11 image.hex
# cp gg xx
BLK 768 read
BLK 769 read
BLK 770 read
BLK 771 read
BLK 772 read
dir: '.          ' attr 10
dir: '..         ' attr 10
dir: 'BIN        ' attr 10
dir: 'IMAGE       ' attr 20
dir: 'IMAGE   HEX ' attr 20
dir: 'GG          ' attr 20
dir: 'HH          ' attr 20
dir: '.          ' attr 10
dir: '..         ' attr 10
dir: 'BIN        ' attr 10
dir: 'IMAGE       ' attr 20
dir: 'IMAGE   HEX ' attr 20
dir: 'GG          ' attr 20
dir: '.          ' attr 10
dir: '..         ' attr 10
dir: 'BIN        ' attr 10
dir: 'IMAGE       ' attr 20
dir: 'IMAGE   HEX ' attr 20
dir: 'GG          ' attr 20
dir: 'HH          ' attr 20
[... 198 lines cut ...]
BLK 14276 read
write_inode block write 14276
Failed to copy gg -> xx
# 
```
By now the file system has been damaged, and and mount is not possible:
```
# cd /mnt/a
BLK 992 read
# 
# 
# umount /mnt/b
BLK 18 read
BLK 5 read
BLK 504 read
BLK 505 read
/mnt/b: Device or resource busy 
# 
```

@ghaerr, is this useful at all?

The other problem, which started this issue, was copying a large file to a FAT file system. The root cause may be the same I guess.

--Mellvik

ghaerr, 2020-11-04:
@Mellvik,

Thanks for the debugging. This is going to take a little while. I haven't got it to repeat yet on smaller (1.44Mb) FAT floppies; I'm not sure if the problem has to do with large block numbers (> 32767?) or just FAT16 EOF. The floppies are FAT12 (I think) so perhaps it is only FAT16 related. Depending on other priorities, this issue may be fixed in the next version.

Mellvik, 2020-11-04:
That's fine with me, @ghaerr.

FWIW - when the problem was first discovered, there was no filsystem EOF issue involved, just BIG files - like 100MB.

I'll do another test asap - almost filling up the file system in DOS, then copy just a small file around and see if it is repeatable. We have have two separate issues.

—Mellvik

Mellvik, 2020-11-05:
Actually @ghaerr - 
more testing uncovered new 'evidence':
- the file copy error documented above is not caused by running out of space. There is ample space on the device.
```
# cd elks
# cp gg xx
gg: I/O error 
Failed to copy gg -> xx
#
```
BTW - I don't think we have a way to `df` a FAT volume at this time, do we?

Further - when the volume is actually running out of space, the behaviour is correct:
```
... ...
BLK 121 read
xx: BLK 16 read
BLK 8 read
BLK 1069 read
BLK 1070 read
No space left on device 
put_inode 456858 count 1 dirty 1
write_inode 456858 0
BLK 14276 read
write_inode block write 14276
Failed to copy image.hex -> xx
# 
```
A possibly useful observation is that the destination file is closed and kept when the `no space left on device` error happens, it may be a better idea to delete it.

At this point, the FAT filesystem is still ok, unmountable, remountable. 

Possibly a minor thing in the debug setup, `cd` to a directory on the mounted FAT drive yields double or quintuple directory listings:
```
# mount -t fat /dev/hdb1 /mnt/b
BLK 9 read
BLK 0 read
FAT: me=f8,csz=16,#f=2,floc=1,fsz=244,rloc=489,#d=512,dloc=521,#s=997857,ts=997857
FAT: 498M, fat16 format
BLK 244 read
# cd /mnt/b/elks
dir: 'IO      SYS' attr 7
dir: 'MSDOS   SYS' attr 7
dir: 'COMMAND COM' attr 1
dir: 'IBMBIO  COM'' attr 27
dir: 'IBMDOS  COM'' attr 27
dir: 'DOS        ' attr 10
dir: 'WINA20  386' attr 1
dir: 'CONFIG  SYS ' attr 20
dir: 'AUTOEXECBAT ' attr 20
dir: 'OLD_DOS 1  ' attr 10
dir: 'AUTOEXECN2K ' attr 20
dir: 'BIN        ' attr 10
dir: 'MTCP       ' attr 10
dir: 'COMPAQ     ' attr 10
dir: 'TMP        ' attr 10
BLK 245 read
dir: 'PKZIP      ' attr 10
dir: 'DRVSPACEBIN!' attr 21
dir: 'NANSI   SYS ' attr 20
dir: 'CONFIG  OLD ' attr 20
dir: 'AUTOEXECOLD ' attr 20
dir: 'PCPLUS     ' attr 10
dir: 'AUTOEXECPE3 ' attr 20
dir: 'PSFILES    ' attr 10
dir: 'MSCDEXT    ' attr 10
dir: 'PLUSDIAG   ' attr 10
dir: 'ELKS       ' attr 10
BLK 3 read
dir: 'IO      SYS' attr 7
dir: 'MSDOS   SYS' attr 7
dir: 'COMMAND COM' attr 1
dir: 'IBMBIO  COM'' attr 27
dir: 'IBMDOS  COM'' attr 27
dir: 'DOS        ' attr 10
dir: 'WINA20  386' attr 1
dir: 'CONFIG  SYS ' attr 20
dir: 'AUTOEXECBAT ' attr 20
dir: 'OLD_DOS 1  ' attr 10
dir: 'AUTOEXECN2K ' attr 20
dir: 'BIN        ' attr 10
dir: 'MTCP       ' attr 10
dir: 'COMPAQ     ' attr 10
dir: 'TMP        ' attr 10
dir: 'PKZIP      ' attr 10
dir: 'DRVSPACEBIN!' attr 21
dir: 'NANSI   SYS ' attr 20
dir: 'CONFIG  OLD ' attr 20
dir: 'AUTOEXECOLD ' attr 20
dir: 'PCPLUS     ' attr 10
dir: 'AUTOEXECPE3 ' attr 20
dir: 'PSFILES    ' attr 10
dir: 'MSCDEXT    ' attr 10
dir: 'PLUSDIAG   ' attr 10
dir: 'ELKS       ' attr 10
# ls
BLK 755 read
BLK 756 read
BLK 757 read
BLK 758 read
BLK 759 read
BLK 760 read
BLK 761 read
BLK 763 read
BLK 762 read
BLK 764 read
BLK 765 read
BLK 766 read
BLK 767 read
BLK 14276 read
dir: '.          ' attr 10
dir: '..         ' attr 10
dir: 'BIN        ' attr 10
dir: 'IMAGE       ' attr 20
dir: 'IMAGE   HEX ' attr 20
dir: 'XX          ' attr 20
dir: 'II          ' attr 20
dir: 'GG          ' attr 20
dir: 'HH          ' attr 20
dir: 'GGG         ' attr 20
dir: '.          ' attr 10
dir: '..         ' attr 10
dir: 'BIN        ' attr 10
dir: '.          ' attr 10
dir: '..         ' attr 10
dir: 'BIN        ' attr 10
dir: 'IMAGE       ' attr 20
dir: 'IMAGE   HEX ' attr 20
dir: 'XX          ' attr 20
dir: 'II          ' attr 20
dir: 'GG          ' attr 20
dir: '.          ' attr 10
dir: '..         ' attr 10
dir: 'BIN        ' attr 10
dir: 'IMAGE       ' attr 20
dir: 'IMAGE   HEX ' attr 20
dir: 'XX          ' attr 20
dir: 'II          ' attr 20
dir: 'GG          ' attr 20
dir: 'HH          ' attr 20
dir: 'GGG         ' attr 20
dir: '.          ' attr 10
dir: '..         ' attr 10
dir: 'BIN        ' attr 10
dir: 'IMAGE       ' attr 20
dir: 'IMAGE   HEX ' attr 20
dir: 'XX          ' attr 20
dir: 'II          ' attr 20
dir: 'GG          ' attr 20
dir: 'HH          ' attr 20
dir: '.          ' attr 10
dir: '..         ' attr 10
dir: 'BIN        ' attr 10
dir: 'IMAGE       ' attr 20
dir: 'IMAGE   HEX ' attr 20
dir: 'XX          ' attr 20
dir: 'II          ' attr 20
dir: '.          ' attr 10
dir: '..         ' attr 10
dir: 'BIN        ' attr 10
dir: 'IMAGE       ' attr 20
dir: '.          ' attr 10
dir: '..         ' attr 10
dir: 'BIN        ' attr 10
dir: 'IMAGE       ' attr 20
dir: 'IMAGE   HEX ' attr 20
bin        gg         ggg        hh         ii         image      image.hex  
dir: '.          ' attr 10
dir: '..         ' attr 10
dir: 'BIN        ' attr 10
dir: 'IMAGE       ' attr 20
dir: 'IMAGE   HEX ' attr 20
dir: 'XX          ' attr 20
xx         
# 
```

At this point, copying a 73MB file causes the error that started this thread ( sudden I/O error). At that point, 30138368 bytes have been copied, 29.432MB. The FS is unmountable, but a remount fails:
```
FAT: me=31,csz=32,#f=32,floc=25395,fsz=8248,rloc=27187,#d=12320,dloc=27189,#s=8242,ts=540029472
FAT: Unsupported format
```

In other words, the indications are that the problem is large files.

ghaerr, 2020-11-05:
@Mellvik : Thanks for the additional debugging.

Looking into the FAT filesystem driver in more detail last night, there are all sorts of potential issues with FAT; back in February I made it work for apparently limited cases, but it is far from finished. For instance, I just found a global variable used in cluster allocations. Long story short, currently, FAT will only work properly when a *single* FAT instance is mounted, not two. This includes booting from FAT.

You're mounting two FAT systems simultaneously, right? That currently doesn't work, even though it might seem too. Booting from FAT and mounting a FAT filesystem won't work either. So we need to restrict the testing to a single mounted FATFS for now.

That global variable was apparently meant to increase speed; I think I can remove it along with another item I've found for testing, but this whole thing is a definite rabbit hole, unfortunately. Here's the code I was looking at, along with the FIXME I added in February:
```
int FATPROC msdos_add_cluster(register struct inode *inode)
{
        static struct wait_queue wait;
        static int lock = 0;
        //FIXME: using previous on booted FAT volume with mounted FAT floppy won't work well
        static long previous = 0; /* works best if one FS is being used */
```

I will need more specific information on the size of 'gg', the initial filesystem and its free space (we don't have a df for FAT at all, yet more work), and how the filesystem is created in order to properly duplicate. I do believe you are on to something with large files, or possibly large block numbers. The best testing scenario would be building ELKS scripts that create a hard drive using our tools from scratch of a certain size, empty, then copy a large file onto it from MINIX. The add_cluster and FAT chain code is quite complicated. This may also be hindered because I think our current tools can't create a FAT filesystem larger than 32MB!

> Possibly a minor thing in the debug setup, cd to a directory on the mounted FAT drive yields double or quintuple directory listings

This is likely the global variable problem, but frankly I have no idea. We definitely need to test on single FAT filesystems only for the present.

I think the best (first) approach would be for me to produce a PR without the global variable. In addition, this whole problem is complicated by the fact that the FAT driver uses different code for FAT12, FAT16 and FAT32, so we need to be quite particular about the problems.

Thank you!

ghaerr, 2020-11-05:
> I think the best (first) approach would be for me to produce a PR without the global variable.

@Mellvik, since you're testing, just deleting the "static" in front of the "long previous = 0" will work in eliminating the restriction of multiple FAT filesystems mounted, for the time being. You might just try doing that (in elks/fs/msdos/misc.c) and rerunning your tests, instead of me producing a PR with that one change.

ghaerr, 2020-11-05:
Hello @Mellvik,

After looking harder at your excellent debug log, I think I've found the reason for the disk trashing with large files:

Remember the problem with the MINIX filesystem being limited to 65Mb? That's because ELKS uses (unsigned) 16-bit block numbers to access the disk. Well, our FAT driver has the same problem. It appears that when writing the large file, the block numbers start small, then get larger, pass 15 bits (when the printks incorrectly show the negative block numbers), then wrap at 16 bits, where the block numbers start from 0 again. But FATFS keeps on writing, which overwrites the boot block, FAT table, etc.

It appears the FAT FS was original written for 32-bit Linux, and converts a 32-bit FAT sector number into a block number, which is being silently truncated to 16 bits in the I/O subsystem.

I will post a fix that will stop the disk trashing, so you can test this theory. I'm not sure what to do about this problem. Using FAT cluster sizes doesn't solve the problem. I am pretty certain this problem also happens when *reading* a file whose disk location is past 65Mb from the start of the partition.

One possible answer is to increase ELKS block size to 32 bits, which I will look in to. Can you tell me the CHS for the 498MB drive you're testing on?
I think the disk size limit will be limited to CHS 1023/16/63 = 1023\*16\*63 = 527MB.

Mellvik, 2020-11-05:
@ghaerr,
you are debugging this faster than I can answer messages :-) 
and it sounds like you're onto the root cause right there. It always seemed like a word length related problem.

BTW remember that file copies, read, writes work fine on the almost full (527MB) FAT file system as long as the files are small. Only large files cause problems and even when the file system is relatively empty.

As to two FATs concurrently mounted (from a previous message): Yes, the observation is correct. But the big file problem was discovered when only one FAT FS was mounted, the 2nd came along while debugging.

As to the origin of the FS: It was created by DOS 6.22 FORMAT on the COMPAQ, imported via ELKS httpd (!), and 'scandisked' by DOS on QEMU. The large files were created under DOS on QEMU. As to the file sizes - here's the dir list:

drwxrwxrwx  3 root     root         8192  Aug 28 13:15 bin
-rwxr-xr-x  1 root     root     73128360  Oct 23 12:24 gg
-rwxr-xr-x  1 root     root     73128360  Oct 23 12:24 ggg
-rwxr-xr-x  1 root     root     30138368  Nov 05 13:19 gggg
-rwxr-xr-x  1 root     root     109692541  Oct 23 12:24 hh
-rwxr-xr-x  1 root     root     109692541  Oct 23 12:24 ii
-rwxr-xr-x  1 root     root      1451520  Oct 08 15:47 image
-rwxr-xr-x  1 root     root      6094030  Oct 14 09:11 image.hex
-rwxr-xr-x  1 root     root      6094030  Nov 05 12:01 xx
# 

I managed to screw up even the copy of this FS so the file sizes will be different in the next run.

Disk params - here's what MacOS thinks:

tmp> file elks.bda.new.img 
elks.bda.new.img: DOS/MBR boot sector MS-MBR,D0S version 3.3-7.0 english at offset 0x8b "Invalid partition table" at offset 0xa3 "Error loading operating system" at offset 0xc2 "Missing operating system"; partition 1 : ID=0x6, active, start-CHS (0x0,1,1), end-CHS (0x3dd,15,63), startsector 63, 997857 sectors

Elks/Qemu:
bioshd: hda BIOS CHS 992,16,63
bioshd: hda  IDE CHS 993,16,63
bioshd: hdb BIOS CHS 992,16,63
bioshd: hdb  IDE CHS 993,16,63
bioshd: 1 floppy drive and 2 hard drives

—Mellvik

ghaerr, 2020-11-05:
> BTW remember that file copies, read, writes work fine on the almost full (527MB) FAT file system as long as the files are small. Only large files cause problems and even when the file system is relatively empty.

Actually, the code I'm looking at shows the above is likely not true. That is, the sector \# is being treated as sector % 65536 for all reads and writes, which means that even small file I/O is mapping large sectors to (sectors % 65536). Reading and writing a small file will appear correct (but use the truncated block number). If the block numbers in the file are > 65535, then they'll likely overwrite another file on disk, but would still read-back correctly (using the truncated block numbers).

This could be checked by verifying that all the large files are still correct using a DOS utility, after writing several small files.

The reason that the "large file" problem is trashing your disk is that the large files you're testing with are guaranteed to wrap a 16-bit block number, which then overwrites low sectors, including the FAT table. Its just a matter of chance whether a small file happens to allocate a block whose (sector % 65536) == [0..20] or so.

I'll produce something you can test with tomorrow.

Mellvik, 2020-11-05:
Makes sense @ghaerr, Sounds like you're closing in on the culprit, awaiting your next test.

Thank you!

M

Mellvik, 2020-11-06:
Thank you @ghaerr, this rocks.

closed 2020-11-06 by Mellvik