ghaerr/elks #815: Mouse unusable in nx GUI on slower XT machines

pawosm-arm, 2020-10-24:
Mouse arrow movements are lagging massively making it unusable in nx GUI. I guess the way mouse is handled isn't very optimal and it's hitting hard the performance on XT machines. The projects like FreeGEM demonstrate that it is possible to achieve smooth mouse pointer movements on those machines.

ghaerr, 2020-10-25:
Lets see if the "fast" serial driver helps solve this problem (assuming you're using a serial mouse on com1: 

Try editing ports.h, uncomment "CONFIG_FAST_IRQ4" and comment out "CONFIG_NEED_IRQ4", then make clean; make.

Making sure the serial port is running as quickly as possible is the first step. Secondly, the mouse threshold and acceleration can be adjusted in elkscmd/nano-X/drivers/mou_ser.c:
```
#define SCALE       3   /* default scaling factor for acceleration */
#define THRESH      5   /* default threshhold for acceleration */
```

pawosm-arm, 2020-10-25:
> Try editing ports.h, uncomment "CONFIG_FAST_IRQ4" and comment out "CONFIG_NEED_IRQ4", then make clean; make.

I've tried this and it does not seem to be sufficient, there's still annoying lag.

> Secondly, the mouse threshold and acceleration can be adjusted in elkscmd/nano-X/drivers/mou_ser.cSecondly, the mouse threshold and acceleration can be adjusted in elkscmd/nano-X/drivers/mou_ser.c

What are meanings and accepted range for those values?

Mellvik, 2020-10-25:
@pawosm-arm,
just in case - did you check the active tty params on the port after the port has been activated (i.e. mouse is active)?

stty -a < /dev/ttyS0

… or whatever the name of the port is. Maybe there is a clue —

—Mellvik

pawosm-arm, 2020-10-25:
@Mellvik before running `nxdemo`:

```
speed 9600 baud; cs8 -parenb -parodd -hupcl -cstopb -cread -clocal
-ignbrk brkint -ignpar -parmrk -inpck -istrip -inlcr -igncr icrnl -ixon
-ixoff -ixany
opost onlcr -xtabs
isig icanon -iexten echo echoe -echok -echonl -noflsh
eof = ^D eol = <undef> erase = ^H intr = ^C kill = ^U quit = ^\ susp = ^Z
start = ^Q stop = ^S min = 1 time = 0
```

after running `nxdemo`:

```
speed 1200 baud; cs8 -parenb -parodd -hupcl -cstopb cread -clocal
-ignbrk -brkint -ignpar -parmrk -inpck -istrip -inlcr -igncr -icrnl -ixon
-ixoff -ixany
opost onlcr -xtabs
-isig -icanon -iexten -echo echoe -echok -echonl -noflsh
eof = ^D eol = <undef> erase = ^H intr = ^C kill = ^U quit = ^\ susp = ^Z
start = ^Q stop = ^S min = 0 time = 0
```

ghaerr, 2020-10-25:
> What are meanings and accepted range for those values?

The scale and threshold are applied as follows to each relative X and Y mouse position reported by the mouse, in engine/devmouse.c:
```
if (x > thresh)
        x = thresh + (x - thresh) * scale;
```
That is, no mouse acceleration happens if the mouse relative position hasn't changed by threshold (5), but when it does, the movement value is then multiple by scale (3).

So, setting threshold to 0 might be a good thing to try. Then setting scale to 1 would report exactly the mouse returned values to Nano-X.

If this doesn't help much, we look into the serial driver, there could be a way to improve things by having the driver wait until all serial mouse data is received before returning. We would need to know your type of mouse and the number of bytes sent per mouse movement. Say it is 6. Then the following code in drivers/mou_ser.c could be changed:
```
        termios.c_cc[VMIN] = 0;
        termios.c_cc[VTIME] = 0;
to
        termios.c_cc[VMIN] = 6;
        termios.c_cc[VTIME] = 1;
```
That effectively says "wait for 6 characters, and/or timeout after 1ms if nothing else received."
The original code says "don't wait for any character, just poll. Return after a single character read." This means that the entire read and mouse processing cycle has to be repeated until all mouse characters have been received. See MOU_Read() in drivers/mou_ser.c for details.

pawosm-arm, 2020-10-25:
> We would need to know your type of mouse and the number of bytes sent per mouse movement.

It's the most typical original Microsoft serial mouse C3KMS1/37964 ( https://fccid.io/C3KMS1 ). Is there a way I could measure number of bytes sent per movement?!

> So, setting threshold to 0 might be a good thing to try. Then setting scale to 1 would report exactly the mouse returned values to Nano-X.

I'm missing something I guess... Where it is reported?

ghaerr, 2020-10-25:
> I'm missing something I guess... Where it is reported?

Sorry - I meant "reported to the Nano-X driver".  You can use `mouse` to test the mouse driver. It doesn't use acceleration, but look at the source for it and you can have it report the byte count to the console.

pawosm-arm, 2020-10-25:
Well, I've made following change:

```
diff --git a/elkscmd/sys_utils/mouse.c b/elkscmd/sys_utils/mouse.c
index 4595312d..0682bbc8 100644
--- a/elkscmd/sys_utils/mouse.c
+++ b/elkscmd/sys_utils/mouse.c
@@ -295,7 +295,7 @@ int main(int argc, char **argv)
        printf("[Mouse test, ^C to quit]\n");
        while(1) {
                if(read_mouse(&x, &y, &z, &b) == 1) {
-                       printf("x:%4d, y:%4d, b:%2d\n", x, y, b);
+                       printf("x:%4d, y:%4d, b:%2d, nbytes: %d\n", x, y, b, nbytes);
                }
        }
        return 0;
```
The result is pasted into attached file.
[mouse.txt](https://github.com/jbruchon/elks/files/5435866/mouse.txt)

ghaerr, 2020-10-25:
Interesting. Due to the way you've coded it, nbytes actually shows the \# bytes remaining (read but not processed by the completed mouse parse routine) after each mouse movement.

I am thinking the actual problem isn't too little mouse data because of your slower system, but too much! That is, nano-X is getting tons of mouse data but not able to process it quickly enough, as every mouse movement ends up reading 16x16 pixels from the graphics screen, XORing a cursor into it, then a MASK operation, then rewriting 16x16 pixels. Your system just can't keep up.

I will have to think a bit about how to fix this. I'm thinking that perhaps nano-X may want to "dump" mouse data when too much has been received. The question is whether to dump the first mouse "droppings" or the latter ones.... pun intended.

Mellvik, 2020-10-25:
I have a compaq slt386 (laptop!) with a ms mouse (386sx processor) that can be pullled out of a shelf for testing if a comparison would be valuable.

M

ghaerr, 2020-10-25:
@pawosm-arm,

One idea would to drop mouse input would be to discard pending input while processing. Try the following mod in drivers/mou_ser.c function MOU_Read():
```
    while (nbytes-- > 0) { 
        if ((*parse)((int) *bp++)) {
            *dx = xd;
            *dy = yd;
            *dz = 0;
            b = 0;
            if(buttons & left)
                b |= LBUTTON;
            if(buttons & right)
                b |= RBUTTON;
            if(buttons & middle)
                b |= MBUTTON;
            *bptr = b;
            while (nbytes >= 3) { <--- add these 4 lines
                nbytes -= 3;
                bp += 3;
            }
            return 1;
```
Since MS mouse position reports are 3 bytes long, this drops all fully received reports after the current one, which will be processed.

pawosm-arm, 2020-10-25:
This code makes huge improvement. It makes `nxlandmine` playable for a first time. Yet in some areas of `nxdemo` the mouse arrow moves slower (especially when moving on 'active' areas, e.g. where pixels are drawn under the mouse arrow). The same with `nxterm`: the mouse movement are OK inside of the white (active) area, while it moves terribly slow in the blue (border) area.

ghaerr, 2020-10-25:
> This code makes huge improvement. 

Well, good, so we now know what the problem is. The code you inserted should be protected with a "if (parse == parseMS) ..." as otherwise it will crash with non-Microsoft mice. (The other mice use 4 byte rather than 3 byte reporting).

If you like, I can package this all up into a PR, but it will need to be contained within a special \#if SLOW_CPU likely set in nano-X/Makefile, or something like that. This is because discarding mouse input (otherwise known henceforth as "cleaning up excess mouse droppings" :)  isn't a good idea on fast systems.

> Yet in some areas of nxdemo the mouse arrow moves slower

This is because of more complex mouse processing under the hood: the application, as well as the GUI framework, also gets the reported mouse position and further delays time with additional internal processing. There isn't much we can do about that.

You might play with the THRESH and SCALE parameters a bit to see whether higher or lower acceleration helps, unless you're generally happy with that.

pawosm-arm, 2020-10-25:
This change makes nx usable hence it's acceptable to me. `SLOW_CPU` is anything 8086/8088 so this `#if` could be based on kernel config.
Things like THRESH and SCALE should be tweakable by userspace utils, similarly one can configure mouse sensitivity in most of the GUI environments.

pawosm-arm, 2020-10-26:
Fixed by #817

closed 2020-10-26 by pawosm-arm