ghaerr/elks #820: nxlandmine cannot be exited after game over

pawosm-arm, 2020-10-26:
When a player loses the game, mouse arrow disappears. The game does not react to mouse actions and after a while it also stops responding to key presses. It is impossible to press the 'quit' button or exit via pressing the Esc key. The only way to get away from it is the Ctrl-Alt-Del salute.

ghaerr, 2020-10-26:
I notice there is a routine with a 256 byte stack buffer, this could be stack overflow.

Trying adding a "-maout-heap=6144" to the link line for bin/landmine in nano-X/Makefile. I'm not getting it to duplicate on QEMU.

pawosm-arm, 2020-10-26:
I've tried this:

```
diff --git a/elkscmd/nano-X/Makefile b/elkscmd/nano-X/Makefile
index b636700f..229a18e5 100644
--- a/elkscmd/nano-X/Makefile
+++ b/elkscmd/nano-X/Makefile
@@ -67,7 +67,7 @@ libnano-X.a: $(CLNTFILES)
 salibs: $(SERVFILES) $(NANOXFILES)
 
 bin/nxlandmine: demos/landmine.o
-       $(LD) $(LDFLAGS) $(NANOXFILES) demos/landmine.o $(SERVFILES) $(NANOLIBS) $(LIBS) -o bin/nxlandmine $(MATHLIB) $(LDLIBS)
+       $(LD) -maout-heap=6144 $(LDFLAGS) $(NANOXFILES) demos/landmine.o $(SERVFILES) $(NANOLIBS) $(LIBS) -o bin/nxlandmine $(MATHLIB) $(LDLIBS)
 
 bin/nxworld: demos/world.o
        $(LD) $(LDFLAGS) $(NANOXFILES) demos/world.o $(SERVFILES) $(NANOLIBS) $(LIBS) -o bin/nxworld $(MATHLIB) $(LDLIBS)
```

and nothing changed, wrong behavior is still observed.

ghaerr, 2020-10-26:
Thanks, that's correct, providing you either "make clean" or removed nxlandmine, otherwise the binary won't be rebuilt.

I will test this on a real machine when I get back from traveling, since it doesn't repeat under QEMU for me.

ghaerr, 2020-11-11:
Hello @pawosm-arm,

> I will test this on a real machine 

I'm running this on my 386 desktop... and no problems, everything runs fine through multiple games. 

Perhaps something to do with your machine's memory configuration? Would it be possible to test this on other machines you have, to see what they do?

Thanks!

toncho11, 2020-11-14:
So I tested the nxlandmine and once you hit a mine: it shows the red circle, then nothing for 10 seconds, the it shows again the red cicle, then nothing again ... some kind endless loop. And you can not break the loop to exit I suppose. I am using a serial mouse.

ghaerr, 2020-11-14:
> .. some kind of endless loop

The red/no-red cycle is a lot slower on older systems, but is supposed to "blink" the cell about 5 times or so, then the game continues, with the cursor showing one less leg on the "minesweeper", with the mouse operational and able to step on another mine. Does your system stay in the blink cycle forever, or does it stop after 20-30 seconds?

@pawosm-arm is saying that the game stops working complete after the game is over, which happens after hitting three mines. I can't get it to reproduce.

pawosm-arm, 2020-11-15:
> The red/no-red cycle is a lot slower on older systems, but is supposed to "blink" the cell about 5 times or so, then the game continues, with the cursor showing one less leg on the "minesweeper", [...]

Ok, you've solved the mystery. I was just not patient enough. I've started this game on a remote telnet session (as expected, the graphics display is still used on the local VGA card), and I could stop the game with Ctrl+C at any time (even during red/no-red cycle which I thought was a crash).

Then I've made this video (using TVOne CS-320R Connect): https://www.youtube.com/watch?v=DdUrkYRSEjw it shows, the game doesn't really crash, it's just enters red/no-red cycle which is prohibitively slow!

ghaerr, 2020-11-15:
Nice video! Wow, it is truly amazing how far CPU's have come since the early days. Given the speed of your CPU, one can even see the cursor mask being updated as the mouse is being moved.

I'll decrease the blink wait time drastically so that the game works on early systems. Unfortunately, I don't have any precisely timed code available to work better for all CPUs, that will have to wait until another version.

pawosm-arm, 2020-11-15:
> Nice video! Wow, it is truly amazing how far CPU's have come since the early days. Given the speed of your CPU, one can even see the cursor mask being updated as the mouse is being moved.

I've made one more video (publicly listed this time) showing ramdisk, networking and even more of the nx in action: https://youtu.be/eooviN1SdQ8

ghaerr, 2020-11-16:
@pawosm-arm,

Please give this a try again with fix #868 in. Feel free to adjust the countdown in landmine.c so that it looks better.

closed 2020-11-16 by ghaerr

toncho11, 2020-11-16:
@ghaerr And how to you set the code optimization level -02 in ELKS? How to you disable the debug messages?

Maybe recompiling like that will make the game faster (including everything else). If there was "./build.sh release" for example?

ghaerr, 2020-11-16:
> And how to you set the code optimization level -02 in ELKS? 

The optimization level is set in a single Makefile to -Os (smallest size). The kernel won't compile with any other setting. (This is on my long list of things to look into). The optimization at -Os is surprisingly good, IMO.

> How to you disable the debug messages?

Which ones, specifically, are you referring to?
Debug messages are controlled in the kernel include file elks/include/linuxmt/debug.h.

> Maybe recompiling like that will make the game faster (including everything else).

No, the reason for it being slow is just the amount of processing required for graphics. (Even moving the cursor, which involves 256 XOR operations on an EGA, involves lots of read/writes and bit twiddling). Microwindows does the twiddling as fast as possible, in most cases.

toncho11, 2020-11-16:
Which Makefile for the -Os? Won't compile or it won't run?

pawosm-arm, 2020-11-16:
nxlandmine speed seems to be acceptable now: https://www.youtube.com/watch?v=00vSyyJWoWE

ghaerr, 2020-11-16:
> nxlandmine speed seems to be acceptable now

Great, thanks for testing! Looks like I guessed the 32000 value correctly :) 

Would you mind if we linked a few of your videos from the ELKS home page? I plan on updating the home page slightly with the v0.4.0 release, and would like to emphasize some of the usability a bit more by showing recent videos of ELKS capabilities, and put the long build instructions behind a link, so as not to scare off newbies that might want to try ELKS without having to build it. The v0.4.0 release should have some screenshots as well, and links to floppy disk images.

pawosm-arm, 2020-11-16:
Sure, go ahead and use them

toncho11, 2020-11-16:
For the front page I think you need to add links to both the wiki and [https://htmlpreview.github.io/?https://github.com/jbruchon/elks/blob/master/Documentation/index.html](https://htmlpreview.github.io/?https://github.com/jbruchon/elks/blob/master/Documentation/index.html)

ghaerr, 2020-11-16:
> Which Makefile for the -Os? Won't compile or it won't run?

You need my "ELKS source grepping tool":
```
alias gr='grep -r -I --exclude-dir cross'
gr Os .       (don't use -Os here because it would be treated as an option)
```
And you'll see the -Os is in elks/Makefile-rules. I couldn't live without `gr` for ELKS.

The kernel make mechanism and certain networking programs break when other -O options are used.

ghaerr, 2020-11-16:
> For the front page I think you need to add links to both the wiki and https://htmlpreview.github.io/?https://github.com/jbruchon/elks/blob/master/Documentation/index.html Now both seem out of sight.

Thanks, I've made a note of it. I plan on finalizing the release documents and other items for v0.4.0 shortly. Enhancing the front page is at the top of the list.

PR #868 (2020-11-16): [nano-X] Lower nxlandmine blink delay
Attempts to fix #820 by lowering blink time on nxlandmine display.