ghaerr/elks #829: FAT 32 does not mount

toncho11, 2020-11-02:
I have created the fat32 partition using Disk Management on Windows 10.

I have used the command:

[image]

ghaerr, 2020-11-02:
Hello @toncho11, 

I see you're back with more bug fix requests :)

I have a test fix for you. I see above you're booting from HD. Can I send you a 360k floppy boot to test the fix with?

toncho11, 2020-11-02:
Yes I am booting from 360 kb floppy. This is an old image where it says 0 floppy even if I am booting from it. Please send me an image with at least the mount command and not xx yy zz.

ghaerr, 2020-11-02:
Ok, the one I just uploaded should work, even though it still has xx yy zz on it.

toncho11, 2020-11-02:
Yes it works!!!

But did not mention that now it is not "bda2", but "hda2". There are no "b"s now.
Also without a sync command nothing is saved. But with sync I changed a file on ELKS and then I verified it on Windows 10.

Thanks!!!
Please make me a standard 360 KB image that I will be using from now on (please with the floppy patch also).

ghaerr, 2020-11-02:
> But you did not mention that now it is not "bda2", but "hda2". There are no "b"s now.

Sorry - the "BIOS disk" (bd) naming was changed to "Hard disk" (hd) to be easier to understand for users a few weeks ago.

> Also without a sync command nothing is saved. 

Yes, there is no auto-sync capability at the moment.

Here's an up-to-date 360K version with the official fix included for this issue (and no xx yy zz). 

[fd360-minix.bin.zip](https://github.com/jbruchon/elks/files/5478550/fd360-minix.bin.zip)

Please test and you can close this, thanks!!

toncho11, 2020-11-03:
Thanks. This last version does work.

How hard is it to replace minix with bootable fat32? Fat32 does not have things like links for example ?

closed 2020-11-03 by toncho11

ghaerr, 2020-11-03:
> How hard is it to replace minix with bootable fat32? 

We added bootable FAT32 support some time ago, and it's easy. We have a FAT boot block which needs to be added to the ELKS partition sector 1, and then if booting from a hard drive I advise using our own MBR as well, which is a boot block for the physical sector 1. 

Are you looking for a FAT32 bootable floppy?

> Fat32 does not have things like links for example ?

That doesn't matter much, we've removed any requirement for ELKS to have links. In the distribution disks, the FAT versions just copy the file twice since links aren't supported.

toncho11, 2020-11-03:
Yes, I would like to test a bootable Fat32 floppy.

ghaerr, 2020-11-03:
> Yes, I would like to test a bootable Fat32 floppy.

Here you go. Let me know how it works.

[fd360-fat.bin.zip](https://github.com/jbruchon/elks/files/5483073/fd360-fat.bin.zip)

toncho11, 2020-11-03:
It works with fat32 and this is really nice. Because it will allow for easier exchange of files with windows.
Good work!!!