ghaerr/elks #835: Sys tool

toncho11, 2020-11-03:
Hi,

I am suggesting creating a tool similar to DOS sys tool. DOS won also because it was simple. You do sys c: or a: and you are done. 

This new tool called "sys" should:
- copy the bootloader code from the current media (or the one given as parameter)
- create folders and copy the kernel and the most important folders (or everything from the source media) to the destination media

ELKS is bootable on fat32. So one can create easily a FAT 32 partition on a hdd using windows or fdisk. Next from a bootable fat32 ELKS you do:
`sys /dev/fd0 /dev/hda1`
and you are done. You can boot from your hdd and add more programs or compile your own and add them easily.

Or maybe not a program but a script? First dd the bootloader, then mount and copy?

ghaerr, 2020-11-04:
Hello @toncho11,

This is a pretty good idea, which really makes ELKS easier to use. Implementing it will be a bit complicated since ELKS supports two different types of boot sectors, MINIX and FAT, as well as other issues:

Supporting only either MINIX or FAT may not be a big deal in a first version, which would only "sys" a new volume with the same type of boot sector the system was booted from. That means if you booted from FAT, you could only "sys" another volume with FAT boot. Same with MINIX. The suggested first  parameter isn't needed, sys would use the boot drive to get the system.

A second issue is that ELKS doesn't currently support changing floppy media. This means that, in order to "sys" when booted from floppy on /dev/fd0, one would only be able to sys a hard disk partition or /dev/fd1, *not* /dev/fd0. So sys would not work well to create floppy disks on single-floppy systems. One could, however, sys a hard disk, boot, then sys a floppy, if one had a single floppy, single hard disk system.

The third issue is a bit more technical: ELKS needs to match the media/drive CHS (number of cylinders/heads/sectors), and the CHS is stamped into the boot sector. For dual-floppy systems, sys would assume the same CHS when creating a system on a second drive, which may not be what is wanted. When sys'ing from a floppy boot to a hard drive, the CHS can be read from the HD partition table, so that would work. However, sys'ing from a hard disk boot to a floppy, sys might not have a way of knowing the floppy CHS, so that might not work.

What are your (and other's) thoughts on these possible limitations? There are possible ways around some of them, we can discuss further if interested.

Thank you!

toncho11, 2020-11-04:
I think sys'ing fd0->hdd and hdd->fd0 is enough for now. 
Maybe mount/unmount can be used as quick solution to gather information about the target partition. We try try with minix and then fat32 and we get a detailed output of mount.

The tools can ask questions to be confirmed by the user:
- "FAT32 detected with CHS .... on partition /dev/hda1"
- "Do you wish to proceed?
- "ERROR: Target partition is FAT32, but minix is required"(or the opposite)

Still a boot loader might be needed. How do you tell on which partition ELKS is? Or maybe in the beginning it will use the whole drive. Ranish Partition Manager (RPM) for DOS is able to install graphical bootloader. RPM is a good tool. I think RPM can handle the creation of minix and fat32,etc.

The install can be:
- boot DOS with fdisk (for fat32 only) or RPM from floppy
- create minix or fat32 (for ELKS)
- put a bootloader with RPM
- boot into ELKS from floppy
- sys'ing the partition previously created for ELKS from the ELKS floppy

Mellvik, 2020-11-04:
@toncho11, @ghaerr,

FWIW - I like the idea. Such a utility will make life easier for all, newbies and experienced, not having to look up the exact procedure to make a bootable volume, regardless of media.

To me it would be really useful if I could boot fd0 as usual and easily create a bootable fd1, which is a different format and size floppy.
And I think it makes sense to expect a ready made file system on the destination media, just like DOS SYS.

Further, I'd like to propose we revitalize /sbin to contain the minimal set of binaries required to boot a system, ie. init, getty, login, sash, mount, umount, fsck, maybe others.

I've had this on my list for a while - brought up by practical needs: ELKS is now fully networked. Development and debugging is no longer (necessarily) a 'build new floppy image and boot' cycle, but a rebuild/transfer files via network/boot cycle. Which in many cases means transferring the kernel and affected binaries - typically those closest to the kernel/boot process. Long story short, it would make life easier to have this small group of files in their own directory which always gets transferred and avoid errors (memory lapses). Needless to say, missing (well, forgetting) one or more of these files in a cycle has bitten me a number of times of the past months.

—Mellvik

tkchia, 2020-11-04:
Hello all,

Putting the `/linux` file on an arbitrary FAT filesystem in the right way will be kind of tricky (though still possible).

Because of the (current) size constraints of the FAT bootloader, it will only correctly load `/linux` if it appears as the very first root directory entry, and if it is laid out in contiguous sectors.

But yeah, the technical problems aside, I agree that having a `sys` utility would be nice.

Thank you!

ghaerr, 2020-11-05:
Hello @tkchia,

> Because of the (current) size constraints of the FAT bootloader, it will only correctly load /linux if it appears as the very first root directory entry, and if it is laid out in contiguous sectors.

Thank you for the reminder on those two limitations, I had forgotten about that.

A sys command could still be useful with the above limitations if used on a freshly created or empty disk... Do you recall if the DOS SYS command has (or had) the same limitations? That is, did the hidden IBMBIO.SYS (and/or MSDOS.BIN whatever they were called) have to be in the first (few) directory entries as well? Or was that limitation removed in later versions?

Thank you!

ghaerr, 2020-11-05:
> ELKS needs to match the media/drive CHS (number of cylinders/heads/sectors), and the CHS is stamped into the boot sector.

>  it would be really useful if I could boot fd0 as usual and easily create a bootable fd1, which is a different format and size floppy.

It is possible that a sys command could use a kernel ioctl that executed a BIOS INT 13h function 8 (Get Drive Parameters), which will return the BIOS's idea of drive CHS, as well as the CMOS drive type. This could solve the problem of ELKS not knowing what the drive CHS is for a non-probed floppy drive. For systems that don't implement INT 13h,8, like early XT's, the CHS for a 360k floppy could be returned.

Using this along with @tkchia's information above should allow a sys command to be written that would work for any installed second floppy drive, as well as any hard drive that had an MBR, provided that the floppy or partition was empty. A later update to either the ELKS FAT boot sector, or a more complex sys command, could write a system onto a floppy or partition with existing files already on it.

Mellvik, 2020-11-05:
@ghaerr,
DOS does indeed have this same limitation, the (hidden) OS files (IO.SYS, MSDOS.SYS) need to be in specific positions on the file system in order for boot to work (actually the first two files in the root directory).
 
In newer DOS versions, the SYS command will move files around on the disk to ensure the correct placement of the files. On older systems, SYS will just complain and exit.

--M

ghaerr, 2020-11-07:
Hello,

After creating an almost-fully-workng sys program which should work copying a bootblack and /linux, I finally realized something: it won't work!

The problem is, ELKS is Linux, it isn't like DOS, where a simple system image and boot can be transferred and the resulting floppy will boot. ELKS needs a full /dev directory, /etc/inittab, /etc/rc.d/rc.sys, /bin/init, /bin/sh, etc etc or it won't boot.

So, the process of "duplicating" ELKS from one drive to another is quite a bit more complicated. In some sense, the entire disk needs to be copied, which will be extremely slow on ELKS. There are additional concerns if creating an HD system from FD, or a 360k system from 1440k, etc.

This can all be made to work, but it is more complicated that our initial thoughts of just having a sys utility that takes one argument and works.

My technical thoughts at this point are to seperate this into two steps: making a bootable image with just a proper bootblock and /linux in the proper places; and a seperate shell script that copies loads of files. The whole thing would be started in a "sys" shell script that would take a couple arguments, and call "makeboot" to create a bootable image, then copy all the files.

Thoughts? I am mostly concerned about the speed of operation on floppies... and whether the issue is just getting a boot sector transferred, or more likely a tool that essentially copies an entire disk worth of files, and the resulting complexity.

Thank you!

toncho11, 2020-11-07:
I still think it is worth it. 
Even having just fd0->hdd will be quite useful. 
What is the procedure right now to make a bootable HDD? Does anybody know exactly? Download a hdd image of ELKS and ... dd this image to a hdd partition? It is not documented! It means the disk must be attached to another computer with Linux and ... it is cumbersome.

Taking too long to copy all the files ... not a problem ... as long as it is automatic and saves me the procedure above.
A sys script and "makeboot" sounds good.

ghaerr, 2020-11-07:
> I still think it is worth it.

Agreed. Especially since I've basically finished with the initial sys command!

> What is the procedure right now to make a bootable HDD? Does anybody know exactly? Download a hdd image of ELKS and ... dd this image to a hdd partition? It is not documented!

Yes, we know exactly. But it isn't documented much, and its kind of complicated. It also varies a bit based on certain factors.

> It means the disk must be attached to another computer with Linux and ... it is cumbersome.

Yes.

I appreciate your ideas for usability. Computer systems should be easier to use, since most people aren't into the technical details, they just want to use the system.

> Taking too long to copy all the files ... not a problem ... as long as it is automatic and saves me the procedure above.
A sys script and "makeboot" sounds good.

I'll have something shortly. I just wanted to give a heads up that since ELKS/Linux require quite a few more files than a bootstrap and /linux to run, there may be cases where more user knowledge or input may be needed.

Thank you!

ghaerr, 2020-11-08:
Hi @toncho11,

I've created a first version of the sys tool in #846.

I've attached a 360k floppy binary you can use to create some bootable images on your systems. Be careful, this will overwrite the boot blocks on your MBR partitions.

[fd360-fat.bin.zip](https://github.com/jbruchon/elks/files/5505774/fd360-fat.bin.zip)

To use, boot from floppy and:
```
sys /dev/fd1    creates a system on floppy B
sys /dev/hda1   creates a system on hard drive partition 1
sys /dev/hda2   creates a system on hard drive partition 2
```
Right now, full debugging output is on, and only files required for a small standalone system are copied. More will be added later, once we know the basic approach works.

Thanks!

toncho11, 2020-11-08:
[image]

Mellvik, 2020-11-08:
Thank you @ghaerr - 

I think this is going to make a real difference. Incidentally I just bought another compaq portable and need an easy way to create bootable 1.2M floppies :-) This will save me a bunchload of time.

It may be a good idea to use sash instead of sh - thus getting many of the basic utilities for free. It may not be good for scripts but the basic bootable may not have any of that??

--Mellvik

ghaerr, 2020-11-08:
> Excellent job!!! :) I see you are already handling a long list of possible errors in the code!

Thanks, it has been a lot of work, and its also quickly showing a number of ELKS vulnerabilities.
I would have liked to have seen a better result than an immediate kernel panic, but OTOH, I have seen this a few times on my system, and this may answer a few very strange issues I'm seeing. I'll have another version with an increased kernel stack size shortly. (If you're now building ELKS, the kernel stack for each process can be expanded by changing KSTACK_BYTES in elks/include/linuxmt/limits.h. I'm going to increase it from 512 to 640. Make sure you make clean if changing anything and build ELKS all over again.)

> It can also fail on copying the /dev.

Yes, that's only for MINIX. In the FAT filesystem, /dev/ is faked.

> What is the current stack size limit? 512

> maybe sync must be called at some point to free the buffers?

Sync is called automatically by umount.

> it does copy the file for a while like 20 seconds

Yes, it will be slow on floppy reading... that's why I've limited the number of files for now.

> sys uses /mnt, but makeboot uses /tmp/mnt , but it should not be a problem

Yes, makeboot creates then removes /tmp/mnt, since something may already be mounted on /mnt.
I can see I didn't make that assumption on /mnt, so it should also use /tmp/mnt. I'll make that change, thanks.

> one needs to check if the destination is mountable

That's done first by makeboot, which will return an error and the whole sys will stop.

> Windows 10 format might put "System Volume", so it won't be an empty disk.

Hmmm, we will see what happens. /linux needs to be in the first sector of the root directory, so we should be OK.

> I used freedos to format to make sure is empty used fat32, would fat16 work ?

Yes, FAT32, FAT16 and FAT12 should all work.

> can check if /tmp/mnt is already created before creating it
> check if /tmp/mnt is empty (to verify destination is empty) ?

/tmp/mnt should not be present, but could be, if sys or makeboot died in the middle of things.
I'll add some code so that makeboot doesn't die if /tmp/mnt already present.

Thanks for your comments!

toncho11, 2020-11-08:
I was trying to say that the copy file in makeboot might be exhausting some write buffers and thus the kernel panic in the end? So by making periodic syncs during the copy kernel file that might resolve the problem? Umount might be too late. This is just an idea ...

ghaerr, 2020-11-08:
> I was trying to say that the copy file in makeboot might be exhausting some write buffers and thus the kernel panic in the end? 

No, I think the problem is that the FAT filesystem code uses a large amount of stack space. And we're booted from FAT and running on a mounted FAT.

Here's a version with increased kernel stack. Try it out.

[fd360-fat.bin.zip](https://github.com/jbruchon/elks/files/5506665/fd360-fat.bin.zip)

ghaerr, 2020-11-08:
> Here's a version with increased kernel stack. Try it out.

You don't need to recreate your HD partition, I've tried to allow some cases of working when its not entirely fresh. So just boot the new kernel and try sys, lets see whether it will at least complete.

toncho11, 2020-11-08:
Good news! It does complete at 100%. 

I did not format this time, just removed all files.

But it does not boot. It loads ELKS Boot manager. It says "no partition is bootable".
I have only one hda1 partition. 
Then I type 1, it says "ELKS3! press any key" and then it restarts instead of booting (I suppose). 

So yes definitely we are getting close. Maybe I really need to format the partition next time ?

ghaerr, 2020-11-08:
> But it does not boot. It loads ELKS Boot manager. It says "no partition is bootable".

I see. So you have the ELKS MBR, not Windows MBR?

> Then I type 1, it says "ELKS3! press any key" and then it restarts instead of booting (as I was expecting).

I can never remember what the dang boot failure codes are. I am guessing it can't find /linux. On your machine, running the ELKS MBR, that means likely the newly patched sys boot doesn't know the "starting sector offset" of your partition. 

There is some more work to do with properly setting the partition start sector offset in boot sectors. The current code is dependent on Windows format setting it properly. I need a hex dump of the boot sector in order to tell, but we don't have a tool for that yet.

Can you compile code now? If so, we can set that offset using some code in line 161 of elkscmds/makeboot.c. Change the "\#if 0" to "\#if 1".

I'll send you another fd360-fat.bin, that's probably easier.

ghaerr, 2020-11-08:
Try this one

[fd360-fat.bin.zip](https://github.com/jbruchon/elks/files/5506783/fd360-fat.bin.zip)

tkchia, 2020-11-08:
Hello @ghaerr, hello @toncho11,

I am not sure if the current bootloader code for FAT works with FAT32 --- I recall it currently only really works with FAT12 and FAT16.  We probably need to write a separate bootloader for FAT32.

Thank you!

ghaerr, 2020-11-08:
> it says "ELKS3! press any key"

Error '3' means it can't find /linux on the FAT volume.

This means that either /linux was not found in the first directory entry as @tkchia suggests, or that the starting sector offset isn't correct. The latter will be determined with my last image.

If the "System Volume" is in the first directory entry, then this might need to be removed.

Hello @tkchia,

> I am not sure if the current bootloader code for FAT works with FAT32 --- I recall it currently only really works with FAT12 and FAT16

I see. Do you remember any more details as to why that might be, so I can think a bit more about it? 

While you're here, I'd like to bring up a pending problem: that of the partition start offset. You will notice this as "sect_offset" which was made a .global in #846 for visibility. This value is kept in two different regions (like the sect_max and head_max), different for FAT vs MINIX boot. For the MINIX boot sector, I have temporarily kluged the sys command to use its current location in minix.map. It will need to be moved to the ELKS EPB, as it is now a required value to get MINIX hard drives to boot. This will also enable removing an older kluge I added to the MBR code passing a special magic value in a register, to indicate DS:SI points to the starting offset. I have realized the real way MSDOS handles this is through the BPB stamped into the partition boot sector, NOT through the MBR pointer.

So - we need to do the same thing for MINIX, keeping the start offset in the EPB. Would you like to change that, or should I?
A longer term issue, mentioned a few times, is moving the EPB to the BPB, but that will require more consideration and testing. It has a number of benefits, but we'll get into them later.

Thank you!

ghaerr, 2020-11-08:
Hello @tkchia,

If the boot sector code needs to be modified for a FAT32 boot, it would be nice to keep it within the existing FAT boot sector. As you know, we're down to 1 byte in that, but after moving sect_offset to the EPB and modifying setboot to stamp the sector offset into the EPB and/or FAT BPB, we will be able to remove all of the following code, which gets the sector offset from our MBR code:
```
   // instructions through _next0 not needed for floppy, only MBR boot
    cld
    cmp $0x4c65,%ax         // coming from ELKS MBR?
    jnz _next0
    mov 8(%si),%ax          // get MBR start sector from partion table at DS:SI
    mov 10(%si),%si
    push %cs
    pop %es
    mov $BOOTADDR+sect_offset,%di
    stosw
    xchg %ax,%si
    stosw
_next0:
```
So you see, even with that code removed, we are going to extra measures to keeping the EPB and BPB information separately, and have to make special cases in our tools to handle it. If the MINIX boot code will fit into a boot sector with a BPB, it would be better to adopt locations 11-61 (FAT BPB 4.0) as our information area.

If you can recall much of what might need to be done to get FAT32 booting, I think I can change the MBR code and test everything.

Thank you!

toncho11, 2020-11-08:
I tested again. I did format this time, but it is the same - it does not boot. It says again "ELKS.3!"

I used Win32Imager to make an image of the hd1 partition where I installed ELKS.
[elks_hdd_image.zip](https://github.com/jbruchon/elks/files/5506891/elks_hdd_image.zip)

Unfortunately it also added "System Volume Information" I think before I made the image.
It is only the partition without the MBR I think.

toncho11, 2020-11-08:
Meanwhile I could test with bootable minix image.

Can I use fdisk and mkfs that are already in ELKS to create a minix partition and fs?

ghaerr, 2020-11-08:
> Meanwhile I could test with bootable minix image.

That would be great. I'm analyzing your elks_hdd_image.img right now. The first problem with it is that it definitely has System Volume in the first directory entry, which is a problem, even without the the FAT32 boot changes @tkchia noted might be necessary. I mounted your image and here's the result:

[image]

> Can I use fdisk and mkfs that are already in ELKS to create a minix partition and fs?

Yes, but use small sizes for now, since large filesystems might be buggy as noted by @Mellvik today. I suggest one just 1.44M in size. Try creating this from ELKS:
```
mkfs /dev/hda1 1440
sync
```
This should overwrite your existing partition, and the MBR will not match, but that shouldn't matter for now.

ghaerr, 2020-11-08:
@toncho - you will need a 360k minix boot floppy or the sys won't work, I'll send you one shortly.

toncho11, 2020-11-08:
System Volume was added last. That is for sure, because I did a format before using the ELKS sys tool. System Volume was added when I attached the HDD to Windows later.

Please add mkfs in the image (if not already).

ghaerr, 2020-11-08:
> System Volume was added last. That is for sure, because I did a format before using the ELKS sys tool. System Volume was added when I attached the HDD to Windows later.

Can you format a new Windows partition, just like the last you sent, except without System Volume? I will then test it.

I am realizing now it could be tricky for this to work for you. The following have to be the case or ELKS won't current boot:
- /linux must be the first root directory entry, *and* in consecutive sectors. This means that if anything else is or was on the disk, everything must be removed, or the FAT table won't be entirely free for the /linux copy.
- /dev must be in the first two entries, or it won't work. (/dev is faked in FAT. You'll note its the 3rd entry in the screenshot above.) If the image is completely empty, then this requirement will also be met.
- The FAT partition may need to be FAT12 or FAT16. (@tkchia and I will look into this).

Perhaps you should try creating a much smaller FAT partition, say 1.44M (1440 blocks). That will force your format program to format it FAT16.

If you could send me a blank FAT32 partition, before System Volume, that would be great. And a FAT16 one too, if you can.

> Please add mkfs in the image (if not already).

Working on that, but there isn't much space on 360k, and my source tree is currently under construction trying to chase down all these issues.

If you can build ELKS now, then you can fix this yourself: 
- make clean
- make
- cd images; make images
Look in images/ directory and you'll find all the .bin files you need.
To add mkfs, edit elkscmd/Applications, find mkfs, change ":1440k" to ":360k" and "make images" again.

ghaerr, 2020-11-08:
@toncho11,

I've analyzed your /dev/hda1 partition. It looks like it is, in fact, FAT16, which is good news. So, after creating it the way you did before, can you possibly mount it and then remove System Volumes? That should leave a blank partition, which should work with sys, except that I've also found that the sector offset values are incorrect... stay tuned for a fix for that.

ghaerr, 2020-11-08:
> It looks like it is, in fact, FAT16

Oops, sorry for the confusion: I was looking at the wrong part of the image you sent me. It looks like there used to be a previous partition at offset 63, which was FAT16. The partition is FAT32. I think we can make this work by removing all the files, including System Volume, then running the fixed sys on it. I am checking into that now.

ghaerr, 2020-11-08:
Hello @toncho11,

Wow, what a rabbit hole. I finally debugged what is going on, and there are unfortunately lots of reasons that HD FAT boot isn't working:
- The image has System Volume so isn't blank. (fix by mounting and removing any files after format).
- The image is FAT 32, which likely needs a rewritten boot sector (fix by formatting to less than 63504 blocks to force FAT16).
- The MBR CHS doesn't match the created volume. (fix by running fdisk as shown below. fdisk isn't on 360k floppy though).

[image]

Maybe we better start with getting sys running on MINIX first! :)
That might best be done by creating two working MINIX partitions, /dev/hda1 with all the fd1440-minix.bin tools on it, and /dev/hda2 for testing sys. You will have to use fdisk to bootstrap your HD CHS and then mkfs to create the filesystem, I will try to see if I get can them to fit on a 360k floppy for you.

I can see we need sys more than ever. We also need fdisk and mkfs on a root boot disk, and it would be nice to have a mkfat which creates FAT filesystems as well. Then, one day, all of this could be put into a very fancy sys script. But that will have to wait, since none of this is yet working!

Let me know what you think from your further testing. I need to get my source tree synced, so I will likely be committing #846 followed by a patch for the kernel stack, and then continue working on this.

toncho11, 2020-11-08:
Actually ... it works!!!

I manage to load ELKS after your sys tool on hda1 (fat16). 

This time after I pressed "1" it does load! It still complaints about "no bootable partition", but it does load this time.
So what I changed is that I found DOS 6.22 and created a 8MB partition that is by default fat16 I think. Formatted just to be sure. 
I think Windows 10 does things differently by setting an offset or encoding the MBR differently. On Wikipedia there is:
- classical generic MBR
- modern standard MBR

ghaerr, 2020-11-08:
> Actually ... it works!!!

YES!!! Glad to hear that :)

> I manage to load ELKS after your sys tool on hda1 (fat16).

Good news, I was trying to remove System Volume on the FAT32 partition just for testing and found that ELKS can't remove or cd to directories with spaces in them!

> This time after I pressed "1" it does load! It still complaints about "no bootable partition"

The "no bootable partition can be changed by setting the partition active with fdisk. I am working right now to get you FAT and MINIX boot disks on 360k so you can fix that. A PR will be commited shortly and I'll upload then for you.

> So what I changed is that I found DOS 6.22 and created a 8MB partition that is by default fat16 I think. Formatted just to be sure.

Got it. I take it DOS 6.22 doesn't create a "System Volume"?

> I think Windows 10 does things differently by setting an offset or encoding the MBR differently...

Yes, but I think we can still use that if we can figure how to create a FAT16 volume with it for now, and also get rid of System Volume.

Thanks for your continued testing! There is light at the end of this long tunnel... :)

toncho11, 2020-11-08:
On Windows 10 one needs to stop and disable "Storage Service" to prevent the creation of the "System volume".

This is the HDD that worked:
[elks_hdd_image.zip](https://github.com/jbruchon/elks/files/5507188/elks_hdd_image.zip)

toncho11, 2020-11-08:
When created with Windows 10 the offset was 1024KB and when created with DOS 6.20 the offset was 31KBs.

toncho11, 2020-11-08:
A Windows 10 fat16 formatted partition does not work even if "System volume" does not show any more. ELKS does not detect this fat16 partition. 

The only thing that works is with:
- DOS 6.20 fdisk to create a partition and activate it
- format with DOS 6.20 again as (fat12) 
- 8MB disk
- now there is no need to select the partition manually in ELKS boot loader, it just loads directly

The MINIX version works perfect. One needs to create a minix partition first with mkfs.

ghaerr, 2020-11-08:
> A Windows 10 fat16 formatted partition does not work even if "System volume" does not show any more. ELKS does not detect this fat16 partition.

Interesting. I'd like to see a disk image with the non-mountable FAT16 partition from Windows 10. That's not a sys issue, just a FAT16 compatibility issue.

BTW, on your previous Windows 10 FAT32 partition, I found that even after deleting everything (except System Volume), a close inspection reveals that Windows 10 stuck in "Volume Label" hidden directory entry as the first directory entry. So that's another hidden entry that will prohibit sys from working. I'm not sure there's a way to force Windows to show a volume label entry. ELKS doesn't show volume labels either, but could be modified to, and then deletable by ELKS.

What we really need is "mkfat", which would create FAT partitions directly from ELKS.

On the subject of FAT compatibility, I've just noticed that the serial ports can't be used at all, since /dev/ttyS0 etc have a capital letter in them and the FAT filesystem driver lower-cases everything. I'm thinking I will have to produce a PR to fix rename all ttyS{0,1,2} to ttys{0,1,2} to fix all that, or FAT booted ELKS will be compromised.

> The only thing that works is with:

> DOS 6.20 fdisk to create a partition and activate it

ELKS fdisk should be able to create a partition and activate it.

> format with DOS 6.20 again as (fat12)
> 8MB disk

Why is this needed? Are you saying that sys doesn't work with FAT16? We need to look further into that.

> now there is no need to select the partition manually in ELKS boot loader, it just loads directly

Yes - the MBR boot loader will automatically start if there is an active partition set.

> The MINIX version works perfect. One needs to create a minix partition first with mkfs.

That's great!

Thanks for this information and testing!!!

toncho11, 2020-11-09:
Two things I noticed:

- it is indeed fat 12 that worked above
- BUG: ELKS (fat version) does not put /dev/hda2 or /dev/hda3 even if the kernel detects them (shows them while loading)
- Windows is hell creating an empty fat16 or fat32, so fdisk and mkfat are really needed
- fdisk from ELKS never works: "Unable to read boot sector from /dev/hda"
- I also tried fdisk and mkfat on Ubuntu 20 and managed to install ELKS on fat 16 (when on hda1) but it did not boot! I think ELKS boot manager is missing in this case.
- I also tried fdisk and mkfat on Ubuntu 20 and managed to install ELKS on fat 32 (when on hda1) but it did not boot! I think ELKS boot manager is missing in this case.

ghaerr, 2020-11-09:
> It is indeed fat 12 that worked above

Do you have more information as to why FAT16 failed? If you send me a FAT16 image after formatting, but before running sys, I can debug it.

> BUG: ELKS (fat version) does not put hda2 and hda3 in /dev even if the kernel detects them (shows them while loading)
> fdisk from ELKS never works: "Unable to read boot sector from /dev/hda"

Both these problems have to do with the faked FAT /dev directory, which currently is maxed out at 16 entries, so no room for /dev/hda2, /dev/hda3 or /dev/hda. I am working on fixing that.

> Windows is hell creating an empty fat16 or fat32, so fdisk and mkfat are really needed

Are you saying that mkfs should be automatically run by sys?

The problem with mkfat will be that there are a zillion possible options for creating a FAT image. I suppose ELKS could just use a small set of presets based on requested image size though. This will likely wait until the next version.

> I also tried fdisk and mkfat on Ubuntu 20 and managed to install ELKS with sys on fat 16 (when on hda1) but it did not boot! I think ELKS boot manager is missing in this case.
> I also tried fdisk and mkfat on Ubuntu 20 and managed to install ELKS with sys on fat 32 (when on hda1) but it did not boot! I think ELKS boot manager is missing in this case.
> Where is the phase that puts ELKS boot manager in the beginning of the disk?

I think you're talking about the ELKS MBR boot sector, the one which asks for a partition \#? `sys` doesn't current write that, but could, with a special option. We need to talk through the cases where sys should write an MBR, as well as when it should mkfs/mkfat.

The other reason it might not boot has to do with the way we previously handled getting the partition start sector. My recent PR #848 now allows for any MBR to be used.

Were you using an ELKS MBR for these cases, or someone elses? If the latter, an updated sys from #848 should solve the problem.

> It works only for minix?

The ELKS MBR should work for both FAT and MINIX.

toncho11, 2020-11-09:
Let me know when to test.

ghaerr, 2020-11-10:
@toncho11,

> fdisk from ELKS never works: "Unable to read boot sector from /dev/hda"

> Both these problems have to do with the faked FAT /dev directory, which currently is maxed out at 16 entries, so no room for /dev/hda2, /dev/hda3 or /dev/hda. 

I was incorrect about /dev/hda, which has always existed on FAT. I just added /dev/hda2 in #850.

Can you give me more information on the "fdisk from ELKS never works: Unable to read boot sector from /dev/hda"? I tried it from a ELKS-created booted FAT image (hd32mbr-fat.bin) and it is running fine on QEMU. I'm trying to chase down what the problem is exactly. Does "hd /dev/hda" do anything (hex dump)?

Thanks!

toncho11, 2020-11-10:
I tried the sys from #850, but it did not boot on /hda1 fat32. I did use mkfat before that to clear the partition from Ubuntu.

toncho11, 2020-11-10:
[image]

ghaerr, 2020-11-10:
Thanks, the screenshot helps a lot.

`fdisk` can only be used with base /dev/hda (or /dev/hdb) device, NOT /dev/hda1, etc. It doesn't check, but should.

Funny thing is I booted the elks_hdd_image.img you just sent me, and fdisk works:
[image]
This must mean there's something not working when reading from /dev/hda on your real hardware. Fdisk is only reading the first sector of your HD. However, when you (erroneously) specify reading from /dev/hda1, the reads work. It sounds like ELKS is thinking that the size of /dev/hda is set incorrectly.

Can you screenshot your boot screen? I'll compare it with mine on QEMU:
[image]

ghaerr, 2020-11-10:
@toncho11,

I'm looking at code and wondering whether your HD is large (for ELKS standards, even though partition 1 is small)?
It looks if your HD is > 2GB, this will happen.

The following code in fs/block_dev.c checks the /dev/hda read for size and EOF:
```

    /* Amount we can do I/O over */
    pos = ((loff_t)inode->i_size) - filp->f_pos;
    if (pos <= 0) {
        debug("GENREAD: EOF reached size %ld pos %ld.\n", inode->i_size, filp->f_pos);
        goto mfread;            /* EOF */
    }
    if ((loff_t)count > pos) count = (size_t)pos;
```
The `inode->i_size` is calculated as follows:
```
    inode->i_size = (hdp->nr_sects) << 9;
```
and `hdp->nr_sects` is calculated from CHS:
```
hdp->nr_sects = (sector_t) drivep->sectors * drivep->heads * drivep->cylinders;
```
Thus, if your CHS is such that nr_sects << 9 is greater than 2^30, then bit 31 will be set (i.e. size > 2GB) and ELKS will fail.

toncho11, 2020-11-10:
I always test with 32 GB sd card. So ... much bigger than 2GB 👍

toncho11, 2020-11-10:
[image]

But note that my partition does not start at the very beginning of the disk compared to yours.

ghaerr, 2020-11-10:
> I always test with 32 GB sd card attached to XT-IDE.

Well, @toncho11, you've blown ELKS right out of the water... :0)

What hardware are you using to attach an SD card to your XT? I'd like to try that on my system also.

> But note that my partition does not start at the very beginning of the disk compared to yours.

Yes. The partitions will work, because they are at large cylinder offsets. The issue is that CHS (C\*H\*S can't be such that the overall byte size is greater than 2^30. That's because your approx max CHS (1024/255/63) ~= 2^10 \* 2^8 \* 2^6 = 2^24, which when << 9 is > 2^31, bit 31 is the sign bit so the EOF check on /dev/hda fails. Your partitions start at larger offsets so the CHS doesn't blow out the size.

Fix coming for your (1980's) huuuuge hard drive, which will artificially limit disk and partition sizes to 2^30 - 1 (2GB).

toncho11, 2020-11-10:
SD to IDE card https://fr.aliexpress.com/item/32948941581.html
And then you need a short IDE cable to connect them.

toncho11, 2020-11-10:
It still not booting from hdd. I am trying with fat16.

Here is the output of sys:
[image]

Here is fdisk:
[image]

There is no boot loader for fat?

ghaerr, 2020-11-10:
> Or the partition not set to ACTIVE?

It doesn't show as active, set it active using fdisk.Or you can just press 1 from the prompt.

> Because there is no boot loader installed?

Yes, because I think we've calculated the start CHS to be in the wrong place? Your earlier boot screen showed your virtual HD is CHS 1024/255/63. But the MBR start is showing CHS 1/0/1... Cylinder 1 is 255 \* 63 sectors, not 2048.

ghaerr, 2020-11-10:
> Cylinder 1 is 255 * 63 sectors, not 2048.

Yes - this is the problem. Normally you want to put a partition at a cylinder boundary, but the CHS 1/0/1 is actually start sector 255 \* 63 = 16065. So sys is working properly, writing the supposed boot sector at 2048. But ELKS MBR boot loader is calculating differently.

Run `fdisk` without -l so it shows the disk geometry. I'll bet the CHS isn't matching your SD geometry?

toncho11, 2020-11-10:
When trying to boot it does not print anything. It just freezes.

ghaerr, 2020-11-10:
> When trying to boot it does not print anything. It just freezes.

Yep, its loading in a garbage boot block.

I'm pretty sure I know the problem, but not sure why. We need to verify the boot screen CHS, and the fdisk CHS, shown when run without arguments.

ghaerr, 2020-11-10:
It's a little complicated: we have 3 CHS:
- the floppy CHS 40/2/9, shown in sys
- the HD CHS 1024/255/63, shown at boot and labeled incorrectly as "Boot drive geometry" in the second sys display
- the fdisk CHS, shown when running fdisk with no arguments (needs to be 1024/255/63).

Then we need to calculate a proper start sector for partition 1, which should be a multiple of H times S (255 *63), if you're going to start at CHS 1/0/1.

toncho11, 2020-11-10:
[image].

ghaerr, 2020-11-10:
I am thinking that your Windows or MSDOS formatter was thinking a different CHS than 1024/255/63 when it placed the partition at offset 2048. 

To get to offset 2048, we would need to start at CHS 0/32/32.

ghaerr, 2020-11-10:
Yep - your partition start is incorrect at 1/0/1. Change it to 0/32/33 (or possibly 0/32/32). That will fix it. You'll need to rerun sys after changing your start partition. Make sure fdisk continues to show the start sector as 2048 or everything will be screwed.

toncho11, 2020-11-10:
I am using only mk.fat on ubuntu to create the partition and this tutorial:

https://linuxhint.com/create_fat_filesystem_linux/

It says press 'o' to create a new did partition.
And then I use: mkfs.fat -F 16 /dev/sdb1

[28 more replies]

closed 2020-11-11 by toncho11