ghaerr/elks #841: Minor debug ^P issue ++

Mellvik, 2020-11-06:
A couple of things (really enhancement requests) re. the new dprintk mechanism and debug messages.

- When debug is enabled in /bootopts, enabled debug types, like BLK and FAT, will roll over the screen as expected. TCP debugs will not. ^P will toggle both, turns off those enabled by the debug statement, turns on those in networking (possibly only TCP). I may be wrong, but this cannot be intentional?

- Talking about TCP debug, it would be very helpful if
  - acks and sequence numbers were in decimal
  - acks and sequence numbers were printed as deltas instead of absolute numbers (actual byte count instead of byte count + starting number) (like tcpdump does by default).

--Mellvik

ghaerr, 2020-11-06:
> When debug is enabled in /bootopts, enabled debug types, like BLK and FAT, will roll over the screen as expected. TCP debugs will not. ^P will toggle both, turns off those enabled by the debug [...]

This is a known issue, not intentional. Not sure how to fix it. The problem is that there's a lot under the hood with ^P: the printk option can be enabled via the /bootopts debug statement, but ktcp isn't yet running to send a special signal to, to toggle its debugging. So when the first (or any subsequent) ^P is typed, the kernel printk toggle occurs, and the signal is sent, which ktcp toggles separately. The signaling method doesn't allow a parameter or kernel communication to determine what the debug state is, ktcp just toggles the state with every signal. The SIGURG signal is used, because that is the only signal that won't terminate a process if received without a signal handler.

This issue normally occurs only when trying to debug kernel startup with printk. Otherwise, it is a better idea to just remove the debug keyword in /bootopts, and hit ^P after boot. Then, both ktcp and the kernel debug will be in sync.

> acks and sequence numbers were in decimal

Ok - I thought it would be easier in hex. I'll just change to your delta idea below, or do you want both?

> acks and sequence numbers were printed as deltas instead of absolute numbers (actual byte count instead of byte count + starting number) (like tcpdump does by default).

Which should be subtracted from the other? I guess we would always like to show a positive number, which would be the difference between the current sequence number and the last ack received?

Mellvik, 2020-11-07:
Having spent weeks and months with tcpdump listings, I've come to appreciate the speed at which it is possible to scroll through and check acks against sequence numbers when they are in decimal and (relatively small). No thinking, just a snap. 

What tcpdump does is to always subtract the starting offset - from sequence numbers and acks, so that when actual byte # 237 is acked, the ack in the log says 237. Similarly, when displaying sequence numbers, it includes both there start and end, as in seq=238:437. Also it leaves out sequence numbers when the length is 0.
```
13:12:21.977301 IP (tos 0x0, ttl 64, id 823, offset 0, flags [none], proto TCP (6), length 40)
    10.0.2.15.1024 > 10.0.2.2.ftp: Flags [.], cksum 0x20e9 (correct), ack 463, win 255, length 0
13:12:21.991556 IP (tos 0x0, ttl 64, id 824, offset 0, flags [none], proto TCP (6), length 48)
    10.0.2.15.1024 > 10.0.2.2.ftp: Flags [P.], cksum 0x4ec7 (correct), seq 29:37, ack 463, win 255, length 8: FTP, length: 8
	TYPE i
13:12:21.991634 IP (tos 0x10, ttl 64, id 13509, offset 0, flags [DF], proto TCP (6), length 40)
    10.0.2.2.ftp > 10.0.2.15.1024: Flags [.], cksum 0x182b (incorrect -> 0xafcf), ack 37, win 29200, length 0
13:12:21.991848 IP (tos 0x10, ttl 64, id 13510, offset 0, flags [DF], proto TCP (6), length 60)
    10.0.2.2.ftp > 10.0.2.15.1024: Flags [P.], cksum 0x183f (incorrect -> 0x1cd0), seq 463:483, ack 37, win 29200, length 20: FTP, length: 20
```
Elks tcp log always reports checksum as 0, it's a convenience to have if it's easy to get to the actual checksum.

Also, while on the issue of debug logs, if it's an easy fix it would be nice to have a more compact format on the ICMP logs, without the ports, the tcp flags, ack, win, urg etc, keeping the seq, chk and length.

Thank you!

--M

Mellvik, 2020-11-09:
@ghaerr,
I can think of 3 ways to fix (or remedy) the ^P issue:
- in tcpdev open, test if the debug flag (assuming it's a global flag) is set, then send SIGURG to the calling process - presumably  ktcp - if it's set.
- add a debug flag to ktcp, have the rc.sys script check /bootopts and set the debug flag appropriately (slow)
- without the ktcp debug flag, rc.sys can still check /bootopts and send SIGURG to ktcp after startup (also slow)

No elegance, but might work.

--M

closed 2020-11-09 by Mellvik