ghaerr/elks #845: Build (install) fails

toncho11, 2020-11-06:
```
install /home/toncho/elks/elks/arch/i86/boot/Image /home/toncho/elks/target/linux
rm -f /home/toncho/elks/image/fd360.bin
mfs  /home/toncho/elks/image/fd360.bin genfs -1 -n14 -i192 -s360 /home/toncho/elks/target
Blocks required 1355, available 360

Make.image:63: recipe for target 'minixfs' failed
make[2]: *** [minixfs] Error 255
make[2]: Leaving directory '/home/toncho/elks/image'
Makefile:28: recipe for target 'image' failed
make[1]: *** [image] Error 2
make[1]: Leaving directory '/home/toncho/elks/image'
Makefile:11: recipe for target 'all' failed
make: *** [all] Error 2
Build script has terminated with error 5
toncho@toncho-OptiPlex-755:~/elks$ cat  /home/toncho/elks/target/etc/issue
```

I tried a few things, but to no avail:
- loading minix kernel module (is it needed?)
- sudo ./build.sh

If I choose a bootable fat image as target it says "Disk full".

ghaerr, 2020-11-06:
Hello @toncho11,

Nice to see you're getting the build system going! Sorry for the confusion.

> Blocks required 1355, available 360

Since you're building for a 360k disk, but have all the applications selected, the build fails since there's not enough space. 

To build your 360k bootable floppy, use the method that I built that allows one to select the applications based on the disk size. This is under "make menuconfig": Userland -> Select Applications by Image Size -> Small (360k).

Thanks!

toncho11, 2020-11-06:
Yes. This was the whole problem. 
I do not know how it was the first time, but maybe it should be by default: "Select Applications by Image Size"

closed 2020-11-07 by toncho11