ghaerr/elks #853: Problem with boot after cp folder

toncho11, 2020-11-11:
Using fat16 partition after sys tool.

After 

```
mount -t msdos /dev/fd0 /mnt
cp /mnt/bin/* /bin/
```

ELKS does not complete the boot process. It stops after "Mounted root 0x0301 (msdos filesystem)"

If a copy single file it is ok - it boots.
The problem seem to be only when the entire folder is copied.
After the entire folder is copied - programs appear to be working (before reboot)
The problem comes after rebooting and after that the system is unusable.

I also tried:

```
$ sudo fsck.vfat -w -a /dev/sdb1
fsck.fat 4.1 (2017-01-24)
FATs differ but appear to be intact. Using first FAT.
Performing changes.
/dev/sdb1: 36 files, 61/35117 clusters
```
and this did not help. Still unbootable.

ghaerr, 2020-11-11:
Try erasing /bin/init on the new system. I'm betting the system is trying to run init without the appropriate /etc/inittab files, etc.

toncho11, 2020-11-11:
Yes. That was the problem. 
So maybe the sys tool can have a "-F" option that copies everything including the /etc folder.

closed 2020-11-11 by toncho11

ghaerr, 2020-11-11:
> So maybe the sys tool can have a "-F" option that copies everything including the /etc folder.

Yes, although -F could fail from a hard disk to a smaller floppy. I need to think about what an ELKS "system" is in terms of all the extra files, and whether or not the sys command should be in the business of copying full disk images. Agreed it should be made simple.

However, probably wait until we have a recursive copy command (cp -R), so that the sys script doesn't need to be updated depending on what happens to be in /etc, /bin, etc. Perhaps a cp -Rn that copies everything but doesn't overwrite anything already existing on the destination, would be preferable, then just executing that after sys.