ghaerr/elks #860: pwd command not working

toncho11, 2020-11-13:
pwd says stat: Unknown error 2

Using on fat32 and not on /dev/...

toncho11, 2020-11-13:
```
cd /bin
cd ..
```
it sa says:
`cd: cant cd to ..`

ghaerr, 2020-11-13:
Hi @toncho11,

Geez, we finally get FAT32 boot going and what do you do, go out and find another bug?! :) 

I've got this duplicated, and it seems to only occur on FAT32. It is related to some complicated code having to do with the fact that the `..` directory entries are faked in the ELKS FAT filesystem driver, since there aren't .. directories in actual FAT filesystems.

There has been little real-world testing of running on FAT filesystems. I'm glad you've found this bug, but I'm thinking of prioritizing getting an ELKS 0.4 out ASAP first. In the mean time, keep on using ELKS FAT32 and lets see what other problems there may be.

Thank you!

toncho11, 2020-11-13:
The boot process also does not detect /home folder 
It says `No home directory. Starting in /`
Also the `cd ..` actually works for second level directories such as /mnt/bin, but not from /mnt to /.

ghaerr, 2020-11-13:
> The boot process also does not detect /home folder (I created one manually).
It says No home directory. Starting in /

That's because `sys` doesn't create a /home directory. As mentioned at some point in #835, there are more considerations as to how much of the ELKS system should be copied during a "system transfer", given sys can be run from a wide variety of disk images sizes.

> Also the cd .. actually works for second level directories such as /mnt/bin, but not from /mnt to /.

Yes, I'm looking at that now. The `..` machinery has some special cases I'm looking at now regarding the root directory on FAT32, which is technically a bit different from FAT16 root directories.

toncho11, 2020-11-13:
I created `home ` manually and it still does not find it when booting.

ghaerr, 2020-11-13:
> I created /home manually and it still does not find it when booting.

Oops, I meant that sys doesn't create a /root directory. Root uses /root rather than /home for its home dir. I'll fix that in sys.

toncho11, 2020-11-13:
It works now!

PR #862 (2020-11-13): [fat] Fix get_parent_ino on FAT32, fixes pwd
Fixes #860.

Also copies enough files with sys to enable a sys from the new system.

@toncho11,

Here's a system for confirmation testing, thanks.

[fd360-fat.bin.zip](https://github.com/jbruchon/elks/files/5539198/fd360-fat.bin.zip)
closed 2020-11-13 by toncho11