ghaerr/elks #863: nxworld improvements

toncho11, 2020-11-14:
1) If the map is not there we do not see the message. It clears the screen when exiting, so it was a pure guess that the file was not there

2) the file is searched only in /lib, it will better to search also the current folder. Seems natural.

3) the file is used as  "nxworld.map" at runtime, but it resides as "world.map". So even after I copied the file in /lib it still did not work. So I had to also notice the name difference. 

All these problems can happen to someone else and you need to loo the code to figure it out.
And mouse is extremely slow on 8 Mhz 8086.

ghaerr, 2020-11-14:
Thanks @toncho11, I'll look into fixing these. Not sure before or after v0.4.0 though.

pawosm-arm, 2020-11-15:
The map must have been drawn back in the 80s. Things has changed east to Poland since then...

ghaerr, 2020-11-16:
@toncho11: I've fixed everything in PR #869 except for searching the current folder. If the map file is missing, it will display a message. Searching the current folder really isn't needed and is too much extra logic the way nxworld is written.

Thank you!

toncho11, 2020-11-16:
#define	F_OK	0		/* Test for existence.  */

Thanks.

closed 2020-11-16 by toncho11

ghaerr, 2020-11-16:
Thanks @toncho11, I'll get that F_OK added in another commit.