ghaerr/elks #864: Boots DOS Svarog86 also

toncho11, 2020-11-14:
I have ELSK on /hda1 and DOS Svarog86 on /hda3

Both of them boot now!!!!!!!!!!!!!!!!!!!
The only minor problem was that I had to format /dev/hda3 for Svarog86 with Windows 10.  I noticed that Windows set the partition type to 'c' instead of '4' (set by me) and that is why Svarog86 was able to make /dev/hda3 boot with the DOS sys command.
I choose which one in ELKS boot manager (1 or 3).
For Svarog86 I have to just say that command,com is on 3rd partition and it loads completely.

Wow. That is great. And I can exchange files between these.

toncho11, 2020-11-14:
Dos 6.20 does not see the fat16 drive that Svarog86 is installed.

ghaerr, 2020-11-14:
> I think we should document that it is possible to boot both.

You're welcome to create another page on our Wiki and describe how one goes about creating partitions, loading other OS's, that sort of thing.

> Dos 6.20 does not see the fat16 drive on which Svarog86 is installed.

Has it ever before? How was the partition table created, some OS's are much more sensitive to the partition ID type set in the partition table. Fdisk will allow setting one of many partition types when creating, I don't think afterwards though.

I'm glad to hear that the ELKS MBR boots other OS's, that hadn't been previously tested.

toncho11, 2020-11-16:
I added a wiki page [Dual-boot-ELSK-and-DOS](Dual-boot-ELSK-and-DOS) explaining this.

closed 2020-11-16 by toncho11

ghaerr, 2020-11-16:
> I added a wiki page Dual-boot-ELSK-and-DOS explaining this.

Thanks! Can you fix the "ELSK" typo?