ghaerr/elks #865: Are SMC network cards supported?

toncho11, 2020-11-14:
Mine is 83C690 ISA card. When I tried to build ELKS with SMC network card support , it failed.

What about 3com cards? I have one also.

ghaerr, 2020-11-14:
> When I tried to build ELKS with SMC network card support , it failed.

Oops, fixed by #866.   

> What about 3com cards? 

They probably won't work out of the box, but with some mods should be supported. I have one also, and have been meaning to test it.

toncho11, 2020-11-14:
[image]

[image]
I tried to set up 192.168.1.77 as my ip on my network, but it does not work because of the Mac address I suppose.

Mellvik, 2020-11-14:
Hi @toncho11,
You're right: the fact that the driver is unable to fetch the mac address from the card means that the card is not supported. It is most likely a wd8013 card, possibly a 16bit card to boot.

I belive @ghaerr has a card like that, which increases your odds dramatically. Have some patience and it'll be supported. 

As to 3com - that depends on the age. I'm working on (got started on) a 3c509 driver - simply because I happen to have the card. If you have one, your help in testing would be great.

-Mellvik

toncho11, 2020-11-15:
My SWC is ISA 8 bit. It says 83C690 on the biggest chip. It works on DOS with a specific driver.

toncho11, 2020-11-15:
Yes I have a 16 bit ISA 3Com EthrLink III, 1995, 3C5098-C. There is a DOS driver here:

[https://github.com/hackerb9/3C509B-nestor](https://github.com/hackerb9/3C509B-nestor)

toncho11, 2020-11-15:
And drivers should use 8086 only (not 186 or 286) or it won't be usable on older machines like mine.

toncho11, 2020-11-15:
The driver says that "SMC 8003WC" is supported and my card looks exactly as SMC 8003WC. I mean my card is almost identical. So it could be a bug in wd_get_hw_addr for example ...

toncho11, 2020-11-15:
The driver says that "SMC 8003WC" is supported. Now I took a better look at my card. It says 8003WC B 2692 REV A. So it a supported card. 

Maybe it is a bug in: wd_get_hw_addr

toncho11, 2020-11-15:
On the card it says "280, 3" for I/O IRQ as pin configuration, but the kernel says "0x240, 2". The kernel detection is not correct I think,

Mellvik, 2020-11-15:
Well, isn't this a bunch of good news, @toncho11!!
Your card is supported (and the 3com card is the one I'm working on!).

The problem is definitely the address and IRQ, and the kernel has no means of detecting either, courtesy of the PC architecture. The driver should have - and eventually will have - a means of detecting that there is no card (such as the mac address being all Fs) however.

If the card cannot be reconfigured by jumpers, you have to go into ports.h and make the necessary changes. Let me know if you need advice as to what to change.

You'll be up and running in no time!

--Mellvik

pawosm-arm, 2020-11-15:
Isn't it the same card that I have?
[image]

Anyway, to make it work, I had to configure it first under DOS (namely, set W1 jumper on the card to "SOFT") so it avoids IRQ, I/O ports and I/O mem conflicts. The ELKS driver we worked on reflects this configuration. I guess you either need to configure it the same way I did, or patch the driver with your settings.

toncho11, 2020-11-15:
Yes same card.

Now it works!!!!!!!!!!!!!!!!!!!!!!!! :) I am telnetting to my Ubuntu machine.

./elks/elks/include/arch/ports.h

Changed from:
```
/* wd, wd.c*/
#define WD_IRQ          2
#define WD_PORT         0x240
```

to

/* wd, wd.c*/
#define WD_IRQ          3
#define WD_PORT         0x280

Can we put a config file in /etc for that instead of recompiling ?

pawosm-arm, 2020-11-15:
Make sure you configured I/O mem correctly too (0xd800U according to the snapshot you've sent), on my machine this address is occupied by hardware ramdisk, so I had to change it to 0xce00U (right after XT-IDE's memory):

```
#define WD_SHMEMSEG    0xce00U
```

> Can we put a config file in /etc for that instead of recompiling ?

Due to similarity with ne2k driver (whose configuration is also hardcoded), there is a plan to do a bigger refactor to merge these two into one, until that, I doubt it will change. Even if, keep in mind that ELKS kernel needs to be kept small, hence reading configuration from file on mounted filesystem isn't an option. Adding configuration option to kernel's `.cofnig` file seems more realistic.

toncho11, 2020-11-15:
WD_SHMEMSEG is the ROM address? I think I changed the configuration of my XT-IDE to avoid conflicts. 

So the WD_SHMEMSEG is used to access data from the ROM? Is that correct?
Bu then why is it working if the current value is 0xce00U and my card is configured to 0xd000U ?

pawosm-arm, 2020-11-15:
> WD_SHMEMSEG is the network card ROM address?

Yes, you're right, I've corrected my comment when I realized that I'd read wrong value from the snapshot of your card.

> So the WD_SHMEMSEG is used to access data from the ROM? Is that correct?

No, keep the ROM value aside. The value we're after is 0xd000U on your card.

> But then why is it working if the current value is 0xce00U and my card is configured to 0xd000U ?

You may have been writing someone else's memory. And you were lucky nothing exploded.

toncho11, 2020-11-15:
So you need to put a total of 3 parameters:
- IO base memory address (ports,h)
- IRQ (ports.h)
- ROM location address (wd.c) 

Please correct me if I am wrong.

pawosm-arm, 2020-11-15:
> ROM location address (wd.c)

It's I/O mem, not ROM. As I said before, I've introduced this confusion by mistake.

> And for what is the ROM location address used in the wd.c?

see above, it should be I/O mem, not ROM.

toncho11, 2020-11-15:
I think it is called "[RAM base address](http://www.stack.nl/~marcolz/network/wd80x3.html)", no?

So the right terms are:

- IO base memory address (ports,h) = 280
- IRQ (ports.h) = 3
- RAM base address (wd.c) = 0xD000

It is called "RAM" on the card, jumper w1 and from internet I see it is used for RAM buffer.

pawosm-arm, 2020-11-15:
> I think it is called "RAM base address", no?

yes, that's correct name for it.

closed 2020-11-15 by toncho11

ghaerr, 2020-11-15:
I'm glad to see we've got a fourth person using ELKS networking, which is proving to work quite well, once configured!

> Even if, keep in mind that ELKS kernel needs to be kept small, hence reading configuration from file on mounted filesystem isn't an option. 

Agreed that the kernel shouldn't be doing that, but that's not to say a separate user-mode "netconfig" couldn't be developed, and use an ioctl to talk to the network driver after boot. This program could read a network configuration file and then ioctl the I/O and RAM base address to the driver. Configuring the IRQ would be harder, but possibly doable, if within a range of permitted values. This would require the network card drivers to use a variable, rather than the current constant, every time the I/O or RAM base addresses were used.

Something to think about in 0.5.0, to make ELKS easier to use!

The Wiki should be updated with the three "requirements" ELKS networking needs in order to work, if not done already.

I hope to get a 0.4.0 out shortly, mostly as a marker, since it seems ELKS development continues at a rapid pace. I plan on using Github "assets" to hold all the disk images for download without compilation, although we won't be able to support every version of networking, for the reasons discussed here.

toncho11, 2020-11-15:
Yes. I updated the wiki with the 3 parameters. I also added a new section "Installation" that describes the `sys` command.

PR #866 (2020-11-14): Fix size_t compilation error in memory.h
Fixes #865.