ghaerr/elks #870: Tinyirc problem

toncho11, 2020-11-16:
Are you able to use tinyirc?

There is no DNS in ELKS, right?

So I use `tinyirc 185.30.166.38 6667` to connect to irc.freenode.org and it prints some correct messages, but in never connects.

Also `nslookup` is not working.

ghaerr, 2020-11-16:
There is no DNS in ELKS.

> Are you able to use tinyirc?

I've never used it. I would doubt anyone has run it in the last 20 years. It looks to have been created between 1991-1996.

> Also nslookup is not working.

Have you looked at or changed /etc/resolv.conf? Nslookup uses that to get its DNS data; it likely needs updating.
Due to additional long-name issues with FAT filesystems, this should be tested on a MINIX filesystem. The ".conf" extension can be opened, but not searched with `ls` on FAT, IIRC. Also, netstat has a compiled-in nameserver address when /etc/resolv.conf is missing, which may want to be updated, and perhaps /etc/resolv.conf removed from the distribution.

Volunteers needed for fixing these older utilities. If they can be proven non-operational, perhaps they should be removed from the distribution until fixed?

toncho11, 2020-11-16:
There are not may things one can do with functional network ... so it will be better to fix tinyirc. Maybe I can work on this in the future.

ghaerr, 2020-11-16:
> Maybe I can work on this in the future. 

You might take a stab at nslookup, its a lot simpler. I'm thinking if it could be made working, we could rename /etc/resolv.conf to /etc/resolv.cnf also, that would solve the FAT issues above. If you start on it, I will help you if needed.

> Tinyirc can connect only to plain-text connections, not TLS.

Oh, so tinyirc works otherwise?

ELKS won't likely ever have TLS support, the crypto libraries are way too large.

toncho11, 2020-11-28:
I tried two small IRC clients to compile, but:

- C++ is not supported in ELKS?
- some headers were missing in the compiler when trying a C IRC client

ghaerr, 2020-11-28:
> some headers were missing in the compiler when trying a C IRC client

Which ones? Unable to tell whether they are standard headers or library headers.

ghaerr, 2020-11-29:
Hello @toncho11 and @tkchia,

> C++ is not supported in ELKS?

I've never used C++ on ELKS, so I don't know - @tkchia, can you comment on that?

Thank you!

tkchia, 2020-11-29:
Hello @ghaerr,

> I've never used C++ on ELKS, so I don't know - @tkchia, can you comment on that?

Yes, C++ is not yet supported with the ELKS target --- in fact, even for the MS-DOS target, C++ support is still rather incomplete.  The main problems I am facing for now are these:

  * The GNU C++ Library (`libstdc++`) which comes with GCC is very bloated.  I have been trying to tweak it to get a simple `std::cout << "Hello world\n";` to fit in a small model program, without much success.
  * I am also having trouble getting the medium memory model to work with the C++ language middle-end.  The middle-end thinks that all pointers have the same size, which is not the case with the medium model.

Thank you!

toncho11, 2022-09-22:
One more IRC client:

https://github.com/retrohun/mTCP/tree/master/APPS/IRCJR

Although for DOS it is recent and one that is proven to work on non TLS servers.

closed 2023-03-22 by ghaerr