ghaerr/elks #873: ELKS v0.4.0 Pre-release version history and credits

ghaerr, 2020-11-19:
Following is a first-draft template of all that's happened since v0.3.0, along with the contributors. Please let me know what needs to be added, as the list is long and I'm sure something or someone has been forgotten! I plan to use this text as the v0.4.0 Final Release documentation.

**ELKS v0.4.0 Enhancements**

**Kernel**
- No compiler warnings on kernel build
- Strace working, calculates kernel stack usage
- Implement task zombie state, rewrite wait
- Implement kernel local heap (@mfld-fr)
- Improved memory segment allocator (@mfld-fr)
- Fix userland signals
- Use single far address for kernel process signaling (@tkchia)
- New synchronization primitives
- Remove unneeded and slow kernel locks
- Add race-safe sleep/wait prepare_to_wait etc
- Fix select busylooping, network semaphore lockups
- Much improved kernel debug macros and mechanism
- ^P kernel/application debug mechanism using SIGURG
- Restartable system calls
- Reduce kernel stack to 600 bytes from 988
- Align stack pointer to even boundary to save malloc issues
- Kernel FAR text section, saves 11k bytes of .text space
- Enhance I/O performance w/direct fmemcpy to/from user space
- Rewritten fmemcpy routines to use word copy

**System Call Fixes**: wait, rmdir, access, brk, link, stat, gettimeofday

**Exec**
- Separate heap & stack, v1 a.out header
- chmem command enhanced
- New compiler options for heap and stack (@tkchia)
- Tuned 2k stack / 2k heap for all processes, tuned min memory usage
- Run programs w/historical Minix a.out header (@tkchia)
- Load medium-model programs (@tkchia)

**Filesystems**
- FAT filesystem debugged and working
- Mountable Minix and FAT partitions
- Improved MBR, use sect_offset to support MBR boots to /dev/hda1 etc
- Add VFAT long filename support, virtual /dev directory
- Fix Minix mount/unmount superblock checking/write
- 16-bit or 32-bit inodes in kernel
- Mount readonly, remount options, can boot readonly
- Fix romdisk read directory
- Allow ramdisk size larger than 64k
- Allow preloaded ramdisks in upper memory
- FAT32 fixes, large drive mount fix

**Disk & I/O Subsystem**
- Rename /dev/bdX to /dev/hdX
- Support for USB drives
- Variable number of L2 buffers
- Revise /dev/hda from 4 to 8, implement partitions
- Floppy I/O to use DMASEG only when necessary
- IDE query to BIOS HD driver for reliable CHS data (@Mellvik)
- Replaced DDPT allows running on IBM XT v1 BIOS machines (@tkchia)
- Align buffers on 1K to remove DMASEG requirement
- Floppy track caching, improves floppy speed tremendously
- Block I/O subsystem enhanced, no mapbuffer or buffer copying
- 32-bit block numbers for FAT

**Console**
- Direct Console fnkey sequences
- Direct console CapsLock/Numlock and LED support (@Mellvik)
- Direct Console ANSI sequences for vi and sh line editing
- Combine ANSI code in BIOS and Direct Console
- Direct and BIOS console driver refactoring and cleanup
- Serial console rewritten, improved structure
- Kernel printk allowed before console init
- Add headless console for SBC systems

**Drivers**
- Fast serial driver
- Serial driver FIFO
- Add O_EXCL for serial com/mouse
- Implement VMIN/VMAX, termios cleanup in applications
- Dynamically allocate TTY I/O queues
- Char I/O excess wakeup calls removed for speed
- Character queues don’t have to be power of 2
- Customizable IRQ and I/O ports for all drivers
- Use simpler wait routines in network drivers
- New WD SMC network driver (@pawosm-arm)
- NE2K fully working and debugged (@Mellvik)

**Networking**
- Packet statistics via netstat
- Arp cache list display
- Debug RTO values
- Multiple session support to telnetd
- Tune timeouts for TCP transmissions
- Remove double buffering in network drivers
- Tuned network buffer sizes, implement send window write throttling
- Network throughput increased substantially

**Networking Bug Fixes**
- Tons of TCP/IP fixes to ktcp and kernel
- Async ARP packets
- Fixes to telnet, telnetd and httpd to get them actually working
- Implement SLIP driver
- Fix SLIP buffer overrun
- Fix odd length IP/ICMP checksum calcs
- Networking fixes for connections, PTY fixes, keypress to continue fix

**Boot**
- Boot option parsing in /bootopts
- Serial console via /bootopts
- Init run modes for multi-user/multi-console
- FAT 12/16 boot (@tkchia)
- FAT32 boot
- Improved FAT booter, /linux and /dev first 8 entries (@tkchia)
- Implement ELKS parameter block in boot blocks (@tkchia)
- MBR boot loader
- Revised single stage boot loader design for Minix (@mfld-fr)
- Minix boot loader enhancements (@tkchia)
- Major setup.S cleanup, source cleanups
- Refactor REL_SYS, remove !REL_SYS, CONFIG_ROMCODE working

**New Commands**
- sys - transfer system images
- makeboot command to write boot blocks
- mkfat - create FAT filesystems
- kilo - visual editor enhancements (@georgp24)
- cron - schedule background jobs (@georgp24)
- arp - display ARP cache
- net - network start/stop script
- df - disk free
- time - calculate elapsed time running command (@Mellvik)
- busyelks - BusyBox for ELKS (@marcin-laszewski)
- tty - display current TTY (@Mellvik)
- hd - hex dump
- unreal - switch to 8086 unreal mode (@georgp24)
- uuencode/uudecode - encode / decode binary files for network transmission (@Mellvik)
- fdtest - floppy disk tester
- sercat - serial port tester
- mouse - mouse tester
- Add Nano-X and games into build tree

**Improved Commands**
- Sh line editing (@georgp24)
- BASH command completion
- Sash history (@Mellvik)
- Dynamic window size for vi, kilo
- ANSI /etc/termcap for vi, vi works from serial port
- Enhanced ps, meminfo, miniterm, ls
- Major bug fixes to init, getty
- ^C handling, no shell exit
- Workaround kernel directory rename with improved mv command
- Add recursive option to cp to copy entire disks
- Fixed mkfs and fsck to work
- Fixed fdisk to work

**C Library**
- Debug malloc heap trace
- Libc Routines added or fixed: putenv/basename (@cjsthompson), dirname, chgrp, snprintf/ vsnprintf (@tkchia), getpass, times, tzset, sleep, rand, srand, gmtime, in_aton, in_ntoa, in_gethostbyname
- Header file cleanup and all prototypes
- No warnings on libc compile
- Split all functions to separate .c files (@marcin-laszewski)
- Localtime support and TZ=
- Add stdcall support (@tkchia)
- Add multilib support (@tkchia)

**Build**
- MacOS build
- Disk images built with mfs tool, no root or host filesystem driver needed
- Add setboot for managing boot blocks
- Elkscmd/ reorganization, all commands built w/any config

**Compiler / Tools**
- Toolchain enhancements for medium mode (@tkchia)
- Enhance thunk code for 0-6 argument near calls from fartext (@tkchia)
- Move to new elf linker (@tkchia)
- Use ligbcc from toolchain (@tkchia)
- Add objdump86
- Elksemu enhancements (@tkchia)
- GCC tools update for GCC v10 (@tkchia)
- mfs tool to create Minix images w/o being root

**Images**
- Build all disk images automatically
- Images include 2.88M, 1.44M, 1.2M, 720k and 360k, FAT and Minix, 32MB HD, 32MB MBR
- Config image build system for app category and disk image size
- Select applications by image size as well as application category
- Consolidate commands to /bin

**Documentation**
- Wiki
- Improved HTML documentation (@georgp24)

mfld-fr, 2020-11-19:
This is obviously a huge work that has been done on ELKS for the PC/XT/AT platform. So many fixes & improvements ! That largely deserves the '0.4.0' version.

toncho11, 2020-11-19:
Or the 1.0 :)

Mellvik, 2020-11-19:
This is big, @ghaerr!!

I’m vouching for 1.0 - this is just too big for a decimal-increment. Think about it - if all this is worth a decimal, what would it take to yank it up a whole number?

@ghaerr, your note keeping is impressive. Likewise for the contributions. Thank you.

Comments:
ARP: ARP wasn’t really working until you fixed it. So a reasonable entry would be
Fixed ARP, added ARP cache

clock:
added clock (or fixed?), elks now picks up time and date from the hardware at boot (and can set it).

Networking:
- File transfers both ways now supported.
- WD driver: It may be appropriate to be more specific than 'WD SMC', since there are many very different WD SMC cards out there. Suggest you mention WD8003 specifically.
NE2K: I think ‘rewritten’ is more appropriate than ‘fully working’.

Improved commands:
It may be worthwhile mentioning getty in this context, and that ELKS now works multiuser with 3 (probably 4) serial terminals.

Finally:
Possibly something about ’tested on’ or ‘works on’ - PCs running 8088, 8086, 80286, 80386 … and later

This should be a proud moment for all. Now ELKS is real - not just a half-stable playground for hackers (the original definition) and enthusiasts, but for users looking for a tool to make ’small’ hardware work big.

Unless someone else volunteers, I’ll do the Wikipedia update when the announcement is out.

Thanks again - to everyone and @ghjaerr in particular.

—Mellvik

ghaerr, 2020-11-19:
@Mellvik,

Thanks for your comments, I've updated the release docs (above) from them.

> I’m vouching for 1.0 - this is just too big for a decimal-increment. Think about it - if all this is worth a decimal, what would it take to yank it up a whole number?

Yes, there's a ton of stuff added, seemingly enough for a v1.0.0. However, there's still lots that doesn't work, and no need to rush. I plan on incrementing versions more quickly, and would like to continue to allow ELKS to be easier to use and evaluate. The Wiki page and website also need updating. The combination of that and more frequent versions will lead us to v1.0, just not yet :)

> Unless someone else volunteers, I’ll do the Wikipedia update when the announcement is out.

Thanks. I think it important to update the number of commits that have been made, and the kinds of things that have been added that make ELKS more useful on PC/XT/ATs, since that information is also quite out of date. Hopefully that will impart to readers the kind of activity ELKS has been seeing, and its usefulness.

Thank you!

toncho11, 2020-11-20:
OK for 0.4.
1.0 is also for marketing. Didn't you know that all commercial products start from version 3.0 :). It is because it gives some credit to your product. In case you say ELKS is stable. But it is not important. 

I think for the bin images you should say:
compiled with 2k network driver: IO base address=... IRQ=..., RMA adress=...

ghaerr, 2020-11-20:
> 1.0 is also for marketing. Didn't you know that all commercial products start from version 3.0 :). It is because it gives some credit to your product. In this case you say ELKS is stable.

Yes, I agree that version numbers are also for marketing. 

> I think for the bin images you should say:
compiled with 2k network driver: IO base address=... IRQ=..., RAM base address=...

So here's a good point, and a problem. There currently isn't a good way to produce an ELKS image that will work on a variety of systems. In fact, ELKS was designed with a massive configuration system - the opposite of "it just works". The reasons for this are many, but with the far text kernel, it is now possible to create a larger kernel, which could possibly contain multiple network drivers, detection code, etc. So all that should be added, if the idea were truly to be able to run ELKS from a single distribution image/kernel. All this needs to be thought about more, and another reason we're not really ready for 1.0 - ELKS configuration vs works out-of-the-box design isn't ready for a lot of users yet.

Thank you for your comments!

toncho11, 2020-11-20:
Simply saying that it is currently configured with these settings is helpful. It hints why it might not work when people download it.

toncho11, 2020-11-20:
When is it coming the new release :) ?

ghaerr, 2020-11-21:
> compiled with 2k network driver: IO base address=... IRQ=..., RAM base address=...

Thanks, I have updated the release notes with a download section.

toncho11, 2020-11-21:
@ghaerr From some comments on Youtube, people report that they can not attach the hdd to another computer to put an ELKS partition, so the sys tool comes handy!

closed 2020-11-21 by ghaerr