ghaerr/elks #882: [cmd] ls not showing symbolic links in current directory

Mellvik, 2020-11-26:
I believe this is a recently introduced 'anomaly' to `ls`: 

If listing the current directory, symbolic links show up as their target, if in a different directory (i.e. there is a path), the display is fine.

```
# cd /var/www
# ln -s /dev/hda hdimage
# ls -l 
brw-r--r--  2 root     root       3,   0  Nov 26 09:57 hdimage
-rwxr-xr-x  1 root     root          275  Nov 26 09:57 index.html
# mkdir x
# ln -s /dev/fd1 x/fd
# ls -l x
lrwxrwxrwx  1 root     root            8  Nov 26 14:00 fd -> /dev/fd1
# mv x/fd .
# ls -l 
brw-r--r--  2 root     root       3, 160  Nov 26 09:57 fd
brw-r--r--  2 root     root       3,   0  Nov 26 09:57 hdimage
-rwxr-xr-x  1 root     root          275  Nov 26 09:57 index.html
drwxrwxrwx  2 root     root           48  Nov 26 14:02 x
# 
```

ghaerr, 2020-11-27:
@Mellvik : 

First of all, the display above is not actually what happens on ELKS. Try it. Instead, the following occurs:
```
# cd /var/www
# ln -s /dev/hda hdimage
# ls -l 
brw-r--r--  2 root     root       3,   0  Nov 26 09:57 hdimage -> /dev/hda
-rwxr-xr-x  1 root     root          275  Nov 26 09:57 index.html
#
```
The "bug" hasn't to do with `ls` at all, and ls hasn't changed. Instead, `mv` is not renaming the symlink, but renaming what the symlink references. At the moment, all ELKS commands (including cp, mv etc) do that with a symlink: they operate on what the symlink references, not the symlink itself. To change this behaviour, I think each command might need to be analyzed separately, and a plan developed on each of them as to what it's action on a symlink ought to be.

Mellvik, 2020-11-27:
Thank you @ghaerr.

> The "bug" hasn't to do with ls at all, and ls hasn't changed.
Ah, yes, I get it - this is mv, not ls.
> Instead, mv is not renaming the symlink, but renaming what the symlink references.
Well, not really. What mv is doing (or seems to be doing), is changing the symlink to a hard link, which is definitely a bug afaik.
> At the moment, all ELKS commands (including cp, mv etc) do that with a symlink: they operate on what the symlink references, not the symlink itself.
I believe this is the right behaviour for most utilities, that's the purpose of the symlink afaik. Only utilities dealing with inodes directly should see the difference between a symlink and the referenced file. ls, tar, maybe ar, certainly mv , probably others …

—Mellvik

ghaerr, 2020-12-01:
@Mellvik,

PR #889 fixes this issue in `mv`. Please test when you have time then close this issue, thanks!

Mellvik, 2020-12-01:
Thank you @ghaerr,

testing passed with flying colours. Looking great!

—Mellvik

PR #889 (2020-12-01): [cmds] Enhance mv to support renaming symlinks
Fixes #882.

Also updates `access` to use F_OK as suggested by @toncho11.
closed 2020-12-01 by Mellvik